cyring/CoreFreq #481: coreqd not working on 6.8.6-1-cachyos-echo

LethalManBoob, 2024-04-14:
× corefreqd.service - CoreFreq Daemon
     Loaded: loaded (/usr/lib/systemd/system/corefreqd.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Sun 2024-04-14 14:22:54 BST; 1h 40min ago
   Duration: 14ms
   Main PID: 672 (code=exited, status=203/EXEC)
        CPU: 1ms

Apr 14 14:22:54 archlinux systemd[1]: Started CoreFreq Daemon.
Apr 14 14:22:54 archlinux systemd[1]: corefreqd.service: Main process exited, code=exited, status=203/EXEC
Apr 14 14:22:54 archlinux systemd[1]: corefreqd.service: Failed with result 'exit-code'.

cyring, 2024-04-14:
Perhaps the package you are using is broken.
Can you try from a manual build and run as below ?

```sh
## as a user
git clone https://github.com/cyring/CoreFreq.git
cd CoreFreq
make

## as root
insmod build/corefreqk.ko
./build/corefreqd -d
```
```sh
## as a user, in another terminal 
cd CoreFreq
./build/corefreq-cli -s
``` 

Please show me in Markdown the output of all the above.

ptr1337, 2024-04-14:
> × corefreqd.service - CoreFreq Daemon Loaded: loaded (/usr/lib/systemd/system/corefreqd.service; enabled; preset: disabled) Active: failed (Result: exit-code) since Sun 2024-04-14 14:22:54 BST; 1h [...]

That likely comes from the ECHO Scheduler due not supported several features. You should try it with the default kernel (e.G linux-cachyos, linux-cachyos-eevdf)

LethalManBoob, 2024-04-14:
> That likely comes from the ECHO Scheduler due not supported several features. You should try it with the default kernel (e.G linux-cachyos, linux-cachyos-eevdf)

× corefreqd.service - CoreFreq Daemon
     Loaded: loaded (/usr/lib/systemd/system/corefreqd.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Sun 2024-04-14 17:43:19 BST; 21s ago
   Duration: 10ms
    Process: 677 ExecStart=corefreqd -q (code=exited, status=203/EXEC)
   Main PID: 677 (code=exited, status=203/EXEC)
        CPU: 1ms

Apr 14 17:43:19 archlinux systemd[1]: Started CoreFreq Daemon.
Apr 14 17:43:19 archlinux systemd[1]: corefreqd.service: Main process exited, code=exited, status=203/EXEC
Apr 14 17:43:19 archlinux systemd[1]: corefreqd.service: Failed with result 'exit-code'.

This is what I get on the base cachyos kernel

cyring, 2024-04-14:
> This is what I get on the base cachyos kernel

Please I need to see results from the manual procedure I wrote above.
Because Systemd is masking CoreFreq ...

LethalManBoob, 2024-04-14:
> > This is what I get on the base cachyos kernel
> 
> Please I need to see results from the manual procedure I wrote above. Because Systemd is masking CoreFreq ...

ok it needed reinstalling. somehow the dkms got deleted and didnt reinstall dispite kernel update
unless i misconfigured sdboot i have no clue what caused this

closed 2024-04-14 by LethalManBoob