cyring/CoreFreq #482: When uninstalling any kernel. corefreq will stop working

LethalManBoob, 2024-05-07:
as title suggests. so far the kernel version doesnt matter. happens on any combination i tried. mostly cachyos kernels

only way to make it work is to reinstall corefreq completly

cyring, 2024-05-07:
Hello,

Can you list all commands employed from install t'il Uninstall and up to the issue appearing ?

LethalManBoob, 2024-05-09:
Never mind. Turns out it just seems to stop working on its own. The only thing that seems to be common is the rebuilding of mkinitcpio causes this

cyring, 2024-05-09:
> [> Hello,
> 
> > > When uninstalling any kernel. corefreq will stop working
> > 
> > 
> > Like most kernel modules, any change into the kernel environment implies a module rebuild:
> > ```
> > make clean
> > make
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > > only way to make it work is to reinstall corefreq completly and modprobe
> > 
> > 
> > Can you list all commands employed from install to uninstall and t'il the issue appearing ?
> 
> ](https://paste.cachyos.org/b5a298b.txt)
> 
> All I did was install a kernel update and that broke it. Once again, seems anything running the mkinitcpio command breaks it. The only way to fix is to reinstall and restart the daemon

Happy it works.

Please delete all your past data in this thread, including revisions: your credential is unveiled.

closed 2024-05-09 by cyring

LethalManBoob, 2024-05-09:
Thats been done but also pls remove your reply, also showing my link. Plus its only working until the next update. or probebly even update that rebuilds mkinitcpio. this isnt a solution lol