cyring/CoreFreq #483: Building on Almalinux 9 Redux

fishbonesnl, 2024-05-13:
Hello,

A few months ago I reported problems building on Almalinux 9, which were fixed.

I performed a kernel update and building CoreFreq breaks. I have downloaded the latest master and it still fails. Sorry to be a bother again.

The error message is as follows (home directory anonymised):

`In file included from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:8,
                 from ./include/linux/list.h:9,
                 from ./include/linux/module.h:12,
                 from <home>/CoreFreq/build/module/corefreqk.c:18:
<home>/CoreFreq/build/module/corefreqk.c: In function ‘CoreFreqK_Create_Device_Level_Up’:
./include/linux/export.h:17:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   17 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
<home>/CoreFreq/build/module/corefreqk.c:23685:41: note: in expansion of macro ‘THIS_MODULE’
23685 |         CoreFreqK.clsdev = class_create(THIS_MODULE, DRV_DEVNAME);
      |                                         ^~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:299: <home>/CoreFreq/build/module/corefreqk.o] Error 1
make[1]: *** [Makefile:1934:<home>/CoreFreq/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-427.16.1.el9_4.x86_64'
make: *** [Makefile:102: all] Error 2`

uname -a `Linux 5.14.0-427.16.1.el9_4.x86_64 #1 SMP PREEMPT_DYNAMIC Thu May 9 18:15:59 EDT 2024 x86_64 x86_64 x86_64 GNU/Linux`

Any ideas what's going on?

cyring, 2024-05-13:
Hello,

I'm downloading this version [AlmaLinux OS 9.4 DVD ISO](https://repo.almalinux.org/almalinux/9.4/isos/x86_64/AlmaLinux-9.4-x86_64-dvd.iso) to try building into a VM.

Is it the concerned version ?

fishbonesnl, 2024-05-13:
Yes, my version is 9.4. The kernel version is the latest one (updated today via dnf update).

cyring, 2024-05-13:
Commit 7aa4aa6d9c8abe9a529feb53bc2706287e7ec428 in branch [`develop`](https://github.com/cyring/CoreFreq/tree/develop) should fix the issue.

Can you please give it a try ?

fishbonesnl, 2024-05-13:
I confirm that it builds as intended and that it runs.

Many thanks!

cyring, 2024-05-13:
Stay tune, I have to fix other Linux `6.9` conflicts and will merge into `master` soon.

cyring, 2024-05-14:
Merged into `master`. You can can close your issue.

closed 2024-05-14 by fishbonesnl

cyring, 2024-05-14:
I'm adding a build and run documentation for AlmaLinux
https://github.com/cyring/CoreFreq/tree/develop?tab=readme-ov-file#almalinux

I however had this (_non blocking_) BTF issue during build and installation

[image]
[image]

Any idea how to fix it in Alma ?

fishbonesnl, 2024-05-14:
I don't have this issue. I have it on Ubuntu, and all I remember is that it has to do with where the code expects the kernel image to be located. See here

[BTF issue](https://www.reddit.com/r/openSUSE/comments/pwxs3i/btf_file_not_generated_because_vmlinux_could_not/)