cyring/CoreFreq #484: Corefreq broken on kernel 6.9

bunkbail, 2024-05-13:
```
cd ~/CoreFreq && git pull && make DELAY_TSC=1 clean all
Already up to date.
make[1]: Entering directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
make[1]: Leaving directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreqd.c -o build/corefreqd.o
cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreqm.c -o build/corefreqm.o
cc  -o build/corefreqd \
  build/corefreqd.o build/corefreqm.o -lpthread -lm -lrt -lc
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -c x86_64/corefreq-cli.c -o build/corefreq-cli.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-ui.c -o build/corefreq-ui.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -c x86_64/corefreq-cli-rsc.c -o build/corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-cli-json.c -o build/corefreq-cli-json.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-cli-extra.c -o build/corefreq-cli-extra.o
cc  -o build/corefreq-cli \
  build/corefreq-cli.o build/corefreq-ui.o \
  build/corefreq-cli-rsc.o build/corefreq-cli-json.o \
  build/corefreq-cli-extra.o -lm -lrt -lc
make[1]: Entering directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
  CC [M]  /home/bunkbail/CoreFreq/build/module/corefreqk.o
In file included from /home/bunkbail/CoreFreq/build/module/corefreqk.c:63:
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:99: warning: "MSR_IA32_FRED_STKLVLS" redefined
   99 | #define MSR_IA32_FRED_STKLVLS                   0x000001d0
      | 
In file included from ./arch/x86/include/asm/nospec-branch.h:12,
                 from ./arch/x86/include/asm/paravirt_types.h:12,
                 from ./arch/x86/include/asm/ptrace.h:175,
                 from ./arch/x86/include/asm/math_emu.h:5,
                 from ./arch/x86/include/asm/processor.h:13,
                 from ./arch/x86/include/asm/timex.h:5,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /home/bunkbail/CoreFreq/build/module/corefreqk.c:18:
./arch/x86/include/asm/msr-index.h:44: note: this is the location of the previous definition
   44 | #define MSR_IA32_FRED_STKLVLS   0x1d0                   /* Exception stack levels */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:100: warning: "MSR_IA32_FRED_SSP1" redefined
  100 | #define MSR_IA32_FRED_SSP1                      0x000001d1
      | 
./arch/x86/include/asm/msr-index.h:46: note: this is the location of the previous definition
   46 | #define MSR_IA32_FRED_SSP1      0x1d1                   /* Level 1 shadow stack pointer */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:101: warning: "MSR_IA32_FRED_SSP2" redefined
  101 | #define MSR_IA32_FRED_SSP2                      0x000001d2
      | 
./arch/x86/include/asm/msr-index.h:47: note: this is the location of the previous definition
   47 | #define MSR_IA32_FRED_SSP2      0x1d2                   /* Level 2 shadow stack pointer */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:102: warning: "MSR_IA32_FRED_SSP3" redefined
  102 | #define MSR_IA32_FRED_SSP3                      0x000001d3
      | 
./arch/x86/include/asm/msr-index.h:48: note: this is the location of the previous definition
   48 | #define MSR_IA32_FRED_SSP3      0x1d3                   /* Level 3 shadow stack pointer */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:103: warning: "MSR_IA32_FRED_CONFIG" redefined
  103 | #define MSR_IA32_FRED_CONFIG                    0x000001d4
      | 
./arch/x86/include/asm/msr-index.h:49: note: this is the location of the previous definition
   49 | #define MSR_IA32_FRED_CONFIG    0x1d4                   /* Entrypoint and interrupt stack level */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:104: warning: "MSR_IA32_FRED_RSP0" redefined
  104 | #define MSR_IA32_FRED_RSP0                      0x000001cc
      | 
./arch/x86/include/asm/msr-index.h:40: note: this is the location of the previous definition
   40 | #define MSR_IA32_FRED_RSP0      0x1cc                   /* Level 0 stack pointer */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:105: warning: "MSR_IA32_FRED_RSP1" redefined
  105 | #define MSR_IA32_FRED_RSP1                      0x000001cd
      | 
./arch/x86/include/asm/msr-index.h:41: note: this is the location of the previous definition
   41 | #define MSR_IA32_FRED_RSP1      0x1cd                   /* Level 1 stack pointer */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:106: warning: "MSR_IA32_FRED_RSP2" redefined
  106 | #define MSR_IA32_FRED_RSP2                      0x000001ce
      | 
./arch/x86/include/asm/msr-index.h:42: note: this is the location of the previous definition
   42 | #define MSR_IA32_FRED_RSP2      0x1ce                   /* Level 2 stack pointer */
      | 
/home/bunkbail/CoreFreq/x86_64/intel_reg.h:107: warning: "MSR_IA32_FRED_RSP3" redefined
  107 | #define MSR_IA32_FRED_RSP3                      0x000001cf
      | 
./arch/x86/include/asm/msr-index.h:43: note: this is the location of the previous definition
   43 | #define MSR_IA32_FRED_RSP3      0x1cf                   /* Level 3 stack pointer */
      | 
  LD [M]  /home/bunkbail/CoreFreq/build/corefreqk.o
  MODPOST /home/bunkbail/CoreFreq/build/Module.symvers
ERROR: modpost: "cppc_get_perf_ctrs" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "processors" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "x86_spec_ctrl_base" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "__clocksource_register_scale" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cpufreq_unregister_governor" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "clocksource_unregister" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_get_epp_perf" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_get_desired_perf" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cpuidle_get_driver" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_set_enable" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
WARNING: modpost: suppressed 4 unresolved symbol warnings because there were too many)
make[3]: *** [scripts/Makefile.modpost:145: /home/bunkbail/CoreFreq/build/Module.symvers] Error 1
make[2]: *** [/usr/src/linux-headers-6.9.0-x64v3-bore-ext/Makefile:2029: modpost] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
make: *** [Makefile:102: all] Error 2
```

cyring, 2024-05-13:
Hello,
Thanks for the issue.

Commit a4dbfc9f82a46481d1da24f74653961a01cde3cc in branch [`develop`](https://github.com/cyring/CoreFreq/tree/develop) should fix compilation. 
Can you please give it a try ?

bunkbail, 2024-05-13:
```
make[1]: Entering directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
make[1]: Leaving directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreqd.c -o build/corefreqd.o
cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreqm.c -o build/corefreqm.o
cc  -o build/corefreqd \
  build/corefreqd.o build/corefreqm.o -lpthread -lm -lrt -lc
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -c x86_64/corefreq-cli.c -o build/corefreq-cli.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-ui.c -o build/corefreq-ui.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -c x86_64/corefreq-cli-rsc.c -o build/corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-cli-json.c -o build/corefreq-cli-json.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-cli-extra.c -o build/corefreq-cli-extra.o
cc  -o build/corefreq-cli \
  build/corefreq-cli.o build/corefreq-ui.o \
  build/corefreq-cli-rsc.o build/corefreq-cli-json.o \
  build/corefreq-cli-extra.o -lm -lrt -lc
make[1]: Entering directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
  CC [M]  /home/bunkbail/CoreFreq/build/module/corefreqk.o
  LD [M]  /home/bunkbail/CoreFreq/build/corefreqk.o
  MODPOST /home/bunkbail/CoreFreq/build/Module.symvers
ERROR: modpost: "cppc_get_perf_ctrs" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "processors" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "x86_spec_ctrl_base" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "__clocksource_register_scale" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cpufreq_unregister_governor" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "clocksource_unregister" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_get_epp_perf" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_get_desired_perf" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cpuidle_get_driver" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_set_enable" [/home/bunkbail/CoreFreq/build/corefreqk.ko] undefined!
WARNING: modpost: suppressed 4 unresolved symbol warnings because there were too many)
make[3]: *** [scripts/Makefile.modpost:145: /home/bunkbail/CoreFreq/build/Module.symvers] Error 1
make[2]: *** [/usr/src/linux-headers-6.9.0-x64v3-bore-ext/Makefile:2029: modpost] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.9.0-x64v3-bore-ext'
make: *** [Makefile:102: all] Error 2
```

bunkbail, 2024-05-13:
The output is shorter now but still errors out.

cyring, 2024-05-13:
All these API functions, since `6.9` !
Can't be the mainstream kernel ?

bunkbail, 2024-05-13:
It's linux-xanmod. So that means xanmod disables these API functions? Pretty weird since Corefreq used to be working fine on kernel 6.8 (till the last update of 6.8.9. I updated to the latest stable of 6.9 then this happens.

bunkbail, 2024-05-13:
Oh it's probably due to [sched-ext](https://github.com/sched-ext/sched_ext) since I use sched-ext patchset as well. They mentioned about implementing cpufreq related feature on kernel 6.9.

LethalManBoob, 2024-05-13:
Hi. Latest cachyos-lto 6.9 kernel here with sched-ex enabled. its not broken. it just "as i pointed out in the deleted issue" breaks every time you uninstall or reinstall a kernel

bunkbail, 2024-05-13:
I don't use any daemon/dkms modules. I compile directly from the git repo. Does Corefreq require a specific config in the kernel to be enabled?

LethalManBoob, 2024-05-13:
> I don't use any daemon/dkms modules. I compile directly from the git repo. Does Corefreq require a specific config in the kernel to be enabled?

I deleted my comics. Turns out while the client doesn't work until you do the steps I mentioned. It's overall still does not work on 6.9. it is a shame that he hasn't fixed the issue I found, however, the issue being that the client breaks when you install or uninstall a new kernel

bunkbail, 2024-05-13:
I don't think your issue has anything to do with Corefreq specifically. I think you should report that to who ever packages the dkms module you're using. Sounds like the pacman hook got misconfigured or something. My error is that the kernel module itself errored out during compilation.

LethalManBoob, 2024-05-13:
Alright ill give that a go thanks

cyring, 2024-05-14:
Which one of the xanmod package is used in the [AUR](https://aur.archlinux.org/packages?K=xanmod) and which procedure or `.config` to build kernel ?

bunkbail, 2024-05-14:
I built the kernel myself, from the git repo [branch 6.9](https://github.com/xanmod/linux/tree/6.9). I use the [0001-sched-ext.patch](https://github.com/CachyOS/kernel-patches/blob/master/6.9/sched/0001-sched-ext.patch) from here and this is the [config file](https://gist.github.com/bunkbail/bc06607dfbc5794c79fba0af0a684fd1) I use to build it.

cyring, 2024-05-14:
> I built the kernel myself, from the git repo [branch 6.9](https://github.com/xanmod/linux/tree/6.9). I use the [...]

Thanks. It will take sometimes to reproduce your environment and track the missing _CoreFreq_ prerequisites 

Meanwhile I will suggest you rename the title issue to inform that it is a **patched** 6.9 kernel

bunkbail, 2024-05-14:
Done. I'll also try to use other configurations and try to disect if possible.

bunkbail, 2024-05-14:
Okay I've done bisecting the config file and found the culprit. I had `CONFIG_TRIM_UNUSED_KSYMS=y` and you shouldn't be using that if you want to build out-of-tree modules. It's an oversight on my part but you probably want to add it into the README.

closed 2024-05-14 by bunkbail

cyring, 2024-05-15:
Hello

Fyi, _CoreFreq_ is running OK with freshly built `6.9` Arch Linux

[image]

I still have to experiment `CONFIG_TRIM_UNUSED_KSYMS` ...

Thanks for this tip

cyring, 2024-05-15:
Now building for `TRIM_UNUSED_KSYMS`

[image]

Rebooting ...

cyring, 2024-05-15:
Results are:
* when module previously built for `6.9` without `TRIM_UNUSED_KSYMS`
```sh
insmod corefreqk.ko
```
```
insmod: ERROR: could not insert module corefreqk.ko: Unknown symbol in module
```
* After rebuilding _CoreFreq_
```sh
make -j
```
```
make[1]: Entering directory '/usr/lib/modules/6.9.0-arch1-1-latest/build'
  MODPOST CoreFreq/build/Module.symvers
ERROR: modpost: "cppc_get_perf_ctrs" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "x86_spec_ctrl_base" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "__clocksource_register_scale" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cpufreq_unregister_governor" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "clocksource_unregister" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_get_epp_perf" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_get_desired_perf" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cpuidle_get_driver" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_set_enable" [CoreFreq/build/corefreqk.ko] undefined!
ERROR: modpost: "cppc_set_perf" [CoreFreq/build/corefreqk.ko] undefined!
WARNING: modpost: suppressed 3 unresolved symbol warnings because there were too many)
make[3]: *** [scripts/Makefile.modpost:145: CoreFreq/build/Module.symvers] Error 1
make[2]: *** [/usr/lib/modules/6.9.0-arch1-1-latest/build/Makefile:1871: modpost] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.9.0-arch1-1-latest/build'
make: *** [Makefile:102: all] Error 2
```
```sh
zgrep TRIM_UNUSED_KSYMS /proc/config.gz 
```
```
CONFIG_TRIM_UNUSED_KSYMS=y
```

----

It is now documented in `README.md` since commit 6caa79d69ce49885705db59b6dde650b44c13cb1