cyring/CoreFreq #488: [Intel: Gen 12,13,14] Need your Memory Controller data to debug decoder

cyring, 2024-05-19:
Decoder has reporting issues among DDR4 and DDR5, two or four DIMMs, 32 or 64 bits per channel

Based on **latest version** `1.97.2`; Owners of Intel Processors of 12th, 13th, 14th and above, please post the output of:

```
corefreq-cli -s -n -m -n -M -n -k -n -B
```

cyring, 2024-05-19:
@BugReporterZ: #486 `12th Gen Intel(R) Core(TM) i7-12700K`
```
Processor                                 [12th Gen Intel(R) Core(TM) i7-12700K]
|- Architecture                                                     [Alder Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000035]
|- Signature                                                           [  06_97]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 20/ 20]
|- Base Clock                                                          [100.265]
|- Frequency            (MHz)                      Ratio                        
                 Min    802.12                    <   8 >                       
                 Max   3609.53                    <  36 >                       
|- Factory                                                             [100.000]
                       3600                       [  36 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   1102.91                    <  11 >                       
   |- HWP                                                                       
                 Min    802.13                    <   8 >                       
                 Max   6717.74                    <  67 >                       
                 TGT      AUTO                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   5213.77                    <  52 >                       
                  2C   5213.77                    <  52 >                       
                  3C   5213.77                    <  52 >                       
                  4C   5213.77                    <  52 >                       
                  5C   5213.77                    <  52 >                       
                  6C   5213.77                    <  52 >                       
                  7C   5213.77                    <  52 >                       
                  8C   5213.77                    <  52 >                       
|- Hybrid                                                              [ UNLOCK]
                  1C   4010.63                    <  40 >                       
                  2C   4010.63                    <  40 >                       
                  3C   4010.63                    <  40 >                       
                  4C   4010.63                    <  40 >                       
                  5C   4010.63                    <  40 >                       
                  6C   4010.63                    <  40 >                       
                  7C   4010.63                    <  40 >                       
                  8C   4010.63                    <  40 >                       
|- Uncore                                                              [ UNLOCK]
                 Min    802.13                    <   8 >                       
                 Max   4612.23                    <  46 >                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3609.53                    [  36 ]                       
               Turbo      AUTO                    <   0 >                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [Y] AVX-VNN-INT8 [N] AVX-NE-CONV [N] 
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [Y]   MOVDIR64B [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- ENQCMD       [N]         GFNI [Y]        OSPKE [Y]     WAITPKG [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [Y]      SYSCALL [Y]        RDPID [Y]         SGX [N] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
[... 329 lines cut ...]
[ 2] 03/29/2024                                                                 
[... cut ...]
[19] Corsair                                                                    
[20] Corsair                                                                    
[21] Corsair                                                                    
[22] CMK32GX4M2D3600C18                                                         
[23] CMK32GX4M2D3600C18                                                         
[24] CMK32GX4M2D3600C18                                                         
[25] CMK32GX4M2D3600C18 
```

cyring, 2024-05-25:
In branch [`develop`](https://github.com/cyring/CoreFreq/tree/develop), commit c376448200f46da0444be3f26d93351a13367137 brings a decoder of the **Meteor Lake** memory controller

Decoder has been programmed from the Core Ultra Processor [datasheets](https://www.intel.com/content/www/us/en/products/docs/processors/core/core-technical-resources.html)

I will appreciate if anyone could show me the result, either `corefreq-cli -M` or UI screenshot

BugReporterZ, 2024-05-25:
Not sure if that was meant for Meteor Lake users, but on my own I get this:

```
./build/corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                  
```

cyring, 2024-05-25:
@BugReporterZ thanks;  it was indeed meant for Meteor Lake 

I also want to debug previous generations IMC; your output helps a lot but **I need other combinations** of DDR4, DDR5, 32 or 64 bits DIMM, LP-DDR and so on to solve my decoders.

cyring, 2024-05-28:
Source: [Intel](https://edc.intel.com/content/www/us/en/design/publications/14th-generation-core-processors-cfg-and-mem-registers/d0-f0-host-bridge-and-dram-controller-mchbar-memory-controller-part-2-registers/)

[image]

From `MADCH.DDR_TYPE` we can tell if `LPDDR5` or `DDR5` is set up with [ADL](https://github.com/cyring/CoreFreq/blob/4705b327404ff96fa00736fdad571e4ec5381268/x86_64/corefreqk.c#L6430) and compatible but also [MTL](https://github.com/cyring/CoreFreq/blob/4705b327404ff96fa00736fdad571e4ec5381268/x86_64/corefreqk.c#L6489)

----

Still no clue from register [MAD_DIMM](https://edc.intel.com/content/www/us/en/design/publications/14th-generation-core-processors-cfg-and-mem-registers/channel-0-dimm-characteristics-mad-dimm-ch0-0-0-0-mchbar-offset-d80c/) how to decode the new DIMM kind of 24 or 48 GB size with [ADL](https://github.com/cyring/CoreFreq/blob/4705b327404ff96fa00736fdad571e4ec5381268/x86_64/corefreqd.c#L5684) or [MTL](https://github.com/cyring/CoreFreq/blob/4705b327404ff96fa00736fdad571e4ec5381268/x86_64/corefreqd.c#L6176)

cyring, 2024-05-29:
As shown in MTL datasheet vol 1

[image]

paulzzh, 2024-07-23:
```
root@pve:~# corefreq-cli -s -n -m -n -M -n -k -n -B
Processor                                          [Intel(R) Core(TM) i7-14700K]
|- Architecture                                                    [Raptor Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x0000011d]
|- Signature                                                           [  06_B7]
|- Stepping                                                            [      1]
|- Online CPU                                                          [ 20/ 20]
|- Base Clock                                                          [100.515]
|- Frequency            (MHz)                      Ratio
                 Min    804.12                    <   8 >
                 Max   3417.52                    <  34 >
|- Factory                                                             [100.000]
                       3400                       [  34 ]
|- Performance
                 TGT   7036.07                    <  70 >
   |- HWP
                 Min   4322.21                    <  43 >
                 Max   7036.07                    <  70 >
                 TGT      AUTO                    <   0 >
|- Turbo Boost                                                         [ UNLOCK]
                  1C   5628.85                    <  56 >
                  2C   5628.85                    <  56 >
                  3C   5528.34                    <  55 >
                  4C   5528.34                    <  55 >
                  5C   5528.34                    <  55 >
                  6C   5528.34                    <  55 >
                  7C   5528.34                    <  55 >
                  8C   5528.34                    <  55 >
|- Hybrid                                                              [ UNLOCK]
                  1C   4322.17                    <  43 >
|- Uncore                                                              [ UNLOCK]
                 Min    804.12                    <   8 >
                 Max   5025.78                    <  50 >
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3417.52                    [  34 ]
               Turbo      AUTO                    <   0 >

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N]
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N]
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N]
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N]
|- AVX512-BF16  [N] AVX-VNNI-VEX [Y] AVX-VNN-INT8 [N] AVX-NE-CONV [N]
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [Y]   MOVDIR64B [Y]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- ENQCMD       [N]         GFNI [Y]        OSPKE [Y]     WAITPKG [Y]
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y]
|- SERIALIZE    [Y]      SYSCALL [Y]        RDPID [Y]         SGX [N]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
[... 287 lines cut ...]
[ 2] 09/06/2023
[ 3] Micro-Star International Co., Ltd.
[ 4] MS-7E07
[ 5] 4.0
[ 6] D---u---s---n-
[ 7] Default string
[ 8] Default string
[ 9] Micro-Star International Co., Ltd.
[10] PRO Z790-A MAX WIFI (MS-7E07)
[11] 4.0
[12] 0---7---N---4---5-
[13] Number Of Devices:4\Maximum Capacity:134217728 kilobytes
[14]
[15] Controller0-DIMMA2\BANK 0
[16]
[17] Controller1-DIMMB2\BANK 0
[18]
[19] 0x0B92
[20]
[21] 0x0B92
[22]
[23]
[24]
[25]
```

cyring, 2024-07-23:
@paulzzh Thank you for your Raptor output 
I don't have its Memory Controller device identifier.
Can you please post the output of the following command:
```
lspci -nn
```

paulzzh, 2024-07-23:
> @paulzzh Thank you for your Raptor output I don't have its Memory Controller device identifier. Can you please post the output of the following command:
> 
> ```
> lspci -nn
> ```

```
root@pve:~# lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:a740] (rev 01)
00:02.0 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:02.1 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:02.2 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:02.3 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:02.4 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:02.5 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:02.6 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:02.7 VGA compatible controller [0300]: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] [8086:a780] (rev 04)
00:06.0 PCI bridge [0604]: Intel Corporation Raptor Lake PCIe 4.0 Graphics Port [8086:a74d] (rev 01)
00:08.0 System peripheral [0880]: Intel Corporation GNA Scoring Accelerator module [8086:a74f] (rev 01)
00:0a.0 Signal processing controller [1180]: Intel Corporation Raptor Lake Crashlog and Telemetry [8086:a77d] (rev 01)
00:14.0 USB controller [0c03]: Intel Corporation Raptor Lake USB 3.2 Gen 2x2 (20 Gb/s) XHCI Host Controller [8086:7a60] (rev 11)
00:14.2 RAM memory [0500]: Intel Corporation Raptor Lake-S PCH Shared SRAM [8086:7a27] (rev 11)
00:16.0 Communication controller [0780]: Intel Corporation Raptor Lake CSME HECI [8086:7a68] (rev 11)
00:17.0 SATA controller [0106]: Intel Corporation Raptor Lake SATA AHCI Controller [8086:7a62] (rev 11)
00:1c.0 PCI bridge [0604]: Intel Corporation Raptor Lake PCI Express Root Port [8086:7a38] (rev 11)
00:1c.2 PCI bridge [0604]: Intel Corporation Device [8086:7a3a] (rev 11)
00:1c.3 PCI bridge [0604]: Intel Corporation Raptor Lake PCI Express Root Port [8086:7a3b] (rev 11)
00:1d.0 PCI bridge [0604]: Intel Corporation Raptor Lake PCI Express Root Port [8086:7a30] (rev 11)
00:1f.0 ISA bridge [0601]: Intel Corporation Raptor Lake LPC/eSPI Controller [8086:7a04] (rev 11)
00:1f.3 Audio device [0403]: Intel Corporation Raptor Lake High Definition Audio Controller [8086:7a50] (rev 11)
00:1f.4 SMBus [0c05]: Intel Corporation Raptor Lake-S PCH SMBus Controller [8086:7a23] (rev 11)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Raptor Lake SPI (flash) Controller [8086:7a24] (rev 11)
01:00.0 Non-Volatile memory controller [0108]: Intel Corporation NVMe Optane Memory Series [8086:2522]
03:00.0 Network controller [0280]: Qualcomm Technologies, Inc WCN785x Wi-Fi 7(802.11be) 320MHz 2x2 [FastConnect 7800] [17cb:1107] (rev 01)
04:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V [8086:15f3] (rev 03)
05:00.0 Non-Volatile memory controller [0108]: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1602 (DRAM-less) [1e4b:1602] (rev 01)
```

cyring, 2024-07-23:
@paulzzh 

Your `Host bridge [8086:a740]` has been added to probe the Memory Controller
You can now pull the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch; rebuild and reload _CoreFreq_ (especially its driver `corefreqk.ko`)
Can you then post the output of `corefreq-cli -M`

paulzzh, 2024-07-23:
> @paulzzh
> 
> Your `Host bridge [8086:a740]` has been added to probe the Memory Controller You can now pull the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch; rebuild and reload _CoreFreq_ (especially its driver `corefreqk.ko`) Can you then post the output of `corefreq-cli -s -n -M` and tell how results match with your setup ?

It seems to work.
[image]

```
root@pve:~# corefreq-cli -s -n -M
Processor                                          [Intel(R) Core(TM) i7-14700K]
|- Architecture                                                    [Raptor Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000123]
|- Signature                                                           [  06_B7]
|- Stepping                                                            [      1]
|- Online CPU                                                          [ 20/ 20]
|- Base Clock                                                          [100.509]
|- Frequency            (MHz)                      Ratio
                 Min    804.07                    <   8 >
                 Max   3417.30                    <  34 >
|- Factory                                                             [100.000]
                       3400                       [  34 ]
|- Performance
                 TGT   7236.64                    <  72 >
   |- HWP
                 Min   4321.92                    <  43 >
                 Max   7236.64                    <  72 >
                 TGT      AUTO                    <   0 >
|- Turbo Boost                                                         [ UNLOCK]
                  1C   5628.49                    <  56 >
                  2C   5628.49                    <  56 >
                  3C   5527.99                    <  55 >
                  4C   5527.99                    <  55 >
                  5C   5527.99                    <  55 >
                  6C   5527.99                    <  55 >
                  7C   5527.99                    <  55 >
                  8C   5527.99                    <  55 >
|- Hybrid                                                              [ UNLOCK]
                  1C   4321.97                    <  43 >
|- Uncore                                                              [ UNLOCK]
                 Min    804.09                    <   8 >
                 Max   5025.55                    <  50 >
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3417.30                    [  34 ]
               Turbo      AUTO                    <   0 >

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N]
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N]
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N]
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N]
|- AVX512-BF16  [N] AVX-VNNI-VEX [Y] AVX-VNN-INT8 [N] AVX-NE-CONV [N]
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [Y]   MOVDIR64B [Y]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- ENQCMD       [N]         GFNI [Y]        OSPKE [Y]     WAITPKG [Y]
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y]
|- SERIALIZE    [Y]      SYSCALL [Y]        RDPID [Y]         SGX [N]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
[... 244 lines cut ...]

                             Intel Z790  [7A04]
Controller #0                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32  100   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32  100   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768

```

paulzzh, 2024-07-23:
Perfect.
[image]
[image]

cyring, 2024-07-23:
> Where is `Controller #1 DIMMB2` ?

It did however with AlderLake & DDR4

https://gist.github.com/cyring/f15646816a47a04f8b8f9c3a4cada24c?permalink_comment_id=4454592#gistcomment-4454592

That's why this issue remains opened: I wish new developers could work straight on HW and finalize those IMC decoders.

----

Also notice that Virtualization section is now showing up in "Technologies"

cyring, 2024-07-30:
@BugReporterZ Hello,

Can you please pull the latest commits from the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and show the output of `corefreq-cli -k -n -B -n -M` ?

CC: @paulzzh

BugReporterZ, 2024-07-30:
> @BugReporterZ Hello,
> 
> Can you please pull the latest commits from the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and show the output of `corefreq-cli -k -n -B -n -M` ?

I got this:

```
./corefreq-cli -k -n -B -n -M
Linux:                                                                          
|- Release                                                     [6.9.9-1-default]
|- Version       [#1 SMP PREEMPT_DYNAMIC Thu Jul 11 11:31:54 UTC 2024 (8c0f797)]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         65530220 KB
|- Shared RAM                                                         1691920 KB
|- Free RAM                                                           1244324 KB
|- Buffer RAM                                                             672 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
Clock Source                                                  <             tsc>
CPU-Freq driver                                               [    intel_pstate]
Governor                                                      [         Missing]
CPU-Idle driver                                               [      intel_idle]
|- Idle Limit                                                 [             C10]
   |- State        POLL     C1E      C6      C8     C10                         
   |-           CPUIDLE MWAIT 0 MWAIT 0 MWAIT 0 MWAIT 0                         
   |- Power          -1       0       0       0       0                         
   |- Latency         0       2     220     280     680                         
   |- Residency       0       4     600     800    2000                         

[ 0] American Megatrends International, LLC.                                    
[ 1] 1.H0                                                                       
[ 2] 03/29/2024                                                                 
[ 3] Micro-Star International Co., Ltd.                                         
[ 4] MS-7D25                                                                    
[ 5] 1.0                                                                        
[ 6] D---u---s---n-                                                             
[ 7] Default string                                                             
[ 8] Default string                                                             
[ 9] Micro-Star International Co., Ltd.                                         
[10] PRO Z690-A WIFI DDR4(MS-7D25)                                              
[11] 1.0                                                                        
[12] 0---5---L---0---8-                                                         
[13] Number Of Devices:4\Maximum Capacity:134217728 kilobytes                   
[14] Controller0-DIMMA1\BANK 0                                                  
[15] Controller0-DIMMA2\BANK 0                                                  
[16] Controller1-DIMMB1\BANK 0                                                  
[17] Controller1-DIMMB2\BANK 0                                                  
[18] Corsair                                                                    
[19] Corsair                                                                    
[20] Corsair                                                                    
[21] Corsair                                                                    
[22] CMK32GX4M2D3600C18                                                         
[23] CMK32GX4M2D3600C18                                                         
[24] CMK32GX4M2D3600C18                                                         
[25] CMK32GX4M2D3600C18                                                         

                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
[... 7 lines cut ...]
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                          
```

cyring, 2024-07-30:
> I got this:

Thank you

Can you tell if the `WR` timing of `24` is correct facing its BIOS setting ?

Feel free to post the BIOS screenshot then I can compare all the other timings

paulzzh, 2024-07-30:
> Can you please pull the latest commits from the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and show the output of `corefreq-cli -k -n -B -n -M` ?
> 
> CC: @paulzzh

```
root@pve:~# corefreq-cli -k -n -B -n -M
Linux:                                                                          
|- Release                                                         [6.8.8-4-pve]
|- Version              [#1 SMP PREEMPT_DYNAMIC PMX 6.8.8-4 (2024-07-26T11:15Z)]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         65609688 KB
|- Shared RAM                                                           62128 KB
|- Free RAM                                                          60300304 KB
|- Buffer RAM                                                          131820 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
Clock Source                                                  <             tsc>
CPU-Freq driver                                               [    intel_pstate]
Governor                                                      [         Missing]
CPU-Idle driver                                               [      intel_idle]
|- Idle Limit                                                 [         C2_ACPI]
   |- State        POLL C1_ACPI C2_ACPI                                         
   |-           CPUIDLE ACPI FF ACPI FF                                         
   |- Power          -1       0       0                                         
   |- Latency         0       1     127                                         
   |- Residency       0       1     381                                         

[ 0] American Megatrends International, LLC.                                    
[ 1] M.40                                                                       
[ 2] 04/19/2024                                                                 
[ 3] Micro-Star International Co., Ltd.                                         
[ 4] MS-7E07                                                                    
[ 5] 4.0                                                                        
[ 6] D---u---s---n-                                                             
[ 7] Default string                                                             
[ 8] Default string                                                             
[ 9] Micro-Star International Co., Ltd.                                         
[10] PRO Z790-A MAX WIFI (MS-7E07)                                              
[11] 4.0                                                                        
[12] 0---7---N---4---5-                                                         
[13] Number Of Devices:4\Maximum Capacity:134217728 kilobytes                   
[14]                                                                            
[15] Controller0-DIMMA2\BANK 0                                                  
[16]                                                                            
[17] Controller1-DIMMB2\BANK 0                                                  
[18]                                                                            
[19] 0x0B92                                                                     
[20]                                                                            
[21] 0x0B92                                                                     
[22]                                                                            
[23]                                                                            
[24]                                                                            
[25]                                                                            

                             Intel Z790  [7A04]                            
Controller #0                                                Dual Channel  
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32  100   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32  100   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768
```

BugReporterZ, 2024-07-30:
> > I got this:
> 
> Thank you
> 
> Can you tell if the `WR` timing of `24` is correct facing its BIOS setting ?
> 
> Feel free to post the BIOS screenshot then I can compare all the other timings

[image]

cyring, 2024-07-30:
@BugReporterZ @paulzzh 

My `tWR` formula is working with DDR4 but not with DDR5

```math
tWR = tWRPRE - tCW - 4
```

Source code:
https://github.com/cyring/CoreFreq/blob/0db6f0a18d3b03a26287fd30d27bb3fd24a2a3c2/x86_64/corefreqd.c#L5623

----

According to 13th and 14th [datasheet](https://www.intel.com/content/www/us/en/content-details/743846/13th-generation-intel-core-processors-datasheet-volume-2-of-2.html), `tWRPRE` is expressed in `tCK` delay but in `WCK` for LPDDR5

[image]

cyring, 2024-07-30:
I'm reading [this](https://www.overclock.net/posts/29089330/)
```math
tWRPRE = tWR + tCWL + 8
```

----

@paulzzh @BugReporterZ Can you please pull latest commit 36adf0e855ff485a1f219c2927fe924fd168eb6d and post `corefreq-cli -M`

BugReporterZ, 2024-07-30:
After pulling and recompiling:

```
./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  

```

paulzzh, 2024-07-30:
WR 100 -> 96
```
root@pve:~# corefreq-cli -M
                             Intel Z790  [7A04]                            
Controller #0                                                Dual Channel  
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768
```

cyring, 2024-07-30:
> After pulling and recompiling:

Great; no regression on your `tWR = 24`

----

I still have no clue how to handle the second channel timings

cyring, 2024-07-31:
@paulzzh Hello,
The change below is an attempt to fix the second memory controller.

Can you edit code at this line:
https://github.com/cyring/CoreFreq/blob/36adf0e855ff485a1f219c2927fe924fd168eb6d/x86_64/corefreqk.c#L6579

And replace function `ADL_IMC` with this one:
```c
static PCI_CALLBACK ADL_IMC(struct pci_dev *dev)
{	/* Source: 12th Generation Intel Core Processors datasheet, vol 2 */
	PCI_CALLBACK rc = 0;
	unsigned short mc, cha;

	PUBLIC(RO(Proc))->Uncore.CtrlCount = 0;
	/* MCHBAR matches bits 41 to 17 ; two MC x 64KB memory space	*/
  for (mc = 0; mc < MC_MAX_CTRL; mc++)
  {
	rc = ADL_HOST(dev, Query_ADL_IMC, 0x10000, mc);

   if ( (PCI_CALLBACK) 0 == rc)
   {
    if (PUBLIC(RO(Proc))->Uncore.MC[mc].ADL.MADCH.value != 0xffffffff) {
     switch (PUBLIC(RO(Proc))->Uncore.MC[mc].ADL.MADCH.DDR_TYPE) {
     case 0b01:	/*	DDR5	*/
     case 0b10:	/*	LPDDR5	*/
/*    if (mc & 1) {
	PUBLIC(RO(Proc))->Uncore.MC[mc].ChannelCount = 0;
      } else */{
       for (cha = 0; cha < PUBLIC(RO(Proc))->Uncore.MC[mc].ChannelCount; cha++)
       {
	 PUBLIC(RO(Proc))->Uncore.MC[mc].Channel[cha].ADL.Sched.ReservedBits1=0;
       }
      }
	break;
     }
    }
    if (PUBLIC(RO(Proc))->Uncore.MC[mc].ChannelCount > 0)
    {
	PUBLIC(RO(Proc))->Uncore.CtrlCount = mc + 1;
    }
   }
  }
	return rc;
}
```

Then rebuild and reload _CoreFreq_ and please post the output of `corefreq-cli -M`

paulzzh, 2024-07-31:
> Then rebuild and reload _CoreFreq_ and please post the output of `corefreq-cli -M`

```
root@pve:~/CoreFreq# corefreq-cli -M
                             Intel Z790  [7A04]
Controller #0                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768

Controller #1                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768

```

cyring, 2024-07-31:
@BugReporterZ Can you please show me what you get with the code change [above](https://github.com/cyring/CoreFreq/issues/488#issuecomment-2259398334) ?

BugReporterZ, 2024-07-31:
I'm getting this.

```
./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                
```

cyring, 2024-07-31:
> I'm getting this.

Thank you.
So for your DDR4 case, code change does not make any difference

Issue happens with DDR5 where DIMMs are duplicated among channel 0 and 1 (@paulzzh)

cyring, 2024-07-31:
[image]

cyring, 2024-07-31:
@BugReporterZ I'm trying fix the second channel with garbage values

Can you pull latest `develop` branch then change this block code of code:
https://github.com/cyring/CoreFreq/blob/fdb74e7c1f4a164bd106d1d0140c9c44a773cf1b/x86_64/corefreqk.c#L5568

with this code
```c
	/*	Check for 2 DIMMs Per Channel is enabled		*/
    if (PUBLIC(RO(Proc))->Uncore.Bus.ADL_Cap_A.DDPCD == 0) {
	switch (PUBLIC(RO(Proc))->Uncore.MC[mc].ADL.MADCH.DDR_TYPE) {
	case 0b00:	/*	DDR4	*/
		PUBLIC(RO(Proc))->Uncore.MC[mc].ChannelCount = 1;
		break;
	case 0b11:	/*	LPDDR4	*/
	case 0b01:	/*	DDR5	*/
	case 0b10:	/*	LPDDR5	*/
	default:
		PUBLIC(RO(Proc))->Uncore.MC[mc].ChannelCount = 2;
		break;
	}
    } else {
```

Rebuild, reload and please post `corefreq-cli -M`

BugReporterZ, 2024-07-31:
It changed like this:

```
./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                               Single Channel 
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
                                                                           
Controller #1                                               Single Channel 
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
```

cyring, 2024-07-31:
> It changed like this:

Thanks,
Does it look what you see in your BIOS ?

BugReporterZ, 2024-07-31:
> Thanks, Does it look what you see in your BIOS ?
Seems like so, or at least it matches the composite screenshot I uploaded earlier.

cyring, 2024-07-31:
> > Thanks, Does it look what you see in your BIOS ?
> 
> Seems like so, or at least it matches the composite screenshot I uploaded earlier.

I believe your BIOS says Dual Channel rather than Single ?

BugReporterZ, 2024-07-31:
Yes, my RAM is configured to be in dual channel mode, with 4 DIMM.

cyring, 2024-07-31:
> Yes, my RAM is configured to be in dual channel mode, with 4 DIMM. I was checking out the timings and didn't notice that it changed to "Single Channel" in CoreFreq.

What do you think if we are showing zeros instead of garbage but keep saying Dual Channel ?

Please have a try to this archive and post `corefreq-cli -M`
[CoreFreq_develop.tar.gz](https://github.com/user-attachments/files/16441593/CoreFreq_develop.tar.gz)

BugReporterZ, 2024-07-31:
> @BugReporterZ
> 
> Sorry for all these changes; I found a better way to aggregate timings Please try this archive [CoreFreq_develop.tar.gz](https://github.com/user-attachments/files/16441815/CoreFreq_develop.tar.gz)

```
./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7218 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   0T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7218 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   0T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
```

Perhaps showing `---` instead of `0` would be clearer, like this or similar:

```
./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7218 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1    --   --   --   --   --   --   --   --   --   --   --   --   --   --
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1    --   --   --   --        --   --   --   --        --   --   --   --
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1   --
  #1    --   --   --   --                  --   --   --   --   --   --   --
                                                                           
[... 6 lines cut ...]
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7218 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1    --   --   --   --   --   --   --   --   --   --   --   --   --   --
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
```

paulzzh, 2024-07-31:
> Commit [a049516](https://github.com/cyring/CoreFreq/commit/a0495167073818574b722394ad654fb69d5bc5de) pushed

branch develop a049516
```
root@pve:~/CoreFreq# corefreq-cli -M
                             Intel Z790  [7A04]
Controller #0                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768

Controller #1                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
```

cyring, 2024-07-31:
> branch develop [a049516](https://github.com/cyring/CoreFreq/commit/a0495167073818574b722394ad654fb69d5bc5de)

Thank you.
I think timings data are duplicated because DDR5 is dual channel 32-bits on two DIMMs
Decoder gate confused

cyring, 2024-08-01:
@BugReporterZ @paulzzh Hi,

For your MC testings commit 0516c27578a792700f1e2e646fc6b147009e9b20 is pushed on the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch

paulzzh, 2024-08-01:
> For your MC testings commit [0516c27](https://github.com/cyring/CoreFreq/commit/0516c27578a792700f1e2e646fc6b147009e9b20) is pushed on the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch

```
root@pve:~/CoreFreq# corefreq-cli -M
                             Intel Z790  [7A04]
Controller #0                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768

Controller #1                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    16    2    131072      1024          32768

```

cyring, 2024-08-01:
It's a manual simulation of what would happen if the DIMM numbers designated within the MC Registers are in fact the channel numbers.

We are changing the topology for two MC, two channels, **but one DIMM per channel**

----

# Raptor Lake with DDR5
* Current layout returned by CoreFreq

MC|Cha|DIMM|Size
--|---|----|----
0 |0  |0   |
0 |0  |1 * |32GB
0 |1  |0   |
0 |1  |1 + |32GB
1 |0  |0   |
1 |0  |1 **|32GB
1 |1  |0   |
1 |1  |1 ++|32GB

----

* Simulated topology
 > DIMM mapping is presumed to be the Channel mapping
 > And when two registers duplicate the same info, the last is overwriting the layout

MC|Cha |DIMM|Size
--|----|----|----
0 |0   |0   |
0 |1 + |0   |32GB
1 |0   |0   |
1 |1 ++|0   |32GB

* It would result as below

```
                             Intel Z790  [7A04]
Controller #0                                                Dual Channel
...
 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    2    131072      1024          32768

Controller #1                                                Dual Channel
...
 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    2    131072      1024          32768
```

# Alder Lake with DDR4
* Current layout returned by CoreFreq

MC|Cha|DIMM|Size
--|---|----|----
0 |0  |0 * |16GB
0 |0  |1 **|16GB
0 |1  |0   |
0 |1  |1   |
1 |0  |0 + |16GB
1 |0  |1 ++|16GB
1 |1  |0   |
1 |1  |1   |

----

* Simulated topology

MC|Cha |DIMM|Size
--|----|----|----
0 |0 * |0   |16GB
0 |1 **|0   |16GB
1 |0 + |0   |16GB
1 |1 ++|0   |16GB

```
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
...
 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
                                                                           
Controller #1                                                Dual Channel  
...
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
```

cyring, 2024-08-01:
@Technologicat (i7-12700H)
@SimonFair (12600K)
@ich777 (i7-12700K)
@rushvora (i9-12900K)
@justanerd (i9-13900K)

Hello,

Could you please pull the latest [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and post the output of `corefreq-cli -M`

cyring, 2024-08-02:
@paulzzh @BugReporterZ Hello,

Can you pull the branch [`Intel_12_13_14_IMC`](https://github.com/cyring/CoreFreq/tree/Intel_12_13_14_IMC) where commit 6401e7d2bc35d64ef01d5e1eec7eb7c0b3c8d0bf attempts a [different]((https://github.com/cyring/CoreFreq/issues/488#issuecomment-2264163671)) MC decoding ?

BugReporterZ, 2024-08-02:
I'm getting this:

```
./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     5    8    8    8   28    4    4   16   14    6   24    6    4   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     4    4    4    4                4100  180    0    4    1    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                              
```

cyring, 2024-08-02:
> I'm getting this:

Thank you.
I've an idea what's going on but I'm waiting @paulzzh DDR5 output before changing code

Technologicat, 2024-08-02:
@cyring Here's my output (DDR5):

```
                          Intel ADL PCH-P  [5182]                          
Controller #0                                                Dual Channel  
 Bus Rate  2400 MHz       Bus Speed 2389 MHz           DDR5 Speed 4779 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    40   40   40   40   76    8   12   32   71   17  117   38   18   2T
  #1    40   40   40   40   76    8   12   32   71   17  117   38   18   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    12    8   14   14        18   18   20   22        72   54   12   12
  #1    12    8   14   14        18   18   20   22        72   54   12   12
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    26    8   14   14                4680  383  706   18   12    2    0
  #1    26    8   14   14                4680  383  706   18   12    2    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    1    131072      1024          16384      PSD516G480081S
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  2400 MHz       Bus Speed 2389 MHz           DDR5 Speed 4779 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    40   40   40   40   76    8   12   32   71   17  117   38   18   2T
  #1    40   40   40   40   76    8   12   32   71   17  117   38   18   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    12    8   14   14        18   18   20   22        72   54   12   12
  #1    12    8   14   14        18   18   20   22        72   54   12   12
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    26    8   14   14                4680  383  706   18   12    2    0
  #1    26    8   14   14                4680  383  706   18   12    2    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    1    131072      1024          16384      PSD516G480081S
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0
```

cyring, 2024-08-02:
> **EDIT**: Oops, that's with `Intel_12_13_14_IMC`, not `develop`. Ping me if you want the DDR5 output from `develop`, too.

Thank you.
In fact, results based `Intel_12_13_14_IMC` branch are more important to me.
Compared to your [previous output](https://gist.github.com/cyring/cf6f2f21c30763dc97e88b8b3d118d7f?permalink_comment_id=4479909#gistcomment-4479909), they show that `i7-12700H` with DDR5 has two enabled Memory Controllers, one DIMM per MC

----

Also noticed that `tWR` is now adjusted.

Technologicat, 2024-08-02:
Great! Feel free to ask again if you need something else. I'm rather busy with other things at the moment, but CoreFreq is a nice system monitoring tool and I'm happy to help with small tests like this when I can.

paulzzh, 2024-08-02:
> branch [`Intel_12_13_14_IMC`](https://github.com/cyring/CoreFreq/tree/Intel_12_13_14_IMC)

```
root@pve:~/CoreFreq# corefreq-cli -M
                             Intel Z790  [7A04]
Controller #0                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6432 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    2    131072      1024          32768

Controller #1                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6432 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    2    131072      1024          32768

```

cyring, 2024-08-02:
@BugReporterZ 

To solve the empty DIMM, can you please trace variables ?

Go to the code at this line (still in branch `Intel_12_13_14_IMC`):
https://github.com/cyring/CoreFreq/blob/6401e7d2bc35d64ef01d5e1eec7eb7c0b3c8d0bf/x86_64/corefreqd.c#L5744

And change statement as below

```c
    case 1 ... 4:
/*TODO*/
	printf("D0: %7d %7d %7d %d\n",
		mc, RO(Proc)->Uncore.MC[mc].ADL.MADC0.Dimm_L_Map,
		DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD0.DLW),
		512 * RO(Proc)->Uncore.MC[mc].ADL.MADD0.Dimm_L_Size );
	printf("D1: %7d %7d %7d %d\n",
		mc, !RO(Proc)->Uncore.MC[mc].ADL.MADC0.Dimm_L_Map,
		DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD0.DSW),
		512 * RO(Proc)->Uncore.MC[mc].ADL.MADD0.Dimm_S_Size );
	printf("D2: %7d %7d %7d %d\n",
		mc, RO(Proc)->Uncore.MC[mc].ADL.MADC1.Dimm_L_Map,
		DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD1.DLW),
		512 * RO(Proc)->Uncore.MC[mc].ADL.MADD1.Dimm_L_Size );
	printf("D3: %7d %7d %7d %d\n",
		mc, !RO(Proc)->Uncore.MC[mc].ADL.MADC1.Dimm_L_Map,
		DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD1.DSW),
		512 * RO(Proc)->Uncore.MC[mc].ADL.MADD1.Dimm_S_Size );
/*TODO*/
	RO(Shm)->Uncore.MC[mc].Channel[
		RO(Proc)->Uncore.MC[mc].ADL.MADC0.Dimm_L_Map
	].DIMM[0].Rows = DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD0.DLW);

	RO(Shm)->Uncore.MC[mc].Channel[
		!RO(Proc)->Uncore.MC[mc].ADL.MADC0.Dimm_L_Map
	].DIMM[0].Rows = DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD0.DSW);

	RO(Shm)->Uncore.MC[mc].Channel[
		RO(Proc)->Uncore.MC[mc].ADL.MADC1.Dimm_L_Map
	].DIMM[0].Rows = DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD1.DLW);

	RO(Shm)->Uncore.MC[mc].Channel[
		!RO(Proc)->Uncore.MC[mc].ADL.MADC1.Dimm_L_Map
	].DIMM[0].Rows = DimmWidthToRows(RO(Proc)->Uncore.MC[mc].ADL.MADD1.DSW);

	RO(Shm)->Uncore.MC[mc].Channel[
		RO(Proc)->Uncore.MC[mc].ADL.MADC0.Dimm_L_Map
	].DIMM[0].Size = 512 * RO(Proc)->Uncore.MC[mc].ADL.MADD0.Dimm_L_Size;

	RO(Shm)->Uncore.MC[mc].Channel[
		!RO(Proc)->Uncore.MC[mc].ADL.MADC0.Dimm_L_Map
	].DIMM[0].Size = 512 * RO(Proc)->Uncore.MC[mc].ADL.MADD0.Dimm_S_Size;

	RO(Shm)->Uncore.MC[mc].Channel[
		RO(Proc)->Uncore.MC[mc].ADL.MADC1.Dimm_L_Map
	].DIMM[0].Size = 512 * RO(Proc)->Uncore.MC[mc].ADL.MADD1.Dimm_L_Size;

	RO(Shm)->Uncore.MC[mc].Channel[
		!RO(Proc)->Uncore.MC[mc].ADL.MADC1.Dimm_L_Map
	].DIMM[0].Size = 512 * RO(Proc)->Uncore.MC[mc].ADL.MADD1.Dimm_S_Size;
	break;
```

In the daemon `corefreqd` output you will find the printed variables

Thank you

BugReporterZ, 2024-08-02:
The daemon outputted this

```
CoreFreq Daemon 1.98.1  Copyright (C) 2015-2024 CYRIL COURTIAT
D0:       0       0   65536 16384
D1:       0       1   65536 16384
D2:       0       0   65536 0
D3:       0       1   65536 0
D0:       1       0   65536 16384
D1:       1       1   65536 16384
D2:       1       0   65536 0
D3:       1       1   65536 0
```

cyring, 2024-08-02:
> The daemon outputted this
>
> ```
> CoreFreq Daemon 1.98.1  Copyright (C) 2015-2024 CYRIL COURTIAT
> D0:       0       0   65536 16384
> D1:       0       1   65536 16384
> D2:       0       0   65536 0 [...]

All of these mean that I should not process a second channel with DDR4 but I have to include a second channel with DDR5

cyring, 2024-08-02:
This new commit 54a044fdfa593237b8d27e1917fe055dd53e90b9 now processes the channels decoding with DDR4 differently than DDR5

@BugReporterZ @paulzzh : can you please show me if the DIMMs layout is ok ?

BugReporterZ, 2024-08-02:
Getting this now, appears to make sense to me.
```
corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   0T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     0    2     65536         0          16384  CMK32GX4M2D3600C18
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   0T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     0    2     65536         0          16384  CMK32GX4M2D3600C18
```

cyring, 2024-08-02:
> Getting this now.

It's tuf to debug why the previous time all DIMM slots were blank
I have to reset that commit

paulzzh, 2024-08-02:
> commit [54a044f](https://github.com/cyring/CoreFreq/commit/54a044fdfa593237b8d27e1917fe055dd53e90b9)

```
root@pve:~# corefreq-cli -M
                             Intel Z790  [7A04]
Controller #0                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    2    131072      1024          32768

Controller #1                                                Dual Channel
 Bus Rate  3200 MHz       Bus Speed 3216 MHz           DDR5 Speed 6433 MT/s

 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
  #1    32   39   39   39   89    8   16   32   96   23  136   32   24   2T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0    16    8   16   16        20   20   22   22        76   52   14   14
  #1    16    8   16   16        20   20   22   22        76   52   14   14
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0    34    8   16   16                6250  510  940   24   16    2    0
  #1    34    8   16   16                6250  510  940   24   16    2    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    2    131072      1024          32768
```

cyring, 2024-08-02:
@BugReporterZ 
Can you try this commit 16148198391e0cff58552aa54c0400fa2f1e5325

BugReporterZ, 2024-08-02:
Result:

```
corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   0T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3609 MHz           DDR4 Speed 7219 MT/s
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    18   22   22   22   42    4    9   40   24   12   46   18    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   0T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   16        36   29    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          16384  CMK32GX4M2D3600C18
```

cyring, 2024-08-02:
> Result:

Thanks.
Does it sound correct to you ?

[16 more replies]

closed 2024-10-02 by cyring