cyring/CoreFreq #496: Max value hits 429C after some time on 7950X

tofurky, 2024-06-22:
After running for a while, several cores will eventually get a bad reading of 429C and stick there until corefreqd is restarted.

Ryzen 7950X w/PBO enabled on ASUS ProArt X670E-Creator, kernel 6.6.32 built with gcc 13.2.0 on Ubuntu 24.04.

CoreFreq 117b8c04ef0ef18f877472c268aadc74dd3cffb4 from Wed Feb 28 02:40:18 2024 +0100. Currently now testing with tip of master, but the issue takes some time to show up and I'll update if/when it does.

System PBO offsets were tuned to undervolt most cores and overvolt 1 for stability. RAM is DDR5-6000 with Infinity Fabric at UCLK/1. UEFI is latest available with AGESA 1.1.7.0. No application crashes or other issues seen in dmesg. This issue has been happening as long as I've been letting corefreqd run continuously, so since last August according to .bash_history.

Unsure if this means there's an occasional bad read from the SMU or there's something else at play? Please let me know if you need any other information, thanks.

Screenshot:
[image]

cyring, 2024-06-22:
Hello,

Thank you for reporting this issue.

Sorry for all details that will follow; I need to simulate the call flow of thermal which starts in this Driver function 

https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/corefreqk.c#L16233

Temperature is indeed gotten from SMU; `TCCD_REGISTER`  address interlaced per CCD

https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/amd_reg.h#L1504

The SMU current temperature `CurTmp` is subtracted by `49` whenever `CurTempRangeSel` bit is set to one

Then thermal data are processed by the Daemon:

https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/corefreqd.c#L1203

and maths start here:

https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/corefreqd.c#L316

Temperature is finally computed by this macro:

https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/coretypes.h#L709

----

Question: do you have any other thermal agent running in parallel of _CoreFreq_ ?
For instance, the kernel module `k10twmp` ?

tofurky, 2024-06-23:
> 
> Question: do you have any other thermal agent running in parallel of _CoreFreq_ ? For instance, the kernel module `k10temp` ?
>

Yep, k10temp is loaded and polled every 5m by munin-node and much more frequently by MATE sensors applet (Tctl only). Possibly there's 2 calls at once sometimes? There have never been super high values reported by k10temp by munin, though.

```
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +51.5°C  
Tccd1:        +39.4°C  
Tccd2:        +38.8°C  
```

The graph tends to squash any one time high readings, but each point below represents 15m (3x 5m readings) I think. System uptime from the previous screenshot showing 429C is 18 days uptime.

[image]

[image]

cyring, 2024-06-23:
There's an access conflict to the SMU between `corefreqk.ko` and `k10temp` which has to be unloaded.

Fyi you can quickly reproduce this conflict by lowering the collect interval to `100ms` in the `corefreq-cli`

tofurky, 2024-06-23:
I haven't been able to reproduce it yet doing the following:

- Running the following to poll k10temp Tctl, Tccd1, Tccd2 as quickly as possible:
`while true; do cat /sys/class/hwmon/hwmon4/temp{1,3,4}_input ; done`
- Running corefreq-cli at 100ms interval

Been running for about 5m now, I'll let it run longer and see if I see the high temps.

cyring, 2024-06-23:
> I haven't been able to reproduce it yet doing the following:
> 
> * Running the following to poll k10temp Tctl, Tccd1, Tccd2 as quickly as possible:
>   `while true; do cat /sys/class/hwmon/hwmon4/temp{1,3,4}_input ; done`
> * Running corefreq-cli at 100ms interval
> 
> Been running for about 5m now, I'll let it run longer and see if I see the high temps.
> 
> Edit: after 45m, one core went to 429C. I'll continue testing without k10temp loaded.

I did not noticed your Issue Edit.

If you confirm no other SMU is running in parallel, it may be possible it is visible since Raphael.

As a day to day, I'm programming on AMD Matisse with particular kernel boot parameters documented in [Wiki](https://github.com/cyring/CoreFreq/wiki/CoreFreq-as-the-Clock-Source,-CPU-Freq-and-CPU-Idle-driver)
Once a while, before releasing especially, I'm doing full non regression tests with most kernel modules loaded but `k10temp` and occasionally `amd_pstate`

tofurky, 2024-06-23:
Ah, sorry about the edit, yeah I didn't want to generate too many e-mails.

I've left it running at 100ms overnight and one core has gone to 429C. k10temp isn't loaded.
```
matt@aquos:~$ lsmod |grep k10temp
matt@aquos:~$ 
```

[image]

tofurky, 2024-06-23:
I'm using `amd_pstate=active` for CPU frequency control; I imagine this might interact with the SMU in some way? It is available in kernel 6.5+ I believe.

tofurky, 2024-06-23:
Sorry, make that 2 cores. Needed to scroll down. I was thinking it was a bit odd that if there's 16 physical cores there'd be 32 separate temperature readings. 

[image]

024 in the screenshot is CCD 1, ID 8 and shows 429C, but 008 is also CCD 1, ID 8 and doesn't show the high reading. Maybe this is just a quirk of how the temperature data is retrieved? Seems like they both should be at 429C though, if it's the same core. I guess `CCD_AMD_Family_19h_Zen4_Temp()` mentioned above doesn't take topology into account and lets the SMU return a separate temperature for each "core" even if it's just a thread versus physical core?

[image]

cyring, 2024-06-23:
I have a 100ms session running on my Ryzen 9 3950X and issue has not appeared yet.

But looking at your screenshots, can we say that `429` is only displayed in the Max TMP column ?

tofurky, 2024-06-23:
Yes, it's only shown in the max column. I see 429C in max columns for cores 1, 3, 7, 11, 13, 16, 24, 26 now.

cyring, 2024-06-23:
> I'm using `amd_pstate=active` for CPU frequency control; I imagine this might interact with the SMU in some way? It is available in kernel 6.5+ I believe.

Depending on the firmware implementation [amd-pstate.c](https://elixir.bootlin.com/linux/latest/source/drivers/cpufreq/amd-pstate.c) is requesting either the ACPI CPPC either MSR registers (similar to Intel HWP)
I have not read SMU function references in that driver but it may indirectly.

cyring, 2024-06-23:
> Yes, it's only shown in the max column. I see 429C in max columns for cores 1, 3, 7, 11, 13, 16, 24, 26 now.

Great if I can say; it's narrowing the investigation.
It could be a math side effect in the computing macros.

https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/coretypes.h#L631

tofurky, 2024-06-23:
Well, I haven't been watching it for hours so it's possible that it appears once in the TMP column before appearing in the Max column.

But, it's always 429C. Never in between e.g. 300C. There's not wrong MIN values, either.

I am running 'corefreq-cli -C > corefreq.out' so I can check if the instantaneous values show 429C.

cyring, 2024-06-23:
> Maybe this is just a quirk of how the temperature data is retrieved?

In fact, sensors are retrieved depending on the sensor scope.
In the UI Settings window you can change the thermal scope among 4 possibilities: None, Package, Core or SMT (Thread)

cyring, 2024-06-23:
Tip: at any time press the star key `*` to reset values, including the Min and Max of each CPU sensors, frequencies and so on.

cyring, 2024-06-23:
With an uptime of 11 hours, I have not been able to reproduce the issue.

I'm giving a closed look into source code for certain values reached by your CPU and not mine.

tofurky, 2024-06-24:
Got it, so it's 429(4967296), should've realized it was 2^32.

```
matt@aquos:~$ grep -C2 '4294967283' corefreq.out 
009    6.92   244  1.5250   36  000000000000000276    0.004211426   0.042114258
010    6.73   244  1.5250   36  000000000000000277    0.004226685   0.042266846
011    9.31   244  1.5250  4294967283  000000000000000317    0.004837036   0.048370361
012   18.27   244  1.5250   36  000000000000000449    0.006851196   0.068511963
013   11.92   244  1.5250   36  000000000000000291    0.004440308   0.044403076
```

cyring, 2024-06-24:
Congrats

```
= (4294967295) - 12
= 0xFFFFFFFF - 0xC
= (-1U) - 0xC
```
https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/coretypes.h#L709

----

```math
Temp = (Sensor * 5 / 40) - P1 - P2
```
* `P1` can be `49` or `0` depending on `CurTempRangeSel` bit

* `P2` is sourced from  table and unless a bug has to remain at `0`

https://github.com/cyring/CoreFreq/blob/7c8d354aeed79bcde59d2fcadbab899b60137a74/x86_64/corefreqk.h#L7893

----

I have not found yet but somehow in the formula an unexpected sensor value happened

cyring, 2024-06-24:
I think I got it.
Any sensor value lower than `8` and a thermal offset of `49` lead to a negative temperature. It never happened with my hardware but can with yours.

[image]

I have reproduced the issue by injecting those values in driver source code.

[image]

cyring, 2024-06-24:
@tofurky Hello

Please find a fix for your testings in commit 43b96ae67e05b8d978c72bd0b51f7e261a9f212b of the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch

tofurky, 2024-06-24:
> @tofurky Hello
> 
> Please find a fix for your testings in commit [43b96ae](https://github.com/cyring/CoreFreq/commit/43b96ae67e05b8d978c72bd0b51f7e261a9f212b) of the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch

Thanks very much for looking into this, I will test it now. Yes, the negative number makes a lot of sense.

cyring, 2024-06-25:
> Thanks very much for looking into this, I will test it now. Yes, the negative number makes a lot of sense.

Hi,
Do you have a status and an uptime of the fix running w/o issue ?

tofurky, 2024-06-25:
The issue hasn't happened again in the past 5 hours since updating.

cyring, 2024-06-25:
> The issue hasn't happened again in the past 5 hours since updating.

Excellent, thank you

cyring, 2024-06-25:
Hello,

We have an issue with Zen4 voltage in discussion #439 
Could you join it and post voltage from your Raphael 7950X ?

First the output of the general overview like: 
```
corefreq-cli -s -n -m -n -M -n -B -n -k
```

Next for the 3 cases (idle, all Cores stressed, single core stressed) the voltage output with at least 2 rounds of monitoring :
```
corefreq-cli -V 2
```

Thank you

tofurky, 2024-06-25:
Still OK so far with the fix in place. I reloaded k10temp module to see if that changes things.

Info is here: https://github.com/cyring/CoreFreq/discussions/439#discussioncomment-9874109

cyring, 2024-06-25:
> Still OK so far with the fix in place. I reloaded k10temp module to see if that changes things.
> 
> Info is here: [#439 (comment)](https://github.com/cyring/CoreFreq/discussions/439#discussioncomment-9874109)

Thank you very much
Hope it will last in dual with k10temp

Feel free to close that issue when satisfied

tofurky, 2024-07-26:
Sorry to revive an old issue, but I just saw a couple bad max readings that don't appear to be related to a negative number like previously:

From ltrace of corefreq-cli:
```
...
1750613 snprintf("  20.14 ( 0.20)   0.45%   0.36%  99.64%   0.00%   0.00%   0.00%  30 / 39/128", 106, "%7.2f (%5.2f) %6.2f%% %6.2f%% %6.2f%% %6.2f%% %6.2f%% %6.2f%%  %-3u/%3u/%3u", 20.138857, 0.201392, 0.447537, 0.364435, 99.635571, 0.000000, 0.000000, 0.000000, 30, 39, 128) = 76
...
1750613 snprintf("   3.62 ( 0.04)   0.08%   0.07%  99.93%   0.00%   0.00%   0.00%  30 / 38/206", 106, "%7.2f (%5.2f) %6.2f%% %6.2f%% %6.2f%% %6.2f%% %6.2f%% %6.2f%%  %-3u/%3u/%3u", 3.622833, 0.036229, 0.080509, 0.065753, 99.934260, 0.000000, 0.000000, 0.000000, 30, 38, 206) = 76
...
```

The only thing that has changed recently in my setup is a BIOS update a few days ago; previously, values still looked good after running for about a month.

```
PROART X670E-CREATOR WIFI BIOS 2124
Version 2124
13.74 MB
2024/07/19
"1. Update AGESA version to Combo AM5 PI 1.2.0.0a
2. Improve system stability
```

Unsure if it's related or it's just coincidence. I'm going to update to the tagged release and continue monitoring.

cyring, 2024-07-26:
Found something from AMD I left in code comments

https://github.com/cyring/CoreFreq/blob/5aefb1b6c966f2d1fd973c42e37e7d832c7d5f07/x86_64/corefreqk.c#L16189

cyring, 2024-07-26:
I need to debug the thermal register when sensor is particularly hitting the `206` value
Because I believe one of the bits of [`TCCD_REGISTER`](https://github.com/cyring/CoreFreq/blob/5aefb1b6c966f2d1fd973c42e37e7d832c7d5f07/x86_64/amd_reg.h#L1530) or an alias of [`TCTL_REGISTER`](https://github.com/cyring/CoreFreq/blob/5aefb1b6c966f2d1fd973c42e37e7d832c7d5f07/x86_64/amd_reg.h#L1423C3-L1423C16) is telling us what to do with such value.

https://github.com/cyring/CoreFreq/blob/5aefb1b6c966f2d1fd973c42e37e7d832c7d5f07/x86_64/corefreqk.c#L16250

You can replace function with this one which should dump Register when issue happens 

```c
static void CCD_AMD_Family_19h_Zen4_Temp(CORE_RO *Core)
{
	TCCD_REGISTER TccdSensor = {.value = 0};

	Core_AMD_SMN_Read(	TccdSensor,
				(SMU_AMD_THM_TCTL_CCD_REGISTER_F19H_61H
				+ (Core->T.Cluster.CCD << 2)),
				PRIVATE(OF(Zen)).Device.DF );

	Core->PowerThermal.Sensor = TccdSensor.CurTmp;

	if (Core->PowerThermal.Sensor >= 206) {
		pr_info("TCCD_REGISTER[%x]\n", TccdSensor.value);
	}

	if ((TccdSensor.CurTempRangeSel == 1)
	 && (Core->PowerThermal.Sensor >= (49 << 3)))
	{
		Core->PowerThermal.Param.Offset[THERMAL_OFFSET_P1] = 49;
	} else {
		Core->PowerThermal.Param.Offset[THERMAL_OFFSET_P1] = 0;
	}

	Core_AMD_Family_17h_ThermTrip(Core);
}
```
The `TCCD_REGISTER[%x]` dump should be left in the kernel log (`dmesg`)

tofurky, 2024-07-26:
Thanks, I've reloaded the module with that pr_info patch; I'll update when the issue happens (assuming it wasn't a cosmic ray).

tofurky, 2024-07-26:
Seeing a bunch of this in dmesg:

```
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac7]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac7]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[acf]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[acf]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[acf]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac7]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[acf]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac7]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac3]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[abe]
[Fri Jul 26 10:20:39 2024] TCCD_REGISTER[ac0]
```

Maybe the condition needs updated? Unloaded module for now, since there is too much noise.

cyring, 2024-07-26:
> Maybe the condition needs updated?

True. I'm fixing the condition which should match a sensor value of `1023` to give a temperature around `128 C`

```math
Temp = (Sensor * 5)/40
```
Also when bit 11 (`CurTempRangeSel`) is set we subtract `49` to the computed temperature.

Please try again with this condition:

```c
        if (Core->PowerThermal.Sensor >= 1023) {
		pr_info("TCCD_REGISTER[%x]\n", TccdSensor.value);
	}
```

tofurky, 2024-07-27:
There were a couple bursts of messages in dmesg overnight, but the max value seen in corefreq-cli is 86.

```
matt@aquos:~$ dmesg |grep TCCD_REGISTER|cut -d ' ' -f2|sort -u
TCCD_REGISTER[c00]
TCCD_REGISTER[c01]
TCCD_REGISTER[c02]
TCCD_REGISTER[c03]
TCCD_REGISTER[c04]
TCCD_REGISTER[c08]
TCCD_REGISTER[c0c]
TCCD_REGISTER[c0e]
TCCD_REGISTER[c0f]
TCCD_REGISTER[c10]
TCCD_REGISTER[c11]
TCCD_REGISTER[c13]
TCCD_REGISTER[c17]
TCCD_REGISTER[c18]
TCCD_REGISTER[c1a]
TCCD_REGISTER[c1b]
TCCD_REGISTER[c1d]
TCCD_REGISTER[c1f]
TCCD_REGISTER[c21]
TCCD_REGISTER[c23]
TCCD_REGISTER[c25]
TCCD_REGISTER[c2c]
TCCD_REGISTER[c2e]
TCCD_REGISTER[c30]
TCCD_REGISTER[c33]
TCCD_REGISTER[c37]
TCCD_REGISTER[c39]
```

```
matt@aquos:~$ dmesg -T |grep -c 'Sat Jul 27 03:47'
16
matt@aquos:~$ dmesg -T |grep -c 'Sat Jul 27 05:02'
545
```

Also, it seems I made a slight typo, but I think it shouldn't matter:
```
        if (Core->PowerThermal.Sensor >= 1024) {
                pr_info("TCCD_REGISTER[%x]\n", TccdSensor.value);
        }
```

cyring, 2024-07-27:
> There were a couple bursts of messages in dmesg overnight, but the max value seen in corefreq-cli is 86.

Thank you for these results.

Indeed. the max register values appears to be `c39` which converts to `86 C`
```math
Temp = (1081 * 5 / 40) - 49
```

----

To reach `206 C` as you have recorded, register should fall in of those combinations:

`TCCD_REGISTER`|Sensor|`CurTempRangeSel`
---------------|------|-----------------
`0xff8` | `2040` | `-49`
`0x670` | `1648` | `0`

cyring, 2024-07-27:
> Also, it seems I made a slight typo, but I think it shouldn't matter:

Fine. Tracking a sensor of `1024` is also useful because if `TCCD_REGISTER` is equal to `0x400` then Temperature is `128 C`

----

In short, I still don't see why your processor has reached such temperatures; and from your dumps, I don't find new bits which could warn about a thermal case.
Perhaps throttling was engaged and firmware had encoded some magic values into `TCCD_REGISTER`

cyring, 2024-07-28:
@tofurky 
`206C` bug had happened with other software and other Ryzen and even K8

https://www.hwinfo.com/forum/threads/ryzen-3600-hwinfo-6-20-cpu-ccd2-tdie-206-degree.6102/

https://steamcommunity.com/discussions/forum/11/1729828401668273245/

> [Fix](https://www.guru3d.com/news/category/generic-news/page-9/): AMD K8 would sometimes spike to 206C, the false reading is now ignored

cyring, 2024-07-28:
FreeBSD does Math in Kelvin
[`amdtemp_gettemp17h`](https://cgit.freebsd.org/src/plain/sys/dev/amdtemp/amdtemp.c?id=7508f15ff1183f33a3d44438e4c33e1397b5ed3e)

cyring, 2024-07-28:
@tofurky 
How often do you have those temperature spikes ?

tofurky, 2024-07-28:
> @tofurky How often do you have those temperature spikes ?

I saw the max values once after a couple days uptime, but they have not happened again since I applied the TCCD_REGISTER dmesg patch.

cyring, 2024-07-28:
> I saw the max values once after a couple days uptime, 

Was the system woken up from Suspend To Ram (S3) or Suspend to Disk (S4) ?

tofurky, 2024-07-29:
> > I saw the max values once after a couple days uptime,
> 
> Was the system woken up from Suspend To Ram (S3) or Suspend to Disk (S4) ?

No, I do not use suspend/hibernate on this PC. I checked journalctl in case I accidentally did somehow but don't see any evidence of that happening.

cyring, 2024-07-29:
Solutions:
1. Accept spikes and reset min, max values pressing the `*` key
2. Ignore spikes when the computed temperature exceeds the thermal junction but TjMax may be unknown 
3. Apply a low-pass filter to attenuate spikes

cyring, 2024-07-29:
Hello,

In this block:
https://github.com/cyring/CoreFreq/blob/5aefb1b6c966f2d1fd973c42e37e7d832c7d5f07/x86_64/corefreqk.h#L7904
Can you change the `Offset` as below
```c
	.Param.Offset = {95, 0, 0},
```
Rebuild and reload _CoreFreq_

Finally please post the output of `corefreq-cli -s`

You should now read the `TjMax` which is sourced from [product sheet](https://www.amd.com/en/products/processors/desktops/ryzen/7000-series/amd-ryzen-9-7950x.html)
```console
Power, Current & Thermal                                                        
|- Temperature Offset:Junction                                 TjMax [ 49: 95 C]
```

cyring, 2024-07-30:
@tofurky Hi,

Meanwhile I have already programmed the TjMax for your processor.
You just have to pull and test with the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch
And post the output of `corefreq-cli -s`

tofurky, 2024-07-30:
```
root@aquos:~# corefreq-cli -s
Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601206]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 99.999]
|- Frequency            (MHz)                      Ratio                        
                 Min   2999.96                    <  30 >                       
                 Max   4499.94                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
                 TGT   4499.94                    <  45 >                       
   |- CPPC                                                                      
                 Min   3699.95                    <  37 >                       
                 Max   3699.95                    <  37 >                       
                 TGT      AUTO                    <   0 >                       
   |- Boost                                                            [ UNLOCK]
                 XFR   5799.92                    [  58 ]                       
                 CPB   5699.93                    [  57 ]                       
   |- P-State                                                                   
                 P1    2999.96                    <  30 >                       
|- Uncore                                                              [   LOCK]
                 CLK   1499.98                    [  15 ]                       
                 MEM   2999.96                    [  30 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
|- Fused Multiply Add                                            FMA   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hyper-Threading Technology                                    HTT   [Capable]
[... 151 lines cut ...]
   |- Time Window                                                TW1   [   0 ns]
[... cut ...]
   |- Thermal Monitor Trip                                     Limit   [  115 C]
   |- HTC Temperature Limit                                    Limit   [  127 C]
   |- HTC Temperature Hysteresis                           Threshold   [    2 C]
|- Units                                                                        
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
```

cyring, 2024-07-30:
@tofurky 

The power unit in watt is missing, can you read this RAPL register ?

```sh
rdmsr -a 0xc0010299
```

tofurky, 2024-07-30:
> ```shell
> rdmsr -a 0xc0010299
> ```

```
root@aquos:~# rdmsr -a 0xc0010299
a1000
...
a1000
```

cyring, 2024-07-30:
Power Unit is encoded in the first nibble ; same bits structure as Intel MSR

https://github.com/cyring/CoreFreq/blob/36adf0e855ff485a1f219c2927fe924fd168eb6d/x86_64/intel_reg.h#L1976

With 7950X but also other Zen4, PU is left to zero!

I wonder how RAPL is supposed to work with those AMD architectures ?

cyring, 2024-07-30:
@tofurky 
Does Power monitoring work with your 7950X ?
Can you please show two UI screenshots of the Power view when idle then stressed ?

tofurky, 2024-07-30:
I hadn't noticed any issues with it.

[image]
[image]

tofurky, 2024-08-06:
To update, I haven't seen those odd non-negative values reoccur since the first time it happened.

cyring, 2024-08-06:
> To update, I haven't seen those odd non-negative values reoccur since the first time it happened.

Perhaps Manufacturer knows and could answer. At least thermal conditions and/or any transitional operation (firmware).

I'm thinking of injecting a thermal event to the UI when sensor values are out of bounds.

cyring, 2024-08-06:
@tofurky 

New since commit 68ee04002256fd782d7bb6bd168a91972823da1e

----

Because of this AMD statement:
> Register: SMU::THM::THM_TCON_CUR_TMP - Bit 19: CUR_TEMP_RANGE_SEL
>		0 = Report on 0C to 225C scale range.
>		1 = Report on -49C to 206C scale range.

CoreFreq will log `Th. Threshold1` when negative sensor is detected or `Th. Threshold2` when sensor is either greater than `225C` scale range case 0 either `206C` in scale range case 1

To view those threshold events you'll press the `H` key when `HOT` is highlighted in the bottom bar.

[image]
[image]
[image]

cyring, 2024-08-09:
If your BIOS support Nitro then you can dump register to me using [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c)
```
zencli smu 0x50284
```

tofurky, 2024-08-09:
> If your BIOS support Nitro then you can dump register to me using [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c)
> 
> ```
> zencli smu 0x50284
> ```

Not sure what Nitro is - can't find anything except references to graphics cards via Google. PBO is enabled, though.
```
root@aquos:~# /home/matt/aquos/zencli smu 0x50284
[0x00050284] READ(smu) = 0x00000231 (561)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0010 0011 0001
```

cyring, 2024-08-09:
> Not sure what Nitro is

[image]

[Reddit](https://www.reddit.com/r/Amd/comments/15a7jz8/agesa_1007b_ddr5_nitro_recommended_starting_point/)

----

You don't see it in your BIOS but the first 3 nibbles of the dump register say that you have something!

tofurky, 2024-08-09:
> ![image](https://private-user-images.githubusercontent.com/11563789/356730649-62a18b5f-4416-48b5-a100-49cf729f0d14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

ah, i see.

i'm just using the expo profile (DDR5 6000), along with pbo. i also have memory context restore (and subsequently, ddr5 power down) enabled to speed up boot. most things are still at auto. when i read the voltages here a while ago, IIRC i'd set the bios temporarily back to factory defaults.

cyring, 2024-08-24:
@tofurky Hello,

There's a frequency calculation issue with `9950X` and I would like to verify potential regressions using this [formula](https://github.com/cyring/CoreFreq/issues/505#issuecomment-2308182722)
Can you please dump the following registers ?

```sh
## MSR_AMD_PSTATE_DEF_BASE
## P0
rdmsr -p 0 0xc0010064
## P1
rdmsr -p 0 0xc0010065
## P2
rdmsr -p 0 0xc0010066
## P3
rdmsr -p 0 0xc0010067
## P4
rdmsr -p 0 0xc0010068
## P5
rdmsr -p 0 0xc0010069
## P6
rdmsr -p 0 0xc001006a
## P7
rdmsr -p 0 0xc001006b

## MSR_AMD_PSTATE_CURRENT_LIMIT
rdmsr -p 0 0xc0010061

## MSR_AMD_PERF_CTL
rdmsr -p 0 0xc0010062

## MSR_AMD_PERF_STATUS
rdmsr -p 0 0xc0010063

## MSR_AMD_F17H_HW_PSTATE_STATUS
rdmsr -p 0 0xc0010293
```

cyring, 2024-08-24:
@tofurky Can you try latest commit from the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and tell if you notice or not regression in the frequency computing ?

[8 more replies]

closed 2024-09-09 by cyring