cyring/CoreFreq #497: Unexpected low votage and power on E5-2660

samxiaowastaken, 2024-07-17:
I was checking my NAS and noticed that the CPU power consumption seems to be too low. Below are some images showing the issue:

[image]
[image]

It appears that the power consumption is not displaying correctly. However, considering the voltage and temperature readings, it seems there might be an issue with the machine.

I am using an ITX-designed air cooler, and the temperature for this CPU seems abnormal.

When I installed Windows 7 on a USB drive and used CPU-Z for performance benchmarking, the peak performance was as expected, indicating that there is no issue in the Windows environment. This also rules out many hardware and BIOS-related problems.

Although many indicators suggest that it might be due to a misconfigured BIOS and Unraid, there is no clear reason explaining my situation. I have checked my plugin list, which does not include any performance-tweaking plugins. Even after installing Tips and Tweaks and setting it to Performance Mode, the issue persists. Fix Common Problems also did not diagnose any issues. I have searched for similar issues but couldn't find anything relevant.

Here is some potentially helpful information:

CPU: Intel(R) Xeon(R) CPU E5-2660 v4 @ 2.00GHz
Memory: 16GB*2 REGECC
Motherboard: Cloud Star Matx-CS612 (from an unknown manufacturer)
Power Supply: Confirmed to be working fine (since it performs normally under Windows)

Any assistance or insights would be greatly appreciated. Thank you!

cyring, 2024-07-17:
No Intel RAPL register exists to measure the per Core consumed power.
Only global power is specified by manufacturer.
That's why you see this empty block of data (which however won't be empty with AMD Zen CPUs)

Despite this specification fact, I do agree that `7.9 Watts` is far to low for a full stressed processor.
And I would like to verify the Power Units.

Can you please post, in Markdown format, the output of:
```
corefreq-cli -s -n -m
```

cyring, 2024-07-17:
Googling the E5-2660 V4 lead me to [CPU-Z](https://valid.x86.fr/d1u094) with a very closed voltage Vcore of `0.884 Volts`

samxiaowastaken, 2024-07-18:
Here is what I have:
`
Processor                            [Intel(R) Xeon(R) CPU E5-2660 v4 @ 2.00GHz]
|- PPIN#                                                      [25912a176dfdfba2]
|- Architecture                                          [Broadwell/EP/Grantley]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000000]
|- Signature                                                           [  06_4F]
|- Stepping                                                            [      1]
|- Online CPU                                                          [ 28/ 28]
|- Base Clock                                                          [ 99.768]
|- Frequency            (MHz)                      Ratio                        
                 Min   1197.21                    <  12 >                       
                 Max   1995.35                    <  20 >                       
|- Factory                                                             [100.000]
                       2000                       [  20 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   3192.56                    <  32 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   3192.56                    <  32 >                       
                  2C   3192.56                    <  32 >                       
                  3C   2993.03                    <  30 >                       
                  4C   2893.26                    <  29 >                       
                  5C   2793.49                    <  28 >                       
                  6C   2693.72                    <  27 >                       
                  7C   2593.96                    <  26 >                       
                  8C   2494.19                    <  25 >                       
                  9C   2394.42                    <  24 >                       
                 10C   2394.42                    <  24 >                       
                 11C   2394.42                    <  24 >                       
                 12C   2394.42                    <  24 >                       
                 13C   2394.42                    <  24 >                       
                 14C   2394.42                    <  24 >                       
                 15C   2394.42                    <  24 >                       
                 16C   2394.42                    <  24 >                       
                 17C   2394.42                    <  24 >                       
                 18C   2394.42                    <  24 >                       
|- Uncore                                                              [ UNLOCK]
                 Min   1197.21                    <  12 >                       
                 Max   2693.72                    <  27 >                       
|- TDP                                                           Level <  0:1  >
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   1995.35                    [  20 ]                       
              Level1   1696.05                    [  17 ]                       
               Turbo      AUTO                    <   0 >                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [N] AVX-VNN-INT8 [N] AVX-NE-CONV [N] 
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [N]   MOVDIR64B [N] 
|- BMI1/BMI2  [Y/Y]         CLWB [N]      CLFLUSH [Y] CLFLUSH-OPT [N] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- ENQCMD       [N]         GFNI [N]        OSPKE [N]     WAITPKG [N] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [N]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]         SGX [N] 
|- VAES         [N]   VPCLMULQDQ [N]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
[... 405 lines cut ...]
008    1.13  7006  0.8552   34  000000000000000000    0.000000000   0.000000000
[... cut ...]
024    7.69  7137  0.8712   34  000000000000000000    0.000000000   0.000000000
025   32.41  7119  0.8690   33  000000000000000000    0.000000000   0.000000000
026    1.27  7263  0.8866   33  000000000000000000    0.000000000   0.000000000
027    4.96  7246  0.8845   33  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J):   0.764770508   0.000000000   0.000000000   0.000000000   0.000000000
Power(W) :   0.764770508   0.000000000   0.000000000   0.000000000   0.000000000
`

samxiaowastaken, 2024-07-18:
[wikichip](https://en.wikichip.org/wiki/intel/xeon_e5/e5-2660_v4) says that it will need 1.2v for full load (at least), but when tested again, I got 0.88v on full load.
[image]

I'm going to test it again with hwinfo and windows today, hopefully I can find something.

samxiaowastaken, 2024-07-18:
Problem found, it should be motherboard problem. I got the same voltage from AIDA64, full load, 0.876v and 7.92w. And the benchmark score is close enough to the score from cpuz official (5134/5250)

Idk wtf my mother board is doing💀

closed 2024-07-18 by samxiaowastaken

samxiaowastaken, 2024-07-18:
I just contacted the motherboard manufacturer. They mentioned that the voltage and power consumption errors are due to the outdated power supply scheme, which is common in B560 and X99 motherboards. Some manufacturers use additional sensors to "compensate" for these erroneous readings, but this usually doesn't solve the problem.

If you notice that the voltage and power consumption do not meet the expected values, my suggestion is to install Windows and use specialized tools (such as CPU-Z and AIDA64) to test the machine's performance and compare it with the benchmark. If necessary, please contact the motherboard seller to inquire about any relevant warranty or return policies.

Or, the best solution might be to purchase a better motherboard.