cyring/CoreFreq #498: Complie fail kernel 6.10

mikealanni, 2024-07-20:
Hi was trying to rebuild from the source and I'm getting this Error after kernel 6.10 install

`make -j8                                                   cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreqd.c -o build/corefreqd.o
cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreqm.c -o build/corefreqm.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -c x86_64/corefreq-cli.c -o build/corefreq-cli.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-ui.c -o build/corefreq-ui.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -c x86_64/corefreq-cli-rsc.c -o build/corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-cli-json.c -o build/corefreq-cli-json.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -c x86_64/corefreq-cli-extra.c -o build/corefreq-cli-extra.o
cc  -o build/corefreqd \
  build/corefreqd.o build/corefreqm.o -lpthread -lm -lrt -lc
cc  -o build/corefreq-cli \
  build/corefreq-cli.o build/corefreq-ui.o \
  build/corefreq-cli-rsc.o build/corefreq-cli-json.o \
  build/corefreq-cli-extra.o -lm -lrt -lc
make[1]: Entering directory '/home/majed/linux'
  CC [M]  /home/majed/CoreFreq/build/module/corefreqk.o
/home/majed/CoreFreq/build/module/corefreqk.c:510:35: error: initialization of ‘void (*)(struct cpufreq_policy *)’ from incompatible pointer type ‘int (*)(struct cpufreq_policy *)’ [-Werror=incompatible-pointer-types]
  510 |                         .exit   = CoreFreqK_Policy_Exit,
      |                                   ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:244: /home/majed/CoreFreq/build/module/corefreqk.o] Error 1
make[2]: *** [/home/majed/linux/Makefile:1934: /home/majed/CoreFreq/build] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/home/majed/linux'
make: *** [Makefile:110: all] Error 2`

cyring, 2024-07-20:
I don't understand this issue.

Mainline function [prototype](https://elixir.bootlin.com/linux/v6.10/source/include/linux/cpufreq.h#L399) within Linux 6.10 
```c
int		(*exit)(struct cpufreq_policy *policy);
```

_CoreFreq_ function [prototype](https://github.com/cyring/CoreFreq/blob/master/x86_64/corefreqk.c#L21311)
```c
static int CoreFreqK_Policy_Exit(struct cpufreq_policy *policy)
```

Still the same prototype as previous Linux versions.

Which distribution are you using ?
Is your kernel patched ?

cyring, 2024-07-20:
ArchLinux: build and boot the kernel 6.10.

No compilation issues encountered

```console
cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -c x86_64/corefreqd.c -o build/corefreqd.o
cc  -Wall -Wfatal-errors -pthread -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -c x86_64/corefreqm.c -o build/corefreqm.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC  \
  -c x86_64/corefreq-cli.c -o build/corefreq-cli.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -c x86_64/corefreq-ui.c -o build/corefreq-ui.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC  \
  -c x86_64/corefreq-cli-rsc.c -o build/corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -c x86_64/corefreq-cli-json.c -o build/corefreq-cli-json.o
cc  -Wall -Wfatal-errors -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -c x86_64/corefreq-cli-extra.c -o build/corefreq-cli-extra.o
cc  -o build/corefreqd \
  build/corefreqd.o build/corefreqm.o -lpthread -lm -lrt -lc
cc  -o build/corefreq-cli \
  build/corefreq-cli.o build/corefreq-ui.o \
  build/corefreq-cli-rsc.o build/corefreq-cli-json.o \
  build/corefreq-cli-extra.o -lm -lrt -lc
make[1]: Entering directory '/usr/lib/modules/6.10.0-arch1-1/build'
  CC [M]  CoreFreq/build/module/corefreqk.o
  LD [M]  CoreFreq/build/corefreqk.o
  MODPOST CoreFreq/build/Module.symvers
  CC [M]  CoreFreq/build/corefreqk.mod.o
  LD [M]  CoreFreq/build/corefreqk.ko
  BTF [M] CoreFreq/build/corefreqk.ko
make[1]: Leaving directory '/usr/lib/modules/6.10.0-arch1-1/build'
```

mikealanni, 2024-07-20:
Ubuntu 22.04, no extra kernel patch, just downloaded the source, compiled it and installed it

cyring, 2024-07-20:
> Ubuntu 22.04, no extra kernel patch, just downloaded the source, compiled it and installed it

Link to the kernel source code please ?

mikealanni, 2024-07-20:
ChatGPT offered to change static init to static void and add `&` to the line 510 `.exit   = &CoreFreqK_Policy_Exit`. 

That's fixed the issue but I have no clue what is it and will that affect the app

cyring, 2024-07-20:
> ChatGPT offered to change static init to static void and remove the return line and add `&` to the line 510 `.exit = &CoreFreqK_Policy_Exit`.
> 
> That's fixed the issue but I have no clue what is it and will that affect the app

Sure. But why are you having a Linux source code apparently not official ?

mikealanni, 2024-07-20:
Oh didn't know that. Sorry not expert but I got Linus  git

cyring, 2024-07-20:
> Oh didn't know that. Sorry not expert but I got Linus git

From here ? https://kernel.org/

mikealanni, 2024-07-20:
I did git clone to this git://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git a while back then I did git pull and build it. I'm doing something wrong?

cyring, 2024-07-20:
There is only one prototype in kernel.org so far and up to to `6.10`

```c
int		(*exit)(struct cpufreq_policy *policy);
```

cyring, 2024-07-20:
> I did git clone to this git://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git a while back then I did git pull and build it. I'm doing something wrong?

Since [kernel `3.0`](https://elixir.bootlin.com/linux/v3.0/source/include/linux/cpufreq.h#L235) that `exit` function within structure `cpufreq_driver` has not changed!

cyring, 2024-07-20:
Make sur to clean everything up

1. Clone again Linux kernel `6.10` and build it by hand:
 `make distclean` and `make clean` before your `make defconfig`
 * Install and reboot that kernel, including its kernel header files
 
2. Clone latest CoreFreq version:
* Make sure to uninstall any CoreFreq package if one used;
* Remove any old binaries remaining on PATH `corefreqk.ko corefreqd corefreq-cli`
* Clone straight from the GitHub repository
* Clean and build CoreFreq
```
make -j clean
make -j
```
* Start CoreFreq directly from the work directory
```
insmod build/corefreqk.ko
./build/corefreqd -q &
./buid/corefreq-cli
```

cyring, 2024-07-20:
Wait a minute; I found that major change in [github.com/torvalds](https://github.com/torvalds/linux/blob/3c3ff7be9729959699eb6cbc7fd7303566d74069/include/linux/cpufreq.h#L399)

```c
struct cpufreq_driver {
...
void		(*exit)(struct cpufreq_policy *policy);
...
};

```

cyring, 2024-07-20:
And here's the root cause
https://github.com/torvalds/linux/commit/b4b1ddc9dfe997a5f492fa3a36487f8e7a5de30d

cyring, 2024-07-20:
And it's part of Linux [Next](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/include/linux/cpufreq.h?h=next-20240719#n399)

cyring, 2024-07-20:
Until I'm committing a change you can fix source code by yourself as below

1. Edit `CoreFreqK_Policy_Exit` function in `x86_64/corefreqk.c` file

https://github.com/cyring/CoreFreq/blob/a03489feb3560bd7b1107da9701896ab0cb9ca57/x86_64/corefreqk.c#L21311

and replace with:
```c
static void CoreFreqK_Policy_Exit(struct cpufreq_policy *policy)
{
	UNUSED(policy);
	return 0;
}
```

2. Edit file `x86_64/corefreqk.h`

https://github.com/cyring/CoreFreq/blob/a03489feb3560bd7b1107da9701896ab0cb9ca57/x86_64/corefreqk.h#L8634

and replace with

```c
static void CoreFreqK_Policy_Exit(struct cpufreq_policy *policy) ;
```

3. Rebuild

```sh
make -j clean
make -j
```

4. Start _CoreFreq_

```sh
insmod build/corefreqk.ko
./build/corefreqd
```
```
./build/corefreq-cli
```

mikealanni, 2024-07-20:
Man I just want to thank for your fast response, will try it and feedback

mikealanni, 2024-07-20:
It is working now, and I think this the same I did earlier when chatgpt asks about change int to void and remove return line. don't know that AI do these things tho 😅

cyring, 2024-07-23:
Hello,

In Arch's [CDN](https://cdn.kernel.org/pub/linux/kernel/v6.x/) the latest [testing](https://gitlab.archlinux.org/archlinux/packaging/packages/linux) linux `6.10.0-arch1-2` archive file is still based on the _old_ cpufreq function prototype.

I'm releasing the change as a patch and will wait to commit until Linux-Next  is showing up into Arch

[CoreFreq_linux_6_10_cpufreq.patch.gz](https://github.com/user-attachments/files/16344580/CoreFreq_linux_6_10_cpufreq.patch.gz)

cyring, 2024-07-30:
@mikealanni Hello,

Fix is now available in commit 0db6f0a18d3b03a26287fd30d27bb3fd24a2a3c2

cyring, 2024-08-07:
Now as part of [Releases](https://github.com/cyring/CoreFreq/releases)

closed 2024-08-07 by cyring