cyring/CoreFreq #505: AMD Ryzen 9950x freq incorrect

madoverlord40, 2024-08-17:
When single thread boosting occurs on 9950x, it goes all the way to 61 and seems to be off the chart while reporting 5676 in the upper left corner. I assume you have not yet added support for this cpu as it just released. I will keep using CoreFreq anyway hoping you will release an update soon to correct this.

Thanks.

cyring, 2024-08-17:
@madoverlord40 
Thank you for trying _CoreFreq_ with your 9950X. This is indeed the first report I'm receiving and I will need a lot of traces to debug the state of progress.

Can you post the output of all options of the CLI. List them with option `-h`

At least:
```
corwfreq-cli -s
```

About frequency issue, I need a screenshot of the CLI. Select `Tools` in the Menu and choose `CPU Select`
Select your best Core, you can press `z` to find it the CPPC table (if available)
Your best Core should the max boosted frequency, please take a screenshot at that moment.

madoverlord40, 2024-08-18:
Processor                                  [AMD Ryzen 9 9950X 16-Core Processor]
|- Architecture                                             [Zen5/Granite Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0b40401a]
|- Signature                                                           [  BF_44]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 70.487]
|- Frequency            (MHz)                      Ratio                        
                 Min   6202.86                    <  88 >                       
                 Max   4299.71                    <  61 >                       
|- Factory                                                             [100.000]
                       6100                       [  61 ]                       
|- Performance                                                                  
                 TGT   4299.71                    <  61 >                       
   |- CPPC                                                                      
                 Min   3453.87                    <  49 >                       
                 Max    352.44                    <   5 >                       
                 TGT      AUTO                    <   0 >                       
   |- Boost                                                            [ UNLOCK]
                 XFR   5286.53                    [  75 ]                       
                 CPB   5286.53                    [  75 ]                       
   |- P-State                                                                   
                 P1    6202.86                    <  88 >                       
|- Uncore                                                              [   LOCK]
                 CLK   1057.31                    [  15 ]                       
                 MEM   2114.61                    [  30 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [Y] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [Y]    AVX-FP128 [N]   AVX-FP256 [N] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Missing]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
|- Fused Multiply Add                                            FMA   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hyper-Threading Technology                                    HTT   [Capable]
|- Hardware P-state control                                      HwP   [Capable]
[... cut ...]
   |- Thermal Monitor Trip                                     Limit   [  115 C]
   |- HTC Temperature Limit                                    Limit   [  127 C]
   |- HTC Temperature Hysteresis                           Threshold   [    2 C]
|- Units                                                                        
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]

madoverlord40, 2024-08-18:
I also noticed that if i leave CoreFreq running idle, and i walk away for a while, when i come back, the system is locked up and i have to power off/on. It boots right back into linux mint, but i test this. I left the pc on for 1HR without the program running, then with the program running, less than 30min later mouse wont move. I See [missing] in a lot of places above so i am wondering if that is enough info for you. I will try to get you a screen shot of best core boost when i am able to catch it doing that.

cyring, 2024-08-18:
The baseclock `BCLK` can be fixed to `100 MHz` with a driver parameter
```
insmod build/corefreqk.ko AutoClock=0
```
Or (if installed)
```
modprobe corefreqk AutoClock=0
```

Apparently the baseclock is wrongly updated to `~ 70 MHz`
You can also try `AutoClock=1` for a unique estimation and see if it makes a difference into the frequency issue.

----

Commonly motherboards mitigate electromigration with SPREAD SPECTRUM.
I'm disabling such feature in BIOS onto both CPU and Chipset.
As a result, the `BXLK` estimates closer to 100 MHz`

----

(tbc)

cyring, 2024-08-18:
The frozen system issue could be due to an access conflict onto the SMU between _CoreFreq_ and another Linux driver agent like `k10temp`, `amd_pstate`
You have to unload them or blacklist them during Linux startup.
It will help me if you can list all drivers with command `lsmod`

Also have a look to the Wiki/[CoreFreq as the Clock Source, CPU Freq and CPU Idle driver](https://github.com/cyring/CoreFreq/wiki/CoreFreq-as-the-Clock-Source,-CPU-Freq-and-CPU-Idle-driver)

----

(tbc)

madoverlord40, 2024-08-18:
Im on a new install of Mint 22, all i have installed for additional drivers is LACT, if you are by chance not familiar with that, is a kernel driver level tool for my radeon 7900xtx. For some reason in linux the gpu fans wont kick on and use LACT to force them on. I can still get you an lsmod report if you still wish it, but i dont think it will help you. maybe LACT is conflicting with your software? Im going to go try out "insmod build/corefreqk.ko AutoClock=0" and see what the frequence does after that and ill report back.

madoverlord40, 2024-08-18:
Ok i just downloaded and ran passmarks performance test 11 for linux and ran a CPU benchmark. Here is a screen shot of what your software is reporting.
[image]

madoverlord40, 2024-08-18:
Ok here i used AutoClock=1 during a CPU benchmark:
[image]

madoverlord40, 2024-08-18:
Ok i just realized that passmarks performance test has it wrong too. Check this out:

PassMark PerformanceTest Linux (11.0.1002)
AMD Ryzen 9 9950X 16-Core Processor (x86_64)
16 cores @ 8007 MHz  |  62.4 GiB RAM
Number of Processes: 32  |  Test Iterations: 1  |  Test Duration: Medium

madoverlord40, 2024-08-18:
Ok i just installed latest kernel from mainline kernels 6.10.5 and now passmark shows this:
AMD Ryzen 9 9950X 16-Core Processor (x86_64)
16 cores @ 5752 MHz  |  62.4 GiB RAM
Number of Processes: 32  |  Test Iterations: 1  |  Test Duration: Medium

and your software shows this:
[image]

madoverlord40, 2024-08-18:
I booted into windows 11 and ryzen master is showing this during passmark 11:
[image]

cyring, 2024-08-18:
We need to know what's the reference base clock is ?
Aka `BCLK`, the baseclock has to be displayed in one of your UEFI/BIOS screen.

A `70 MHz` rather than a usual `100 MHz` is not impossible but that will be a premiere since a long time.

I would like also to see the `TSC`,`UCC`, `URC` counters. 
Press `c` for such view or `corefreq-cli -c`

----

Btw, the computed voltage Vcore is wrong. As shown by Ryzen Master, it has to peak at `1.3 V`

Can you replace this line of source code ?

https://github.com/cyring/CoreFreq/blob/8ca842b164f196d6205a401d469c3274315227e0/x86_64/coretypes.h#L611

with this line:

```c
#define VOLTAGE_FORMULA_AMD_1Ah VOLTAGE_FORMULA_AMD_ZEN4
```

Save file then rebuild _CoreFreq_
```
make -j clean
make -j
```
Make sure to fully unload _CoreFreq_, especially its driver (`rmmod corefreqk`), before restarting it for the voltage test in a **single** Core fully stressed which has to trigger the PBO max Vcore-frequency.

cyring, 2024-08-18:
Btw for all your next screenshots please switch to the custom view, press `y` key.

cyring, 2024-08-18:
Also have you noticed that some CPU temperature of `96°C` had exceeded the `95°C` TjMax

cyring, 2024-08-18:
Thanks @InstLatx64 MSR [dump](https://github.com/InstLatx64/InstLatx64/commit/3d181fb6029f327b120dcab051ffd53d5db8ec19#diff-210f04209642dcdf0ced26e1b1145a27c460496bb45bdb1a33cfb9f3dc855ee6R2959) I see than the TSC seems decoralated from the base clock
```
------[ MSR Registers / Logical CPU #0 ]------

CPU Clock (Normal): 5760 MHz
CPU Clock (TSC): 4321 MHz
CPU Multiplier: 57.3x
```

```
4321 (MHz) / 57.3 = 75.4
```

Somehow as with arm processors, _facial_ frequency is not the  same as the TSC counter.
The Window tools above may source the frequency from other sources; I bet the AMD PP tables.

madoverlord40, 2024-08-18:
The base clock in my bios is 100, you are correct. I didnt change that, however i am running EXPO at 6000, but i assume that has nothing to do with the base clock. As for the temp, yeah, i saw that it hit 96 and that should be impossible, i wonder however, im running PBO unlimited, might have pushed it past 95? I read somewhere one time that the sensors in the CPU only report up to 95 but the CPU could actually be above that if pegged out at max at the time and that some times other sensors can report higher. Just a thought, what do you think?

Ok ill go change the source code like you asked and ill report back.
Thanks for your support!

cyring, 2024-08-18:
> other sensors can report higher. Just a thought, what do you think?

True, some are die sensors, others for the whole package. 
It's not that easy to guess which one is from without some Registers specifications.

> Ok ill go change the source code like you asked and ill report back.

Great. Can't wait to see the results. _Manufacturer always change the rules for their own reasons I believe..._

----

Please also let me know about the accuracy of the measured Power in Watt. 
Just press the `w` key and do two screenshots; when system is idle and when it is fully stressed.

madoverlord40, 2024-08-18:
Ok so i removed the module, and recompiled with the changes below:
//#define VOLTAGE_FORMULA_AMD_1Ah VOLTAGE_FORMULA_AMD_19_61h
#define VOLTAGE_FORMULA_AMD_1Ah VOLTAGE_FORMULA_AMD_ZEN4
In header file coretypes.h

I started your software with corefreq-cli -t custom with the below screen shot during passmark single thread test:

[image]

cyring, 2024-08-18:
> Ok so i removed the module, and recompiled with the changes

Thank you.
To me, Vcore appears now in the range from a minimum `0.8` to a maximum `1.3` V
And current voltage looks to be per cluster CCX or CCD. This could explain why a single stressed CPU voltage is repeated among its cluster.
* Can you post the topology ?
```
corefreq-cli -m
```

The power measured per physical Core looks coherent.
I'm astonished by the total of `217 W`. 
9950X is said to be an efficient processor.

> Although i cant really tell if your software is not reporting correctly or if its just not hitting PBO max being at 5.4 single thread, i think its supposed to be 5.7 in single thread.

That's why I claimed to use my integrated CPU stress Tools.
They have been especially designed to trigger the boosted frequency-ies

* Press `O` to open the "Tools" window then choose "CPU Select..." option and select your best CPU.
* If you ignore the best Core then you can choose either option "Round Robin" or option "Random CPU"
* You can then follow the max recorded frequency within the "Custom" view.
* You will see a red dot (right to the top LCD) when stress tools are active.
* You can stop at any time pressing the `F10` function key. Dot will disappear when Tools stop.
* `Conics` is the recommended algorithm to stress FPU. You should read the most consumed energy that you can monitor with the view "Power"

----

I can see that the UI is facing glitches. Those because of abnormal huge values within the absolute frequencies. So far I can't tell if regressions are part of this new architecture or the queried Registers have unpunished changes.

----

What about the memory controller ?
* Can you post the output of below options:
```
corefreq-cli -k -n -B -n -M
```

Regards
CyrIng

madoverlord40, 2024-08-18:
./build/corefreq-cli -m
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0   0  0   0  0      32  8        48 12      1024 16 i   65536 16w 
001:  0    2   0  0   1  0      32  8        48 12      1024 16 i   65536 16w 
002:  0    4   0  0   2  0      32  8        48 12      1024 16 i   65536 16w 
003:  0    6   0  0   3  0      32  8        48 12      1024 16 i   65536 16w 
004:  0    8   0  1   4  0      32  8        48 12      1024 16 i   65536 16w 
005:  0   10   0  1   5  0      32  8        48 12      1024 16 i   65536 16w 
006:  0   12   0  1   6  0      32  8        48 12      1024 16 i   65536 16w 
007:  0   14   0  1   7  0      32  8        48 12      1024 16 i   65536 16w 
008:  0   16   1  2   8  0      32  8        48 12      1024 16 i   65536 16w 
009:  0   18   1  2   9  0      32  8        48 12      1024 16 i   65536 16w 
010:  0   20   1  2  10  0      32  8        48 12      1024 16 i   65536 16w 
011:  0   22   1  2  11  0      32  8        48 12      1024 16 i   65536 16w 
012:  0   24   1  3  12  0      32  8        48 12      1024 16 i   65536 16w 
013:  0   26   1  3  13  0      32  8        48 12      1024 16 i   65536 16w 
014:  0   28   1  3  14  0      32  8        48 12      1024 16 i   65536 16w 
015:  0   30   1  3  15  0      32  8        48 12      1024 16 i   65536 16w 
016:  0    1   0  0   0  1      32  8        48 12      1024 16 i   65536 16w 
017:  0    3   0  0   1  1      32  8        48 12      1024 16 i   65536 16w 
018:  0    5   0  0   2  1      32  8        48 12      1024 16 i   65536 16w 
019:  0    7   0  0   3  1      32  8        48 12      1024 16 i   65536 16w 
020:  0    9   0  1   4  1      32  8        48 12      1024 16 i   65536 16w 
021:  0   11   0  1   5  1      32  8        48 12      1024 16 i   65536 16w 
022:  0   13   0  1   6  1      32  8        48 12      1024 16 i   65536 16w 
023:  0   15   0  1   7  1      32  8        48 12      1024 16 i   65536 16w 
024:  0   17   1  2   8  1      32  8        48 12      1024 16 i   65536 16w 
025:  0   19   1  2   9  1      32  8        48 12      1024 16 i   65536 16w 
026:  0   21   1  2  10  1      32  8        48 12      1024 16 i   65536 16w 
027:  0   23   1  2  11  1      32  8        48 12      1024 16 i   65536 16w 
028:  0   25   1  3  12  1      32  8        48 12      1024 16 i   65536 16w 
029:  0   27   1  3  13  1      32  8        48 12      1024 16 i   65536 16w 
030:  0   29   1  3  14  1      32  8        48 12      1024 16 i   65536 16w 
031:  0   31   1  3  15  1      32  8        48 12      1024 16 i   65536 16w

madoverlord40, 2024-08-18:
./build/corefreq-cli -k -n -B -n -M
Linux:                                                                          
|- Release                                               [6.10.5-061005-generic]
|- Version      [#202408141037 SMP PREEMPT_DYNAMIC Wed Aug 14 14:47:07 UTC 2024]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         65452224 KB
|- Shared RAM                                                          608720 KB
|- Free RAM                                                          20622252 KB
|- Buffer RAM                                                           33384 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
Clock Source                                                  <             tsc>
CPU-Freq driver                                               [  amd-pstate-epp]
Governor                                                      [         Missing]
CPU-Idle driver                                               [       acpi_idle]
|- Idle Limit                                                 [              C3]
   |- State        POLL      C1      C2      C3                                 
   |-           CPUIDLE ACPI FF ACPI IO ACPI IO                                 
   |- Power          -1       0       0       0                                 
   |- Latency         0       1      18     350                                 
   |- Residency       0       2      36     700                                 

[ 0] American Megatrends Inc.                                                   
[ 1] 2204                                                                       
[ 2] 08/02/2024                                                                 
[ 3] ASUS                                                                       
[ 4] System Product Name                                                        
[ 5] System Version                                                             
[ 6] S---e---e---l---m---                                                       
[ 7] SKU                                                                        
[ 8] To be filled by O.E.M.                                                     
[ 9] ASUSTeK COMPUTER INC.                                                      
[10] ROG STRIX X670E-E GAMING WIFI                                              
[11] Rev 1.xx                                                                   
[12] 2---1---1---6--                                                            
[13] Number Of Devices:4\Maximum Capacity:134217728 kilobytes                   
[14]                                                                            
[15] DIMM 1\P0 CHANNEL A                                                        
[16]                                                                            
[17] DIMM 1\P0 CHANNEL B                                                        
[18]                                                                            
[19] G Skill Intl                                                               
[20]                                                                            
[21] G Skill Intl                                                               
[22]                                                                            
[23] F5-6400J3239G32G                                                           
[24]                                                                            
[25] F5-6400J3239G32G                                                           

                              Zen UMC  [14E0]                              
Controller #0                                                Dual Channel  
 Bus Rate  3000 MHz       Bus Speed 2114 MHz           DDR5 Speed 4228 MT/s
                                                                           
 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   32   39   39   39  102  146    8   15   32    8   30   90    8   23 
  #1   32   39   39   39  102  146    8   15   32    8   30   90    8   23 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   30   23   21    8    1   10   10    1   11   11    0    0    0    0 
  #1   30   23   22    8    1   10   10    1   11   11    0    0    0    0 
      REFI RFC1 RFC2 RFCsb RCPB RPPB BGS:Alt  Ban  Page  CKE  CMD  GDM  ECC
  #0 11677  312  192  390   0    0    ON OFF  R0W0   0    0   1T    ON   0 
  #1 11677  312  192  390   0    0    ON OFF  R0W0   0    0   1T    ON   0 
      MRD:PDA   MOD:PDA  WRMPR STAG PDM RDDATA WRD  WRL  RDL  XS   XP CPDED
  #0   42  32    42  32    24    7 0:F:1   20   6   18   36  914   23   15 
  #1   42  32    42  32    24    7 0:F:1   20   6   18   36  914   23   15 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1     65536      1024          16384    F5-6400J3239G32G
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1     65536      1024          16384    F5-6400J3239G32G

cyring, 2024-08-18:
> ./build/corefreq-cli -m [topology.txt](https://github.com/user-attachments/files/16652137/topology.txt)

Thank you.
Looks good to me

madoverlord40, 2024-08-18:
> > ./build/corefreq-cli -m [topology.txt](https://github.com/user-attachments/files/16652137/topology.txt)
> 
> Thank you. Looks good to me

Ok cool, so you have what you need to update your software for 9950x?

cyring, 2024-08-18:
> ./build/corefreq-cli -k -n -B -n -M [memory.txt](https://github.com/user-attachments/files/16652110/memory.txt)

Thanks.

* Memory size is wrongly computed.
I believe the rank count of `F5-6400J3239G32G`
* `4800 MHz` rather than `6400` : does it have to be fixed ?
* Can you tell if the timings are matching your BIOS values ?

----

Btw, in case you are not familiar with Markdown to post the output data;
1. Type 3 anti-quotes followed by carriage return 
2. Type or copy/past the output data
3. Type again 3 anti-quotes

cyring, 2024-08-18:
> Ok cool, so you have what you need to update your software for 9950x?

Yes I have minor fixes to provide in next version

madoverlord40, 2024-08-18:
> Thanks.
>
>     * Memory size is wrongly computed.
>       I believe the rank count of `F5-6400J3239G32G`
>
>     * `4800 MHz` rather than `6400` : does it have to be fixed ?
>
>     * Can you tell if the [...]

The ram is 6400 but im running it at 6000 in bios, the ryzen sweet spot otherwise fabric clock ends up no longer 1:1:1 with mem clk and uclk.

cyring, 2024-08-18:
> ... So it should be reading 3000 a channel.

The main issue is that the estimated BCLK of `~ 70 MHz` is also messing with the DDR5 computed speed.

Can you please post the following:
```
corefreq-cli -c 3
```

madoverlord40, 2024-08-19:
./build/corefreq-cli  -c 3
'''
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
000   39.63 ( 0.56)   0.92   0.95  99.05   0.00   0.00   0.00  32 / 36:680/ 95
001  166.16 ( 2.36)   3.86   3.40  96.60   0.00   0.00   0.00  32 / 36:680/ 95
002   67.90 ( 0.96)   1.58   1.50  98.50   0.00   0.00   0.00  32 / 36:680/ 95
003   14.74 ( 0.21)   0.34   0.47  99.53   0.00   0.00   0.00  32 / 36:680/ 95
004   45.99 ( 0.65)   1.07   1.08  98.92   0.00   0.00   0.00  32 / 36:687/ 95
005   19.18 ( 0.27)   0.45   0.51  99.49   0.00   0.00   0.00  32 / 36:680/ 95
006   10.83 ( 0.15)   0.25   0.32  99.68   0.00   0.00   0.00  32 / 36:680/ 95
007   52.32 ( 0.74)   1.22   0.93  99.07   0.00   0.00   0.00  32 / 36:680/ 95
008    0.45 ( 0.01)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 36:684/ 94
009   67.07 ( 0.95)   1.56   1.26  98.74   0.00   0.00   0.00  32 / 36:684/ 94
010  117.92 ( 1.67)   2.74   2.96  97.04   0.00   0.00   0.00  32 / 36:684/ 94
011   45.76 ( 0.65)   1.06   0.90  99.10   0.00   0.00   0.00  32 / 36:684/ 94
012   27.62 ( 0.39)   0.64   0.74  99.26   0.00   0.00   0.00  32 / 37:688/ 94
013    2.57 ( 0.04)   0.06   0.07  99.93   0.00   0.00   0.00  32 / 36:684/ 94
014    0.43 ( 0.01)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 37:688/ 94
015    0.55 ( 0.01)   0.01   0.02  99.98   0.00   0.00   0.00  32 / 36:684/ 94
016   18.10 ( 0.26)   0.42   0.48  99.52   0.00   0.00   0.00  32 / 36:680/ 95
017   54.33 ( 0.77)   1.26   1.19  98.81   0.00   0.00   0.00  32 / 36:683/ 95
018   19.90 ( 0.28)   0.46   0.44  99.56   0.00   0.00   0.00  32 / 36:687/ 95
019    0.50 ( 0.01)   0.01   0.02  99.98   0.00   0.00   0.00  32 / 36:687/ 95
020   54.28 ( 0.77)   1.26   1.17  98.83   0.00   0.00   0.00  32 / 36:680/ 95
021    2.23 ( 0.03)   0.05   0.08  99.92   0.00   0.00   0.00  32 / 36:683/ 95
022    1.18 ( 0.02)   0.03   0.04  99.96   0.00   0.00   0.00  32 / 36:680/ 95
023    1.46 ( 0.02)   0.03   0.03  99.97   0.00   0.00   0.00  32 / 36:683/ 95
024   33.25 ( 0.47)   0.77   0.65  99.35   0.00   0.00   0.00  32 / 36:684/ 94
025   10.87 ( 0.15)   0.25   0.25  99.75   0.00   0.00   0.00  32 / 36:684/ 94
026    8.32 ( 0.12)   0.19   0.16  99.84   0.00   0.00   0.00  32 / 36:684/ 94
027   19.10 ( 0.27)   0.44   0.43  99.57   0.00   0.00   0.00  32 / 36:684/ 94
028    0.27 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 36:684/ 94
029   42.56 ( 0.60)   0.99   1.23  98.77   0.00   0.00   0.00  32 / 36:684/ 94
030    0.27 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 36:684/ 94
031    5.75 ( 0.08)   0.13   0.18  99.82   0.00   0.00   0.00  32 / 36:684/ 94

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      0.81   0.75  99.25   0.00   0.00   0.00      95 C    37 C

CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
000   22.71 ( 0.32)   0.53   0.58  99.42   0.00   0.00   0.00  32 / 36:687/ 95
001  132.91 ( 1.89)   3.09   2.81  97.19   0.00   0.00   0.00  32 / 36:687/ 95
002   60.41 ( 0.86)   1.40   1.41  98.59   0.00   0.00   0.00  32 / 36:687/ 95
003    2.78 ( 0.04)   0.06   0.09  99.91   0.00   0.00   0.00  32 / 36:687/ 95
004   45.99 ( 0.65)   1.07   1.08  98.92   0.00   0.00   0.00  32 / 36:687/ 95
005    2.27 ( 0.03)   0.05   0.07  99.93   0.00   0.00   0.00  32 / 36:687/ 95
006   41.42 ( 0.59)   0.96   0.74  99.26   0.00   0.00   0.00  32 / 36:687/ 95
007    4.87 ( 0.07)   0.11   0.16  99.84   0.00   0.00   0.00  32 / 36:687/ 95
008    0.38 ( 0.01)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 36:684/ 94
009   40.41 ( 0.57)   0.94   0.84  99.16   0.00   0.00   0.00  32 / 36:684/ 94
010  117.92 ( 1.67)   2.74   2.96  97.04   0.00   0.00   0.00  32 / 36:684/ 94
011   87.98 ( 1.25)   2.05   2.35  97.65   0.00   0.00   0.00  32 / 36:684/ 94
012   27.62 ( 0.39)   0.64   0.74  99.26   0.00   0.00   0.00  32 / 37:688/ 94
013    6.28 ( 0.09)   0.15   0.14  99.86   0.00   0.00   0.00  32 / 37:688/ 94
014    0.43 ( 0.01)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 37:688/ 94
015    0.34 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 37:688/ 94
016   16.46 ( 0.23)   0.38   0.42  99.58   0.00   0.00   0.00  32 / 36:687/ 95
017   53.80 ( 0.76)   1.25   1.14  98.86   0.00   0.00   0.00  32 / 36:687/ 95
018   19.90 ( 0.28)   0.46   0.44  99.56   0.00   0.00   0.00  32 / 36:687/ 95
019    0.50 ( 0.01)   0.01   0.02  99.98   0.00   0.00   0.00  32 / 36:687/ 95
020   46.55 ( 0.66)   1.08   1.10  98.90   0.00   0.00   0.00  32 / 36:687/ 95
021    9.27 ( 0.13)   0.22   0.31  99.69   0.00   0.00   0.00  32 / 36:687/ 95
022    0.60 ( 0.01)   0.01   0.02  99.98   0.00   0.00   0.00  32 / 36:687/ 95
023    0.55 ( 0.01)   0.01   0.02  99.98   0.00   0.00   0.00  32 / 36:687/ 95
024   33.35 ( 0.47)   0.78   0.66  99.34   0.00   0.00   0.00  32 / 36:684/ 94
025   20.64 ( 0.29)   0.48   0.46  99.54   0.00   0.00   0.00  32 / 37:688/ 94
026    3.78 ( 0.05)   0.09   0.11  99.89   0.00   0.00   0.00  32 / 37:688/ 94
027   19.10 ( 0.27)   0.44   0.43  99.57   0.00   0.00   0.00  32 / 36:684/ 94
028    0.31 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 36:684/ 94
029   25.53 ( 0.36)   0.59   0.67  99.33   0.00   0.00   0.00  32 / 37:688/ 94
[... 18 lines cut ...]
011   26.93 ( 0.38)   0.63   0.82  99.18   0.00   0.00   0.00  32 / 37:688/ 94
012    3.65 ( 0.05)   0.08   0.11  99.89   0.00   0.00   0.00  32 / 37:688/ 94
[... cut ...]
027   10.26 ( 0.15)   0.24   0.33  99.67   0.00   0.00   0.00  32 / 37:688/ 94
028    0.25 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 37:688/ 94
029    0.25 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 37:688/ 94
030    0.23 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 37:688/ 94
031    0.25 ( 0.00)   0.01   0.01  99.99   0.00   0.00   0.00  32 / 37:688/ 94

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      0.44   0.45  99.55   0.00   0.00   0.00      95 C    37 C
'''

madoverlord40, 2024-08-19:
Ok so using your turbo feature on core 5, ryzen master shows it to be a fast core with a white dot?
[image]

madoverlord40, 2024-08-19:
As for matching ram timings, i checked in bios, and booted into windows 11, this is ryzen master shows which matches bios:
[image]

madoverlord40, 2024-08-19:
Also while i was in windows 11, i ran Cinebench 23 to see what ryzen master would show for all core workload. You mentioned earlier 217w, but look at what ryzen master says the watts are at full load.
[image]

Score if you are interested:
[image]

madoverlord40, 2024-08-19:
Cinebench 23 runing single core, here is ryzen master:
[image]

score if you are interested:
[image]

madoverlord40, 2024-08-19:
Dont forget I have curve optimizer of -25, i am not sure how that affects the core voltage, but it should lower it quite a bit?

cyring, 2024-08-19:
> Ok so using your turbo feature on core 5, ryzen master shows it to be a fast core with a white dot?

Great!
This time we are reaching the PBO frequency. Even with a quarter frequency above as XFR used to be.

I don't know how RM is marking the Core score from but _CoreFreq_ decodes the CPPC table and translates a register into a max frequency capability per CPU which appears to tell the same as RM dots or stars.
Press `z` to open the CPPC table window.

cyring, 2024-08-19:
> Also while i was in windows 11, i ran Cinebench 23 to see what ryzen master would show for all core workload. You mentioned earlier 217w, but look at what ryzen master says the watts are at full load.

RM displays for `PPT` a `% of 1000W`

Really!!!

----

Give a try to the `Tools > Conics >  Hyperboloid of two sheets" and see if `9950X` is consuming more power than that.

cyring, 2024-08-19:
> score if you are interested: [...]

This is such an IPC improvement from AMD.
Here is my 3950X score from CineBench v20
[image]

cyring, 2024-08-19:
# Factor from Ryzen 1700X 

`3950X`|`9950X`
-------|-------
`521/378`|`2287/959`
`x 1.38` | `x 2.38`

cyring, 2024-08-19:
> * F5-6400J3239G32G

Indeed I'm supposed to decode this DIMM [geometry](https://www.techpowerup.com/review/g-skill-trident-z5-rgb-ddr5-6400-cl32-2x-32gb/) : `Dual Rank`

cyring, 2024-08-19:
> As for matching ram timings, i checked in bios, and booted into windows 11, this is ryzen master shows which matches bios: [...]

_CoreFreq_ is matching on Pri, Sec, and Ter Timings

`RFC1 RFC2` are differing because I just know how to decode default values

cyring, 2024-08-19:
@madoverlord40 

You can now pull commit 0e37a134988bb00db8a2cb1934eefb5b0737ef67 from [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch

Can you please take this `Core cycles` view screenshot ?

[image]

madoverlord40, 2024-08-19:
> > Also while i was in windows 11, i ran Cinebench 23 to see what ryzen master would show for all core workload. You mentioned earlier 217w, but look at what ryzen master says the watts are at full load.
> 
> RM displays for `PPT` a `% of 1000W`
> 
> Really!!!

Thats what my mainboard does when i put it on PBO max, it sets the power limits to 1k.

madoverlord40, 2024-08-19:
> # Factor from Ryzen 1700X
> `3950X` 	`9950X`
> `521/378` 	`2287/959`
> `x 1.38` 	`x 2.38`

I also have an intel 14900kf setup and the 9950x is still not quite there. single core when pushed to 320w is still higher than 9950x PBO max at 230w. All-core however, the 9950x is slightly ahead of the 14900KS. Where the 9950x truely shines i think is the watt to freq comparison. The 9950x does the basically the same job as intel 14900k at 100watt less.

madoverlord40, 2024-08-19:
Ok here are the core cycles view, after compiling latest from develop.
[image]

cyring, 2024-08-19:
> Ok here are the core cycles view, after compiling latest from develop. ![Screenshot from 2024-08-19 07-56-57](https://private-user-images.githubusercontent.com/25307356/359147799-19c1cfc0-c376-4f78-b8ac-c5fa862911ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQwODQ0NTgsIm5iZiI6MTcyNDA4NDE1OCwicGF0aCI6Ii8yNTMwNzM1Ni8zNTkxNDc3OTktMTljMWNmYzAtYzM3Ni00Zjc4LWI4YWMtYzVmYTg2MjkxMWVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE5VDE2MTU1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUyODlmMDU3ZGJhZDMwMWQyMjgyMmVmMjRkNTZmZjA0YjM3Y2JiMmIxNTlhNDQ0ZjU0Yzk4ZDkxNTcyMjEzMGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._3-QoI_efDwxeCwIsYkiAqKHUnQcLOWdOcmEGzS3p58)

I was not expecting the `TSC` to be equal to the CPU Base Clock of `4.3 GHz`
I'm now back to square one and I don't see what's going with the `BCLK` estimation ?

madoverlord40, 2024-08-19:
> I was not expecting the `TSC` to be equal to the CPU Base Clock of `4.3 GHz` I'm now back to square one and I don't see what's going with the `BCLK` estimation ?

At this point i think its a kernel level issue. The kernel is probably not reporting the CPU correctly.

cyring, 2024-08-19:
> At this point i think its a kernel level issue. The kernel is probably not reporting the CPU correctly.

The estimation is computed from a one second interval between two `TSC` collects. _CoreFreq_ relies on kernel just for the delay; the `TSC` being collected directly onto CPU register.

A kernel issue would be its own clock source but you would have noticed it if one second is not accurate.

Can you however try the following build option and tell if it makes a difference ?
```
make -j clean
make -j DELAY_TSC=0
```

Also, just in case of a kernel issue,  can you post the output of `/proc/cpuinfo` 
_(first CPU will be enough)_

madoverlord40, 2024-08-19:
> The estimation is computed from a one second interval between two `TSC` collects. _CoreFreq_ relies on kernel just for the delay; the `TSC` being collected directly onto CPU register.
>
> A kernel [...]

OH! Ok then, i didnt realize you actually talk directly to CPU, thats cool.
Ok here is output of only core0:
```
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 26
model		: 68
model name	: AMD Ryzen 9 9950X 16-Core Processor
stepping	: 0
microcode	: 0xb40401a
cpu MHz		: 5720.447
cache size	: 1024 KB
physical id	: 0
siblings	: 32
core id		: 0
cpu cores	: 16
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 16
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good amd_lbr_v2 nopl xtopology nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba perfmon_v2 ibrs ibpb stibp ibrs_enhanced vmmcall fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid cqm rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local user_shstk avx_vnni avx512_bf16 clzero irperf xsaveerptr rdpru wbnoinvd cppc arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold v_vmsave_vmload vgif v_spec_ctrl vnmi avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid bus_lock_detect movdiri movdir64b overflow_recov succor smca fsrm avx512_vp2intersect flush_l1d amd_lbr_pmc_freeze
bugs		: sysret_ss_attrs spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 8599.98
TLB size	: 192 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate cpb eff_freq_ro [13] [14]
```

cyring, 2024-08-19:
> So maybe you should adjust how the multipliers work

This is what also puzzles me.
The minimum ratio is shown as `88` which is impossible.
Ratios are computed from P-State registers.
The AMD formula of the Coefficient Of Frequency(COF) could have be changed by manufacturer; making my function inaccurate to extract the ratio

https://github.com/cyring/CoreFreq/blob/8ca842b164f196d6205a401d469c3274315227e0/x86_64/corefreqk.c#L8110

madoverlord40, 2024-08-19:
> This is what also puzzles me. The minimum ratio is shown as `88` which is impossible. Ratios are computed from P-State registers. The AMD formula of the Coefficient Of Frequency(COF) could have be [...]

Thats why i was thinking it could be a kernel issue, even though you are talking to the cpu, doesnt it need to go through the OS to do that? What if the kernel is miss-reporting the registers or the P-State values? Sorry if this has nothing to do with it, trying to help trouble shoot.

madoverlord40, 2024-08-19:
> The estimation is computed from a one second interval between two `TSC` collects. _CoreFreq_ relies on kernel just for the delay; the `TSC` being collected directly onto CPU register.
>
> A kernel [...]

Ok so i just rebuilt your software using make -j DELAY_TSC=0 what info are you needing with that change?

cyring, 2024-08-20:
> doesnt it need to go through the OS to do that?

Nope, all hardware primitives are made by my own.

> Ok so i just rebuilt your software using make -j DELAY_TSC=0 what info are you needing with that change?

Just want to know about BCLK. If it's still `70` or what we want `100 MHz`

----

But how can it process so differently from the previous architectures, especially [7950X](https://github.com/cyring/CoreFreq/wiki/CPU-support)

cyring, 2024-08-20:
@madoverlord40 : With the MSR-Tools installed, can you dump that series of [P-State](https://github.com/cyring/CoreFreq/blob/8ca842b164f196d6205a401d469c3274315227e0/x86_64/amd_reg.h#L544) registers ?

```sh
## MSR_AMD_PSTATE_DEF_BASE
## P0
rdmsr -p 0 0xc0010064
## P1
rdmsr -p 0 0xc0010065
## P2
rdmsr -p 0 0xc0010066
## P3
rdmsr -p 0 0xc0010067
## P4
rdmsr -p 0 0xc0010068
## P5
rdmsr -p 0 0xc0010069
## P6
rdmsr -p 0 0xc001006a
## P7
rdmsr -p 0 0xc001006b

## MSR_AMD_PSTATE_CURRENT_LIMIT
rdmsr -p 0 0xc0010061

## MSR_AMD_PERF_CTL
rdmsr -p 0 0xc0010062

## MSR_AMD_PERF_STATUS
rdmsr -p 0 0xc0010063

## MSR_AMD_F17H_HW_PSTATE_STATUS
rdmsr -p 0 0xc0010293

cyring, 2024-08-20:
> Ok so using your turbo feature on core 5, ryzen master shows it to be a fast core with a white dot?

Remark: _CoreFreq_ counts CPU number on a zero base whilst RM is one based

madoverlord40, 2024-08-20:
> > Ok so using your turbo feature on core 5, ryzen master shows it to be a fast core with a white dot?
> 
> Remark: _CoreFreq_ counts CPU number on a zero base whilst RM is one based

Oh crap lol does that invalidate the PBO max test i did for you?

madoverlord40, 2024-08-20:
> > doesnt it need to go through the OS to do that?
> 
> Nope, all hardware primitives are made by my own.
> 
> > Ok so i just rebuilt your software using make -j DELAY_TSC=0 what info are you needing with that change?
> 
> Just want to know about BCLK. If it's still `70` or what we want `100 MHz`
> 
> But how can it process so differently from the previous architectures, especially [7950X](https://github.com/cyring/CoreFreq/wiki/CPU-support)
> 
> BCLK `70 MHz` is my main issue. TSC looks OK and counts invariantly at processor base frequency but my estimation function takes a ratio as an input parameter; ratio is broken.

ITs still 70 and still showing a boost off the chart past 61.

cyring, 2024-08-20:
> > > Ok so using your turbo feature on core 5, ryzen master shows it to be a fast core with a white dot?
> > 
> > 
> > Remark: _CoreFreq_ counts CPU number on a zero base whilst RM is one based
> 
> Oh crap lol does that invalidate the PBO max test i did for you?

Luckily those adjacent cores are providing the same PBO score

cyring, 2024-08-20:
> ITs still 70 and still showing a boost off the chart past 61.

Thanks for your answer. 
I will debug the Coefficient Of Frequency as soon as I received the MSR dump I claimed previously.

> Also, my system still locks up if i leave your software running ilde for more than like 30min.

I can suggest you boot my **`archlinux-corefreq.iso`** to download from [version 1.98.2](https://github.com/cyring/CoreFreq/releases/tag/1.98.2)

ISO can be flashed on an USB key with a plain `dd` command. See instructions in [wiki/Live-CD](https://github.com/cyring/CoreFreq/wiki/Live-CD)

Boot the key and when facing the startup menu, choose `[Bare Metal]`

[image]

And let it go...

Once _CoreFreq_ is launched, press `k` for the kernel window

1. Case no driver registered at all
[image]
2. You can register all drivers. Press `s` for the Settings window and proceed to the Governor, CPU-Freq and CPU-idle registration
[image]
3. _CoreFreq_ is now governing Kernel
[image]

Q: In that configuration please let me know if system is locking up or not ?

madoverlord40, 2024-08-20:
Ok so I booted to the thumb drive with arch linux and it installed the corefreq.ko but governor will not activate in settings windows. It shows 'off' and when I try to turn it on it says error file exists. It did not lock up however I wonder if not having governor on is causing that?

cyring, 2024-08-20:
> when I try to turn it on it says error file exists

This means a Linux default driver is already acting as CPU-Freq.

That's why there is that `[Bare Metal]` entry you have to select immediately when thumb drive boots. You have a few seconds to make the selection.

Fyi, that boot menu entry attempts to blacklist all default CPU-Freq and CPU-Idle kernel modules to leave the seat to _CoreFreq_

Next, press `k` for Kernel window: you should read `Missing` for the name of drivers.
If true then you are allowed to register from the Settings window.

To verify things are OK, reopen the Kernel window: drivers are now assigned to `corefreq`

In short, select `[Bare Metal]` during the thumb drive startup.

> It did not lock up however I wonder if not having freq driver on is causing that?

That's the point of the above procedure:  to find which Linux kernel module(s) _CoreFreq_ conflicts with. 
Most of the time, it has been the `k10temp` module. 

----

Q: while running the ISO, had you read a `BCLK` of `100` or `70 MHz` ?

cyring, 2024-08-20:
As soin as you can please provide me the [MSR dump](https://github.com/cyring/CoreFreq/issues/505#issuecomment-2297923634) as it should give me some clues to debug the frequency ratios.

[80 more replies]

closed 2024-11-27 by cyring