cyring/CoreFreq #506: [AMD][VMR/RPL/GNR] Voltage Curve Optimizer

cyring, 2024-08-25:
For AMD Vermeer, Raphael and Granite Ridge is introduced the VCO formula as a build argument:
```sh
make -j HWM_CHIPSET=AMD_VCO
```

Remarks:
* without the `HWM_CHIPSET` argument, the _CoreFreq_ build will fallback to the AMD P-State voltage formula.
* the P-State formula is preferred with EPYC and any platform whom BIOS excludes voltage optimization

cyring, 2024-08-25:
<details>
<summary>Contributors</summary>
@leinardi [5800X]
@LethalManBoob [5800X] [5700X]
@thor2002ro [5900X]
@Slaviusz [5950X]
@WildPenquin [5950X]
@ppascher [5950X]
@olejon [5950X]
@Jon0 [7950X]
@KeithMyers [7950X]
@amfern [7950X]
@tofurky [7950X]
@mikealanni [7945X]
@madoverlord40 [9950X]
</details>

VCO change is available since 85b1e626183986c3bdfcc9b2bb046e8f24a13882 in [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch

If you have chosen the Voltage Optimizer within BIOS option then you have to specify `HWM_CHIPSET=AMD_VCO` to get an appropriate voltage Core reading.

Please post screenshots of the UI `View > Sensors > Voltage` in various cases: Idle and Stressed CPU

[image]

Tips:
* If you don't need the UI bar charts then you can build this way
```sh
make -j HWM_CHIPSET=AMD_VCO NO_UPPER=1
```

----

Thanks anyone for helping

madoverlord40, 2024-08-27:
Please post screenshots of the UI View > Sensors > Voltage in various cases: Idle and Stressed CPU (not showing Menu)

Do I do this while having curve optimizer active, or not active?

cyring, 2024-08-27:
>> Please post screenshots of the UI View > Sensors > Voltage in various cases: Idle and Stressed CPU (not showing Menu)
> 
> Do I do this while having curve optimizer active, or not active?

You do this while having curve optimizer active

madoverlord40, 2024-08-27:
Ok so i cleaned and rebuilt with make -j HWM_CHIPSET=AMD_VCO
Here is idle voltage view:
[image]

All core stress using your software voltage view:

[image]

cyring, 2024-08-28:
@madoverlord40 Thank you for your tests. CO is really odd: from idle to stress, the `VID` delta is just about `4` for most cores.
And unfortunately `9950X` results don't behave as `7950X` ones 

`VID` to voltage formula has to be customized directly in source code and as a function of the curve parameters left in BIOS
https://github.com/cyring/CoreFreq/blob/85b1e626183986c3bdfcc9b2bb046e8f24a13882/x86_64/coretypes.h#L788

----

Now BIOS can be configured with no CO, voltage offset and tweaks. Just the plain P-State. PBO is ok.
A simple `make` (without `HWM_CHIPSET` parameter) will now build _CoreFreq_ with standard formula: please let ne know if voltage is accurate

KeithMyers, 2024-08-28:
Not using Curve Optimizer.  Using per CCX with 52.0/50.5 @ 1.28 VID
Here is idle and stress screenshots under my Boinc load.
[image]
[image]

madoverlord40, 2024-08-28:
I still have CO on and have not testing with it off but as having curve optimizer on at -20 and level 5 CPU LLC in bios, i dont think this voltage is correct?

[image]

cyring, 2024-08-28:
> I still have CO on and have not testing with it off but as having curve optimizer on at -20 and level 5 CPU LLC in bios, i dont think this voltage is correct?

For this screenshot how was made the build ?

cyring, 2024-08-28:
> Ok so i cleaned and rebuilt with make -j HWM_CHIPSET=AMD_VCO
> Im using a CO in bios of -20.

When CO is activated, the `VID` values in both case, idle and stressed, are so closed that it can't compute the low and high Vcore!

cyring, 2024-08-28:
Let me show you what we have with Zen2/Matisse
1. **No** voltage optimizer or whatever LLC or performance tricks are activated
2. Only PBO, in its default mode, is allowed

* Idle case
[image]

* All CPUs stressed
[image]

* Single CPU stressed
[image]

The `VID` values (obtained from current P-State register) are inversely proportionate to the frequency 

Applying the voltage [formula](https://github.com/cyring/CoreFreq/blob/85b1e626183986c3bdfcc9b2bb046e8f24a13882/x86_64/coretypes.h#L782) (specified in AMD datasheet), we are getting an accurate voltage (for both CPU and SoC)

----

* Zen1, Zen2 and Zen3
```math
Vcore = 1.550 - (0.00625 * VID)
```

----

* Zen3+ and Zen4: I have discovered this [formula](https://github.com/cyring/CoreFreq/blob/85b1e626183986c3bdfcc9b2bb046e8f24a13882/x86_64/coretypes.h#L785)
```math
Vcore = 0.00625 * VID
```

----

* Zen5: no datasheet; we have to guess a formula or we can bet a formula same as previous generations, especially when no voltage optimization is enabled.

madoverlord40, 2024-08-28:
> > I still have CO on and have not testing with it off but as having curve optimizer on at -20 and level 5 CPU LLC in bios, i dont think this voltage is correct?
> 
> For this screenshot how was made the build ?

I think its still built with HWM_CHIPSET=AMD_VCO.

madoverlord40, 2024-08-28:
You know i think these voltages im getting right now are good, not that i know what the formulas should be. But i rebuilt with just -j on develop, and then started to build HLOD in unreal engine 5, and voltages were between 1.29 and 1.36 while boosting. Seeing that i have a CO of -20 and LLC of level 5, its doing well. Even if the voltages are not correct, they are i think quite close!

cyring, 2024-08-28:
> You know i think these voltages im getting right now are good, not that i know what the formulas should be. But i rebuilt with just -j on develop, and then started to build HLOD in unreal engine 5, and voltages were between 1.29 and 1.36 while boosting. Seeing that i have a CO of -20 and LLC of level 5, its doing well. Even if the voltages are not correct, they are i think quite close!
> 
> Yeah, its close, im in RM now during a C++ all core compile, RM is averaging 1.33vcore, you are so close!

Great to see that.
`make -j` brings the default voltage formula.
I'm going to merge last commit into `master`
Thank you for your help.

cyring, 2024-10-02:
Closing issue

To sum up:
* Voltage curve is an optional compilation directive. User has to specialize the formula macro source code by her/him self.
* The default build computes the voltage Core with the standard p-state formula.

closed 2024-10-02 by cyring