cyring/CoreFreq #508: ArchLinux: "Remove DKMS modules" issue

cyring, 2024-08-29:
Issue raised in corefreq package [comments](https://aur.archlinux.org/pkgbase/corefreq)

https://github.com/cyring/CoreFreq/blob/85b1e626183986c3bdfcc9b2bb046e8f24a13882/dkms.conf#L26

This PKGBUILD [change](https://aur.archlinux.org/cgit/aur.git/commit/?h=corefreq&id=7a8c163499e8740bf6aa0b6d83f862c9712b8b25) is fixing the issue

cyring, 2024-08-29:
[`dkms.conf`](https://github.com/cyring/CoreFreq/blob/master/dkms.conf) has been made for all distributions which use it.
Commenting the `POST_REMOVE` will impact all of them.
Need to do non-regression tests among all various packaging.

inglor, 2024-08-30:
I maintain dkms.conf specifically for Archlinux but if you want to do that feel free. This is cause Archlinux use split packages for client/server/dkms.

cyring, 2024-08-30:
> I maintain dkms.conf specifically for Archlinux but if you want to do that feel free. This is cause Archlinux use split packages for client/server/dkms.

OK, I'm fine with this.
Thanks.

closed 2024-08-30 by cyring