cyring/CoreFreq #511: [AMD][Intel] Non-binary DDR5

cyring, 2024-10-10:
# DDR5 Query
* [What is “non-binary” memory?](https://www.kingston.com/en/blog/system-builder/24gbit-memory-faq)
* To decode the 24GB or 48GB DIMM geometry, can anyone dump some registers ?

# [AMD] Zen 4 or 5
* You will have to compile the [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c)  

```sh
# Compile the tool
cc zencli.c -o zencli
```

* Run it as root on the first four channels to dump the memory controller configuration
```sh
# channel 0
./zencli smu 0x50100
# channel 1
./zencli smu 0x150100
# channel 2
./zencli smu 0x250100
# channel 3
./zencli smu 0x350100
```

[image]

* In the `DdrType` bits-field I expect to find some new DDR5 type values or something in the `Reserved` part

# [Intel]
* I still have found no IMC register about non-binary DDR5</div>

cyring, 2024-10-10:
[Micron](https://www.mouser.com/datasheet/2/671/24gb_ddr5_sdram_dierevb-3412405.pdf)
> Notes: 1. For non-binary densities, a quarter of the row address space is invalid. When the MSB address bit is HIGH, the MSB-1 address must be LOW.

nooberfsh, 2024-11-10:
Hi. 
9950x, 2 * 48G
```
# channel 0
[0x00050100] READ(smu) = 0x80000701 (2147485441)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1000 0000 0000 0000 0000 0111 0000 0001
# channel 1
[0x00150100] READ(smu) = 0x80000701 (2147485441)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1000 0000 0000 0000 0000 0111 0000 0001
```

cyring, 2024-11-10:
@nooberfsh Thanks for your dump. Unfortunately I'm finding no hint in this register.

Can you show me how your 9950X memory controller is decoded with:
```
corefreq-cli -k -n -B -n -M
```

nooberfsh, 2024-11-11:
@cyring Sure
```
Linux:                                                                          
|- Release                                                      [6.11.6-arch1-1]
|- Version              [#1 SMP PREEMPT_DYNAMIC Fri, 01 Nov 2024 03:30:41 +0000]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         96393508 KB
|- Shared RAM                                                          379760 KB
|- Free RAM                                                          89599260 KB
|- Buffer RAM                                                          139252 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
Clock Source                                                  <             tsc>
CPU-Freq driver                                               [  amd-pstate-epp]
Governor                                                      [         Missing]
CPU-Idle driver                                               [       acpi_idle]
|- Idle Limit                                                 [              C3]
   |- State        POLL      C1      C2      C3                                 
   |-           CPUIDLE ACPI FF ACPI IO ACPI IO                                 
   |- Power          -1       0       0       0                                 
   |- Latency         0       1      18     350                                 
   |- Residency       0       2      36     700                                 

[ 0] American Megatrends International, LLC.                                    
[ 1] 1.L0                                                                       
[ 2] 10/16/2024                                                                 
[ 3] Micro-Star International Co., Ltd.                                         
[ 4] MS-7D70                                                                    
[ 5] 1.0                                                                        
[ 6] T---e---l--- ---O---M-                                                     
[ 7] To be filled by O.E.M.                                                     
[ 8] To be filled by O.E.M.                                                     
[ 9] Micro-Star International Co., Ltd.                                         
[10] MPG X670E CARBON WIFI (MS-7D70)                                            
[11] 1.0                                                                        
[12] 0---0---O---3---0-                                                         
[13] Number Of Devices:4\Maximum Capacity:134217728 kilobytes                   
[14]                                                                            
[15] DIMMA2\P0 CHANNEL A                                                        
[16]                                                                            
[17] DIMMB2\P0 CHANNEL B                                                        
[18]                                                                            
[19] Unknown                                                                    
[20]                                                                            
[21] Unknown                                                                    
[22]                                                                            
[23]                                                                            
[24]                                                                            
[25]                                                                            

                              Zen UMC  [14E0]                              
Controller #0                                                Dual Channel  
 Bus Rate  3000 MHz       Bus Speed 2994 MHz           DDR5 Speed 5988 MT/s
                                                                           
 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   30   36   36   36   76  115    8   15   32    8   30   90    8   23 
  #1   30   36   36   36   76  115    8   15   32    8   30   90    8   23 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   28   23   20    8    1   10   10    1   11   11    0    0    0    0 
  #1   28   23   21    8    1   10   10    1   11   11    0    0    0    0 
      REFI RFC1 RFC2 RFCsb RCPB RPPB BGS:Alt  Ban  Page  CKE  CMD  GDM  ECC
  #0 11677  312  192  390   0    0    ON OFF  R0W0   0    0   1T    ON   0 
  #1 11677  312  192  390   0    0    ON OFF  R0W0   0    0   1T    ON   0 
      MRD:PDA   MOD:PDA  WRMPR STAG PDM RDDATA WRD  WRL  RDL  XS   XP CPDED
  #0   42  32    42  32    24    7 0:F:0   18   6   16   36  914   23   15 
  #1   42  32    42  32    24    7 0:F:0   18   6   16   36  914   23   15 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768     
```

cyring, 2024-11-12:
> Hi. 9950x, 2 * 48G

Hello,
Can I have this long list of dump registers based on these specifications
[image]
[image]

```sh
# UMC::BaseAddr
## Channel 0
./zencli smu 0x50000
./zencli smu 0x50004
./zencli smu 0x50008
./zencli smu 0x5000c
## Channel 1
./zencli smu 0x150000
./zencli smu 0x150004
./zencli smu 0x150008
./zencli smu 0x15000c
# UMC::BaseAddrSec
## Channel 0
./zencli smu 0x50010
./zencli smu 0x50014
./zencli smu 0x50018
./zencli smu 0x5001c
## Channel 1
./zencli smu 0x150010
./zencli smu 0x150014
./zencli smu 0x150018
./zencli smu 0x15001c
# UMC::AddrMask
## Channel 0
./zencli smu 0x50020
./zencli smu 0x50024
./zencli smu 0x50028
./zencli smu 0x5002c
## Channel 1
./zencli smu 0x150020
./zencli smu 0x150024
./zencli smu 0x150028
./zencli smu 0x15002c
# UMC::AddrMaskSec
## Channel 0
./zencli smu 0x50030
./zencli smu 0x50034
./zencli smu 0x50038
./zencli smu 0x5003c
## Channel 1
./zencli smu 0x150030
./zencli smu 0x150034
./zencli smu 0x150038
./zencli smu 0x15003c
```
And also
```sh
# UMC::AddrCfg
## Channel 0
./zencli smu 0x50040
./zencli smu 0x50044
./zencli smu 0x50048
./zencli smu 0x5004c
## Channel 1
./zencli smu 0x150040
./zencli smu 0x150044
./zencli smu 0x150048
./zencli smu 0x15004c
# UMC::DimmCfg
## Channel 0
./zencli smu 0x50090
./zencli smu 0x50094
## Channel 1
./zencli smu 0x150090
./zencli smu 0x150094
```

nooberfsh, 2024-11-12:
Sure, but I can't access my PC until this weekend, I'll post the result as soon as possible.

cyring, 2024-11-13:
> Sure, but I can't access my PC until this weekend, I'll post the result as soon as possible.

Will you also please have a try to the branch [develop_amd_ext_apic_id](https://github.com/cyring/CoreFreq/tree/develop_amd_ext_apic_id) and post the output of `corefreq-cli -s -n -m`

nooberfsh, 2024-11-17:
```
[0x00050000] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050004] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050008] READ(smu) = 0x00000001 (1)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001
[0x0005000c] READ(smu) = 0x00000401 (1025)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0100 0000 0001
[0x00150000] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150004] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150008] READ(smu) = 0x00000001 (1)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001
[0x0015000c] READ(smu) = 0x00000401 (1025)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0100 0000 0001
[0x00050010] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050014] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050018] READ(smu) = 0x08000001 (134217729)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 1000 0000 0000 0000 0000 0000 0001
[0x0005001c] READ(smu) = 0x08000401 (134218753)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 1000 0000 0000 0000 0100 0000 0001
[0x00150010] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150014] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150018] READ(smu) = 0x08000001 (134217729)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 1000 0000 0000 0000 0000 0000 0001
[0x0015001c] READ(smu) = 0x08000401 (134218753)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 1000 0000 0000 0000 0100 0000 0001
[0x00050020] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050024] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050028] READ(smu) = 0x07fffbfe (134216702)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0111 1111 1111 1111 1011 1111 1110
[0x0005002c] READ(smu) = 0x07fffbfe (134216702)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0111 1111 1111 1111 1011 1111 1110
[0x00150020] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150024] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150028] READ(smu) = 0x07fffbfe (134216702)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0111 1111 1111 1111 1011 1111 1110
[... 44 lines cut ...]
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000
[0x00150044] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000
[0x00150048] READ(smu) = 0x0025070c (2426636)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0010 0101 0000 0111 0000 1100
[0x0015004c] READ(smu) = 0x0025070c (2426636)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
[... cut ...]
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150090] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00150094] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

```

nooberfsh, 2024-11-17:
> > Sure, but I can't access my PC until this weekend, I'll post the result as soon as possible.
> 
> Will you also please have a try to the branch [develop_amd_ext_apic_id](https://github.com/cyring/CoreFreq/tree/develop_amd_ext_apic_id) and post the output of `corefreq-cli -s -n -m`

It seems the branch doesn't exist

cyring, 2024-11-17:
> It seems the branch doesn't exist

Everything has been merged, you can now pull latest commits from master.

nooberfsh, 2024-11-17:
```
Processor                                  [AMD Ryzen 9 9950X 16-Core Processor]
|- Architecture                                             [Zen5/Granite Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0b404022]
|- Signature                                                           [  BF_44]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 99.813]
|- Frequency            (MHz)                      Ratio                        
                 Min   2994.38                    <  30 >                       
                 Max   4291.95                    <  43 >                       
|- Factory                                                             [100.000]
                       4300                       [  43 ]                       
|- Performance                                                                  
                 TGT   4291.95                    <  43 >                       
   |- CPPC                                                                      
                 Min   3693.07                    <  37 >                       
                 Max    399.25                    <   4 >                       
                 TGT      AUTO                    <   0 >                       
   |- Boost                                                            [ UNLOCK]
                 XFR   5689.33                    [  57 ]                       
                 CPB   5689.33                    [  57 ]                       
   |- P-State                                                                   
                 P1    2994.38                    <  30 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [Y] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [Y]    AVX-FP128 [N]   AVX-FP256 [N] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Missing]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP CMPSB|SCASB                                   FSRC   [Capable]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast Short REP STOSB                                         FSRS   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
|- Fused Multiply Add                                            FMA   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hyper-Threading Technology                                    HTT   [Capable]
|- Hardware P-state control                                      HwP   [Capable]
[... 189 lines cut ...]
[... cut ...]
025:  0   19   1  2   9  1      32  8        48 12      1024 16 i   65536 16w 
026:  0   21   1  2  10  1      32  8        48 12      1024 16 i   65536 16w 
027:  0   23   1  2  11  1      32  8        48 12      1024 16 i   65536 16w 
028:  0   25   1  3  12  1      32  8        48 12      1024 16 i   65536 16w 
029:  0   27   1  3  13  1      32  8        48 12      1024 16 i   65536 16w 
030:  0   29   1  3  14  1      32  8        48 12      1024 16 i   65536 16w 
031:  0   31   1  3  15  1      32  8        48 12      1024 16 i   65536 16w 
```

cyring, 2024-11-17:
@nooberfsh Thank you for your answers.
* Still analyzing UMC Registers which will take some time.
* Can you post the output of `lspci -nn`: I need to probe the IOMMU id to show its version ?

nooberfsh, 2024-11-17:
Sure
```
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14d8]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:14d9]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:01.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:02.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:02.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:08.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 71)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e0]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e1]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e2]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e3]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e4]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e5]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e6]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e7]
01:00.0 Ethernet controller [0200]: Mellanox Technologies MT27800 Family [ConnectX-5] [15b3:1017]
02:00.0 Non-Volatile memory controller [0108]: Solidigm P44 Pro NVMe SSD [Hollywood Beach] [025e:f1ac]
03:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Upstream Port [1022:43f4] (rev 01)
04:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
04:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
04:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
04:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
04:07.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
04:08.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
04:0c.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
04:0d.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0a:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Upstream Port [1022:43f4] (rev 01)
0b:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0b:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0b:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0b:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0b:07.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0b:08.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0b:0c.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0b:0d.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset PCIe Switch Downstream Port [1022:43f5] (rev 01)
0d:00.0 Network controller [0280]: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter [14c3:0616]
0e:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)
12:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset USB 3.2 Controller [1022:43f7] (rev 01)
13:00.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset SATA Controller [1022:43f6] (rev 01)
14:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset USB 3.2 Controller [1022:43f7] (rev 01)
15:00.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] 600 Series Chipset SATA Controller [1022:43f6] (rev 01)
16:00.0 Non-Volatile memory controller [0108]: Solidigm P44 Pro NVMe SSD [Hollywood Beach] [025e:f1ac]
17:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Granite Ridge [Radeon Graphics] [1002:13c0] (rev c1)
[... cut ...] io Controller [1002:1640]
17:00.2 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Family 19h PSP/CCP [1022:1649]
17:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b6]
17:00.4 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b7]
17:00.6 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Family 17h/19h HD Audio Controller [1022:15e3]
18:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b8]

```

cyring, 2024-11-17:
> Sure

Thank you.

Can you switch to the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and pull latest commit 7ee38f9284dfa830ccae73728dd7dce40bc0b03a which queries IOMMU `[1022:14d9]` from Raphael list

Make sure to unload previous version.
Next, build and start this version `1.98.7`

I would like to see the output of:
```
corefreq-cli -s -n -z -n -m -n -k -n -n -B -n -M -n -C 1 -n -c 1 -n -i 1 -n -R
```

nooberfsh, 2024-11-17:
```
Processor                                  [AMD Ryzen 9 9950X 16-Core Processor]
|- Architecture                                             [Zen5/Granite Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0b404022]
|- Signature                                                           [  BF_44]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 99.813]
|- Frequency            (MHz)                      Ratio                        
                 Min   2994.39                    <  30 >                       
                 Max   4291.96                    <  43 >                       
|- Factory                                                             [100.000]
                       4300                       [  43 ]                       
|- Performance                                                                  
                 TGT   4291.96                    <  43 >                       
   |- CPPC                                                                      
                 Min   3693.08                    <  37 >                       
                 Max    399.25                    <   4 >                       
                 TGT      AUTO                    <   0 >                       
   |- Boost                                                            [ UNLOCK]
                 XFR   5689.34                    [  57 ]                       
                 CPB   5689.34                    [  57 ]                       
   |- P-State                                                                   
                 P1    2994.39                    <  30 >                       
|- Uncore                                                              [   LOCK]
                 CLK   1497.19                    [  15 ]                       
                 MEM   2994.39                    [  30 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [Y] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [Y]    AVX-FP128 [N]   AVX-FP256 [N] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Missing]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP CMPSB|SCASB                                   FSRC   [Capable]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast Short REP STOSB                                         FSRS   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
|- Fused Multiply Add                                            FMA   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
[... 741 lines cut ...]
[... cut ...]
#24   0   0   1   0   0   1   0   0   1   1   1   0   0             
#25   0   0   1   0   0   1   0   0   1   1   1   0   0             
#26   0   0   1   0   0   1   0   0   1   1   1   0   0             
#27   0   0   1   0   0   1   0   0   1   1   1   0   0             
#28   0   0   1   0   0   1   0   0   1   1   1   0   0             
#29   0   0   1   0   0   1   0   0   1   1   1   0   0             
#30   0   0   1   0   0   1   0   0   1   1   1   0   0             
#31   0   0   1   0   0   1   0   0   1   1   1   0   0   
```

cyring, 2024-11-17:
@nooberfsh Thanks. 

* DDR size is still a work in progress
* AMD-V and IO/MMU are now showing up
* `TSME` has toggled to enabled since your previous output.
   I'm not sure if it comes from my last commit ?
* According to AMD whitepaper, `AGENPICK` and `XPROC_LEAK` [features](https://github.com/cyring/CoreFreq/blob/7ee38f9284dfa830ccae73728dd7dce40bc0b03a/x86_64/corefreqk.c#L12712) have to be limited to Zen-2 but `BTC-NOBR` is a question.
  Can you read this [`MSR_AMD_DE_CFG2`](https://github.com/cyring/CoreFreq/blob/c75af85b5bc148c364e2e55e92e79a98eab4ce8e/x86_64/amd_reg.h#L133) register ?
```
rdmsr -ax 0xc00110e3
```

----

Fyi, your output is stored in Wiki/CPU support/[Ryzen 9 9950X](https://gist.github.com/cyring/1d43eef7dada2854f5fcdbe39ce0e5d8)/MPG X670E CARBON WIFI

nooberfsh, 2024-11-17:
```
tom@arch ~/h/p/CoreFreq ((7ee38f92))> sudo rdmsr -ax 0xc00110e3
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000
610bae00bc800000

```

nooberfsh, 2024-11-17:
Thanks for making a such great project.

cyring, 2024-11-17:
> ```
> tom@arch ~/h/p/CoreFreq ((7ee38f92))> sudo rdmsr -ax 0xc00110e3
> 610bae00bc800000
> ```
Although register is accessible and `SuppressBPOnNonBr` bit is clear, I'm finding no specification update about such vulnerability within Zen-5.

As a fix, I'm reserving to Zen-2 the aggregation in new commit 09a1d36a929b027a50a271456910a2935793feb8 which should state `[Missing]` for `BTC-NOBR` on your processor.