cyring/CoreFreq #512: AMD Ryzen 5 7600G

chendo, 2024-10-21:
Thank you for this project.

I've supplied the `corefreq-cli -s` for a Ryzen 5 7600G. Let me know if you'd like any other output.

```
Processor                                    [AMD Ryzen 5 7600 6-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601206]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [100.001]
|- Frequency            (MHz)                      Ratio                        
                 Min   3000.04                    <  30 >                       
                 Max   3800.05                    <  38 >                       
|- Factory                                                             [100.000]
                       3800                       [  38 ]                       
|- Performance                                                                  
                 TGT   3800.05                    <  38 >                       
   |- CPPC                                                                      
                 Min   3300.05                    <  33 >                       
                 Max    300.00                    <   3 >                       
                 TGT    300.00                    <   3 >                       
   |- Boost                                                            [ UNLOCK]
                 XFR   5100.07                    [  51 ]                       
                 CPB   5000.07                    [  50 ]                       
   |- P-State                                                                   
                 P1    3000.04                    <  30 >                       
|- Uncore                                                              [   LOCK]
                 CLK   1200.02                    [  12 ]                       
                 MEM   2400.03                    [  24 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
[... 154 lines cut ...]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                       Uncore   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                         DRAM   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                     Platform   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
   |- Power Limit                                                PL2   [    0 W]
   |- Time Window                                                TW2   [   0 ns]
|- Package Power Tracking                                        PPT   [Missing]
|- Electrical Design Current                                     EDC   [Missing]
|- Thermal Design Current                                        TDC   [Missing]
|- Core Thermal Point                                                           
|- Package Thermal Point                                                        
   |- Thermal Monitor Trip                                     Limit   [  115 C]
   |- HTC Temperature Limit                                    Limit   [  127 C]
   |- HTC Temperature Hysteresis                           Threshold   [    2 C]
|- Units                                                                        
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
```

cyring, 2024-10-22:
Hi
Thank you very much for your help.

Can you provide screenshots of the Sensors view (Vcore, Power, Temperature) when system is idle; single core stressed; all cores stressed ?
Please use one of the integrated `Tools > Atomic` and `Tools > Conics` 

I would like also to check the Memory Controller and other results:
```
corefreq-cli -k -n -B -n -M
```

Also various Caches and CPPC
```
corefreq-cli -m -n -Z
```

cyring, 2024-10-22:
Also some new feature bits are pending in the [`develop`](https://github.com/cyring/CoreFreq/commits/develop/) branch. Can you please show results of `corefreq-cli -s -n -m`

cyring, 2024-10-23:
By the way `7600G` with a `G` letter is a very mysterious processor name. I don't find its reference at AMD !

chendo, 2024-10-28:
Apologies for late reply, it's been a busy week!

Under Atomic load:

[image]

Under Conics load:

[image]

```
# corefreq-cli -k -n -B -n -M
Linux:                                                                          
|- Release                                                      [6.1.106-Unraid]
|- Version                 [#1 SMP PREEMPT_DYNAMIC Wed Aug 21 23:36:07 PDT 2024]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         31987648 KB
|- Shared RAM                                                          407348 KB
|- Free RAM                                                          28458664 KB
|- Buffer RAM                                                           13424 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
Clock Source                                                  <            hpet>
CPU-Freq driver                                               [      amd-pstate]
Governor                                                      [    conservative]
CPU-Idle driver                                               [         Missing]
|- Idle Limit                                                 [             N/A]

[ 0] American Megatrends Inc.                                                   
[ 1] 3056                                                                       
[ 2] 10/26/2024                                                                 
[ 3] ASUS                                                                       
[ 4] System Product Name                                                        
[ 5] System Version                                                             
[ 6] S---e---e---l---m---                                                       
[ 7] SKU                                                                        
[ 8] To be filled by O.E.M.                                                     
[ 9] ASUSTeK COMPUTER INC.                                                      
[10] PRIME X670-P WIFI                                                          
[11] Rev 1.xx                                                                   
[12] 2---3---9---1--                                                            
[13] Number Of Devices:4\Maximum Capacity:134217728 kilobytes                   
[14]                                                                            
[15] DIMM 1\P0 CHANNEL A                                                        
[16]                                                                            
[17]                                                                            
[18]                                                                            
[19] Kingston                                                                   
[20]                                                                            
[21]                                                                            
[22]                                                                            
[23] 9965794-016.A00G                                                           
[24]                                                                            
[25]                                                                            

                              Zen UMC  [14E0]                              
Controller #0                                               Single Channel 
 Bus Rate  2400 MHz       Bus Speed 2396 MHz           DDR5 Speed 4792 MT/s
                                                                           
 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   40   39   39   39   77  116    8   12   32    6   24   72    5   17 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   38   18   20    7    1    9    9    1    9    9    0    0    0    0 
      REFI RFC1 RFC2 RFCsb RCPB RPPB BGS:Alt  Ban  Page  CKE  CMD  GDM  ECC
  #0  9347  312  192  312   0    0    ON OFF  R0W0   0    0   1T    ON   0 
      MRD:PDA   MOD:PDA  WRMPR STAG PDM RDDATA WRD  WRL  RDL  XS   XP CPDED
  #0   34  32    34  32    24    7 0:F:1   28   6   26   34  732   18   12 
                                                                           
 DIMM Geometry for channel #0                                              
[... 51 lines cut ...]
        -R      Print System Registers
        -m      Print Topology
        -u      Print CPUID
        -B      Print SMBIOS
        -k      Print Kernel
        -n      New line
        -h      Print out this message
        -v      Print the version number

Exit status:
        0       SUCCESS         Successful execution
        3       CMD_SYNTAX      Command syntax error
        4       SHM_FILE        Shared memory file error
        5       SHM_MMAP        Shared memory mapping error
        6       PERM_ERR        Execution not permitted
        7       MEM_ERR         Memory operation error
        8       EXEC_ERR        General execution error
        9       SYS_CALL        System call error

Report bugs to labs[at]cyring.fr
```

I'm on Unraid so no easy way for me to get `develop` I don't think. 

My bad, I thought `G` is meant to signify onboard GPU. Could have sworn it was part of the model number at least some generations ago.

cyring, 2024-10-28:
Thank you very much

Sorry for the command line mistake to read the CPPC which uses a lower `z`
```
corefreq-cli -m -n -z
```

----

You may also have noticed the wrong DIMM size decoded
```
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1     65536      1024          16384    9965794-016.A00G
```
Size has to be `32 GB` and according to [Kingston datasheet](https://media.kingston.com/pcn/PCN_KSM48E40BD8KM-32HM_N.pdf) I should have decoded `2` ranks rather than `1`

I need to dump your 7600 Memory Controller Registers to understand things about DDR5 based on AMD Family 1A [specification](https://www.amd.com/content/dam/amd/en/documents/epyc-technical-docs/programmer-references/57238.zip) 57238_C1_pub_4.pdf chapter 9.3 UMC Registers

I made [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c) to read such registers.

Can you please compile and dump those addresses ?

```sh
## Compilation
cc zencli.c -o zencli

## Read as root
## (up to 4 theorical channels)

### DRAM Address Configuration 
./zencli smu 0x50040
./zencli smu 0x50044
./zencli smu 0x50048
./zencli smu 0x5004c

### DIMM Configuration
./zencli smu 0x50090
./zencli smu 0x50094

### UMC Configuration
./zencli smu 0x50100

### SDP Control
./zencli smu 0x50104
```

chendo, 2024-10-28:
```
# corefreq-cli -m -n -z
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0   0  0   0  0      32  8        32  8      1024  8 i   32768 16w 
001:  0    2   0  0   1  0      32  8        32  8      1024  8 i   32768 16w 
002:  0    4   0  0   2  0      32  8        32  8      1024  8 i   32768 16w 
003:  0    6   0  0   3  0      32  8        32  8      1024  8 i   32768 16w 
004:  0    8   0  1   4  0      32  8        32  8      1024  8 i   32768 16w 
005:  0   10   0  1   5  0      32  8        32  8      1024  8 i   32768 16w 
006:  0    1   0  0   0  1      32  8        32  8      1024  8 i   32768 16w 
007:  0    3   0  0   1  1      32  8        32  8      1024  8 i   32768 16w 
008:  0    5   0  0   2  1      32  8        32  8      1024  8 i   32768 16w 
009:  0    7   0  0   3  1      32  8        32  8      1024  8 i   32768 16w 
010:  0    9   0  1   4  1      32  8        32  8      1024  8 i   32768 16w 
011:  0   11   0  1   5  1      32  8        32  8      1024  8 i   32768 16w 

|- Collaborative Processor Performance Control                  CPPC       < ON>
   |- Capabilities     Lowest      Efficient     Guaranteed        Highest      
   |- CPU #0     299.35 (  3)  1895.89 ( 19)  2394.80 ( 24)  3492.42 ( 35)      
   |- CPU #1     299.36 (  3)  1895.93 ( 19)  2394.86 ( 24)  3592.30 ( 36)      
   |- CPU #2     299.44 (  3)  1896.45 ( 19)  2395.51 ( 24)  3593.26 ( 36)      
   |- CPU #3     299.37 (  3)  1895.99 ( 19)  2394.94 ( 24)  3492.62 ( 35)      
   |- CPU #4     299.37 (  3)  1896.02 ( 19)  2394.97 ( 24)  3293.08 ( 33)      
   |- CPU #5     299.44 (  3)  1896.44 ( 19)  2395.50 ( 24)  3393.63 ( 34)      
   |- CPU #6     299.40 (  3)  1896.20 ( 19)  2395.20 ( 24)  3493.00 ( 35)      
   |- CPU #7     299.40 (  3)  1896.22 ( 19)  2395.23 ( 24)  3592.84 ( 36)      
   |- CPU #8     299.44 (  3)  1896.46 ( 19)  2395.53 ( 24)  3593.30 ( 36)      
   |- CPU #9     299.41 (  3)  1896.25 ( 19)  2395.26 ( 24)  3493.09 ( 35)      
   |- CPU #10    299.41 (  3)  1896.24 ( 19)  2395.25 ( 24)  3293.47 ( 33)      
   |- CPU #11    299.44 (  3)  1896.46 ( 19)  2395.53 ( 24)  3393.67 ( 34)
```

Yeah, the RAM is meant to be 32GB. I'll have a crack at compiling `zencli` another time when I'm not fighting Docker

cyring, 2024-10-28:
Thank you.
Fyi, your outputs are listed at `Ryzen 5 7600G` of the [`Wiki/CPU-support`](https://github.com/cyring/CoreFreq/wiki/CPU-support)

cyring, 2024-10-28:
Could you please print the CPUID using `corefreq-cli -u`
The tree of `CPU #0` will be enough

_Because I want to decode the `CPUID_Fn80000026_ECX_x0[0...3] [Extended CPU Topology]` for CCX enumeration_

chendo, 2024-11-01:
```
CPU #0   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000010     68747541     444d4163     69746e65         
   |- Largest Standard Function=00000010                                        
|- 80000000:00000000    80000028     68747541     444d4163     69746e65         
   |- Largest Extended Function=80000028                                        
|- 00000001:00000000    00a60f12     000c0800     7ed8320b     178bfbff         
|- 00000002:00000000    00000000     00000000     00000000     00000000         
|- 00000003:00000000    00000000     00000000     00000000     00000000         
|- 00000004:00000000    00000000     00000000     00000000     00000000         
|- 00000004:00000001    00000000     00000000     00000000     00000000         
|- 00000004:00000002    00000000     00000000     00000000     00000000         
|- 00000004:00000003    00000000     00000000     00000000     00000000         
|- 00000005:00000000    00000040     00000040     00000003     00000011         
|- 00000006:00000000    00000004     00000000     00000001     00000000         
|- 00000007:00000000    00000001     f1bf97a9     00405fde     10000010         
|- 00000007:00000001    00000020     00000000     00000000     00000000         
|- 00000007:00000002    00000000     00000000     00000000     00000000         
|- 00000009:00000000    00000000     00000000     00000000     00000000         
|- 0000000a:00000000    00000000     00000000     00000000     00000000         
|- 0000000b:00000000    00000001     00000002     00000100     00000000         
|- 0000000d:00000000    000002e7     00000988     00000988     00000000         
|- 0000000d:00000001    0000000f     00000988     00001800     00000000         
|- 0000000d:00000002    00000100     00000240     00000000     00000000         
|- 0000000d:00000003    00000000     00000000     00000000     00000000         
|- 0000000d:00000004    00000000     00000000     00000000     00000000         
|- 0000000d:00000009    00000008     00000980     00000000     00000000         
|- 0000000d:0000000b    00000010     00000000     00000001     00000000         
|- 0000000d:0000000c    00000018     00000000     00000001     00000000         
|- 0000000d:0000003e    00000000     00000000     00000000     00000000         
|- 0000000f:00000000    00000000     000000ff     00000000     00000002         
|- 0000000f:00000001    00000014     00000040     000000ff     00000007         
|- 00000010:00000000    00000000     00000002     00000000     00000000         
|- 00000010:00000001    0000000f     00000000     00000004     0000000f         
|- 00000010:00000002    00000000     00000000     00000000     00000000         
|- 00000010:00000003    00000000     00000000     00000000     00000000         
|- 00000012:00000000    00000000     00000000     00000000     00000000         
|- 00000012:00000001    00000000     00000000     00000000     00000000         
|- 00000012:00000002    00000000     00000000     00000000     00000000         
|- 00000014:00000000    00000000     00000000     00000000     00000000         
|- 00000014:00000001    00000000     00000000     00000000     00000000         
|- 00000015:00000000    00000000     00000000     00000000     00000000         
|- 00000016:00000000    00000000     00000000     00000000     00000000         
|- 00000017:00000000    00000000     00000000     00000000     00000000         
|- 00000017:00000001    00000000     00000000     00000000     00000000         
|- 00000017:00000002    00000000     00000000     00000000     00000000         
|- 00000017:00000003    00000000     00000000     00000000     00000000         
|- 00000018:00000000    00000000     00000000     00000000     00000000         
|- 00000018:00000001    00000000     00000000     00000000     00000000         
|- 00000018:00000002    00000000     00000000     00000000     00000000         
|- 00000018:00000003    00000000     00000000     00000000     00000000         
|- 00000018:00000004    00000000     00000000     00000000     00000000         
|- 00000019:00000000    00000000     00000000     00000000     00000000         
|- 0000001a:00000000    00000000     00000000     00000000     00000000         
|- 0000001b:00000000    00000000     00000000     00000000     00000000         
|- 0000001c:00000000    00000000     00000000     00000000     00000000         
|- 0000001d:00000000    00000000     00000000     00000000     00000000         
|- 0000001d:00000001    00000000     00000000     00000000     00000000         
|- 0000001e:00000000    00000000     00000000     00000000     00000000         
|- 0000001f:00000000    00000000     00000000     00000000     00000000         
|- 00000020:00000000    00000000     00000000     00000000     00000000         
|- 00000023:00000000    00000000     00000000     00000000     00000000         
|- 00000023:00000001    00000000     00000000     00000000     00000000         
|- 00000023:00000002    00000000     00000000     00000000     00000000         
|- 00000023:00000003    00000000     00000000     00000000     00000000         
|- 80000001:00000000    00a60f12     00000000     75c237ff     2fd3fbff         
|- 80000002:00000000    20444d41     657a7952     2035206e     30303637         
|- 80000003:00000000    432d3620     2065726f     636f7250     6f737365         
[... cut ...]
|- 40000000:00000000    00000000     00000000     00000000     00000000         
|- 40000001:00000000    00000000     00000000     00000000     00000000         
|- 40000002:00000000    00000000     00000000     00000000     00000000         
|- 40000003:00000000    00000000     00000000     00000000     00000000         
|- 40000004:00000000    00000000     00000000     00000000     00000000         
|- 40000005:00000000    00000000     00000000     00000000     00000000         
|- 40000006:00000000    00000000     00000000     00000000     00000000         
```

cyring, 2024-11-01:
Thank you.
May ask you CPUID for all CPUs ?

cyring, 2024-11-01:
> " how to determine the number of CCD " question has been asked at [community.amd.com](https://community.amd.com/t5/pc-processors/how-to-determine-the-number-of-ccd/td-p/589275)

Refer to CPUID `Fn8000_0026` specs [1](https://www.amd.com/content/dam/amd/en/documents/processor-tech-docs/programmer-references/24593.pdf) [2](https://www.amd.com/content/dam/amd/en/documents/processor-tech-docs/programmer-references/24594.pdf)

cyring, 2024-11-01:
My understanding is that Core Complex in Raphael is available with 7950X but not with 7600

@tofurky and @amfern : can you please post all CPUID dump from your 7950X ?

```
corefreq-cli -u
```

tofurky, 2024-11-01:
> My understanding is that Core Complex in Raphael is available with 7950X but not with 7600
> 
> @tofurky and @amfern : can you please post all CPUID dump from your 7950X ?
> 
> ```
> corefreq-cli -u
> ```

[corefreq-cli_-u_7950x.txt](https://github.com/user-attachments/files/17601650/corefreq-cli_-u_7950x.txt)

amfern, 2024-11-01:
[corefreq-cli_-u_7950x.txt](https://github.com/user-attachments/files/17602361/corefreq-cli_-u_7950x.txt)

cyring, 2024-11-10:
@chendo 
Hello,
I really need the whole CPUID of your 7600 to debug the CPU cluster map
```
corefreq-cli -u
```
Thank you

cyring, 2024-11-11:
Since first Zen families (Ryzen, TR, EPYC) with less than 256 CPUs, I could _simply_ compute `CCD = (APIC_ID & 0xf0) >> 4`

In [wiki](https://github.com/cyring/CoreFreq/wiki/CPU-support), especially among EPYC, I don't have samples with more than 256 CPUs to evaluate the APIC ID, but if it relies in a range mask of `[0xfff - 0x100]` then `CCD = (APIC_ID & 0xf00) >> 8`

CCX should be derived from something. So far code is employing the Core ID but CCD would be more appropriated to derive cases of 4 or 8 cluster size.

I don't see how I can introduce code changes without those EPYC for non regression or improvement tests ?

cyring, 2024-11-13:
@amfern @tofurky @chendo 
Can you please have a try to the branch [develop_amd_ext_apic_id](https://github.com/cyring/CoreFreq/tree/develop_amd_ext_apic_id) and post the output of `corefreq-cli -m`

tofurky, 2024-11-13:
> @amfern @tofurky @chendo Can you please have a try to the branch [develop_amd_ext_apic_id](https://github.com/cyring/CoreFreq/tree/develop_amd_ext_apic_id) and post the output of `corefreq-cli -m`

```
root@aquos:/usr/src/corefreqk-1.98# corefreq-cli -m
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0   0  0   0  0      32  8        32  8      1024  8 i   65536 16w 
001:  0    2   0  0   1  0      32  8        32  8      1024  8 i   65536 16w 
002:  0    4   0  0   2  0      32  8        32  8      1024  8 i   65536 16w 
003:  0    6   0  0   3  0      32  8        32  8      1024  8 i   65536 16w 
004:  0    8   0  1   4  0      32  8        32  8      1024  8 i   65536 16w 
005:  0   10   0  1   5  0      32  8        32  8      1024  8 i   65536 16w 
006:  0   12   0  1   6  0      32  8        32  8      1024  8 i   65536 16w 
007:  0   14   0  1   7  0      32  8        32  8      1024  8 i   65536 16w 
008:  0   16   1  2   8  0      32  8        32  8      1024  8 i   65536 16w 
009:  0   18   1  2   9  0      32  8        32  8      1024  8 i   65536 16w 
010:  0   20   1  2  10  0      32  8        32  8      1024  8 i   65536 16w 
011:  0   22   1  2  11  0      32  8        32  8      1024  8 i   65536 16w 
012:  0   24   1  3  12  0      32  8        32  8      1024  8 i   65536 16w 
013:  0   26   1  3  13  0      32  8        32  8      1024  8 i   65536 16w 
014:  0   28   1  3  14  0      32  8        32  8      1024  8 i   65536 16w 
015:  0   30   1  3  15  0      32  8        32  8      1024  8 i   65536 16w 
016:  0    1   0  0   0  1      32  8        32  8      1024  8 i   65536 16w 
017:  0    3   0  0   1  1      32  8        32  8      1024  8 i   65536 16w 
018:  0    5   0  0   2  1      32  8        32  8      1024  8 i   65536 16w 
019:  0    7   0  0   3  1      32  8        32  8      1024  8 i   65536 16w 
020:  0    9   0  1   4  1      32  8        32  8      1024  8 i   65536 16w 
021:  0   11   0  1   5  1      32  8        32  8      1024  8 i   65536 16w 
022:  0   13   0  1   6  1      32  8        32  8      1024  8 i   65536 16w 
023:  0   15   0  1   7  1      32  8        32  8      1024  8 i   65536 16w 
024:  0   17   1  2   8  1      32  8        32  8      1024  8 i   65536 16w 
025:  0   19   1  2   9  1      32  8        32  8      1024  8 i   65536 16w 
026:  0   21   1  2  10  1      32  8        32  8      1024  8 i   65536 16w 
027:  0   23   1  2  11  1      32  8        32  8      1024  8 i   65536 16w 
028:  0   25   1  3  12  1      32  8        32  8      1024  8 i   65536 16w 
029:  0   27   1  3  13  1      32  8        32  8      1024  8 i   65536 16w 
030:  0   29   1  3  14  1      32  8        32  8      1024  8 i   65536 16w 
031:  0   31   1  3  15  1      32  8        32  8      1024  8 i   65536 16w 
```

cyring, 2024-11-13:
@tofurky Thank you 
Result is what I was expecting among CCD numbering
Please let me know if sensors (temps, voltage) are aligned with CCD clusters

tofurky, 2024-11-14:
> @tofurky Thank you Result is what I was expecting among CCD numbering Please let me know if sensors (temps, voltage) are not aligned with CCD clusters

In some brief testing when loading specific cores, it seemed correct.

cyring, 2024-11-14:
> > @tofurky Thank you Result is what I was expecting among CCD numbering Please let me know if sensors (temps, voltage) are not aligned with CCD clusters
> 
> In some brief testing when loading specific cores, it seemed correct.

Thank you for your answer

cyring, 2024-11-27:
One day refactoring the 6 Cores cluster topology will be necessary

Thanks

closed 2024-11-27 by cyring