cyring/CoreFreq #514: Intel(R) Core(TM) i9-14900K

valentin-barbotin, 2024-11-03:
Microcode 0x129
Z790 Aorus Elite AX (rev 1.1) with FJh BIOS version
https://www.aorus.com/motherboards/Z790-AORUS-ELITE-AX-rev-1x/Support
BIOS Pictures : https://drive.google.com/drive/folders/1QK8Sr8AuSifU_IauHk4uHpnr5zDtK9JB?usp=sharing

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

cyring, 2024-11-03:
_CoreFreq_ looks great with your processor.

**Thank you**

* You should try enabling CC6 and MC6 for further C-States and less consumed energy

* HWP Highest ratio of `73` has not been computed into a frequency.  

   Which _CoreFreq_ version are you using ?
   If not greater than `1.98.4`, can you update _CoreFreq_ and check again for this issue ?
   Also try to output in text using `corefreq-cli -z` to see if HWP frequencies are all printed or not ?

cyring, 2024-11-04:
Can you start or change for `Experimental` in `Settings` ?
Once enabled please find option `ODCM`, On Demand Clock Modulation, in "Power" window.
Activate this option and set the percent in the `BIAS HINT`
The processor frequency, when stressed, should reduce as much as the percent set in the bias.
Just deactivate `ODCM` to restore frequency to its nominal.

If the `ODCM` test succeed then I will mark as permanent for Raptor Lake in next CoreFreq version.

cyring, 2024-11-09:
> * HWP Highest ratio of `73` has not been computed into a frequency.
>   Which _CoreFreq_ version are you using ?
>   If not greater than `1.98.4`, can you update _CoreFreq_ and check again for this [...]

This UI issue is now fixed in commit cb55c1b72686772c1cec104ea269ef8f8cfd82c8

valentin-barbotin, 2024-11-12:
Corefreq version was 1.98.0, now using 1.98.4

corefreq-cli -z output :
```
|- Hardware-Controlled Performance States                        HWP       < ON>
   |- Capabilities     Lowest      Efficient     Guaranteed        Highest      
   |- CPU #0      99.60 (  1)  1792.73 ( 18)  4083.45 ( 41)     0.00 ( 73)      
   |- CPU #1      99.60 (  1)  1792.73 ( 18)  4083.45 ( 41)     0.00 ( 73)      
   |- CPU #2      99.59 (  1)  1792.58 ( 18)  4083.09 ( 41)     0.00 ( 73)      
   |- CPU #3      99.59 (  1)  1792.63 ( 18)  4083.20 ( 41)     0.00 ( 73)      
   |- CPU #4      99.59 (  1)  1792.60 ( 18)  4083.14 ( 41)     0.00 ( 73)      
   |- CPU #5      99.59 (  1)  1792.62 ( 18)  4083.20 ( 41)     0.00 ( 73)      
   |- CPU #6      99.59 (  1)  1792.63 ( 18)  4083.20 ( 41)     0.00 ( 73)      
   |- CPU #7      99.59 (  1)  1792.67 ( 18)  4083.30 ( 41)     0.00 ( 73)      
   |- CPU #8      99.60 (  1)  1792.73 ( 18)  4083.45 ( 41)     0.00 ( 77)      
   |- CPU #9      99.59 (  1)  1792.69 ( 18)  4083.36 ( 41)     0.00 ( 77)      
   |- CPU #10     99.60 (  1)  1792.73 ( 18)  4083.45 ( 41)     0.00 ( 77)      
   |- CPU #11     99.60 (  1)  1792.73 ( 18)  4083.45 ( 41)     0.00 ( 77)      
   |- CPU #12     99.60 (  1)  1792.84 ( 18)  4083.68 ( 41)     0.00 ( 73)      
   |- CPU #13     99.60 (  1)  1792.81 ( 18)  4083.62 ( 41)     0.00 ( 73)      
   |- CPU #14     99.60 (  1)  1792.73 ( 18)  4083.45 ( 41)     0.00 ( 73)      
   |- CPU #15     99.60 (  1)  1792.73 ( 18)  4083.45 ( 41)     0.00 ( 73)      
   |- CPU #16     99.60 (  1)  1394.35 ( 14)  2390.31 ( 24)  4382.23 ( 44)      
   |- CPU #17     99.60 (  1)  1394.35 ( 14)  2390.32 ( 24)  4382.25 ( 44)      
   |- CPU #18     99.60 (  1)  1394.38 ( 14)  2390.37 ( 24)  4382.34 ( 44)      
   |- CPU #19     99.59 (  1)  1394.33 ( 14)  2390.27 ( 24)  4382.17 ( 44)      
   |- CPU #20     99.60 (  1)  1394.35 ( 14)  2390.31 ( 24)  4382.24 ( 44)      
   |- CPU #21     99.60 (  1)  1394.35 ( 14)  2390.31 ( 24)  4382.24 ( 44)      
   |- CPU #22     99.60 (  1)  1394.36 ( 14)  2390.33 ( 24)  4382.27 ( 44)      
   |- CPU #23     99.60 (  1)  1394.36 ( 14)  2390.34 ( 24)  4382.28 ( 44)      
   |- CPU #24     99.60 (  1)  1394.35 ( 14)  2390.32 ( 24)  4382.25 ( 44)      
   |- CPU #25     99.60 (  1)  1394.35 ( 14)  2390.31 ( 24)  4382.24 ( 44)      
   |- CPU #26     99.60 (  1)  1394.45 ( 14)  2390.49 ( 24)  4382.56 ( 44)      
   |- CPU #27     99.60 (  1)  1394.36 ( 14)  2390.33 ( 24)  4382.26 ( 44)      
   |- CPU #28     99.60 (  1)  1394.35 ( 14)  2390.31 ( 24)  4382.24 ( 44)      
   |- CPU #29     99.60 (  1)  1394.35 ( 14)  2390.32 ( 24)  4382.25 ( 44)      
   |- CPU #30     99.60 (  1)  1394.34 ( 14)  2390.30 ( 24)  4382.21 ( 44)      
   |- CPU #31     99.60 (  1)  1394.35 ( 14)  2390.32 ( 24)  4382.25 ( 44)      

```

[corefreq_screens_odcm.zip](https://github.com/user-attachments/files/17723900/corefreq_screens_odcm.zip)

cyring, 2024-11-13:
> Corefreq version was 1.98.0, now using 1.98.4
* Thanks for your return.
* For your testings, a fix is available since commit cb55c1b72686772c1cec104ea269ef8f8cfd82c8

----

ODCM can be safely enable/disable
Per `12.5%` step and apparently only on Ecores whereas Pcores remain at `7 MHz`
ODCM feature will be marked as no more experimental in next release.

valentin-barbotin, 2024-11-14:
[image]

I am now using version 1.98.5, thanks to the package maintainer
I haven't rebooted yet but I have reloaded the service and the module and it seems the problem still persists
Here is the proof with the version of CoreFreq displayed

cyring, 2024-11-15:
> I am now using version 1.98.5, thanks to the package maintainer I haven't rebooted yet but I have reloaded the service and the module and it seems the problem still persists Here is the proof with the version of CoreFreq displayed

Is it possible that you try directly from the repository rather than a package ?
Make sure to unload/uninstall all previous versions before
```sh
git clone -b develop https://github.com/cyring/CoreFreq.git

cd CoreFreq
make
```
* As root
```
insmod build/corefreqk.ko
./build/corefreqd
```
* As User
```
./build/corefreq-cli
```

cyring, 2024-11-15:
@valentin-barbotin 
Sorry I should have tell you that fix is available since two commits above master
So as shown just above, you have to pull code from the [develop branch](https://github.com/cyring/CoreFreq/commits/develop)

cyring, 2024-11-16:
> It's better with right commit :)

Thank you very much

cyring, 2024-12-24:
For production

closed 2024-12-24 by cyring