cyring/CoreFreq #517: AMD Ryzen 7 9800X3D

notpentadactyl, 2024-11-19:
`corefreq-cli -s`
```
Processor                                 [AMD Ryzen 7 9800X3D 8-Core Processor]
|- Architecture                                             [Zen5/Granite Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0b404023]
|- Signature                                                           [  BF_44]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 16/ 16]
|- Base Clock                                                          [100.003]
|- Frequency            (MHz)                      Ratio
                 Min   3000.09                    <  30 >
                 Max   4700.14                    <  47 >
|- Factory                                                             [100.000]
                       4700                       [  47 ]
|- Performance
                 TGT   4700.14                    <  47 >
   |- CPPC
                 Min   3400.10                    <  34 >
                 Max   2200.09                    <  22 >
                 TGT      AUTO                    <   0 >
   |- Boost                                                            [ UNLOCK]
                 XFR   5200.16                    [  52 ]
                 CPB   5200.16                    [  52 ]
   |- P-State
                 P1    3000.09                    <  30 >
|- Uncore                                                              [   LOCK]

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y]
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y]
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [Y]
|- AVX512-BF16  [Y] AVX-VNNI-VEX [Y]    AVX-FP128 [N]   AVX-FP256 [N]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y]
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP CMPSB|SCASB                                   FSRC   [Capable]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast Short REP STOSB                                         FSRS   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
|- Fused Multiply Add                                            FMA   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hyper-Threading Technology                                    HTT   [Capable]
[... 265 lines cut ...]
   |- CPU #7     400.01 (  4)  1900.03 ( 19)  3000.04 ( 30)  3400.05 ( 34)
   |- CPU #8     400.00 (  4)  1900.01 ( 19)  3000.01 ( 30)  4000.02 ( 40)
   |- CPU #9     400.00 (  4)  1900.01 ( 19)  3000.02 ( 30)  4000.02 ( 40)
   |- CPU #10    400.00 (  4)  1900.01 ( 19)  3000.01 ( 30)  3600.01 ( 36)
   |- CPU #11    400.00 (  4)  1899.99 ( 19)  2999.99 ( 30)  3699.98 ( 37)
   |- CPU #12    400.00 (  4)  1900.01 ( 19)  3000.02 ( 30)  3800.02 ( 38)
   |- CPU #13    400.00 (  4)  1900.01 ( 19)  3000.01 ( 30)  3900.02 ( 39)
   |- CPU #14    400.00 (  4)  1899.99 ( 19)  2999.99 ( 30)  3499.99 ( 35)
   |- CPU #15    400.00 (  4)  1900.00 ( 19)  3000.01 ( 30)  3400.01 ( 34)
```

`corefreq-cli -u`
[corefreq-cli-u.txt](https://github.com/user-attachments/files/17815100/corefreq-cli-u.txt)

Idle:

[image]

Atomic:

[image]

Turbo CPU0:

[image]

cyring, 2024-11-19:
@notpentadactyl **Thank you very much** for this first 9800X3D report.

* Unfortunately, no Power Limit shown because no HSMP protocol on this model
* Memory Controller and IOMMU are not printed 
  1- Can you try from the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch in which I have recently pushed a PCI list change
  2- Rebuild, reload module and post `corefreq-cli -s -n -M`
  3- In case, there is still no output from Memory Controller, please post the PCI identifiers from  command `lspci -nn`

notpentadactyl, 2024-11-19:
@cyring no problem!

https://github.com/cyring/CoreFreq/tree/develop doesn't seem to like my kernel `linux-cachyos-autofdo 6.12.rc7-1`: 
[make.log](https://github.com/user-attachments/files/17816518/make.log)

I need 6.12 for ethernet on this mainboard.

That being said: I could build it on kernel `linux-cachyos 6.11.9-2`:
`corefreq-cli -s -n -M`
```
Processor                                 [AMD Ryzen 7 9800X3D 8-Core Processor]
|- Architecture                                             [Zen5/Granite Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0b404023]
|- Signature                                                           [  BF_44]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 16/ 16]
|- Base Clock                                                          [ 41.229]
|- Frequency            (MHz)                      Ratio
                 Min   3628.17                    <  88 >
                 Max   4700.13                    < 114 >
|- Factory                                                             [100.000]
                      11400                       [ 114 ]
|- Performance
                 TGT   4700.13                    < 114 >
   |- CPPC
                 Min   3504.49                    <  85 >
                 Max   2020.25                    <  49 >
                 TGT      AUTO                    <   0 >
   |- Boost                                                            [   LOCK]
                 XFR   4700.13                    [ 114 ]
                 CPB   4700.13                    [ 114 ]
   |- P-State
                 P1    3628.17                    <  88 >
|- Uncore                                                              [   LOCK]
                 CLK    618.44                    [  15 ]
                 MEM   1236.88                    [  30 ]

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y]
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y]
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [Y]
|- AVX512-BF16  [Y] AVX-VNNI-VEX [Y]    AVX-FP128 [N]   AVX-FP256 [N]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y]
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
[... 185 lines cut ...]
Controller #0                                                Dual Channel
 Bus Rate  3000 MHz       Bus Speed 1236 MHz           DDR5 Speed 2472 MT/s

 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   30   40   40   40   96  137    8   15   32    8   30   96    8   23
  #1   30   40   40   40   96  137    8   15   32    8   30   96    8   23
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   28   23   22    8    1   11   11    1   11   11    0    0    0    0
  #1   28   23   23    8    1   11   11    1   11   11    0    0    0    0
      REFI RFC1 RFC2 RFCsb RCPB RPPB BGS:Alt  Ban  Page  CKE  CMD  GDM  ECC
  #0 11677  312  192  391   0    0    ON OFF  R0W0   0    0   1T    ON   0
  #1 11677  312  192  391   0    0    ON OFF  R0W0   0    0   1T    ON   0
      MRD:PDA   MOD:PDA  WRMPR STAG PDM RDDATA WRD  WRL  RDL  XS   XP CPDED
  #0   42  32    42  32    24    7 0:F:0   18   6   16   36  914   23   15
  #1   42  32    42  32    24    7 0:F:0   18   6   16   36  914   23   15

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    32    1     65536      1024          16384    F5-6000J3040G32G
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    32    1     65536      1024          16384    F5-6000J3040G32G
```

cyring, 2024-11-19:
> https://github.com/cyring/CoreFreq/tree/develop doesn't seem to like my kernel `linux-cachyos-autofdo 6.12.rc7-1`: [make.log](https://github.com/user-attachments/files/17816518/make.log)

The make log shows '/usr/src/corefreq-1.98.1.r0.ge8fe368'
But the develop branch is currently `1.98.6`
However I will try some non-regression tests with latest Cachy

I can see that your last build is now showing the UMC, some Uncore ratios, the IOMMU.

## Remaining issues
* The DIMM size is wrong (compared to kernel reported size)

* The estimated baseclock has suddenly fall down to `41 Mhz`
  This is something I had fixed for 9950X in this [issue](https://github.com/cyring/CoreFreq/issues/505#issuecomment-2308182722)
  
  Since Zen5, specification of coefficient of frequency [`CpuFid`](https://github.com/cyring/CoreFreq/blob/c75af85b5bc148c364e2e55e92e79a98eab4ce8e/x86_64/amd_reg.h#L549) has been expand to 12 bits and it is handled by function [`AMD_Zen_CoreCOF()`](https://github.com/cyring/CoreFreq/blob/ed5feaa4be1dc5de3958280af1d3e80f203fa002/x86_64/corefreqk.c#L8097C15-L8097C31) 
  
  But for some reasons your Processor has not follow the `switch()` case of Family `0xB` as I have programmed it for 9950X
  
  Can you dump the first three P-State registers ?
```sh
rdmsr -ax 0xC0010064
rdmsr -ax 0xC0010065
rdmsr -ax 0xC0010066
```

notpentadactyl, 2024-11-19:
>       But for some reasons your Processor has not follow the `switch()` case of Family `0xB` as I have programmed it for 9950X
>       Can you dump the first three P-State registers ?
> 
> 
> ```shell
> rdmsr -ax 0xC0010064
> ```
```
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
800000004bf243ac
```
> ```shell
> rdmsr -ax 0xC0010065
> ```
```
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
80000000479e4258
```
> ```shell
> rdmsr -ax 0xC0010066
> ```
```
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
```

cyring, 2024-11-19:
@notpentadactyl 

Found this issue in `0x800000004bf243ac`
```
1000000000000000000000000000000001001011111100100100001110101100
```
where I was expecting bits `[14-8]` to be greater than `0b111` but 9800X3D is `0b000011`
I will submit a fix soon.

cyring, 2024-11-19:
> edit: the reported `DDR5 Speed 2472 MT/s`

Yes, it is linked with the COF function whom fix is in progress.

cyring, 2024-11-19:
@notpentadactyl Hello,

For your testings, a fix is available in commit c574ab115b5aea6328377d651fc84dadc3a93ba6 of `develop` branch
It provides both reading and programming the frequency ratio within Zen5 P-State.

Save you files before this test. Processor may freeze.

## To modify a P-State:
1- Disable Core Performance Boost in the Technologies window
[image]

2- In Processor window, go to the P-State and press `[Enter]`
[image]

3- In the opened window, keep the top row `1C` selected for whole processor (aka all cores)
[image]

4- At this time of the test, just add carefully one bin to the ratio.
  (My case for example, I'm going from `28` to `29`) 
  Press `[Enter]` to submit ratio.
[image]

5- Window is refreshed with new ratio. Press `[Escape]`
[image]

6- P-State is also updated in Processor window
[image]

7- CPB can now be restored to enabled
[image]

If things are OK, P-state have been programmed; Base Clock should still be evaluated to `100 Mhz`

cyring, 2024-11-20:
> https://github.com/cyring/CoreFreq/tree/develop doesn't seem to like my kernel `linux-cachyos-autofdo 6.12.rc7-1`:
> [make.log](https://github.com/user-attachments/files/17816518/make.log)

About CachyOS; here is how I have built and run _CoreFreq_ with.

1. CachyOS on top of ArchLinux
```sh
git clone git@github.com:CachyOS/linux-cachyos.git

cd linux-cachyos/linux-cachyos-autofdo/

makepkg -cCsfir

ls /boot/
```
```
initramfs-linux-cachyos-autofdo-fallback.img
initramfs-linux-cachyos-autofdo.img
vmlinuz-linux-cachyos-autofdo
```

2. Install CachyOS in EFI
```sh
cd /efi/loader/entries/

cat cachyos.conf
```
```
title   Arch Linux [CachyOS]
linux   /EFI/Linux/vmlinuz-linux-cachyos-autofdo
initrd  /EFI/Linux/amd-ucode.img
initrd  /EFI/Linux/initramfs-linux-cachyos-autofdo.img
options root=/dev/disk/by-label/root rw quiet break=n add_efi_memmap loglevel=3 rd.systemd.show_status=auto rd.udev.log-priority=3 consoleblank=0 vt.color=0x03 modprobe.blacklist=pcspkr,joydev,mousedev audit=0 amdgpu.ppfeaturemask=0xffffffff initcall_blacklist=acpi_cpufreq_init amd_pstate.shared_mem=1 idle=halt amd_pstate=guided
```
* Boot CachyOS kernel

3. Build _CoreFreq_
```sh
git clone -b develop git@github.com:cyring/CoreFreq.git

cd CoreFreq
make clean
make -j CC=clang LD=ld.lld
```

4. Run _CoreFreq_
```sh
## as root
insmod build/corefreqk.ko
./build/corefreqd -d

## as User
./build/corefreq-cli
```

[image]

notpentadactyl, 2024-11-20:
@cyring thanks, seem to work, in this example I went from `30` to `31`.

[image]

Memory controller also looks better:

[image]

Anything else I should check?

cyring, 2024-11-20:
> @cyring thanks, seem to work, in this example I went from `30` to `31`.
> 
> ![image](https://private-user-images.githubusercontent.com/21175961/388103598-98a631b2-a139-4509-9ee1-98a5e4a4fa8c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxMTE2NzAsIm5iZiI6MTczMjExMTM3MCwicGF0aCI6Ii8yMTE3NTk2MS8zODgxMDM1OTgtOThhNjMxYjItYTEzOS00NTA5LTllZTEtOThhNWU0YTRmYThjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIwVDE0MDI1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE3OTUxNmNmM2U0NGE2OWQ3YzVjMjZmMTI1NzZjMGNjYjk3ZWRhYzNhNjYyMzMxOGY4YjdlNmZhZjBlNWY5OWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nVU-61jISn4cpgPSaKiCKI-zMAbvjSKkT7xKkkzjyMs)
> 
> Memory controller also looks better:
> 
> ![image](https://private-user-images.githubusercontent.com/21175961/388103827-cb523b87-e5b8-4ba5-9521-132245eb74a5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxMTE2NzAsIm5iZiI6MTczMjExMTM3MCwicGF0aCI6Ii8yMTE3NTk2MS8zODgxMDM4MjctY2I1MjNiODctZTViOC00YmE1LTk1MjEtMTMyMjQ1ZWI3NGE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIwVDE0MDI1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyNTQ3YjRkZWQ3NmNjYzk4MDM4ODVjODAyNTIwMjQ0ZWE3N2FjZGE3NTJlN2NkNDk1OTU5YjJlZTdjZTY3OWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VOuVwxnUZskRwBTtogHHt7KMCxo2ct67GE34KOFvIHc)
> 
> Anything else I should check?
> 
> edit: I guess the individual core temps are wrong, they all show the same, hwinfo on windows looks more accurate.

Happy to see the change worked fine.
I'm gonna merge it into `master` branch

DIMM size is still something I have to fix.

About temperature sensors, at most _CoreFreq_ can provide values per CCX or CCD cluster.
Could you stress various single CPU and check if high temp is the same for the same cluster group ?

notpentadactyl, 2024-11-20:
> DIMM size is still something I have to fix.
> 
> About temperature sensors, at most _CoreFreq_ can provide values per CCX or CCD cluster. Could you stress various single CPU and check if high temp is the same for the same cluster group ?

[image]

I guess it kinda works.

Bus Rate on Memory Controller is empty, unsure what it even is, but probably should show 3000 MHz?

cyring, 2024-11-20:
> ![image](https://private-user-images.githubusercontent.com/21175961/388130806-7ec3d2e3-e887-40ce-bd48-734a36e814f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

* Unfortunately we are only getting the same temperature for all cores. They belong to a single CCD. 
About accuracy, I wonder if your temperature reading with _CoreFreq_ differs a lot or not from your Windows tools ?

* Bus rate: indeed it has to show `3000 MHz` like your previous [output](https://github.com/cyring/CoreFreq/issues/517#issuecomment-2485866099).
  * Is it also missing in `corefreq-cli -M` ?
  * Can it be a UI bug only ?

notpentadactyl, 2024-11-20:
> * Unfortunately we are only getting the same temperature for all cores. They belong to a single CCD.
>       About accuracy, I wonder if your temperature reading with _CoreFreq_ differs a lot or not [...]

No, temps are close, I was just talking about per core temps, which is something hwinfo can do, but not too important. :+1: 

>     * Bus rate: indeed it has to show `3000 MHz` like your previous [output](https://github.com/cyring/CoreFreq/issues/517#issuecomment-2485866099).
>       
>       * Is it also missing in `corefreq-cli -M` ?
>       * Can it be a UI bug only ?

You are correct!

`./build/corefreq-cli -M`
```
                              Zen UMC  [14E0]
Controller #0                                                Dual Channel
 Bus Rate  3000 MHz       Bus Speed 3000 MHz           DDR5 Speed 6000 MT/s

 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   30   40   40   40   96  137    8   15   32    8   30   96    8   23
  #1   30   40   40   40   96  137    8   15   32    8   30   96    8   23
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   28   23   22    8    1   11   11    1   11   11    0    0    0    0
  #1   28   23   23    8    1   11   11    1   11   11    0    0    0    0
      REFI RFC1 RFC2 RFCsb RCPB RPPB BGS:Alt  Ban  Page  CKE  CMD  GDM  ECC
  #0 11677  312  192  391   0    0    ON OFF  R0W0   0    0   1T    ON   0
  #1 11677  312  192  391   0    0    ON OFF  R0W0   0    0   1T    ON   0
      MRD:PDA   MOD:PDA  WRMPR STAG PDM RDDATA WRD  WRL  RDL  XS   XP CPDED
  #0   42  32    42  32    24    7 0:F:0   18   6   16   36  914   23   15
  #1   42  32    42  32    24    7 0:F:0   18   6   16   36  914   23   15

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    32    1     65536      1024          16384    F5-6000J3040G32G
 DIMM Geometry for channel #1
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0
       #1    32    1     65536      1024          16384    F5-6000J3040G32G
```
>
>     * Edit: was my above CachyOS procedure helpful to solve your build issue ?
Yeah, I was tying to use the aur git package but failed. I think it is mostly the make patch that is broken with current `develop` branch.

cyring, 2024-11-20:
@notpentadactyl 
I don't how to reproduce the UI issue with missing `Bus Rate`

Can you tell a bit more about which Terminal, which Desktop Environment or Window Manager ?

Is moving selection cursor to missing cell makes value appeared ?

Can try `corefreq-cli` in a plain XTerm and see if issue happens ?

notpentadactyl, 2024-11-20:
> @notpentadactyl I don't how to reproduce the UI issue with missing `Bus Rate`
> 
> Can you tell a bit more about which Terminal, which Desktop Environment or Window Manager ?

alacritty 0.14.0 (22a44757)
KDE Plasma 6.2.3 (kwin 6.2.3)

Happens in Konsole 24.08.3 too.

> Is moving selection cursor to missing cell makes value appeared ?
> 
No.

> Can try `corefreq-cli` in a plain XTerm and see if issue happens ?

https://github.com/user-attachments/assets/e1a96d56-435e-4cd6-b499-86587600c731

Also unsure why it is no longer reporting 3000 MHz/6000 MT/s (`corefreq-cli -M` does the same).
```
~/git/CoreFreq develop ❯ ./build/corefreq-cli -M | grep Bus
 Bus Rate  3000 MHz       Bus Speed 2997 MHz           DDR5 Speed 5994 MT/s
```

Here is `dmidecode`:
```
	Speed: 4800 MT/s
	Configured Memory Speed: 6000 MT/s
	Speed: 4800 MT/s
	Configured Memory Speed: 6000 MT/s
```

cyring, 2024-11-20:
> > @notpentadactyl I don't how to reproduce the UI issue with missing `Bus Rate`
> > Can you tell a bit more about which Terminal, which Desktop Environment or Window Manager ?
> 
> alacritty 0.14.0 (22a44757) KDE Plasma 6.2.3 (kwin 6.2.3)
> 
> Happens in Konsole 24.08.3 too.
> 
> > Is moving selection cursor to missing cell makes value appeared ?
> 
> No.
> 
> > Can try `corefreq-cli` in a plain XTerm and see if issue happens ?
> 
>  corefreq-cli_xterm.mp4 
> Also unsure why it is no longer reporting 3000 MHz/6000 MT/s (`corefreq-cli -M` does the same).
> 
> ```
> ~/git/CoreFreq develop ❯ ./build/corefreq-cli -M | grep Bus
>  Bus Rate  3000 MHz       Bus Speed 2997 MHz           DDR5 Speed 5994 MT/s
> ```
> 
> Here is `dmidecode`:
> 
> ```
> 	Speed: 4800 MT/s
> 	Configured Memory Speed: 6000 MT/s
> 	Speed: 4800 MT/s
> 	Configured Memory Speed: 6000 MT/s
> ```

Thanks, it helps a lot.

I think UI is rendering an empty cell where `freq` is the word behind the window and it has fused in foreground.
I have to find a way to reproduce this issue.

Bus speed and DDR speed are based on Bus rate, normalized with the Processor  base clock. That's why you're getting respectively `2997` and `5994`

cyring, 2024-11-21:
@notpentadactyl Hi,

Looking at the video:
Had the monitoring been stopped and started while Memory Controller window being displayed ?
For example, using shortkeys `]` and `[`

LethalManBoob, 2024-11-29:
Hi, Sorry this is a lot to read. Is the 9800x3d currently working or is this some isolated bug?

cyring, 2024-11-30:
> Hi, Sorry this is a lot to read. Is the 9800x3d currently working or is this some isolated bug?

Hi
Beside a non-reproductible UI glitch and wrong DIMM geometry, `9800X3D` is [supported](https://gist.github.com/cyring/1d43eef7dada2854f5fcdbe39ce0e5d8)

cyring, 2024-12-24:
For production

closed 2024-12-24 by cyring