cyring/CoreFreq #519: Almalinux 9 (again): error: assignment of read-only member ‘vm_flags’

fishbonesnl, 2024-12-19:
Hello again,

I downloaded the latest master.zip and tried to compile it on the latest Almalinux 9, starting with a clean install.
I get the follow error message during make:

`~/CoreFreq/build/module/corefreqk.c: In function ‘CoreFreqK_mmap’:
~/CoreFreq/build/module/corefreqk.c:23538:23: error: assignment of read-only member ‘vm_flags’
23538 |         vma->vm_flags = VM_READ;
      |                       ^
compilation terminated due to -Wfatal-errors.
`

cyring, 2024-12-19:
> Hello again,
> 
> I downloaded the latest master.zip and tried to compile it on the latest Almalinux 9, starting with a clean install. I get the follow error message during make:
> ...

Please pull `master` again to get the fix for "Teal Serval"

[image]

"Seofoam Ocelot" non regression test

[image]

fishbonesnl, 2024-12-20:
I confirm that it works, thanks :+1:

closed 2024-12-22 by cyring