cyring/CoreFreq #530: Why the Makefile is so unstable?

inglor, 2025-02-22:
Hello,

I've been maintaining corefreq in AUR for some time now and latest releases have the Makefile modified. (1.97.1, 1.98.1 1.98.5). One of those time it was re-written completely? Why is it so unstable?

I'm asking because I've been maintaining a patch on top of it to honor the distribution specific (sane) default and it was no problem when the Makefile is stable but recently it's not and it's a pain to figure out again the changes, especially once it's re-written.

Also I understand if you still figuring out Makefile syntax and config, just trying to understand why.

Thanks,
artafinde

cyring, 2025-02-23:
Hello,

Main changes from current `master` branch explained bellow.

## `1.98.5`

* From `git diff 1.98.5 Makefile`

### Explanations

1. Commit 8597fe1dca6ec57d429e28cf107436a967f34851 to allow User change the `WARNING`
1. `make info` prints other variables 

<details close>

```patch
diff --git a/Makefile b/Makefile
index 887ffea..baa0901 100644
--- a/Makefile
+++ b/Makefile
@@ -1,13 +1,13 @@
 # CoreFreq
-# Copyright (C) 2015-2024 CYRIL COURTIAT
+# Copyright (C) 2015-2025 CYRIL COURTIAT
 # Licenses: GPL2
 
-COREFREQ_MAJOR = 1
-COREFREQ_MINOR = 98
-COREFREQ_REV = 5
+COREFREQ_MAJOR = 2
+COREFREQ_MINOR = 0
+COREFREQ_REV = 1
 HW = $(shell uname -m)
 CC ?= cc
-WARNING = -Wall -Wfatal-errors
+WARNING ?= -Wall -Wfatal-errors
 SYMLINK ?= ln -s
 INSTALL ?= install
 DEPMOD ?= depmod
@@ -318,6 +318,10 @@ info:
        $(info FEAT_DBG [$(FEAT_DBG)])
        $(info DELAY_TSC [$(DELAY_TSC)])
        $(info OPTIM_LVL [$(OPTIM_LVL)])
+       $(info CORE_COUNT [$(CORE_COUNT)])
+       $(info TASK_ORDER [$(TASK_ORDER)])
+       $(info MAX_FREQ_HZ [$(MAX_FREQ_HZ)])
+       $(info HWM_CHIPSET [$(HWM_CHIPSET)])
        $(info MSR_CORE_PERF_UCC [$(MSR_CORE_PERF_UCC)])
        $(info MSR_CORE_PERF_URC [$(MSR_CORE_PERF_URC)])
        $(info ARCH_PMC [$(ARCH_PMC)])
@@ -341,7 +345,7 @@ help:
        "|    where <COMPILER> is cc, gcc, clang                         |\n"\
        "|                                                               |\n"\
        "|  WARNING=<ARG>                                                |\n"\
-       "|    where default argument is -Wall                            |\n"\
+       "|    where default argument is -Wall -Wfatal-errors             |\n"\
        "|                                                               |\n"\
        "|  KERNELDIR=<PATH>                                             |\n"\
        "|    where <PATH> is the Kernel source directory                |\n"\
```

</details>

----

## `1.98.1` 

* From `git diff 1.98.1 Makefile`

### Explanations
[... 421 lines cut ...]
        "|  CC=<COMPILER>                                                |\n"\
        "|    where <COMPILER> is cc, gcc, clang                         |\n"\
        "|                                                               |\n"\
@@ -378,13 +332,12 @@ help:
        "|      when <F> is 1: don't build and display this area part    |\n"\
        "|    UI_TRANSPARENCY=<F>                                        |\n"\
        "|      when <F> is 1: build with background transparency        |\n"\
-       "|    UI_RULER_MINIMUM=<N>, UI_RULER_MAXIMUM=<N>                 |\n"\
-       "|      set ruler left or right bound to <N> frequency ratio     |\n"\
        "|                                                               |\n"\
        "|  Example:                                                     |\n"\
-       "|    make CC=gcc OPTIM_LVL=3 FEAT_DBG=1 ARCH_PMC=PCU \\          |\n"\
-       "|         MSR_CORE_PERF_UCC=MSR_CORE_PERF_FIXED_CTR1 \\          |\n"\
-       "|         MSR_CORE_PERF_URC=MSR_CORE_PERF_FIXED_CTR2 \\          |\n"\
-       "|         HWM_CHIPSET=W83627 MAX_FREQ_HZ=5350000000  \\          |\n"\
+       "|    make CC=gcc OPTIM_LVL=3 FEAT_DBG=1 ARCH_PMC=PCU            |\n"\
+       "|         MSR_CORE_PERF_UCC=MSR_CORE_PERF_FIXED_CTR1            |\n"\
+       "|         MSR_CORE_PERF_URC=MSR_CORE_PERF_FIXED_CTR2            |\n"\
+       "|         HWM_CHIPSET=W83627 MAX_FREQ_HZ=5350000000             |\n"\
        "|         CORE_COUNT=1024 NO_FOOTER=1 NO_UPPER=1                |\n"\
+       "|         clean all                                             |\n"\
        "o---------------------------------------------------------------o"
```

</detail>

cyring, 2025-03-02:
@inglor: Hello
Do you still have an issue ?

cyring, 2025-03-02:
> [@cyring](https://github.com/cyring) You provided context on what changes are doing on Makefile which I could (and was) already knew from reading the commits. In fact I already knew what each of those are doing because I needed to re-write my patch for each of them.
> 
> My issue (question) was more why the Makefile is not mostly static as per most of projects. This raises another topic (again) about why this project doesn't take into account environmental variables set from user when calling the make commands which is something most projects do. I've raised this before and didn't get a clear answer hence I was patching the Makefile myself. I understand if you want it your way of doing things but this just raises questions downstream (packaging in distros) on why this is the case, hence my question.

I tried to find a way for a building solution which pleases the various distros.
Pull Requests are welcomed on this project ; although proposed changes to the Makefile have to cover more than one disto and be retro-compatible down to old version 3 kernels. 
The Makefile is also used by the Live CD.
I'm truly opened to any PR

cyring, 2025-03-16:
Hello,

Among new changes, Makefile is impacted at these lines
https://github.com/cyring/CoreFreq/blob/357b4b8c280731cd615013df95c242b67d112cf1/Makefile#L99
This a list of device tree strings which identify a virtual machine. List is shared between the 3 ISA: aarch64, riscv, ppc64.
Purpose is to extend that list at will in future.

CyrIng