cyring/CoreFreq #532: Kernel module crashes on Debian 12 VM running under Android 16

bubbleguuum, 2025-03-11:
Hi,

Android 16 added a [Terminal app ](https://www.androidauthority.com/android-linux-terminal-app-available-3532999/) which runs a full blown ARM Debian 12 VM  (thus with its own Debian kernel). It uses Google own virtualization framework on Android.
On my Pixel 8 Pro on Android 16 Beta 2.1, I compiled CoreFreq 2.0.1 successfully in that VM.
But when I `insmod codefreqk.ko`, it crashes on an "Undefined instruction" with dmesg reporting:

```
 [  +0.010350] Internal error: Oops - Undefined instruction: 0000000002000000 [#1] SMP
[  +0.007406] Modules linked in: corefreqk(OE+) binfmt_misc(E) nls_ascii(E) nls_cp437(E) vfat(E) fat(E) aes_ce_blk(E) aes_ce_cipher(E) crct10dif_ce(E) polyval_ce(E) polyval_generic(E) ghash_ce(E) gf128mul(E) sha3_ce(E) sha3_generic(E) sha512_ce(E) sha512_arm64(E) sha2_ce(E) virtio_gpu(E) virtio_dma_buf(E) vmw_vsock_virtio_transport(E) sha256_arm64(E) vmw_vsock_virtio_transport_common(E) drm_shmem_helper(E) sha1_ce(E) drm_kms_helper(E) vsock(E) arm_smccc_trng(E) virtiofs(E) virtio_console(E) virtio_rng(E) virtio_balloon(E) loop(E) dm_mod(E) drm(E) fuse(E) dax(E) i2c_core(E) efi_pstore(E) configfs(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) virtio_net(E) net_failover(E) failover(E) virtio_blk(E) virtio_pci(E) virtio_pci_legacy_dev(E) virtio_pci_modern_dev(E) virtio(E) virtio_ring(E)
[  +0.066684] CPU: 8 PID: 1155 Comm: insmod Tainted: G           OE      6.1.0-29-avf-arm64 #1  Ubuntu 6.1.123-1
[  +0.010631] Hardware name: linux /, BIOS 2024.04-g6dacd71e00fa-ab12812074 04/01/2024
[  +0.005743] pstate: 62400005 (nZCv daif +PAN -UAO +TCO -DIT -SSBS BTYPE=--)
[  +0.005536] pc : Query_Features+0x8c/0x1c00 [corefreqk]
[  +0.004491] lr : CoreFreqK_Query_Features_Level_Up+0xd8/0x110 [corefreqk]
[  +0.005440] sp : ffff800009143610
[  +0.003084] x29: ffff800009143610 x28: 0000000000000000 x27: ffffda40acb6c580
[  +0.005475] x26: ffffda40627ce7d8 x25: ffffda40627ce7c0 x24: 0000000000000000
[  +0.005698] x23: ffff800009143a60 x22: 0000000000000000 x21: ffff800009143780
[  +0.005554] x20: ffffda40627ce298 x19: ffff800009143768 x18: 0000000000000006
[  +0.005709] x17: ffffda40627c2cc0 x16: ffffda40abb48b80 x15: 00000000411fd4d0
[  +0.005719] x14: 0000000001770000 x13: 000001a53f2eaa1d x12: ffffda40627c6740
[  +0.005712] x11: ffffda40627c3120 x10: ffffda40627c3060 x9 : ffffda40627c6478
[  +0.005805] x8 : ffff7c86fb4031b8 x7 : 0000000000000000 x6 : 000000000000003f
[  +0.005615] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 00000000fffffffe
[  +0.005611] x2 : ffffa24693177000 x1 : ffff7c86fb403100 x0 : 0000000000000001
[  +0.005793] Call trace:
[  +0.002262]  Query_Features+0x8c/0x1c00 [corefreqk]
[  +0.003885]  CoreFreqK_Query_Features_Level_Up+0xd8/0x110 [corefreqk]
[  +0.004785]  CoreFreqK_StartUp+0x68/0xc4 [corefreqk]
[  +0.003951]  CoreFreqK_Init+0x1c/0x1000 [corefreqk]
[  +0.003896]  do_one_initcall+0x60/0x260
[  +0.003180]  do_init_module+0x50/0x1fc
[  +0.003152]  load_module+0x1cb8/0x2220
[  +0.003180]  __do_sys_finit_module+0xac/0x130
[  +0.003608]  __arm64_sys_finit_module+0x28/0x34
[  +0.003777]  invoke_syscall+0x78/0x100
[  +0.003151]  el0_svc_common.constprop.0+0x4c/0xf4
[  +0.003721]  do_el0_svc+0x2c/0x9c
[  +0.002837]  el0_svc+0x34/0xd4
[  +0.002637]  el0t_64_sync_handler+0xf4/0x120
[  +0.003432]  el0t_64_sync+0x18c/0x190
[  +0.003004] Code: f90067ff d538000f d53be00e d53be02d (d53b9c0c) 
[  +0.004833] ---[ end trace 0000000000000000 ]---
```

The characteristics of the CPU in the VM are:

```
lscpu
Architecture:             aarch64
  CPU op-mode(s):         64-bit
  Byte Order:             Little Endian
CPU(s):                   9
  On-line CPU(s) list:    0-8
Vendor ID:                ARM
  Model name:             Cortex-A510
    BIOS Model name:        CPU @ 0.0GHz
    BIOS CPU family:      2
    Model:                1
    Thread(s) per core:   1
    Core(s) per socket:   4
    Socket(s):            1
    Stepping:             r1p1
    BogoMIPS:             49.15
    Flags:                fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm jscvt fcma lrcpc dcpop sha3 sm3 sm4 asimddp sha512 asimdfhm dit uscat ilrcpc flagm sb paca pacg dcpodp flagm2 frint i8mm bti
  Model name:             Cortex-A715
    BIOS Model name:        CPU @ 0.0GHz
    BIOS CPU family:      2
    Model:                0
    Thread(s) per core:   1
    Core(s) per socket:   4
    Socket(s):            1
    Stepping:             r1p0
    BogoMIPS:             49.15
    Flags:                fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm jscvt fcma lrcpc dcpop sha3 sm3 sm4 asimddp sha512 asimdfhm dit uscat ilrcpc flagm sb paca pacg dcpodp flagm2 frint i8mm bti
  Model name:             -
    BIOS Model name:        CPU @ 0.0GHz
    BIOS CPU family:      2
    Model:                0
    Thread(s) per core:   1
    Core(s) per socket:   1
    Socket(s):            1
NUMA:                     
  NUMA node(s):           1
  NUMA node0 CPU(s):      0-8
Vulnerabilities:          
  Gather data sampling:   Not affected
  Itlb multihit:          Not affected
  L1tf:                   Not affected
  Mds:                    Not affected
  Meltdown:               Not affected
  Mmio stale data:        Not affected
  Reg file data sampling: Not affected
  Retbleed:               Not affected
  Spec rstack overflow:   Not affected
  Spec store bypass:      Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:             Mitigation; __user pointer sanitization
  Spectre v2:             Not affected
  Srbds:                  Not affected
  Tsx async abort:        Not affected
```

In this VM, I was investigating some CPU performance issue where Geekbench 6 scores are lower than the Geekbench 6 app on Android.
I was looking to display CPU frequency during the benchmark but to my surprise no tool can display it (as well as min an max frequency). There is nothing in `/proc` either.
So I wanted to check if CoreFreq would display it by some other mean.

cyring, 2025-03-11:
@bubbleguuum Hello,

I'm 100% I went a bit too far in the registers queries.
I bet some, especially those read by `SysRegRead` in [`Query_Features`](https://github.com/cyring/CoreFreq/blob/57c331e6c919a2e1d792e1830e171249b0adf395/aarch64/corefreqk.c#L509C13-L509C27) function, have to be commented out and then made available one by one until crash happens.
I'm not finding an easy-to-read Registers matrix in [developer documentation](https://developer.arm.com/documentation/ddi0601/2024-12/AArch64-Registers?lang=en) which would tell me which bit is accessible or not on [`Cortex-A510`](https://developer.arm.com/Processors/Cortex-A510) and [`Cortex-A715`](https://developer.arm.com/documentation/101590/0103/The-Cortex-A715--core?lang=en); although they both belong to `Armv9`

cyring, 2025-03-11:
I just have a try with the closest ARMv9-A available in Qemu, the Cortex-A710 where I encounter no crash but data are incomplete like Caches

[image]

cyring, 2025-03-11:
> This phone model also has `1x3.0 GHz Cortex-X3`

Could be or not but I'm reading that X3 is classified in the same group of ["64b support only"](https://developer.arm.com/documentation/ka006014/1-0/?lang=en)

AArch64 CoreFreq has been tested against hardware Cortex A55 + A76 (hybrid); A53 and the remainders are virtualized 64-bits architectures from `qemu-system-aarch64 --version` `9.2.0`

cyring, 2025-03-15:
> On my Pixel 8 Pro on Android 16 Beta 2.1

I have a Pixel 5 w/ Android 14 but it is my production phone.
Is it safe to install this emulator ?
No other way to install that Terminal app onto a PC VM ?

cyring, 2025-03-15:
> It uses Google own virtualization framework on Android.

And because we are talking about a new hypervisor; it has to be included in **many** places of the driver to avoid some registers; for example `CLUSTERCFR_EL1`

bubbleguuum, 2025-03-15:
That VM is only available on Android 16 Beta and on Android 15 for some Pixel with the latest "feature drop".
I will do some tests today to find where the module chokes.

cyring, 2025-03-15:
I have been able to produce a crash with a Device Tree only QEMU VM; previously was ACPI BIOS based.
Crash happen in:
```
[   90.969122] Call trace:
[   90.969311]  Query_DynamIQ+0x30/0x50 [corefreqk]
[   90.969639]  CoreFreqK_Ignition_Level_Up+0x3b4/0x9f0 [corefreqk]
[   90.969898]  CoreFreqK_StartUp+0x68/0xc4 [corefreqk]
[   90.970193]  CoreFreqK_Init+0x1c/0x1000 [corefreqk]
```
where `CLUSTERCFR_EL1` registers are accessed

I'm fixing that issue ...

bubbleguuum, 2025-03-15:
More details about the Android Virtualization Framework (AVF) that might help:

https://source.android.com/docs/core/virtualization/architecture

cyring, 2025-03-15:
> More details about the Android Virtualization Framework (AVF) that might help:
> 
> https://source.android.com/docs/core/virtualization/architecture

What do you read here ?
```
cat /sys/firmware/devicetree/base/compatible
```

bubbleguuum, 2025-03-15:
```
root@localhost:/# cat /sys/firmware/devicetree/base/compatible
linux,dummy-virt
```

bubbleguuum, 2025-03-15:
As a side note this kernel is compiled with `CONFIG_CPU_FREQ` disabled for some reason.
I suppose it explains why no tool can report min/max/cur CPU frequencies.
And maybe the low Geekbench6 performance I observed as there is no scaling governor.

cyring, 2025-03-15:
> As a side note this kernel is compiled with `CONFIG_CPU_FREQ` disabled for some reason. I suppose it explains why no tool can report min/max/cur CPU frequencies. And maybe the low Geekbench6 performance I observed as there is no scaling governor.

Reporting for sure; especially if the Operating Points (OPP) are not declared into the device tree; there may have no other mean to detect the frequencies from specs

Can you please pull and test the latest commit from `master` with virtualization detection improved in DT mode ?

bubbleguuum, 2025-03-15:
It still crashes with trace below. Interestingly, `insmod` itself segfaults, but is not able to produce a core dump.

```
[   94.178304] corefreqk: loading out-of-tree module taints kernel.
[   94.185577] Internal error: Oops - Undefined instruction: 0000000002000000 [#1] SMP
[   94.192500] Modules linked in: corefreqk(OE+) binfmt_misc(E) aes_ce_blk(E) aes_ce_cipher(E) crct10dif_ce(E) polyval_ce(E) polyval_generic(E) ghash_ce(E) nls_ascii(E) gf128mul(E) nls_cp437(E) sha3_ce(E) sha3_generic(E) sha512_ce(E) vfat(E) fat(E) sha512_arm64(E) virtio_gpu(E) vmw_vsock_virtio_transport(E) sha2_ce(E) virtio_dma_buf(E) drm_shmem_helper(E) vmw_vsock_virtio_transport_common(E) sha256_arm64(E) sha1_ce(E) virtio_balloon(E) arm_smccc_trng(E) drm_kms_helper(E) virtiofs(E) virtio_console(E) vsock(E) virtio_rng(E) fuse(E) drm(E) dm_mod(E) loop(E) i2c_core(E) dax(E) efi_pstore(E) configfs(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) virtio_net(E) net_failover(E) failover(E) virtio_blk(E) virtio_pci(E) virtio_pci_legacy_dev(E) virtio_pci_modern_dev(E) virtio(E) virtio_ring(E)
[   94.229647] CPU: 0 PID: 836 Comm: insmod Tainted: G           OE      6.1.0-29-avf-arm64 #1  Ubuntu 6.1.123-1
[   94.234438] Hardware name: Unknown Unknown Product/Unknown Product, BIOS 2024.01-g67e23f7c077b-ab12577904 01/01/2024
[   94.239152] pstate: 62400005 (nZCv daif +PAN -UAO +TCO -DIT -SSBS BTYPE=--)
[   94.242810] pc : Query_Features+0x8c/0x1c00 [corefreqk]
[   94.245554] lr : CoreFreqK_Query_Features_Level_Up+0xd8/0x110 [corefreqk]
[   94.249120] sp : ffff8000091bb620
[   94.250949] x29: ffff8000091bb620 x28: 0000000000000000 x27: ffffaf69a840c580
[   94.254786] x26: ffffaf6959fe07d8 x25: ffffaf6959fe07c0 x24: 0000000000000000
[   94.258601] x23: ffff8000091bba70 x22: 0000000000000000 x21: ffff8000091bb790
[   94.260297] x20: ffffaf6959fe02a0 x19: ffff8000091bb778 x18: 0000000000000006
[   94.263439] x17: ffffaf6959fd4cc0 x16: ffffaf69a73e8b80 x15: 00000000411fd461
[   94.266863] x14: 0000000001770000 x13: 00000234b99740b0 x12: ffffaf6959fd8b80
[   94.270372] x11: ffffaf6959fd5120 x10: ffffaf6959fd5060 x9 : ffffaf6959fd88b8
[   94.273759] x8 : ffff6001c32799b8 x7 : 0000000000000000 x6 : 000000000000003f
[   94.277147] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 00000000fffffffe
[   94.280698] x2 : ffffb098977e6000 x1 : ffff6001c3279900 x0 : 0000000000000001
[   94.284448] Call trace:
[   94.285743]  Query_Features+0x8c/0x1c00 [corefreqk]
[   94.288237]  CoreFreqK_Query_Features_Level_Up+0xd8/0x110 [corefreqk]
[   94.292590]  CoreFreqK_StartUp+0x68/0xc4 [corefreqk]
[   94.295048]  CoreFreqK_Init+0x1c/0x1000 [corefreqk]
[   94.297518]  do_one_initcall+0x60/0x260
[   94.299312]  do_init_module+0x50/0x1fc
[   94.301359]  load_module+0x1cb8/0x2220
[   94.303454]  __do_sys_finit_module+0xac/0x130
[   94.305766]  __arm64_sys_finit_module+0x28/0x34
[   94.308108]  invoke_syscall+0x78/0x100
[   94.310106]  el0_svc_common.constprop.0+0x4c/0xf4
[   94.312536]  do_el0_svc+0x2c/0x9c
[   94.314375]  el0_svc+0x34/0xd4
[   94.315565]  el0t_64_sync_handler+0xf4/0x120
[   94.317904]  el0t_64_sync+0x18c/0x190
[   94.319803] Code: f90067ff d538000f d53be00e d53be02d (d53b9c0c) 
[   94.322952] ---[ end trace 0000000000000000 ]---
```

bubbleguuum, 2025-03-15:
I added some traces and it crashes before `CoreFreqK_Ignition_Level_Up` is called, with the new added CONFIG_OF block.

cyring, 2025-03-15:
> It still crashes with trace below. Interestingly, `insmod` itself segfaults (it did before, I just did not mention it), but is not able to produce a core dump.
> 
> ```
> [   94.178304] corefreqk: loading out-of-tree module taints kernel.
> [   94.185577] Internal error: Oops - Undefined instruction: 0000000002000000 [#1] SMP
> [   94.192500] Modules linked in: corefreqk(OE+) binfmt_misc(E) aes_ce_blk(E) aes_ce_cipher(E) crct10dif_ce(E) polyval_ce(E) polyval_generic(E) ghash_ce(E) nls_ascii(E) gf128mul(E) nls_cp437(E) sha3_ce(E) sha3_generic(E) sha512_ce(E) vfat(E) fat(E) sha512_arm64(E) virtio_gpu(E) vmw_vsock_virtio_transport(E) sha2_ce(E) virtio_dma_buf(E) drm_shmem_helper(E) vmw_vsock_virtio_transport_common(E) sha256_arm64(E) sha1_ce(E) virtio_balloon(E) arm_smccc_trng(E) drm_kms_helper(E) virtiofs(E) virtio_console(E) vsock(E) virtio_rng(E) fuse(E) drm(E) dm_mod(E) loop(E) i2c_core(E) dax(E) efi_pstore(E) configfs(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) virtio_net(E) net_failover(E) failover(E) virtio_blk(E) virtio_pci(E) virtio_pci_legacy_dev(E) virtio_pci_modern_dev(E) virtio(E) virtio_ring(E)
> [   94.229647] CPU: 0 PID: 836 Comm: insmod Tainted: G           OE      6.1.0-29-avf-arm64 #1  Ubuntu 6.1.123-1
> [   94.234438] Hardware name: Unknown Unknown Product/Unknown Product, BIOS 2024.01-g67e23f7c077b-ab12577904 01/01/2024
> [   94.239152] pstate: 62400005 (nZCv daif +PAN -UAO +TCO -DIT -SSBS BTYPE=--)
> [   94.242810] pc : Query_Features+0x8c/0x1c00 [corefreqk]
> [   94.245554] lr : CoreFreqK_Query_Features_Level_Up+0xd8/0x110 [corefreqk]
> [   94.249120] sp : ffff8000091bb620
> [   94.250949] x29: ffff8000091bb620 x28: 0000000000000000 x27: ffffaf69a840c580
> [   94.254786] x26: ffffaf6959fe07d8 x25: ffffaf6959fe07c0 x24: 0000000000000000
> [   94.258601] x23: ffff8000091bba70 x22: 0000000000000000 x21: ffff8000091bb790
> [   94.260297] x20: ffffaf6959fe02a0 x19: ffff8000091bb778 x18: 0000000000000006
> [   94.263439] x17: ffffaf6959fd4cc0 x16: ffffaf69a73e8b80 x15: 00000000411fd461
> [   94.266863] x14: 0000000001770000 x13: 00000234b99740b0 x12: ffffaf6959fd8b80
> [   94.270372] x11: ffffaf6959fd5120 x10: ffffaf6959fd5060 x9 : ffffaf6959fd88b8
> [   94.273759] x8 : ffff6001c32799b8 x7 : 0000000000000000 x6 : 000000000000003f
> [   94.277147] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 00000000fffffffe
> [   94.280698] x2 : ffffb098977e6000 x1 : ffff6001c3279900 x0 : 0000000000000001
> [   94.284448] Call trace:
> [   94.285743]  Query_Features+0x8c/0x1c00 [corefreqk]
> [   94.288237]  CoreFreqK_Query_Features_Level_Up+0xd8/0x110 [corefreqk]
> [   94.292590]  CoreFreqK_StartUp+0x68/0xc4 [corefreqk]
> [   94.295048]  CoreFreqK_Init+0x1c/0x1000 [corefreqk]
> [   94.297518]  do_one_initcall+0x60/0x260
> [   94.299312]  do_init_module+0x50/0x1fc
> [   94.301359]  load_module+0x1cb8/0x2220
> [   94.303454]  __do_sys_finit_module+0xac/0x130
> [   94.305766]  __arm64_sys_finit_module+0x28/0x34
> [   94.308108]  invoke_syscall+0x78/0x100
> [   94.310106]  el0_svc_common.constprop.0+0x4c/0xf4
> [   94.312536]  do_el0_svc+0x2c/0x9c
> [   94.314375]  el0_svc+0x34/0xd4
> [   94.315565]  el0t_64_sync_handler+0xf4/0x120
> [   94.317904]  el0t_64_sync+0x18c/0x190
> [   94.319803] Code: f90067ff d538000f d53be00e d53be02d (d53b9c0c) 
> [   94.322952] ---[ end trace 0000000000000000 ]---
> ```

That `Oops - Undefined instruction: 0000000002000000 [#1] SMP` puzzles me.
Does it mean one or some instructions assembled into `Query_Features()` function is/are unsupported ?

Can you comment all those lines in `aarch64/corefreqk.c`

https://github.com/cyring/CoreFreq/blob/260d8af2ff22829481bd593d0ca376c12ff172e5/aarch64/corefreqk.c#L509

```c
/*
__asm__ __volatile__(
		"mrs	%[midr] ,	midr_el1"	"\n\t"
		"mrs	%[cntfrq],	cntfrq_el0"	"\n\t"
		"mrs	%[cntpct],	cntpct_el0"	"\n\t"
		"mrs	%[pmcr] ,	pmcr_el0"	"\n\t"
		"mrs	%[dfr0] ,	id_aa64dfr0_el1""\n\t"
		"mrs	%[dfr1] ,	id_aa64dfr1_el1""\n\t"
		"mrs	%[isar0],	id_aa64isar0_el1""\n\t"
		"mrs	%[isar1],	id_aa64isar1_el1""\n\t"
		"mrs	%[mmfr0],	id_aa64mmfr0_el1""\n\t"
		"mrs	%[mmfr1],	id_aa64mmfr1_el1""\n\t"
		"mrs	%[pfr0] ,	id_aa64pfr0_el1""\n\t"
		"mrs	%[pfr1] ,	id_aa64pfr1_el1""\n\t"
		"mrs	%[mvfr0],	mvfr0_el1"	"\n\t"
		"mrs	%[mvfr1],	mvfr1_el1"	"\n\t"
		"mrs	%[mvfr2],	mvfr2_el1"	"\n\t"
		"isb"
		: [midr]	"=r" (midr),
		  [cntfrq]	"=r" (cntfrq),
		  [cntpct]	"=r" (cntpct),
		  [pmcr]	"=r" (pmcr),
		  [dfr0]	"=r" (dfr0),
		  [dfr1]	"=r" (dfr1),
		  [isar0]	"=r" (isar0),
		  [isar1]	"=r" (isar1),
		  [mmfr0]	"=r" (mmfr0),
		  [mmfr1]	"=r" (mmfr1),
		  [pfr0]	"=r" (pfr0),
		  [pfr1]	"=r" (pfr1),
		  [mvfr0]	"=r" (mvfr0),
		  [mvfr1]	"=r" (mvfr1),
		  [mvfr2]	"=r" (mvfr2)
		:
		: "memory"
	);

	isar2.value = SysRegRead(ID_AA64ISAR2_EL1);
	isar3.value = SysRegRead(ID_AA64ISAR3_EL1);
	mmfr2.value = SysRegRead(ID_AA64MMFR2_EL1);

	iArg->Features->Info.Signature.Stepping = midr.Revision
						| (midr.Variant << 4);
	iArg->Features->Info.Signature.Family = midr.PartNum & 0x00f;
	iArg->Features->Info.Signature.ExtFamily = (midr.PartNum & 0xff0) >> 4;
	iArg->Features->Info.Signature.Model = pmcr.IDcode & 0x0f;
	iArg->Features->Info.Signature.ExtModel = (pmcr.IDcode & 0xf0) >> 4;

	VendorFromMainID(midr, iArg->Features->Info.Vendor.ID,
			&iArg->Features->Info.Vendor.CRC, &iArg->HypervisorID);

	iArg->Features->Factory.Freq = cntfrq.ClockFreq_Hz;
	iArg->Features->Factory.Freq = iArg->Features->Factory.Freq / 10000;
*/
iArg->Features->Factory.Freq=1000;
```

After change, rebuild all code
```sh
make clean
make -j
```

Load `corefreqk.ko` and let see if we are going further ?

bubbleguuum, 2025-03-15:
It makes the kernel hard crash :/

bubbleguuum, 2025-03-15:
With added traces, I confirm that it is the asm block that is causing the illegal instruction.

bubbleguuum, 2025-03-15:
Even if that asm block can be changed so it does not crash, I would not be surprised of a hard kernel crash later on. Maybe this kernel cannot work CoreFreq... Speaking of which it is 6.1.123 (which is fairly recent) patched by Google for AVF and who knows what they did to it.

bubbleguuum, 2025-03-15:
> > With added traces, I confirm that it is the asm block that is causing the illegal instruction:
> > ```
> > pr_info("LOL1");
> > __asm__ __volatile__(  
> > ...
> > );
> > pr_info("LOL2");    <= NOT PRINTED
> > ```
> 
> Was it before you put the whole `asm` statement into comments (as claimed above) ?

Not sure what you mena but here's what I did:

- tested commenting the asm block as you instructed: it caused a hard kernel crash (nothing responding, ssh connection gone, ...)
- rebooted a fresh VM, undid modification above and added the 2 LOL info lines. On `insmod`, only LOL1 is printed in `dmesg`

cyring, 2025-03-15:
> > > With added traces, I confirm that it is the asm block that is causing the illegal instruction:
> > > ```
> > > pr_info("LOL1");
> > > __asm__ __volatile__(  
> > > ...
> > > );
> > > pr_info("LOL2");    <= NOT PRINTED
> > > ```
> > 
> > 
> > Was it before you put the whole `asm` statement into comments (as claimed above) ?
> 
> Not sure what you mean but here's what I did:
> 
> * tested commenting the asm block as you instructed: it caused a hard kernel crash (nothing responding, ssh connection gone, ...)
> * rebooted a fresh VM, undid modifications above and added the 2 LOL info lines. On `insmod`, only LOL1 is printed in `dmesg`

I'm searching for unsupported registers into the Cortex A715 Core r1p3 Technical Reference Manual [[link](https://developer.arm.com/documentation/101590/0103/?lang=en)]

Meanwhile I did non regression tests with my SBC Orange-Pi 5+ Rockchip (dual A55; A76) and other QEMU VM of Neoverse-N2 running RHEL : no crash.
Don't you have other hardware to show me ?

cyring, 2025-03-15:
I think the issue is that your processors are 64-bits only whereas mines are both 32/64

* your `lscpu`
```
Architecture:             aarch64
  CPU op-mode(s):         64-bit
CPU(s):                   9
  Model name:             Cortex-A510
  Model name:             Cortex-A715
  Model name:             -
```
* my `lscpu`
```
Architecture:           aarch64
  CPU op-mode(s):       32-bit, 64-bit
CPU(s):                 8
  Model name:           Cortex-A55
  Model name:           Cortex-A76
```

cyring, 2025-03-15:
In the A76 TRM I see that AArch32 registers are supported this way:

[image]

But in the A715 TRM no such thing but this subset of the Registers list:

[image]

Thus can you please give a try commenting only those 32-bits registers as below:
```c
        __asm__ __volatile__(
                "mrs    %[midr] ,       midr_el1"       "\n\t"
                "mrs    %[cntfrq],      cntfrq_el0"     "\n\t"
                "mrs    %[cntpct],      cntpct_el0"     "\n\t"
                "mrs    %[pmcr] ,       pmcr_el0"       "\n\t"
                "mrs    %[dfr0] ,       id_aa64dfr0_el1""\n\t"
                "mrs    %[dfr1] ,       id_aa64dfr1_el1""\n\t"
                "mrs    %[isar0],       id_aa64isar0_el1""\n\t"
                "mrs    %[isar1],       id_aa64isar1_el1""\n\t"
                "mrs    %[mmfr0],       id_aa64mmfr0_el1""\n\t"
                "mrs    %[mmfr1],       id_aa64mmfr1_el1""\n\t"
                "mrs    %[pfr0] ,       id_aa64pfr0_el1""\n\t"
                "mrs    %[pfr1] ,       id_aa64pfr1_el1""\n\t"
/*              "mrs    %[mvfr0],       mvfr0_el1"      "\n\t"
                "mrs    %[mvfr1],       mvfr1_el1"      "\n\t"
                "mrs    %[mvfr2],       mvfr2_el1"      "\n\t"  */
                "isb"
                : [midr]        "=r" (midr),
                  [cntfrq]      "=r" (cntfrq),
                  [cntpct]      "=r" (cntpct),
                  [pmcr]        "=r" (pmcr),
                  [dfr0]        "=r" (dfr0),
                  [dfr1]        "=r" (dfr1),
                  [isar0]       "=r" (isar0),
                  [isar1]       "=r" (isar1),
                  [mmfr0]       "=r" (mmfr0),
                  [mmfr1]       "=r" (mmfr1),
                  [pfr0]        "=r" (pfr0),
                  [pfr1]        "=r" (pfr1),
                  [mvfr0]       "=r" (mvfr0),
                  [mvfr1]       "=r" (mvfr1),
                  [mvfr2]       "=r" (mvfr2)
                :
                : "memory"
        );
```

bubbleguuum, 2025-03-15:
Still crashing :/

```
[  131.375252] Internal error: Oops - Undefined instruction: 0000000002000000 [#1] SMP
[  131.389515] Modules linked in: corefreqk(OE+) binfmt_misc(E) aes_ce_blk(E) nls_ascii(E) nls_cp437(E) aes_ce_cipher(E) crct10dif_ce(E) polyval_ce(E) vfat(E) polyval_generic(E) fat(E) ghash_ce(E) gf128mul(E) sha3_ce(E) sha3_generic(E) sha512_ce(E) sha512_arm64(E) virtio_gpu(E) sha2_ce(E) vmw_vsock_virtio_transport(E) sha256_arm64(E) vmw_vsock_virtio_transport_common(E) sha1_ce(E) virtio_dma_buf(E) drm_shmem_helper(E) vsock(E) virtio_balloon(E) virtio_console(E) virtiofs(E) drm_kms_helper(E) arm_smccc_trng(E) virtio_rng(E) loop(E) dm_mod(E) drm(E) fuse(E) efi_pstore(E) dax(E) i2c_core(E) configfs(E) efivarfs(E) ip_tables(E) x_tables(E) autofs4(E) virtio_net(E) net_failover(E) failover(E) virtio_blk(E) virtio_pci(E) virtio_pci_legacy_dev(E) virtio_pci_modern_dev(E) virtio(E) virtio_ring(E)
[  131.441164] CPU: 6 PID: 1089 Comm: insmod Tainted: G           OE      6.1.0-29-avf-arm64 #1  Ubuntu 6.1.123-1
[  131.448036] Hardware name: Unknown Unknown Product/Unknown Product, BIOS 2024.01-g67e23f7c077b-ab12577904 01/01/2024
[  131.455602] pstate: 62400005 (nZCv daif +PAN -UAO +TCO -DIT -SSBS BTYPE=--)
[  131.456193] pc : Query_Features+0x9c/0x1bfc [corefreqk]
[  131.456644] lr : Query_Features+0x7c/0x1bfc [corefreqk]
[  131.458438] sp : ffff80000931b630
[  131.460860] x29: ffff80000931b630 x28: 0000000000000000 x27: ffffbba0d0a0c580
[  131.465459] x26: ffffbba05910d7d8 x25: ffffbba05910d7c0 x24: 0000000000000000
[  131.469971] x23: ffff80000931ba80 x22: 0000000000000000 x21: ffff80000931b7a0
[  131.474645] x20: ffffbba05910d2a8 x19: ffff80000931b788 x18: 0000000000000006
[  131.479345] x17: ffffbba059101bc0 x16: ffffbba0cfa147ec x15: 0000000001770000
[  131.483898] x14: 00000000fdebe9a5 x13: ffffbba059102b30 x12: ffffbba059107c1c
[  131.488794] x11: 00000000ffffefff x10: ffffbba0d09f05e0 x9 : ffffbba0cefc949c
[  131.493441] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000000
[  131.498145] x5 : ffff5069bf774a50 x4 : ffff5069bf774a50 x3 : 0000000000000000
[  131.502700] x2 : 0000000000000000 x1 : ffff5069ba704d00 x0 : 00000000411fd4d0
[  131.507386] Call trace:
[  131.509027]  Query_Features+0x9c/0x1bfc [corefreqk]
[  131.512214]  CoreFreqK_Query_Features_Level_Up+0xd8/0x110 [corefreqk]
[  131.516428]  CoreFreqK_StartUp+0x68/0xc4 [corefreqk]
[  131.519670]  CoreFreqK_Init+0x1c/0x1000 [corefreqk]
```

bubbleguuum, 2025-03-15:
After some more testing proceeding by elimination, it turns out it is the second instruction that cause the Undefined instruction:

```
mrs    %[cntfrq],      cntfrq_el0" 
```

cyring, 2025-03-15:
> Still crashing :/ (I double checked that I made the modification properly).

Thank you for your tests and sorry for all this mess.

I made a match between processor documentations and the registers in use in _CoreFreq_.

Those are not listed or partially in the A715 TRM:
```console
cntfrq_el0 cntpct_el0 SCTLR_EL1 fpsr cpacr_el1 pmuserenr_el0 pmcntenset_el0 pmselr_el0 pmxevtyper_el0 pmccfiltr_el0
```

FPU registers (`fpsr`) and System (`SCTLR_EL1`, `cpacr_el1`) are not necessary but the others (`cnt*`, `pmc*`) are the free running counters and the PMU registers which are mandatory to program the Performance counters.

I believe I need such hardware or a simulator to re-program specific functions for your Cortex

cyring, 2025-03-15:
> After some more testing proceeding by elimination, it turns out it is the second instruction that cause the Undefined instruction:
> 
> ```
> mrs    %[cntfrq],      cntfrq_el0" 
> ```

Well done. So far which faulty registers are they ?

bubbleguuum, 2025-03-15:
If I must test them one by one it will take some time as I have to reboot the VM after each failed `insmod`: otherwise subsequent `insmod` never complete getting stuck in kernel function module related that I forgot the name (found this with `strace`).
And when I comment some asm code in a way no 'Undefined instruction' crash happen, instead a hard kernel crash happens (supposedly somewhere in the code after the asm block) and I have to reboot my Android device (Android itself does not crash but it is not possible to restart the VM without a reboot). Quite painful and slow to debug. I'll do some test tomorrow to find other eventual instructions leading to that dreaded 'Undefined 'Undefined instruction'

Shouldn't that `cntfrq_el0`  always work ? Do you think it is possibly caused by a deficiency of the VM or some undocumented restriction added by Google ?

cyring, 2025-03-15:
It's my fault, I should have read the beginner exam [Q&A](https://developer.arm.com/documentation/102379/0104/Check-your-knowledge?lang=en) !

[image]

cyring, 2025-03-15:
> If I must test them one by one it will take some time as I have to reboot the VM after each failed `insmod`: otherwise subsequent `insmod` never complete getting stuck in a kernel function that is module related but for which I forgot the name (found this with `strace`). And when I comment some asm code in a way no 'Undefined instruction' crash happen, instead a hard kernel crash happens (supposedly somewhere in the code after the asm block) and I have to reboot my Android device (Android itself does not crash but it is not possible to restart the VM without a reboot). Quite painful and slow to debug. I'll do some test tomorrow to find other eventual instructions leading to that dreaded 'Undefined instruction'
> 
> Shouldn't that `cntfrq_el0` always work ? Do you think it is possibly caused by a deficiency of the VM or some undocumented restriction added by Google ?

Thanks a lot. I'm aware how tedious it is to circumvent an architecture.
Indeed, to my great surprise, `cntfrq_el0` not always work.

Evening starts here. Time to shutdown ...

bubbleguuum, 2025-03-15:
Some more progress:

The only other problematic register causing 'Undefined instruction' is `pmcr_el0`.

So commenting the 2 asm lines for `cntfrq_el0` and `pmcr_el0` make subsequent code to execute, but the kernel crashes hard as I mentioned.
The hard crash happens in  `CoreFreqK_Ignition_Level_Up` somewhere between the end of the  `#ifdef CONFIG_OF`  block and the end of the function.
Where exactly is left as an exercise for tomorrow :/. Amusingly, to debug this, I cannot rely on `dmesg` traces, but instead have to make the code crash de-referencing a null pointer.

To address the fact that variables `cntfrq` and `pmcr` are not filled with proper data, I did these changes:

- forced fake frequency:

`iArg->Features->Factory.Freq = 1000;`       

- this change:

```
 //iArg->Features->PerfMon.MonCtrs = pmcr.NumEvtCtrs;                                                                                                                                                                                   
 iArg->Features->PerfMon.MonCtrs = 0;    
```

- I left this code as is despite `pmcr` uninitialized, not sure if it matters:
 
```
iArg->Features->Info.Signature.Model = pmcr.IDcode & 0x0f;                                                                                                                                                                             
iArg->Features->Info.Signature.ExtModel = (pmcr.IDcode & 0xf0) >> 4;   
``

cyring, 2025-03-15:
> Some more progress:
> 
> The only other problematic register causing 'Undefined instruction' is `pmcr_el0`.
> 
> So commenting the 2 asm lines for `cntfrq_el0` and `pmcr_el0` make subsequent code to execute, but the kernel hard crashes as I mentioned. The hard crash happens in `CoreFreqK_Ignition_Level_Up` somewhere between the end of the `#ifdef CONFIG_OF` block and the end of the function. Where exactly is left as an exercise for tomorrow :/. Amusingly, to debug this, I cannot rely on `pr_info`, but instead have to make the code crash by de-referencing a null pointer to make the module soft crash before the kernel hard crash.
> 
> To address the fact that variables `cntfrq` and `pmcr` are not filled with proper data, I did these changes:
> 
> * forced fake frequency:
> 
> `iArg->Features->Factory.Freq = 1000;`
> 
> * this change:
> 
> ```
>  //iArg->Features->PerfMon.MonCtrs = pmcr.NumEvtCtrs;                                                                                                                                                                                   
>  iArg->Features->PerfMon.MonCtrs = 0;    
> ```
> 
> * I left this code as is despite `pmcr` uninitialized, not sure if it matters:
> 
> ```
> iArg->Features->Info.Signature.Model = pmcr.IDcode & 0x0f;                                                                                                                                                                             
> iArg->Features->Info.Signature.ExtModel = (pmcr.IDcode & 0xf0) >> 4;   
> ```

You did a tremendous job to find these registers.
* `pmcr` is especially useful to enable/disable PMU
* `cntfrq` could be workaround with DT or ACPI

In both cases it would more appropriate to use arm v9 regusters
CoreFreq has an entry for each function and per processor architecture.
https://github.com/cyring/CoreFreq/blob/260d8af2ff22829481bd593d0ca376c12ff172e5/aarch64/corefreqk.h#L834
So my coming work is to specialize functions for your processors.

bubbleguuum, 2025-03-16:
I investigated the kernel lockup.

It is caused by `Start_GenericMachine` locking up. It is called by `smp_call_function_single` called by `Controller_Start` called by `CoreFreqK_Ignition_Level_Up`.

I mean `smp_call_function_single` is never returning because `Start_GenericMachine` is locking up. I could verify this commenting all the code in `Start_GenericMachine` and in that case `smp_call_function_single` returns.

cyring, 2025-03-16:
> EDIT2: it is this asm block that cause the lock up in `PerCore_GenericMachine`:
> 
> ```
>  __asm__ __volatile__(                                                                                                                                                                                                                      
>                 "mrs    %[pmuser],   pmuserenr_el0"  "\n\t"                                                                                                                                                                                 
>                 "mrs   %[enset],       pmcntenset_el0" "\n\t"                                                                                                                                                                                 
>                 "mrs    %[enclr],       pmcntenclr_el0" "\n\t"                                                                                                                                                                                 
>                 "isb"                                                                                                                                                                                                                          
>                 : [pmuser]      "=r" (pmuser),                                                                                                                                                                                                 
>                   [enset]       "=r" (enset),                                                                                                                                                                                                  
>                   [enclr]       "=r" (enclr)                                                                                                                                                                                                   
>                 :                                                                                                                                                                                                                              
>                 : "memory"                                                                                                                                                                                                                     
>         );       
> ```
> 
> I checked that each individual instruction cause a lockup.

Awesome but how can we measure counters if those PMU registers are not implemented ?
Is this a change in arm v9 ? I doubt so.

bubbleguuum, 2025-03-16:
This is happening for the first enumerated CPU which is  of type `Cortex_A715`.

bubbleguuum, 2025-03-16:
Another surprise:

The `cntfrq_el0` that does not work in `Query_Features` works in  `Query_DeviceTree` (called much later) and returns the seemingly valid value of `2457600` for `Cortex_A510`

bubbleguuum, 2025-03-16:
Reading `cntfrq_el0` individually in `Query_Features` weirdly works:

   __asm__ __volatile__(                                                                                                                                                                                                                  
                "mrs    %[cntfrq],      cntfrq_el0"     "\n\t"                                                                                                                                                                                 
                "isb"                                                                                                                                                                                                                          
                : [cntfrq]      "=r" (cntfrq)                                                                                                                                                                                                  
                :                                                                                                                                                                                                                             
                : "memory"                                                                                                                                                                                                                    
        );                                                                                                                                                                                                                                     
                       

This does not work if attempting the same for `pmcr_el0` (Undefined instruction).

cyring, 2025-03-16:
Thank you so much for what you are doing.

To summarize only `pmc*` registers concerning the PMU setup are involved in the crash and probably only with Cortex-A715 but safe with Cortex-A510

If you are able to build and run _CoreFreq_ (commenting those PMC registers) can you then post the output of `corefreq-cli -s -n -R` because I would like to check the Current EL level with the Android simulator

bubbleguuum, 2025-03-16:
I could finally insmod the module, but had to comment most of the code in `Start_GenericMachine` because there is some additional asm code in `Generic_Core_Counters_Set` that crashes. That function is reduce  to code below and with the problematic asm block in `PerCore_GenericMachine` commented (`Update` function pointer call below):

```
static void Start_GenericMachine(void *arg)                                                                                                                                                                                                    
{                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                               
   unsigned int cpu = smp_processor_id();                                                                                                                                                                                                      
        CORE_RO *Core = (CORE_RO *) PUBLIC(RO(Core, AT(cpu)));                                                                                                                                                                                 
        union SAVE_AREA_CORE *Save = &PRIVATE(OF(Core, AT(cpu)))->SaveArea;                                                                                                                                                                    
        UNUSED(arg);                                                                                                                                                                                                                           
                                                                                                                                                                                                                                               
                                                                                                                                                                                                                                               
        if (Arch[PUBLIC(RO(Proc))->ArchID].Update != NULL) {                                                                                                                                                                                   
                Arch[PUBLIC(RO(Proc))->ArchID].Update(Core);                                                                                                                                                                                   
        }
}                   
```

here's the output of `corefreq-cli -s -n -R` attached.

[output.txt](https://github.com/user-attachments/files/19272659/output.txt)

cyring, 2025-03-16:
> here's the output of `corefreq-cli -s -n -R` attached:

Thanks.
I don't find much differences with all my QEMU simulations
Same Current `EL` level of `1` within the FLAGS

I wonder if the Perf-Tools are able to read the hardware counters ?

bubbleguuum, 2025-03-16:
Any specific `perf` invocation you want me to try ?

bubbleguuum, 2025-03-16:
Output of `perf stat`:

```
Performance counter stats for 'system wide':

          66156.86 msec cpu-clock                        #    9.000 CPUs utilized          
               447      context-switches                 #    6.757 /sec                   
                12      cpu-migrations                   #    0.181 /sec                   
                 0      page-faults                      #    0.000 /sec                   
   <not supported>      cycles                                                      
   <not supported>      instructions                                                
   <not supported>      branches                                                    
   <not supported>      branch-misses
```

It seems some counters have been disabled.

It looks more and more like this VM is a dead-end to work properly with CoreFreq. Probably stuff disabled for security reasons...

cyring, 2025-03-16:
> It looks more and more like this VM is a dead-end to work properly with CoreFreq. Probably stuff disabled for security reasons...

You totally right but I'm still astonished they managed to crash some assembly instructions. And I wonder if it is compliant to ARM ?

cyring, 2025-03-16:
Since, I have tried to start QEMU with a disabled PMU
```sh
qemu-system-aarch64 -machine virt -m 3G -vga \
std -display none -serial mon:stdio \
-smp 8 -cpu neoverse-n2,pmu=off
```
And _CoreFreq_ managed to start without.
No Performance and Instructions counters but a TSC at least

[image]

[image]

<details>

```console
Processor                                                          [Neoverse N2]
|- Architecture                                                        [ARMv9-A]
|- Vendor ID                                                               [Arm]
|- Revision                                                         [0x00000000]
|- Signature                                                           [ D49_00]
|- Stepping                                                            [   r0p3]
|- Online CPU                                                          [  8/  8]
|- Base Clock                                                          [100.757]
|- Frequency            (MHz)                      Ratio                        
                 Min    403.03                    <   4 >                       
                 Max   6297.31                    <  62 >                       
|- Factory                                                             [100.000]
                       6250                       [  63 ]                       
|- Performance                                                                  
                 TGT    403.03                    <   4 >                       
|- OPP                                                                          
|- Uncore                                                              [   LOCK]
                                                                               
Instruction Set Extensions                                                      
|-          AES [Y]        ASIMD [Y]        ATS1A [N]         BF16 [Y]
|-       CLRBHB [N]  CONSTPACFLD [N]          CPA [N]        CRC32 [Y]
|-         CSSC [N]          DGH [Y]           DP [Y]          DPB [Y]
|-         DPB2 [Y]        EBF16 [N]         EPAC [N]     FAMINMAX [N]
|-         FCMA [Y]          FHM [Y]        FlagM [Y]       FlagM2 [Y]
|-           FP [Y]         FPAC [N]  FPACCOMBINE [Y]     FP_ROUND [Y]
|-    FP_Sh_Vec [N]      FP_SQRT [Y]    FP_DIVIDE [Y]      FP_TRAP [N]
|-        FP_DP [Y]        FP_SP [Y]        FP_HP [Y]       FP_NaN [Y]
|-       FP_FtZ [Y]      FP_MISC [Y]       FPRCVT [N]      FRINTTS [Y]
|-          HBC [N]         I8MM [Y]        JSCVT [Y]        LRCPC [Y]
|-       LRCPC2 [Y]       LRCPC3 [N]         LS64 [N]       LS64_V [N]
|- LS64_ACCDATA [N]          LSE [Y]       LSE128 [N]         LSFE [N]
|-         LSUI [N]          LUT [N]         MOPS [N]        OCCMO [N]
|-        PACGA [N]    PACQARMA3 [N]    PACQARMA5 [Y]        PAuth [N]
|-       PAuth2 [N]     PAuth_LR [N]         PACM [N]     PCDPHINT [N]
|-      PRFMSLC [N]        PMULL [Y]         RAND [Y]         RDMA [Y]
|-     RNG-TRAP [N]        RPRES [N]        RPRFM [N]           SB [Y]
|-         SHA1 [Y]       SHA256 [Y]       SHA512 [Y]         SHA3 [Y]
|-     SIMD_REG [Y]     SIMD_FMA [Y]      SIMD_HP [Y]      SIMD_SP [Y]
|-     SIMD_INT [Y]      SIMD_LS [Y]    SIMD_MISC [Y]          SM3 [Y]
|-          SM4 [Y]          SME [N]         SME2 [N]       SME2p1 [N]
|-     SME_FA64 [N]    SME_LUTv2 [N]   SME_I16I64 [N]   SME_F64F64 [N]
|-   SME_I16I32 [N]   SME_B16B16 [N]   SME_F16F16 [N]    SME_F8F16 [N]
|-    SME_F8F32 [N]    SME_I8I32 [N]   SME_F16F32 [N]   SME_B16F32 [N]
|-  SME_BI32I32 [N]   SME_F32F32 [N]   SME_SF8FMA [N]   SME_SF8DP4 [N]
|-   SME_SF8DP2 [N]      SPECRES [Y]     SPECRES2 [N]          SVE [Y]
|-         SVE2 [Y]    SVE_F64MM [N]    SVE_F32MM [N]     SVE_I8MM [Y]
|-      SVE_SM4 [Y]     SVE_SHA3 [Y]     SVE_BF16 [Y]    SVE_EBF16 [N]
|-  SVE_BitPerm [Y]      SVE_AES [Y] SVE_PMULL128 [Y]    SYSREG128 [N]
|-  SYSINSTR128 [N]        TLBIW [N]         WFxT [N]           XS [N]
                                                                                
Features                                                                        
|- Advanced Configuration & Power Interface                     ACPI   [Missing]
|- Activity Monitor Unit                                         AMU   [Missing]
|- Mixed-Endianness EL0                                       BigEnd   [Missing]
[... 84 lines cut ...]
|- Thermal Design Power                                         Core   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                       Uncore   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                         DRAM   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                     Platform   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
   |- Power Limit                                                PL2   [    0 W]
[... cut ...]
|- Core Thermal Point                                                           
|- Package Thermal Point                                                        
|- Units                                                                        
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [      Missing]
   |- Window                                            second   [      Missing]
```

</details>

bubbleguuum, 2025-03-16:
I will report the crashing asm instructions on the Google issue tracker, with a minimal C sample. We'll see if they respond, but hopefully the report will be forwarded to the right people.

bubbleguuum, 2025-03-17:
Looks like the PM registers are not accessible at all by design. They are not either in an Android compiled program (produces `SIGILL`  as well). And I do not believe the Debian VM (even as root) has more privilege that an Android user program.
Somewhat confirmed by this FFmpeg code:
https://github.com/FFmpeg/FFmpeg/blob/master/libavutil/aarch64/timer.h#L36

So I'm closing this issue. Sorry for wasting your time on it as it was a dead-end from the start.

closed 2025-03-17 by bubbleguuum

cyring, 2025-03-17:
> Looks like the PM registers are not accessible at all by design. They are not either in an Android compiled program (produces `SIGILL` as well). And I do not believe the Debian VM (even as root) has more privilege that an Android user program. Somewhat confirmed by this FFmpeg code with `pmccntr_el0` that cannot be used on Android: https://github.com/FFmpeg/FFmpeg/blob/master/libavutil/aarch64/timer.h#L36
> 
> So I'm closing this issue. Sorry for wasting your time on it as it was a dead-end from the start.

It was not a waste of time because we learnt many things from this issue. Are you still going to report it to the Android project ?

bubbleguuum, 2025-03-17:
Yup I learn a few things also along the way !
I'm not reporting it because I'm certain it is expected behavior on Android.
I tested that sample C code with the problematic asm instructions compiled for Android results in `SIGILL` as well.

cyring, 2025-03-17:
I have had a pleasant chat with AI which states that Android hypervisor is based on pKVM with first, `HCR_EL2.TPM` set to trap PMU and second, a fatal exception raised when VM accesses PMU registers.

Apparently the Qemu option `-cpu cortex,pmu=off` may trigger the same KVM code; the exception less however.

Also to prevent a crash, _CoreFreq_ could monitor `HCR_EL2.TPM` but level was incompatible in your Debian VM. Fyi, it's readable on hardware Orange Pi 5+

Perhaps a `HCR_EL2` shadow register exists to EL1, at least read access. Or a kind of Supervisor call to request info.
Or if we have a Device Tree identifier within is unique in Android VM then I will manage gracefully the issue.

cyring, 2025-03-17:
Could you please post the output of:
```sh
cat /sys/firmware/devicetree/base/model
```

bubbleguuum, 2025-03-18:
There is no `model` file. Here's the file list in that directory:

```
root@localhost:/sys/firmware/devicetree/base# ls -l
total 0
-r--r--r--  1 root root  4 Mar 18 10:07 '#address-cells'
-r--r--r--  1 root root  4 Mar 18 10:07 '#size-cells'
drwxr-xr-x  2 root root  0 Mar 18 10:07  U6_16550A@2e8
drwxr-xr-x  2 root root  0 Mar 18 10:07  U6_16550A@2f8
drwxr-xr-x  2 root root  0 Mar 18 10:07  U6_16550A@3e8
drwxr-xr-x  2 root root  0 Mar 18 10:07  U6_16550A@3f8
drwxr-xr-x  2 root root  0 Mar 18 10:07  __symbols__
drwxr-xr-x  3 root root  0 Mar 18 10:07  avf
drwxr-xr-x  2 root root  0 Mar 18 10:07  chosen
-r--r--r--  1 root root 17 Mar  6 14:56  compatible
drwxr-xr-x  2 root root  0 Mar 18 10:07  config
drwxr-xr-x  2 root root  0 Mar 18 10:07  cpufreq
drwxr-xr-x 12 root root  0 Mar 18 10:07  cpus
drwxr-xr-x  2 root root  0 Mar 18 10:07  intc
-r--r--r--  1 root root  4 Mar 18 10:07  interrupt-parent
drwxr-xr-x  2 root root  0 Mar 18 10:07  memory
-r--r--r--  1 root root 10 Mar 18 10:06  name
drwxr-xr-x  2 root root  0 Mar 18 10:07  pci
drwxr-xr-x  2 root root  0 Mar 18 10:07  pclk@3M
drwxr-xr-x  2 root root  0 Mar 18 10:07  psci
drwxr-xr-x  2 root root  0 Mar 18 10:07  rtc@2000
drwxr-xr-x  2 root root  0 Mar 18 10:07  timer
drwxr-xr-x  2 root root  0 Mar 18 10:07  vmwdt@3000
```

bubbleguuum, 2025-03-18:
Note that this VM can be identified by the presence of the `avf` folder (Android Virtualization Framewor)

cyring, 2025-03-18:
> Note that this VM can be identified by the presence of the `avf` folder (Android Virtualization Framework)

I'm finding evocation of "avp" node into Android source code:
https://android.googlesource.com/platform/packages/modules/Virtualization/+/refs/heads/main/guest/pvmfw/src/fdt.rs
Especially `avf/untrusted` if you see the same ?

bubbleguuum, 2025-03-18:
Yes, there is not much there:

```
root@localhost:/home/droid# ls -l /sys/firmware/devicetree/base/avf/untrusted/
total 0
-r--r--r-- 1 root root  0 Mar 18 20:22 defer-rollback-protection
-r--r--r-- 1 root root 64 Mar 18 20:22 instance-id
-r--r--r-- 1 root root 10 Mar 18 20:22 name
```

bubbleguuum, 2025-03-19:
`dmesg` also show some warning but I supposed they are expected:

```
[  175.271761] CoreFreq(4:-1:8): Processor [D4D_00] Architecture [ARMv9-A] CPU [9/9]
[  175.282960] cpu cpu0: _find_key: OPP table not found (-19)
[  175.285879] cpu cpu1: _find_key: OPP table not found (-19)
[  175.288278] cpu cpu2: _find_key: OPP table not found (-19)
[  175.290368] cpu cpu3: _find_key: OPP table not found (-19)
[  175.293800] cpu cpu4: _find_key: OPP table not found (-19)
[  175.296312] cpu cpu5: _find_key: OPP table not found (-19)
[  175.298460] cpu cpu6: _find_key: OPP table not found (-19)
[  175.300811] cpu cpu7: _find_key: OPP table not found (-19)
[  175.302807] cpu cpu8: _find_key: OPP table not found (-19)
```

cyring, 2025-03-19:
> `dmesg` shows some warning but I supposed they are expected:
> 
> ```
> [  175.271761] CoreFreq(4:-1:8): Processor [D4D_00] Architecture [ARMv9-A] CPU [9/9]
> [  175.282960] cpu cpu0: _find_key: OPP table not found (-19)
> [  175.285879] cpu cpu1: _find_key: OPP table not found (-19)
> [  175.288278] cpu cpu2: _find_key: OPP table not found (-19)
> [  175.290368] cpu cpu3: _find_key: OPP table not found (-19)
> [  175.293800] cpu cpu4: _find_key: OPP table not found (-19)
> [  175.296312] cpu cpu5: _find_key: OPP table not found (-19)
> [  175.298460] cpu cpu6: _find_key: OPP table not found (-19)
> [  175.300811] cpu cpu7: _find_key: OPP table not found (-19)
> [  175.302807] cpu cpu8: _find_key: OPP table not found (-19)
> ```

Screenshot is great. You can at most evaluate the Base Clock.

Those OPP traces are normal. It means _CoreFreq_ has requested those tables from Kernel but this last has not retrieved them from Device Tree for some reasons: probably no [Operating Performance Points](https://www.kernel.org/doc/Documentation/devicetree/bindings/opp/opp.txt) declared

cyring, 2025-03-19:
Could you share the output of "Topology" (shortcut `m`)

bubbleguuum, 2025-03-19:
[image]

bubbleguuum, 2025-03-19:
`Base clock` seems to always remain in the 98-99 MHz range, whatever the load on the CPU.

cyring, 2025-03-19:
> [image]

Thank you.
Issue is that a true hardware reports Caches in bytes whereas here, the VM reports in KB

cyring, 2025-03-19:
@bubbleguuum Hi

Can you please show Topology since commit 33b50d0a2bb07055a2ba1b0c35fcf469a91d3b8c

[8 more replies]