cyring/CoreFreq #535: [RISC-V] ACPI code prevents CoreFreq from compiling for OrangePi RV2

jhart99, 2025-03-25:
When compiling for the OrangePi RV2 using the linux headers they provide, builds fail with the following error:

```
/home/orangepi/CoreFreq/build/module/corefreqk.c: In function ‘Get_EPP_ACPI_CPPC’:
/home/orangepi/CoreFreq/build/module/corefreqk.c:1239:19: error: implicit declaration of function ‘cppc_get_epp_perf’ [-Werror=implicit-function-declaration]
 1239 |         if ((rc = cppc_get_epp_perf((signed int) cpu, &epp_perf)) == 0)
      |                   ^~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /home/orangepi/CoreFreq/build/module/corefreqk.o] Error 1
make[1]: *** [Makefile:1921: /home/orangepi/CoreFreq/build] Error 2
make: *** [Makefile:136: all] Error 2
```
There are two issues first the build system is setting CONFIG_ACPI_CPPC_LIB which isn't supported.  Second, two functions need to be surrounded with a conditional group Get_EPP_ACPI_CPPC and Put_EPP_ACPI_CPPC.  This allows the module to build and run.

cyring, 2025-03-25:
Understood I will provide you the fix.

cyring, 2025-03-25:
Btw, can you tell if your SBC is concerned by Linux 6.14 [patch](https://lore.kernel.org/linux-serial/20240730-k1-01-basic-dt-v5-0-98263aae83be@gentoo.org/T/) ?
For development mattera, I'm looking for RISV 64-bits SBC supported by mainline or next release kernel.
With enough horse power to build a kernel straight inside the board.

cyring, 2025-03-26:
> When compiling for the OrangePi RV2 using the linux headers they provide, builds fail with the following error:
>
> ```
> /home/orangepi/CoreFreq/build/module/corefreqk.c: In function [...]

Hello,

Commit 9a9a9651e8988e960791ca9076ca799f5295b62f is now available. Hope it will let you build with no issue.

closed 2025-03-28 by cyring