cyring/CoreFreq #539: Intel Core Ultra 7 265K

valery-kirichenko, 2025-03-29:
Note: power limit was set to 159 W by me (because of small cooler 🙃)

`corefreq-cli -s`
```
Processor                                       [Intel(R) Core(TM) Ultra 7 265K]
|- Architecture                                                     [Arrow Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000117]
|- Signature                                                           [  06_C6]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 20/ 20]
|- Base Clock                                                          [ 99.448]
|- Frequency            (MHz)                      Ratio                        
                 Min    795.58                    <   8 >                       
                 Max   3878.47                    <  39 >                       
|- Factory                                                             [100.000]
                       3900                       [  39 ]                       
|- Performance                                                                  
                 TGT    994.46                    <  10 >                       
   |- HWP                                                                       
                 Min    795.58                    <   8 >                       
                 Max   6364.65                    <  64 >                       
                 TGT      AUTO                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   4574.61                    <  46 >                       
                  2C   4574.61                    <  46 >                       
                  3C   4574.61                    <  46 >                       
                  4C   4574.61                    <  46 >                       
                  5C   4574.61                    <  46 >                       
                  6C   4574.61                    <  46 >                       
                  7C   4574.61                    <  46 >                       
                  8C   4574.61                    <  46 >                       
|- Hybrid                                                              [ UNLOCK]
                  1C   4574.83                    <  46 >                       
                  2C   4574.83                    <  46 >                       
                  3C   4574.83                    <  46 >                       
                  4C   4574.83                    <  46 >                       
                  5C   4574.83                    <  46 >                       
                  6C   4574.83                    <  46 >                       
                  7C   4574.83                    <  46 >                       
                  8C   4574.83                    <  46 >                       
|- Uncore                                                              [ UNLOCK]
                 Min    795.62                    <   8 >                       
                 Max   3779.21                    <  38 >                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3878.47                    [  39 ]                       
               Turbo      AUTO                    <   0 >                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [Y] AVX-VNN-INT8 [Y] AVX-NE-CONV [Y] 
|- AVX-IFMA     [Y]    CMPccXADD [Y]      MOVDIRI [Y]   MOVDIR64B [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- ENQCMD       [N]         GFNI [Y]        OSPKE [Y]     WAITPKG [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [Y]      SYSCALL [Y]        RDPID [Y]         SGX [N] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
[... 375 lines cut ...]
   |- CPU #7      99.44 (  1)  2585.37 ( 26)  4474.68 ( 45)  6264.56 ( 63)      
   |- CPU #8      99.45 (  1)  1491.69 ( 15)  3281.72 ( 33)  4574.52 ( 46)      
   |- CPU #9      99.45 (  1)  1491.72 ( 15)  3281.79 ( 33)  4574.61 ( 46)      
   |- CPU #10     99.45 (  1)  1491.69 ( 15)  3281.72 ( 33)  4574.52 ( 46)      
   |- CPU #11     99.45 (  1)  1491.69 ( 15)  3281.72 ( 33)  4574.52 ( 46)      
   |- CPU #12     99.43 (  1)  1491.52 ( 15)  3281.35 ( 33)  4574.00 ( 46)      
   |- CPU #13     99.45 (  1)  1491.69 ( 15)  3281.72 ( 33)  4574.52 ( 46)      
   |- CPU #14     99.44 (  1)  1491.53 ( 15)  3281.37 ( 33)  4574.02 ( 46)      
   |- CPU #15     99.45 (  1)  1491.69 ( 15)  3281.72 ( 33)  4574.52 ( 46)      
   |- CPU #16     99.46 (  1)  1491.94 ( 15)  3282.26 ( 33)  4575.27 ( 46)      
   |- CPU #17     99.46 (  1)  1491.83 ( 15)  3282.02 ( 33)  4574.94 ( 46)      
   |- CPU #18     99.45 (  1)  1491.69 ( 15)  3281.72 ( 33)  4574.52 ( 46)      
   |- CPU #19     99.45 (  1)  1491.69 ( 15)  3281.71 ( 33)  4574.51 ( 46)
```

`corefreq-cli -u`: [corefreq-u.txt](https://github.com/user-attachments/files/19521871/corefreq-u.txt)

Idle
[image]

Atomic
[image]

Power is missing :(
[image]

cyring, 2025-03-29:
Thank you very much.
This my first Arrow Lake.
I have to check outputs but 2 other issues  observed:
1. Package temperature is stuck to TjMax
2. Memory controller bus rate and speed are missing

In Settings window,  changing the Power Scope to SMT may give some values ?

valery-kirichenko, 2025-03-29:
Changing Power Scope to SMT only gives zeros. I couldn't find anything about RAPL in the BIOS. If you have any ideas where it could be located in the ASUS bios, I can look again.

[image]

cyring, 2025-03-29:
> Changing Power Scope to SMT only gives zeros. I couldn't find anything about RAPL in the BIOS. If you have any ideas where it could be located in the ASUS bios, I can look again

I have reviewed code and it has not changed since Alder Lake.
RAPL unit registers have been found within your Arrow Lake
```
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]
```
Probably the RAPL Accu MSR registers have changed but I miss the datasheet of your processor.

Also, perhaps changing the Power Limit may have disabled the ACCU.
Does it make any difference if you restore setting to nominal ?

valery-kirichenko, 2025-03-29:
Nope, same thing with stock settings

valery-kirichenko, 2025-03-29:
> but I miss the datasheet of your processor

Maybe this can help
https://edc.intel.com/content/www/us/en/design/publications/core-ultra-200s-series-processors-datasheet-volume-2-of-2/

cyring, 2025-03-29:
> Maybe this can help

These are the MSR I'm reading for the consumed energy in joule 

https://github.com/cyring/CoreFreq/blob/f97a8b41e9275a54ed919e0359ce5c7af6f3ed23/x86_64/corefreqk.c#L15938

cyring, 2025-03-29:
@valery-kirichenko 
Can you try reading  registers from the command MSR-Toola
```sh
## MSR_PKG_ENERGY_STATUS
rdmsr -ax 0x00000611

## MSR_PP0_ENERGY_STATUS
rdmsr -ax 0x00000639
```

valery-kirichenko, 2025-03-29:
rdmsr -a 0x00000611
```
1e8f086
1e8f086
1e8f086
1e8f086
1e8f086
1e8f086
1e8f086
1e8f086
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
1e8f23f
```
rdmsr -a 0x00000639
```
f695ec
f695ec
f696bf
f696bf
f696bf
f696bf
f696bf
f696bf
f696bf
f696bf
f696bf
f696bf
f69739
f69739
f69739
f69739
f69739
f69739
f69739
f69739
```

cyring, 2025-03-30:
I guess I see what's going on. 
With Arrow Lake we are facing a new Hybrid identifier and all Cores have been launched with the Ecore monitoring loop.
But only the Pcore loop is collecting the Power Accumulator
https://github.com/cyring/CoreFreq/blob/f97a8b41e9275a54ed919e0359ce5c7af6f3ed23/x86_64/corefreqk.c#L19500

cyring, 2025-03-30:
@valery-kirichenko Hello,

Can you please dump the CPUID ?
```
corefreq-cli -u
```

cyring, 2025-03-30:
> I guess I see what's going on. With Arrow Lake we are facing a new Hybrid identifier and all Cores have been launched with the Ecore monitoring loop. But only the Pcore loop is collecting the Power Accumulator
> 
> [CoreFreq/x86_64/corefreqk.c](https://github.com/cyring/CoreFreq/blob/f97a8b41e9275a54ed919e0359ce5c7af6f3ed23/x86_64/corefreqk.c#L19500)
> 
> Line 19500 in [f97a8b4](/cyring/CoreFreq/commit/f97a8b41e9275a54ed919e0359ce5c7af6f3ed23)
> 
>  static void InitTimer_Alderlake(unsigned int cpu)

Within CPUDID dump, the [Hybrid](https://github.com/cyring/CoreFreq/blob/f97a8b41e9275a54ed919e0359ce5c7af6f3ed23/x86_64/coretypes.h#L1641) `CoreType` are still the same as previous architectures:
```
CPU #0   function         EAX          EBX          ECX          EDX            
|- 0000001a:00000000    40000003     00000000     00000000     00000000         

CPU #8   function         EAX          EBX          ECX          EDX            
|- 0000001a:00000000    20000003     00000000     00000000     00000000         
```

* `0x40` for P-core
* `0x20` for E-core

What's new here is a Model ID of `3`

cyring, 2025-03-30:
Can you start driver this way:
```
 insmod corefreqk.ko ServiceProcessor=0
```

valery-kirichenko, 2025-03-30:
My system hangs if I do it live or doesn't boot if I do it through modprobe config 😅

cyring, 2025-03-30:
> My system hangs if I do it live or doesn't boot if I do it through modprobe.d config 😅

I don't understand. You meant it crashed because of `insmod` ?
How did make successfully _CoreFreq_ started so far ? And can't you had the `ServiceProcessor=0` argument to the module ?

valery-kirichenko, 2025-03-30:
Yeah, at first it crashed because of insmod.
I installed it through AUR (corefreq-dkms) and created /etc/modules-load.d/corefreq.conf with `corefreqk` inside. This way it works just fine.

Then if I add `options corefreqk ServiceProcessor=0` to /etc/modprobe.d/corefreq.conf, my system will be unable to boot. I also tried setting it without modprobe.d through kernel argument `corefreqk.ServiceProcessor=0` and the result was the same.

cyring, 2025-03-30:
> I installed it through AUR (corefreq-dkms) and created /etc/modules-load.d/corefreq.conf with `corefreqk` inside. This way it works just fine.
> 
> After that I tried using insmod and crashed. Then I tried adding `options corefreqk ServiceProcessor=0` to /etc/modprobe.d/corefreq.conf and my system was unable to boot. I also tried setting it without modprobe.d through kernel argument `corefreqk.ServiceProcessor=0` and the result was the same.

You can't mix two installations; if it is the case here.

`corefreq-dkms` is still based on tag [`2.0.1`](https://github.com/cyring/CoreFreq/releases/tag/2.0.1)

You have to uninstall the package with all of the leftover binaries `corefreq-cli , corefreqd , corefreqk.ko , corefreqd.service`

and then you can git clone `CoreFreq` from its repository

cyring, 2025-03-30:
@valery-kirichenko 

But since tag `2.0.1` there have been no change on Intel core.

Meanwhile looking at your screenshots I see, because its color is highlighted, that the Service Processor is assigned to CPU `#19`. And that CPU belongs to the E-cores.

Does it crash when using `ServiceProcessor=19` or `ServiceProcessor=8`?

valery-kirichenko, 2025-03-30:
> You can't mix two installations; if it is the case here.

I didn't. All testing was done with `corefreq-dkms`

> Does it crash when using ServiceProcessor=19 or ServiceProcessor=8?

No, both options work just fine

cyring, 2025-03-30:
> No, both options work just fine

This is it.

Because Service is bound to E-core, some MSR registers have not been accessed by P-core.

I have to establish a list of registers to test with the MSR-Tools which should trap any access fault.
With a list of MSR known to work on Arrow Lake, I will then add specific monitoring loops for this architecture.
All of these will take some days of programming and testings if you are ok to do so ?

valery-kirichenko, 2025-03-30:
I'm here to help :)

cyring, 2025-03-31:
Fyi, I found in the [SDM](https://cdrdv2.intel.com/v1/dl/getContent/671200) the registers reference tables for Core Ultra.

[image]

Your architecture of CPUID `06_C6` is not mentioned but this concerns the Ultra series anyway.

cyring, 2025-03-31:
@valery-kirichenko Hello,

Please find the list of registers to test.

```sh
# hexadecimal
## MSR_ADL_UNCORE_PERF_GLOBAL_CTRL
rdmsr -aX 0x00002ff0
## MSR_ADL_UNCORE_PERF_GLOBAL_STATUS
rdmsr -aX 0x00002ff2
## MSR_ADL_UNCORE_PERF_FIXED_CTR_CTRL
rdmsr -aX 0x00002fde
## MSR_UNCORE_PERF_GLOBAL_OVF_CTRL
rdmsr -aX 0x00000393

# decimal
## MSR_PKG_C2_RESIDENCY
rdmsr -a 0x0000060d
## MSR_PKG_C3_RESIDENCY
rdmsr -a 0x000003f8
## MSR_PKG_C6_RESIDENCY
rdmsr -a 0x000003f9
## MSR_PKG_C7_RESIDENCY
rdmsr -a 0x000003fa
## MSR_PKG_C8_RESIDENCY
rdmsr -a 0x00000630
## MSR_PKG_C9_RESIDENCY
rdmsr -a 0x00000631
## MSR_ADL_UNCORE_PERF_FIXED_CTR0
rdmsr -aX 0x00002fdf
## MSR_PP1_ENERGY_STATUS
rdmsr -a 0x00000641
## MSR_DRAM_ENERGY_STATUS
rdmsr -a 0x00000619
## MSR_PLATFORM_ENERGY_STATUS
rdmsr -a 0x0000064d
## MSR_SMI_COUNT
rdmsr -a 0x00000034

# hexadecimal
## MSR_MISC_FEATURE_CONTROL
rdmsr -aX 0x000001a4
## MSR_CORE_UARCH_CTL
rdmsr -aX 0x00000541

## MSR_IA32_PERF_STATUS
rdmsr -aX 0x00000198
## MSR_IA32_PERF_CTL
rdmsr -aX 0x00000199
## MSR_IA32_THERM_STATUS
rdmsr -aX 0x0000019c
## MSR_IA32_PACKAGE_THERM_STATUS
rdmsr -aX 0x000001b1
## MSR_SKL_CORE_PERF_LIMIT_REASONS
rdmsr -aX 0x0000064f
## MSR_GRAPHICS_PERF_LIMIT_REASONS
rdmsr -aX 0x000006b0
## MSR_RING_PERF_LIMIT_REASONS
rdmsr -aX 0x000006b1

## MSR_FLEX_RATIO
rdmsr -aX 0x00000194
## MSR_IA32_MISC_PACKAGE_CTLS
rdmsr -aX 0x000000bc
```

Remarks:
* `-X` option of `rdmsr` should print hexadecimal in capitals. If not, use default.
* please keep the MSR names in your output list

Thank you

cyring, 2025-04-02:
@valery-kirichenko Hi,

Can you try clearing Uncore Overflow ?

```sh
## MSR_UNCORE_PERF_GLOBAL_OVF_CTRL (0x00000393) is a write-only register
wrmsr -a 0x393 0x100000000
```

valery-kirichenko, 2025-04-02:
@cyring wrmsr: CPU 0 cannot set MSR 0x00000393 to 0x0000000100000000

cyring, 2025-04-02:
> [@cyring](https://github.com/cyring) wrmsr: CPU 0 cannot set MSR 0x00000393 to 0x0000000100000000

Oh! this one is not supported at all on Ultra

Can you read this one ?
```sh
## MSR_DRAM_POWER_LIMIT
rdmsr -aX 0x00000618
```

cyring, 2025-04-02:
And this one:
```sh
## MSR_SECONDARY_TURBO_RATIO_LIMIT
rdmsr -aX 0x00000650
```

valery-kirichenko, 2025-04-02:
```
## MSR_DRAM_POWER_LIMIT
rdmsr -aX 0x00000618
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0

## MSR_SECONDARY_TURBO_RATIO_LIMIT
rdmsr -aX 0x00000650
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E
2E2E2E2E2E2E2E2E

rdmsr -aX 0x00000651
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
C0C080604030201
```

cyring, 2025-04-02:
@valery-kirichenko Hi

Can you test the new branch [Arrow_Lake](https://github.com/cyring/CoreFreq/tree/Arrow_Lake)
Make sure any previous _CoreFreq_ binary is removed then clone that branch
```sh
git pull -b Arrow_Lake https://github.com/cyring/CoreFreq.git
cd CoreFreq
make -j

## Save your files and stop your applications
## Now recommend to switch to single Root user
systemctl rescue

cd CoreFreq
## Start driver bound to a P-core
insmod build/corefreqk.ko ServiceProcessor=6
```

* If crash happens, please provide last trapped registers from `dmesg -t`

* If still running OK then exit back to multi-users; next start the Daemon and the Client

* Please make a screenshot, the **CPU `#006` has to be highlighted** but also one of the E-core, logically `#019`

## Test ODCM
* Go in the "`Settings`" window and enable the "`Experimental`" mode
* Go to "`Power, Current & Thermal`" window ; enable the "`Clock Modulation(ODCM)`" and set the "`DutyCycle`" to `50%`. Run a System stress from the "`Tools`" window, you should observe that frequency is capped to its half.

## Results
Please provide a refresh of the CLI output
```sh
./build/corefreq-cli -s -n -m -n -C 1 -n -W 1 -n -c 1 -n -g 1 -n -i 1 -n -M
```

valery-kirichenko, 2025-04-02:
@cyring it crashed again. I don't think dmesg -t provides logs from the previous boot (the one that crashed) but I will upload it for the sake of completeness. [dmesg-t.txt](https://github.com/user-attachments/files/19568525/dmesg-t.txt)

I tried using journalctl to get previous boot logs but it has nothing at the moment (or after that) of me loading the kernel module.

valery-kirichenko, 2025-04-02:
Stock

[image]

ODCM at 50%

[image]

cyring, 2025-04-02:
Thanks
When running stock the UI in a Xterm, do you see which CPUs are highlighted?

valery-kirichenko, 2025-04-02:
CPU 19

cyring, 2025-04-02:
> CPU 19

That's really odd. P-core has not been selected to service. But luckily it prevents from crashing.

What I'm finding different from previous architectures is that your CPU is listed with Hyperthreading capable but it is deactivated. That's why the Topology shows no Thread id.

Usually when SMT is enabled, two CPU will service: one physical Core and its sibling logical Core.
And if Hybrid then we have 3 CPUs servicing.

My understanding is some part of code, usually registers are crashing if accessed from a P-core aka CPU id `0` to `7`

We have verified most MSR registers, the remainders are the CSR registers, those are accessed through PCI remapping read-write. No tools for verify them one by one. PCI code has to be comment until the faulty is localized. Not an easy way to do remotely.

cyring, 2025-04-02:
> > What I'm finding different from previous architectures is that your CPU is listed with Hyperthreading capable but it is deactivated.
> 
> That's weird. Where do you see it? I thought hyperthreading support was removed in this architecture. That's why it has 20 threads total with 8 P-cores and 12 E-cores as per [Intel Specifications](https://www.intel.com/content/www/us/en/products/sku/241063/intel-core-ultra-7-processor-265k-30m-cache-up-to-5-50-ghz/specifications.html)
> 
> I will try to setup kdump later today to catch kernel panic

In [corefreq-cli](https://github.com/cyring/CoreFreq/blob/5c620b7f423f65abf4e328f873346d0ce9baeb44/x86_64/corefreq-cli.c#L2688)
```
Hyper-Threading Technology                                    HTT   [Capable]           
```
reflects the CPUID feature bit but indeed the driver has [enumerated](https://github.com/cyring/CoreFreq/blob/5c620b7f423f65abf4e328f873346d0ce9baeb44/x86_64/corefreqk.c#L2336) *no thread identifier** within the APIC number. Thus CoreFreq will deduce that SMT is OFF.

valery-kirichenko, 2025-04-03:
Well, at least I got call trace
[image]

cyring, 2025-04-03:
> Well, at least I got call trace

Excellent.

`RCX` register has a hexadecimal value of `2ff0` which is [`MSR_ADL_UNCORE_PERF_GLOBAL_CTRL`](https://github.com/cyring/CoreFreq/blob/46007a9fc872028101e221185d75efa46b9577c1/x86_64/intel_reg.h#L160C9-L160C41)
This register is read then write into the macro function [`Uncore_Counters_Set`](https://github.com/cyring/CoreFreq/blob/46007a9fc872028101e221185d75efa46b9577c1/x86_64/corefreqk.c#L14883`) called with argument `PMU` equal `ADL` in function [`Start_Uncore_Arrowlake`](https://github.com/cyring/CoreFreq/blob/46007a9fc872028101e221185d75efa46b9577c1/x86_64/corefreqk.c#L19599)

I believe if you try to do the same with MSR-Tools, it will raise an exception

```sh
rdmsr -aX 0x00002ff0
## Set bit 29 to enable the Uncore Fixed Counter (according to AlderLake specs)
wrmsr -a 0x00002ff0 0x20000000
```

cyring, 2025-04-03:
Logically `MSR_ADL_UNCORE_PERF_FIXED_CTR_CTRL(0x00002fde)` , `MSR_ADL_UNCORE_PERF_FIXED_CTR0(0x00002fdf)` and `MSR_ADL_UNCORE_PERF_GLOBAL_STATUS(0x00002ff2)` should also fail

cyring, 2025-04-03:
* Can you pull from branch `Arrow_Lake` then comment sections of code as bellow

https://github.com/cyring/CoreFreq/blob/cd9d961dd8508df3d8171551d2cd05e7074d706d/x86_64/corefreqk.c#L19592
```c
static void Stop_Uncore_Alderlake(void *arg)
{
	UNUSED(arg);
/*
	Uncore_Counters_Clear(ADL);
*/
}
```

https://github.com/cyring/CoreFreq/blob/cd9d961dd8508df3d8171551d2cd05e7074d706d/x86_64/corefreqk.c#L19599
```c
static void Start_Uncore_Arrowlake(void *arg)
{
	UNUSED(arg);
/*
	Uncore_Counters_Set(ADL);
*/
}
```

https://github.com/cyring/CoreFreq/blob/cd9d961dd8508df3d8171551d2cd05e7074d706d/x86_64/corefreqk.c#L15395
```c
#define PKG_Counters_Alderlake_Pcore(Core, T)				\
({/*									\
    RDTSCP_COUNTERx7(PUBLIC(RO(Proc))->Counter[T].PCLK ,		\
	MSR_PKG_C2_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC02,	\
	MSR_PKG_C3_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC03,	\
	MSR_PKG_C6_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC06,	\
	MSR_PKG_C7_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC07,	\
	MSR_PKG_C8_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC08,	\
	MSR_PKG_C9_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC09,	\
	MSR_ADL_UNCORE_PERF_FIXED_CTR0 ,				\
				PUBLIC(RO(Proc))->Counter[T].Uncore.FC0);\
*/})
```

* Rebuild all
```sh
make clean
make -j8
```

* Then start the module with `ServiceProcessor=1`

valery-kirichenko, 2025-04-03:
> Logically `MSR_ADL_UNCORE_PERF_FIXED_CTR_CTRL(0x00002fde)` , `MSR_ADL_UNCORE_PERF_FIXED_CTR0(0x00002fdf)` and `MSR_ADL_UNCORE_PERF_GLOBAL_STATUS(0x00002ff2)` should also fail

Not sure what I should set them to but..
```
root@fedora:~# rdmsr -aX 0x00002ff0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
root@fedora:~# wrmsr -a 0x00002ff0 0x20000000
wrmsr: CPU 0 cannot set MSR 0x00002ff0 to 0x0000000020000000
root@fedora:~# rdmsr -aX 0x00002fde
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
root@fedora:~# wrmsr -a 0x00002fde 0x20000000
root@fedora:~# wrmsr -a 0x00002fdf 0x20000000
root@fedora:~# wrmsr -a 0x00002ff2 0x20000000
```

valery-kirichenko, 2025-04-03:
> * Can you pull from branch `Arrow_Lake` then comment sections of code as bellow
> 
> 
> [CoreFreq/x86_64/corefreqk.c](https://github.com/cyring/CoreFreq/blob/cd9d961dd8508df3d8171551d2cd05e7074d706d/x86_64/corefreqk.c#L19592)
> 
> Line 19592 in [cd9d961](/cyring/CoreFreq/commit/cd9d961dd8508df3d8171551d2cd05e7074d706d)
>  static void Stop_Uncore_Alderlake(void *arg) 
> 
> static void Stop_Uncore_Alderlake(void *arg)
> {
> 	UNUSED(arg);
> /*
> 	Uncore_Counters_Clear(ADL);
> */
> }
> 
> [CoreFreq/x86_64/corefreqk.c](https://github.com/cyring/CoreFreq/blob/cd9d961dd8508df3d8171551d2cd05e7074d706d/x86_64/corefreqk.c#L19599)
> 
> Line 19599 in [cd9d961](/cyring/CoreFreq/commit/cd9d961dd8508df3d8171551d2cd05e7074d706d)
>  static void Start_Uncore_Arrowlake(void *arg) 
> 
> static void Start_Uncore_Arrowlake(void *arg)
> {
> 	UNUSED(arg);
> /*
> 	Uncore_Counters_Set(ADL);
> */
> }
> 
> [CoreFreq/x86_64/corefreqk.c](https://github.com/cyring/CoreFreq/blob/cd9d961dd8508df3d8171551d2cd05e7074d706d/x86_64/corefreqk.c#L15395)
> 
> Line 15395 in [cd9d961](/cyring/CoreFreq/commit/cd9d961dd8508df3d8171551d2cd05e7074d706d)
>  #define PKG_Counters_Alderlake_Pcore(Core, T)				\ 
> 
> #define PKG_Counters_Alderlake_Pcore(Core, T)				\
> ({/*									\
>     RDTSCP_COUNTERx7(PUBLIC(RO(Proc))->Counter[T].PCLK ,		\
> 	MSR_PKG_C2_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC02,	\
> 	MSR_PKG_C3_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC03,	\
> 	MSR_PKG_C6_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC06,	\
> 	MSR_PKG_C7_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC07,	\
> 	MSR_PKG_C8_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC08,	\
> 	MSR_PKG_C9_RESIDENCY,	PUBLIC(RO(Proc))->Counter[T].PC09,	\
> 	MSR_ADL_UNCORE_PERF_FIXED_CTR0 ,				\
> 				PUBLIC(RO(Proc))->Counter[T].Uncore.FC0);\
> */})
> 
>     * Rebuild all
> 
> 
> make clean
> make -j8
> 
>     * Then start the module with `ServiceProcessor=1`

It works :)
I can see total package power and total cores power

[image]

[corefreq.txt](https://github.com/user-attachments/files/19594077/corefreq.txt)

cyring, 2025-04-03:
> It works :)
> I can see total package power and total cores power

This is great!
I will push the changes pretty soon.
I don't need to comment the `MSR_PKG_C?_RESIDENCY`. I remembered we had tested them fine.

To sum up, the Arrow Lake Uncore PMU is unknown.

cyring, 2025-04-03:
@valery-kirichenko Hi

Commit a63ed85cf0298011de3f4fd452cc423d809531e5 is now providing the new functions for Arrow Lake. Those are also shared with Lunar Lake which could be closed ARL. But I have receive no report Lunar Lake architecture.

Without the `ServiceProcessor` option, the module should pick up one P-core and one E-core to service Package and Hybrid monitoring.

Once again, **test this commit in a safe condition** in case some MSR have been left behind.

----

Your `corefreq.txt` shows things differently and in a more expected way.
* Turbo Ratios of P-core are more coherent.
* Power, voltage and temperatures: you should stress your system and check if _CoreFreq_ displays the max values provided by specs
* In "Technologies": `L1 NPP`, `L1 Scrubbing`, `L2 AMP`, `L2 NLP` are `<OFF>`
  I don't know if it is a BIOS choice or a register support but you can try to toggle them to `< ON>`. (still in a safe condition)
* The same with `EEO` and `R2H` but also `C3A`, `C3U`, `HDC`

cyring, 2025-04-03:
```
Watchdog Timer                                                    TCO   <OFF>
```

If not disabled in BIOS, _CoreFreq_ has not detect it.  
In that case I will need the PCI identifiers list: `lspci -nn`

Thank you.

valery-kirichenko, 2025-04-04:
> Without the ServiceProcessor option, the module should pick up one P-core and one E-core to service Package and Hybrid monitoring.

This works:

[image]

> Power

It doesn't show up :(

[image]

Toggling L1 NPP, L2 AMP doesn't work. L1 scrubbing, L2 NLP, EEO, R2H, C3A, C3U does.

[image]

HDC can't be toggled

[image]

> If not disabled in BIOS, CoreFreq has not detect it.

I can't find an option to enable or disable Watchdog Timer, so here's lspci:

```
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:7d1b] (rev 01)
00:02.0 VGA compatible controller [0300]: Intel Corporation Arrow Lake-S [Intel Graphics] [8086:7d67] (rev 06)
00:04.0 Signal processing controller [1180]: Intel Corporation Device [8086:ad03] (rev 01)
00:06.0 PCI bridge [0604]: Intel Corporation Device [8086:ae4d] (rev 10)
00:07.0 PCI bridge [0604]: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #0 [8086:7ec4] (rev 10)
00:07.1 PCI bridge [0604]: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #1 [8086:7ec5] (rev 10)
00:08.0 System peripheral [0880]: Intel Corporation Device [8086:ae4c] (rev 10)
00:0a.0 Signal processing controller [1180]: Intel Corporation Device [8086:ad0d] (rev 01)
00:0b.0 Processing accelerators [1200]: Intel Corporation Arrow Lake NPU [8086:ad1d] (rev 01)
00:0d.0 USB controller [0c03]: Intel Corporation Meteor Lake-P Thunderbolt 4 USB Controller [8086:7ec0] (rev 10)
00:0d.2 USB controller [0c03]: Intel Corporation Meteor Lake-P Thunderbolt 4 NHI #0 [8086:7ec2] (rev 10)
00:0e.0 RAID bus controller [0104]: Intel Corporation Volume Management Device NVMe RAID Controller Intel Corporation [8086:ad0b]
00:14.0 RAM memory [0500]: Intel Corporation Device [8086:ae7f] (rev 10)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:ae0d] (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device [8086:ae23] (rev 10)
01:00.0 Ethernet controller [0200]: Aquantia Corp. AQC113C NBase-T/IEEE 802.3an Ethernet Controller [Marvell Scalable mGig] [1d6a:14c0] (rev 03)
80:14.0 USB controller [0c03]: Intel Corporation Device [8086:7f6e] (rev 10)
80:14.5 Non-VGA unclassified device [0000]: Intel Corporation Device [8086:7f2f] (rev 10)
80:15.0 Serial bus controller [0c80]: Intel Corporation Device [8086:7f4c] (rev 10)
80:15.1 Serial bus controller [0c80]: Intel Corporation Device [8086:7f4d] (rev 10)
80:15.2 Serial bus controller [0c80]: Intel Corporation Device [8086:7f4e] (rev 10)
80:16.0 Communication controller [0780]: Intel Corporation Device [8086:7f68] (rev 10)
80:17.0 SATA controller [0106]: Intel Corporation Device [8086:7f62] (rev 10)
80:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:7f40] (rev 10)
80:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:7f38] (rev 10)
80:1c.2 PCI bridge [0604]: Intel Corporation Device [8086:7f3a] (rev 10)
80:1c.3 PCI bridge [0604]: Intel Corporation Device [8086:7f3b] (rev 10)
80:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:7f30] (rev 10)
80:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:7f04] (rev 10)
80:1f.4 SMBus [0c05]: Intel Corporation Device [8086:7f23] (rev 10)
80:1f.5 Serial bus controller [0c80]: Intel Corporation Device [8086:7f24] (rev 10)
83:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7925]
84:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 0c)
85:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983 [144d:a808]
```

cyring, 2025-04-04:
> > Power
> 
> It doesn't show up :(

CoreFreq has selected CPU `16` and `19` to service.
But `Power` has to be monitored by a P-core

Not being on a P-core also impacts PC counters

```console
		Cycles		State(%)
PC02	                 0	   -nan
PC03	                 0	   -nan
PC04	                 0	   -nan
PC06	                 0	   -nan
PC07	                 0	   -nan
PC08	                 0	   -nan
PC09	                 0	   -nan
PC10	                 0	   -nan
MC6	        2313772135	    inf
PTSC	                 0
UNCORE	                 0
```

Since ARL is no more Hyperthreaded but without removing the CPUID capability, the driver makes some decisions which avoid P-cores

Solution: let the Power be also monitored by E-core ; but:
* value in watt **slightly differs between P-core and E-core**. See Energy MSR we [traced ](https://github.com/cyring/CoreFreq/issues/539#issuecomment-2764260272) previously
* value may be counted twice if service is fortunately shared between a P-core and a E-core
Workarround: use `ServiceProcessor` option until I find a fix

> Toggling L1 NPP, L2 AMP doesn't work

I suspect those feature bits has changed. But no Arrow Lake specs to confirm

> HDC can't be toggled

Probably HDC is now a legacy feature

> I can't find an option to enable or disable Watchdog Timer, so here's lspci:

Thank you. [`0x7f23`](https://github.com/cyring/CoreFreq/blob/5c620b7f423f65abf4e328f873346d0ce9baeb44/x86_64/corefreq-api.h#L1585C9-L1585C38) is the one I have into table
```
80:1f.4 SMBus [0c05]: Intel Corporation Device [8086:7f23] (rev 10)
```

... and it it was supposed to be probe at this line
https://github.com/cyring/CoreFreq/blob/5c620b7f423f65abf4e328f873346d0ce9baeb44/x86_64/corefreqk.c#L10510

cyring, 2025-04-04:
> > Power
> 
> It doesn't show up :(

Can you try latest commit b37b7c0a886104911fdb721e70032060bd6578a2 in any kind of `ServiceProcessor` combination. 
Especially when it happens that both Services are E-core ; like your screenshot case where CPU `16` and `19` are highlighted.

You should read the Power but also the Package temperature in the bottom-right ( and not `T[105C]` )

cyring, 2025-04-04:
> HDC can't be toggled

That's right. EAX bit 13 in CPUID has no HDC capabilty
```
CPU #0   function         EAX          EBX          ECX          EDX            
|- 00000006:00000000    00dfcff7     00000002     00000401     00000003         

CPU #19  function         EAX          EBX          ECX          EDX            
|- 00000006:00000000    00dfcff7     00000002     00000401     00130003         
```
```
00dfcff7 = 1101 1111 11[0]0 1111 1111 0111
```

cyring, 2025-04-04:
> Toggling L1 NPP, L2 AMP doesn't work.

* Can you edit this line of code:
https://github.com/cyring/CoreFreq/blob/b37b7c0a886104911fdb721e70032060bd6578a2/x86_64/corefreqk.h#L1683
... and replace with the following
```c
#define PerCore_Arrowlake_Query PerCore_Meteorlake_Query
```

* Rebuild, reload and then check L1 L2 features

valery-kirichenko, 2025-04-04:
Here's latest commit with suggested code change. Temperature shows up but still no luck with power (no ServiceProcessor was specified). Interestingly, now L1 NPP was on when I opened corefreq and it can be toggled. L2 AMP though was off and can't be toggled on.

[image]

Starting the driver with ServiceProcessor=0 allows for power to be read

[image]

cyring, 2025-04-04:
> Here's latest commit with suggested code change. Temperature shows up but still no luck with power (no ServiceProcessor was specified).

Since commit a8489b58584e0685e3dfca37f26bddf7632a0ffb I'm refactoring the monitoring for both P-core and E-core.

Please let me if you are getting the Frequency, Sensors (incl. Power, Thermal, Voltage), Instruction cycles, Core cycles, Idle C-States and Package cycles ?
(all reachable from the `[F3] View` menu)

> Interestingly, now L1 NPP was on when I opened corefreq and it can be toggled. L2 AMP though was off and can't be toggled on (nothing happens when I do that, UI allows for toggling).

L1 NPP is now part of the branch. L2 AMP is not supported.

cyring, 2025-04-05:
About Uncore, can you try those legacy registers ?
```
## MSR_SNB_UNCORE_PERF_GLOBAL_CTRL
rdmsr -aX 0x00000391

## MSR_SNB_UNCORE_PERF_GLOBAL_STATUS
rdmsr -aX 0x00000392

## MSR_SNB_UNCORE_PERF_FIXED_CTR_CTRL or MSR_SKL_UNCORE_PERF_FIXED_CTR_CTRL
rdmsr -aX 0x00000394

## MSR_SNB_UNCORE_PERF_FIXED_CTR0 or MSR_SKL_UNCORE_PERF_FIXED_CTR0
rdmsr -a 0x00000395

## MSR_HSW_UNCORE_RATIO_LIMIT
rdmsr -aX 0x00000620

## MSR_HSW_UNCORE_PERF_STATUS
rdmsr -aX 0x00000621
```

cyring, 2025-04-05:
For the same reasons of Arrow Lake, I'm convinced Meteor Lake is crashing with same Uncore MSR registers.

To anyone, I need _CoreFreq_ tests with [Core Ultra Series 1 processors](https://en.wikipedia.org/wiki/Meteor_Lake#List_of_Core_Ultra_Series_1_processors)

valery-kirichenko, 2025-04-05:
```
## MSR_SNB_UNCORE_PERF_GLOBAL_CTRL
rdmsr -aX 0x00000391
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0

## MSR_SNB_UNCORE_PERF_GLOBAL_STATUS
rdmsr -aX 0x00000392
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000
8000000300000000

## MSR_SNB_UNCORE_PERF_FIXED_CTR_CTRL or MSR_SKL_UNCORE_PERF_FIXED_CTR_CTRL
rdmsr -aX 0x00000394
rdmsr: CPU 0 cannot read MSR 0x00000394

## MSR_SNB_UNCORE_PERF_FIXED_CTR0 or MSR_SKL_UNCORE_PERF_FIXED_CTR0
rdmsr -a 0x00000395
rdmsr: CPU 0 cannot read MSR 0x00000395

## MSR_HSW_UNCORE_RATIO_LIMIT
rdmsr -aX 0x00000620
826
826
826
826
826
826
826
826
826
826
826
826
826
826
826
826
826
826
826
826

## MSR_HSW_UNCORE_PERF_STATUS
rdmsr -aX 0x00000621
345F0016
345F0016
345F0016
345F0016
345F0016
345F0016
345F0016
345F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
1A2F0016
```

valery-kirichenko, 2025-04-05:
Here are all sensors screenshots with the latest update

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

cyring, 2025-04-05:
> Here are all sensors screenshots with the latest update

Thank you very much. Great screenshots.

So the refactoring gives better results. To me, it also easier to maintain the source code of a single loop.

I'm a bit hesitant to employ the legacy Uncore MSR: some OK; others are crashing. I will wait for next SDM; hopefully Intel will more about Arrow Lake MSR.

However MSR_HSW_UNCORE_PERF_STATUS(0x00000621)
`345F0016` : P-core
`1A2F0016` : E-core
...  is read with last byte equal `0x16` which is a [Uncore frequency ratio](https://github.com/cyring/CoreFreq/blob/a8489b58584e0685e3dfca37f26bddf7632a0ffb/x86_64/intel_reg.h#L1762) of `22` when multiplied by a BCLK of `100 MHz`, we have something in the Uncore of a `2200 MHz`  frequency. Do you see such value in your BIOS ?

valery-kirichenko, 2025-04-05:
> Uncore of a 2200 MHz frequency. Do you see such value in your BIOS ?

Not really. Most my settings are set to auto and BIOS doesn't show what's used for many of them but where it does, there's nothing close to 2200 MHz (well, maybe except for iGPU)

[image]

[image]

[image]

[image]

Also, interestingly, BIOS shows HT as unsupported :)

[image]

cyring, 2025-04-05:
> Not really. Most my settings are set to auto and BIOS doesn't show what's used for many of them but where it does, there's nothing close to 2200 MHz (well, maybe except for iGPU)

Thank you.

MSR_HSW_UNCORE_PERF_STATUS has [voltage VID](https://github.com/cyring/CoreFreq/blob/a8489b58584e0685e3dfca37f26bddf7632a0ffb/x86_64/intel_reg.h#L1759) among the upper 16 bits

| Core   | MSR        | VID(hex) | VID(dec) | `VID / 8192`   | Voltage   |
|:------:|:----------:|---------:|---------:|---------------:|:---------:|
| P-core | `345F0016` | `345F`   | `13407`  | `13407 / 8192` | `1.6366V` |
| E-core | `1A2F0016` | `1A2F`   | `6703`   | `6703 / 8192`  | `0.8182V` |

But I have not seen such voltage within the screenshots.

valery-kirichenko, 2025-04-05:
Here are voltages I can find in BIOS

[image]
[image]
[image]
[image]

And also HWiNFO screenshot

[image]

cyring, 2025-04-05:
> And also HWiNFO screenshot

Could be that _CoreFreq_ Uncore Min and Max frequencies are the HWiNFO Ring/LLC Clock.

cyring, 2025-04-05:
We also have the BIOS and HWiNFO "PCH 0.82V" which is very closed to the computed `0.8182V`

[45 more replies]

closed 2025-05-22 by cyring