cyring/CoreFreq #540: Unable to start CoreFreq daemon on Ubuntu 22.04.5 with kernel 6.8.0-50-generic

antoine-leclerc-dev, 2025-04-10:
I'm encountering an error while trying to use CoreFreq on my Ubuntu 22.04.5 LTS system with kernel 6.8.0-50-generic.

The module compiles and loads successfully, but when attempting to start the daemon, it fails with:

```
Driver connection error code 22
corefreq-ro-shm: 'Invalid argument' @ line 9758
```

lscpu:

```
root@server:~#  lscpu
Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          52 bits physical, 57 bits virtual
  Byte Order:             Little Endian
CPU(s):                   384
  On-line CPU(s) list:    0-383
Vendor ID:                AuthenticAMD
  Model name:             AMD EPYC 9654 96-Core Processor
    CPU family:           25
    Model:                17
    Thread(s) per core:   2
    Core(s) per socket:   96
    Socket(s):            2
    Stepping:             1
    Frequency boost:      enabled
    CPU max MHz:          3707.8120
    CPU min MHz:          1500.0000
    BogoMIPS:             4799.88
    Flags:                fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good amd_lbr_v2 nopl nonstop_tsc cpuid extd_a
                          picid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw i
                          bs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba perfmon_v2 ibrs ibpb stibp ibrs_enhanced vmmcall fsgsbase bmi1 avx2 smep bmi2 erms invpcid cqm r
                          dt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local user_shstk avx512_bf16 clzer
                          o irperf xsaveerptr rdpru wbnoinvd amd_ppin cppc arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif x2avic v_spec_ctrl vnmi avx51
                          2vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq la57 rdpid overflow_recov succor smca fsrm flush_l1d debug_swap
Virtualization features:  
  Virtualization:         AMD-V
Caches (sum of all):      
  L1d:                    6 MiB (192 instances)
  L1i:                    6 MiB (192 instances)
  L2:                     192 MiB (192 instances)
  L3:                     768 MiB (24 instances)
NUMA:                     
  NUMA node(s):           2
  NUMA node0 CPU(s):      0-95,192-287
  NUMA node1 CPU(s):      96-191,288-383
Vulnerabilities:          
  Gather data sampling:   Not affected
  Itlb multihit:          Not affected
  L1tf:                   Not affected
  Mds:                    Not affected
  Meltdown:               Not affected
  Mmio stale data:        Not affected
  Reg file data sampling: Not affected
  Retbleed:               Not affected
  Spec rstack overflow:   Mitigation; Safe RET
  Spec store bypass:      Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:             Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:             Mitigation; Enhanced / Automatic IBRS; IBPB conditional; STIBP always-on; RSB filling; PBRSB-eIBRS Not affected; BHI Not affected
  Srbds:                  Not affected
  Tsx async abort:        Not affected
```

Full cmd history:
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
root@server:~#  uname -a
Linux server 6.8.0-50-generic #51~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 21 12:03:03 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
root@server:~#  git clone https://github.com/cyring/CoreFreq.git
Cloning into 'CoreFreq'...
remote: Enumerating objects: 13864, done.
remote: Counting objects: 100% (726/726), done.
remote: Compressing objects: 100% (339/339), done.
remote: Total 13864 (delta 431), reused 402 (delta 387), pack-reused 13138 (from 3)
Receiving objects: 100% (13864/13864), 8.25 MiB | 21.71 MiB/s, done.
Resolving deltas: 100% (10961/10961), done.
root@server:~#  cd CoreFreq/
root@server:~/CoreFreq#   git checkout -b 2.0.1
Switched to a new branch '2.0.1'
root@server:~/CoreFreq#  make -j
  CC [build/corefreqd.o]
  CC [build/corefreqm.o]
  CC [build/corefreq-cli.o]
  CC [build/corefreq-ui.o]
  CC [build/corefreq-cli-rsc.o]
  CC [build/corefreq-cli-json.o]
  CC [build/corefreq-cli-extra.o]
  LD [build/corefreqd]
  LD [build/corefreq-cli]
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /root/CoreFreq/build/module/corefreqk.o
/root/CoreFreq/build/module/corefreqk.c: In function ‘Intel_Watchdog’:
/root/CoreFreq/build/module/corefreqk.c:10521:1: warning: the frame size of 1096 bytes is larger than 1024 bytes [-Wframe-larger-than=]
10521 | }
      | ^
  LD [M]  /root/CoreFreq/build/corefreqk.o
  MODPOST /root/CoreFreq/build/Module.symvers
  CC [M]  /root/CoreFreq/build/corefreqk.mod.o
  LD [M]  /root/CoreFreq/build/corefreqk.ko
  BTF [M] /root/CoreFreq/build/corefreqk.ko
Skipping BTF generation for /root/CoreFreq/build/corefreqk.ko due to unavailability of vmlinux
root@server:~/CoreFreq#  make install
make -C /lib/modules/6.8.0-50-generic/build M=/root/CoreFreq/build modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-50-generic'
  INSTALL /lib/modules/6.8.0-50-generic/updates/corefreqk.ko
  SIGN    /lib/modules/6.8.0-50-generic/updates/corefreqk.ko
  DEPMOD  /lib/modules/6.8.0-50-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-50-generic'
root@server:~/CoreFreq#  insmod build/corefreqk.ko
root@server:~/CoreFreq#  dmesg | tail
...
[8276021.395508] CoreFreq(127:319:-1): Processor [ AF_11] Architecture [EPYC/Genoa] SMT [384/384]
root@server:~/CoreFreq#  ./build/corefreqd
Driver connection error code 22
corefreq-ro-shm: 'Invalid argument' @ line 9758
root@server:~/CoreFreq#  ls -l /dev | grep corefreq
crw-------   1 root root    509,   0 Mar 28 13:15 corefreqk
root@server:~/CoreFreq#   strace -e trace=open,ioctl,shmget,shmat ./build/corefreqd
Driver connection error code 22
corefreq-ro-shm: 'Invalid argument' @ line 9758
+++ exited with 5 +++
root@server:~/CoreFreq#  modinfo corefreqk
filename:       /lib/modules/6.8.0-50-generic/updates/corefreqk.ko
version:        2.0.1
license:        GPL
description:    CoreFreq Processor Driver
author:         CYRIL COURTIAT <labs[at]cyring[dot]fr>
srcversion:     A529122121574DA40EE65CA
depends:        
retpoline:      Y
name:           corefreqk
vermagic:       6.8.0-50-generic SMP preempt mod_unload modversions 
parm:           ArchID:Force an architecture (ID) (int)
.....
root@server:~/CoreFreq# 

```

cyring, 2025-04-10:
> I'm encountering an error while trying to use CoreFreq on my Ubuntu 22.04.5 LTS system with kernel 6.8.0-50-generic.
> 
> The module compiles and loads successfully, but when attempting to start the daemon, it fails with:
> 
> Full cmd history:
> 
> ```
>   CC [M]  /root/CoreFreq/build/module/corefreqk.o
> /root/CoreFreq/build/module/corefreqk.c: In function ‘Intel_Watchdog’:
> /root/CoreFreq/build/module/corefreqk.c:10521:1: warning: the frame size of 1096 bytes is larger than 1024 bytes [-Wframe-larger-than=]
> 10521 | }
>       | ^
> ```

Hello,

This [issue](https://github.com/cyring/CoreFreq/discussions/523#discussioncomment-11880668) comes from the stack size 
```sh
make clean
make -j CC='cc -Wframe-larger-than=12288'
## Or to avoid warnings, explicitly specify gcc
make -j CC='gcc -Wframe-larger-than=12288'
```

If you are still getting size warnings during build then try to increase `-Wframe-larger-than` until it is accepted with no issue.

antoine-leclerc-dev, 2025-04-10:
Hello,

I tried building with:

`make -j CC='gcc-12 -Wframe-larger-than=16384'`
or
`make -j CC="gcc-12" KBUILD_CC="gcc-12" KBUILD_CFLAGS="-Wframe-larger-than=16384"`

Warnings still appeared during compilation of the kernel module. It became clear that the CLI flags were not being respected by the kernel module build.

To resolve this, I edited the top-level Makefile and added:

`ccflags-y += -Wframe-larger-than=16384`
Then ran:

```
# make clean
# make -j CC=gcc-12

  CC [M]  /root/CoreFreq/build/module/corefreqk.o
  LD [M]  /root/CoreFreq/build/corefreqk.o

  Skipping BTF generation for /root/CoreFreq/build/corefreqk.ko due to unavailability of vmlinux
```
No more frame-larger-than warnings, and build completed successfully.

`# insmod build/corefreqk.ko`
No output or errors were shown, and dmesg did not display any major issues related to corefreqk.

```
# ./build/corefreqd

Driver connection error code 22
corefreq-ro-shm: 'Invalid argument' @ line 9738
```

I also understand that -Wframe-larger-than is purely for compile-time diagnostics and has no impact on runtime stack size or behavior. Thus, I think it's unlikely to be the root cause of the current error.

[bash_history.txt](https://github.com/user-attachments/files/19688453/bash_history.txt)

Thanks for your time and support on this great project

cyring, 2025-04-10:
* I have been thinking that you have a high core count processor. 
It's worth to help the build with the following:
```
make CORE_COUNT=512
```
*  Default being `256`, I have to check how code has automatically managed that ...

* My stack frame usage is a bit to much. I will move this PCI list to global allocation. Fix to come...

* You have a spectacular processor. Feel free to share screenshots and Cli outputs like in the Wiki/[CPU support](https://github.com/cyring/CoreFreq/wiki/CPU-support)
  It happens that I notice things to polish or fix within the screenshots

Enjoy _CoreFreq_

Regards

antoine-leclerc-dev, 2025-04-10:
It was exactly that. Thank you for your help :)

Let me know if there's anything else you need, I'd be glad to help

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2025-04-10 by antoine-leclerc-dev

cyring, 2025-04-10:
Your System is huge! Terminal is not big enough to show all cores -;)

* You have UI build options to exclude panels of your choice
```sh
make clean
make -j CORE_COUNT=512 NO_FOOTER=1 NO_UPPER=1 NO_HEADER==1
```

[image]

* You can build the UMC frequency monitoring as below
This is PMU programming ; try this in a safe way: save your files, close your apps.
```sh
make clean
make -j CORE_COUNT=512 NO_UPPER=1 NO_FOOTER=1 ARCH_PMC=UMC
```
UMC is at third bottom row. As an example, my 3950X which has two counters:

[image]

* Bottom right of your screenshot, the Package voltage is wrongly displayed at `V[0.01]`
It may be fixed if you bind the _CoreFreq_ service to one CPU of the first package (aka socket)
```sh
insmod ./build/corefreqk.ko ServiceProcessor=0
```

* I would also appreciate the output of these CLI options
```sh
corefreq-cli -s -n -z -n -B -C 1
```

antoine-leclerc-dev, 2025-04-10:
[corefreq-cli.txt](https://github.com/user-attachments/files/19695053/corefreq-cli.txt)

[dmidecode-ram.txt](https://github.com/user-attachments/files/19695054/dmidecode-ram.txt)

[corefreq-cli (umc+1socket param).txt](https://github.com/user-attachments/files/19695058/corefreq-cli.umc%2B1socket.param.txt)

[module_load_1socket_dmesg.txt](https://github.com/user-attachments/files/19695061/module_load_1socket_dmesg.txt)

I compiled it with: `root@server:~/CoreFreq#  make -j CORE_COUNT=512 NO_UPPER=1 NO_FOOTER=1 ARCH_PMC=UMC`
but I cannot see the UMC bottom row
[image]

antoine-leclerc-dev, 2025-04-10:
The  parameter `insmod ./build/corefreqk.ko ServiceProcessor=0` doesn't seem to fix the voltage problem on the footer

[image]

cyring, 2025-04-11:
Post Zen3, I still have an unsolved issue to count the number of DIMM ranks.
It should be 2 ranks and thus a DIMM size of `32GB` in each geometry.

cyring, 2025-04-11:
> The parameter `insmod ./build/corefreqk.ko ServiceProcessor=0` doesn't seem to fix the voltage problem on the footer
> 
> [image]

The past [tests](https://gist.github.com/cyring/427d90c133afd88bfc8bd69a05fc045d) made with a Genoa EPYC 9274F don't show any evidence that Package voltage was correctly retrieved or computed
Thus with this architecture I'm back to square one to find specs of Package and SOC voltage registers.

I made a special command tool to dig into Zen registers: [`zencli`](https://github.com/cyring/Tips/blob/master/C/zencli.c)
You grab the source code, compile and run as root
```sh
cc zencli.c -o zencli
```

Based on all [SMU addresses](https://github.com/cyring/CoreFreq/blob/6509cc4ba82b95026403cf81ea11246d29a5501c/x86_64/amd_reg.h#L2260) known so far, please dump the followings (keeping the comments with your results)

```sh
## SMU_AMD_F17H_CORE_VID
### plane 0
./zencli smu 0x0005a04c
### plane 1
./zencli smu 0x0005a050

## SMU_AMD_F17H_SVI
### plane 0
./zencli smu 0x0005a00c
### plane 1
./zencli smu 0x0005a010

## SMU_AMD_F17_60H_SVI
### plane 0
./zencli smu 0x0006f038
### plane 1
./zencli smu 0x0006f03c

## SMU_AMD_RMB_SVI
### plane 0
./zencli smu 0x0006f010
### plane 1
./zencli smu 0x0006f014
```

Thank you.

cyring, 2025-04-11:
@antoine-leclerc-dev 

Next commit will warn that message and stop loading the module

[image]

What do you think about it ?

antoine-leclerc-dev, 2025-04-11:
I find the message very helpful for future cases like mine.

Out of curiosity and ignorance, what would be the drawback of supporting 1024 by default during compilation?

```
root@server:~#  ## SMU_AMD_F17H_CORE_VID
 ### plane 0
 ./zencli smu 0x0005a04c
[0x0005a04c] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
root@server:~#  ### plane 1
 ./zencli smu 0x0005a050
[0x0005a050] READ(smu) = 0x00000b07 (2823)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 1011 0000 0111
root@server:~#  ## SMU_AMD_F17H_SVI
 ### plane 0
 ./zencli smu 0x0005a00c
[0x0005a00c] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
root@server:~#  ### plane 1
 ./zencli smu 0x0005a010
[0x0005a010] READ(smu) = 0x00009e41 (40513)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 1001 1110 0100 0001
root@server:~#  ## SMU_AMD_F17_60H_SVI
 ### plane 0
 ./zencli smu 0x0006f038
[0x0006f038] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
root@server:~#  ### plane 1
 ./zencli smu 0x0006f03c
[0x0006f03c] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
root@server:~#  ## SMU_AMD_RMB_SVI
 ### plane 0
 ./zencli smu 0x0006f010
[0x0006f010] READ(smu) = 0x00000001 (1)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001
root@server:~#  ### plane 1
 ./zencli smu 0x0006f014
[0x0006f014] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
root@server:~# 
```

cyring, 2025-04-12:
> Out of curiosity and ignorance, what would be the drawback of supporting 1024 by default during compilation?

Kernel memory footprint and lost CPU cycles.
* Most data are per CPU and
for the 3 binaries driver, daemon, client
* bitmaps will be 1024 length
  and operations like wide compare will wast a few instructions 

Allocated memory in User space is not an issue but in Kernel you have to care about sharing space with other drivers.

128-bits compare and exchange is available in x86_64. That's why `CORE_COUNT` is a multiple of 128 CPUs
Exception to the 64 CPUs mode which employs a basic  atomic comparaison which I believe will cover most processors.

Times a go I have been  querying that question and EPYC had quickly shown that 256 CPUs is becoming the servers trend.

cyring, 2025-04-13:
@antoine-leclerc-dev Hello,

I have found nothing meaningful in old SMU registers.

Commit 70649701734124fe3f17660aa52d24d73e34ebaf is now using default formula.

Can you please take a screenshot ?  
The package temperature should be equal to the highest Core voltage.  
Using `Tools > Turbo Random CPU`, you can apply random stress to verify the above.

The views Voltage and Power have no more SoC (or Uncore) computed. They have to be zero.

antoine-leclerc-dev, 2025-04-13:
[image]
[image]

[image]

antoine-leclerc-dev, 2025-04-13:
I'm also encountering a rather strange bug, though I'm not sure if it's related to my system.

Inside any dropdown menu—for example, "Settings"—if you follow this sequence, the SMBIOS overlay appears: After a pause of 5–7 seconds without pressing any keys, hold down the down arrow key.
After holding it for about 3–4 seconds, the SMBIOS overlay pops up, regardless of which menu you're in.

[image]

cyring, 2025-04-14:
Could you dump the P-State registers with the MSR-Tools ?
1. When System is idling
2. When stressing the System
    * All Cores
    * Turbo: a single Core on package 0

```sh
## MSR_AMD_F17H_HW_PSTATE_STATUS
rdmsr -aX 0xc0010293
```

zaccaro, 2025-04-14:
I think I didn’t explain myself well.
The SMBIOS window appears as a banner, even though it shouldn't.
It shows up on top of any other open window.
What’s peculiar is the trigger that causes it to appear:
From any menu, you need to hold down the down arrow key — but it only happens if no key has been pressed in the last 5–7 seconds.

The simplest way to reproduce it is:
Open the settings menu, then press the down arrow once, wait 5–7 seconds, then hold the down arrow key for 3–4 seconds. At that point, the SMBIOS banner will appear, overlaying the settings window.

cyring, 2025-04-14:
> I think I didn’t explain myself well. The SMBIOS window appears as a banner, even though it shouldn't. It shows up on top of any other open window. What’s peculiar is the trigger that causes it to [...]

Thank you for the details.
I can reproduce such issue in very slow drawing circonstancies like virtualization which messes up sending _phantom_ keystrokes to the ring buffer.

Is it a VM ?
What is the terminal ?

antoine-leclerc-dev, 2025-04-14:
It's a bare-metal server, and I'm accessing it through the internet using a VPN from my Ubuntu desktop

cyring, 2025-04-14:
> It's a bare-metal server, and I'm accessing it through the internet via SSH using the GNOME Terminal from my Ubuntu desktop

Thanks

Terminal may be faster with a XTerm; next, other way to do with ssh
```sh
ssh -t corefreq-cli
```

But I wonder if you are encountered the same issue when using a VT console directly on server ?

cyring, 2025-04-14:
My `/etc/ssh/sshd_config` allow these options
```ini
X11Forwarding yes
X11DisplayOffset 10
X11UseLocalhost yes
```

And session connects with `ssh -Y`

cyring, 2025-04-14:
> It's a bare-metal server, and I'm accessing it through the internet via SSH using the GNOME Terminal from my Ubuntu desktop

I'm now able to reproduce the bug; especially using a _slow_ link, when sliding any UI windows with [Shift]+[Arrows], SMBIOS may show up

antoine-leclerc-dev, 2025-04-14:
[MSR_AMD_F17H_HW_PSTATE_STATUS.txt](https://github.com/user-attachments/files/19734951/MSR_AMD_F17H_HW_PSTATE_STATUS.txt)
[sensors.txt](https://github.com/user-attachments/files/19734950/sensors.txt)

cyring, 2025-04-15:
Hello,
Can you please show what you get from this SMU register ?
```sh
## Idle
./zencli smu 0x73014

## Stressed
./zencli smu 0x73014
```

antoine-leclerc-dev, 2025-04-15:
```
## STRESSED
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a281 (107137)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0010 1000 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a241 (107073)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0010 0100 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a241 (107073)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0010 0100 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a241 (107073)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0010 0100 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a241 (107073)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0010 0100 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a241 (107073)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0010 0100 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a241 (107073)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0010 0100 0001
root@server:~# 

## IDLE
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a342 (107330)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0011 0100 0010
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001aa01 (109057)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 1010 0000 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a302 (107266)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0011 0000 0010
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a342 (107330)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0011 0100 0010
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a0c2 (106690)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0000 1100 0010
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a0c2 (106690)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0000 1100 0010
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a342 (107330)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0011 0100 0010
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x00019bc1 (105409)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1001 1011 1100 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a342 (107330)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0011 0100 0010
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x00019c01 (105473)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1001 1100 0000 0001
root@server:~#  ./zencli smu 0x73014
[0x00073014] READ(smu) = 0x0001a342 (107330)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1010 0011 0100 0010
root@server:~# 
```

antoine-leclerc-dev, 2025-04-15:
let me know if I can do any more tests for you. I'd be happy to help

cyring, 2025-04-15:
> let me know if I can do any more tests for you. I'd be happy to help

I will. Thank you very much for your help.

cyring, 2025-04-17:
I'm reading that `0x5A150` may give VID on Zen4/Zen5 ; can you please dump in idle and stressed system load ?
```sh
./zencli smu 0x5A150
```

antoine-leclerc-dev, 2025-04-17:
```
## IDLE smu 0x5A150
root@server:~#  while true; do ./zencli smu 0x5A150; sleep 1; done;
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

## UNDER LOAD smu 0x5A150
root@server:~#  while true; do ./zencli smu 0x5A150; sleep 1; done;
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x0005a150] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

## IDLE smu 0x5A054
root@server:~#   while true; do ./zencli smu 0x5A054; sleep 1; done
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
[... 23 lines cut ...]
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
[... cut ...] 04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
[0x0005a054] READ(smu) = 0x00000101 (257)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0000 0001
```

cyring, 2025-04-17:
<details>

> ```
> ## STRESSED
> root@server:~#  ./zencli smu 0x73014
> [0x00073014] READ(smu) = 0x0001a281 (107137)
>    60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
>  0000 0000 0000 [...]

</details>

## STRESSED

| Value      | VID `[15-6]`bits | `VID x 0.00625 V` | `1.55 - (VID x 0.00625) V` |
|------------|------------------|-------------------|----------------------------|
|`0x0001a281`|`0x8a (138)`      |`0,8625`           |`0,6875`                    |
|`0x0001a241`|`0x89 (137)`      |`0,85625`          |`0,69375`                   |

## IDLE

| Value      | VID `[15-6]`bits | `VID x 0.00625 V` | `1.55 - (VID x 0.00625) V` |
|------------|------------------|-------------------|----------------------------|
|`0x0001aa01`|`0xa8 (168)`      |`1,05`             |`0,5`                       |
|`0x00019bc1`|`0x6f (111)`      |`0,69375`          |`0,85625`                   |

* It does not flutuate as coherente Vcore

cyring, 2025-04-17:
> [MSR_AMD_F17H_HW_PSTATE_STATUS.txt](https://github.com/user-attachments/files/19734951/MSR_AMD_F17H_HW_PSTATE_STATUS.txt) [sensors.txt](https://github.com/user-attachments/files/19734950/sensors.txt)

## Idling

| Value      | VID `[22-14]`(dec)| `VID x 0.00625 V` | `1.55 - (VID x 0.00625) V` |
|------------|-------------------|-------------------|----------------------------|
|`0x9CCC5A`  |`115`              |`0,71875`          |`-`                         |
|`0xAA9494`  |`170` `!`          |`1.0625`           |`-`                         |

## All Cores

| Value      | VID `[22-14]`(dec)| `VID x 0.00625 V` | `1.55 - (VID x 0.00625) V` |
|------------|-------------------|-------------------|----------------------------|
|`0x21086B`  |`132`              |`0,825`            |`-`                         |
|`0x20C86B`  |`131`              |`0.81875`          |`-`                         |

cyring, 2025-04-17:
@antoine-leclerc-dev Hi,

I have other tests to carry on, please contact me by mail
`labs[at]cyring[dot]fr`

cyring, 2025-04-27:
Hello,

I have recently fixed the DIMM Geometry on Zen4/[Hawk Point](https://gist.github.com/cyring/8d2caa132674fbab321d7aa7b9003abc)
And I believe with your EPYC you can try different addresses until you observe good values.

In function [`AMD_DataFabric_Genoa`](https://github.com/cyring/CoreFreq/blob/adf8cbadc5588f81ab1cca44e4b4a42c1761d9bd/x86_64/corefreqk.c#L7339) you will change the two addresses `0x40, 0x90,` within `AMD_17h_DataFabric`, with those combinations

```c
0x40, 0x94,
0x40, 0x98,
0x44, 0x94,
0x48, 0x98,
0x4c, 0x98,
```

Each time, rebuilding and reloading _CoreFreq_ to display the UMC `corefreq-cli -M`

For your information, Turin specs about these two SMU registers

[image]

[image]

* Genoa EPYC may have a larger address map that's why I'm testing `0x4c, 0x98,`

cyring, 2025-05-10:
I'm finding this potential voltage of `1.2125` at `0x0005a010`

| Value      | IDD `[7-0]` | `(VID x 0.00625) V` | `1.55 - (VID x 0.00625) V`|
|------------|-------------|---------------------|---------------------------|
|`IDLE`      |`194`        |**`1.2125`**         |`0.3375`                   |
|`STRESS`    |`1`          |`0.0063`             |`1.5437`                   |

Any idea what it could be within BIOS of ASRockRack: CPU, SoC, SMU, MEM ?

antoine-leclerc-dev, 2025-05-10:
Seems promising. Not sure which one it corresponds to, but note that DDR5 modules have their own VRM and take 12V directly