cyring/CoreFreq #548: No temperature info on 9950X, "Unable to find AMD Northbridge id"

vient, 2025-05-21:
```
[ 1048.213438] CoreFreq: AMD_SMN_Read(0, 59808) TryLock
[ 1048.213440] ------------[ cut here ]------------
[ 1048.213440] Unable to find AMD Northbridge id for 0000:00:18.0
[ 1048.213442] WARNING: CPU: 11 PID: 0 at arch/x86/include/asm/amd_nb.h:100 CCD_AMD_Family_19h_Zen4_Temp+0x2c4/0x400 [corefreqk]
[ 1048.213445] Modules linked in: corefreqk(OE) cpuid msr k10temp exasock(OE) kvm_amd kvm crct10dif_pclmul ghash_clmulni_intel sha256_ssse3 sha1_ssse3 nls_iso8859_1 aesni_intel crypto_simd btusb cryptd btrtl btbcm btintel rapl eeepc_wmi snd_hda_codec_hdmi bluetooth wmi_bmof snd_usb_audio snd_usbmidi_lib snd_hda_intel snd_rawmidi snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec snd_seq_device snd_hda_core joydev snd_hwdep ccp input_leds mc snd_pcm snd_timer snd soundcore ecdh_generic ecc exanic(OE) i2c_algo_bit ucsi_acpi typec_ucsi typec mac_hid sch_fq_codel binfmt_misc drm ramoops pstore_blk reed_solomon efi_pstore pstore_zone ip_tables x_tables autofs4 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid0 multipath linear raid1 uas hid_generic usbhid usb_storage hid mfd_aaeon asus_wmi sparse_keymap platform_profile crc32_pclmul i2c_piix4 thunderbolt ahci libahci igc xhci_pci xhci_pci_renesas wmi video
[ 1048.213459] CPU: 11 PID: 0 Comm: swapper/11 Tainted: G        W  OE     5.15.0-139-generic #149~20.04.1-Ubuntu
[ 1048.213460] Hardware name: ASUS System Product Name/ROG CROSSHAIR X870E HERO, BIOS 1203 03/04/2025
[ 1048.213460] RIP: 0010:CCD_AMD_Family_19h_Zen4_Temp+0x2c4/0x400 [corefreqk]
[ 1048.213463] Code: 9a d4 09 00 e8 ed 2e e7 dc eb 9f 49 8b b7 20 01 00 00 48 85 f6 75 07 49 8b b7 d0 00 00 00 48 c7 c7 88 22 c5 c0 e8 59 8a 4b dd <0f> 0b 31 db e8 f3 6b 86 dc 66 39 c3 0f 83 2a fe ff ff 48 8b 05 0b
[ 1048.213464] RSP: 0018:ffffadc440448e38 EFLAGS: 00010082
[ 1048.213464] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000027
[ 1048.213465] RDX: 0000000000000027 RSI: 0000000000000003 RDI: ffffa0083e0e0588
[ 1048.213465] RBP: ffffadc440448e80 R08: ffffa0083e0e0580 R09: 0000000000000001
[ 1048.213465] R10: ffffffffa066f774 R11: 00000000a066f765 R12: 0000000000000001
[ 1048.213466] R13: 0000000000000190 R14: ffffa0010d2e7000 R15: ffffa001019b3000
[ 1048.213466] FS:  0000000000000000(0000) GS:ffffa0083e0c0000(0000) knlGS:0000000000000000
[ 1048.213467] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1048.213468] CR2: 00007fffb8000010 CR3: 000000010cc42000 CR4: 0000000000750ee0
[ 1048.213468] PKRU: 55555554
[ 1048.213468] Call Trace:
[ 1048.213468]  <IRQ>
[ 1048.213469]  ? show_regs.cold+0x1a/0x1f
[ 1048.213470]  ? CCD_AMD_Family_19h_Zen4_Temp+0x2c4/0x400 [corefreqk]
[ 1048.213472]  ? __warn+0x8b/0xe0
[ 1048.213473]  ? CCD_AMD_Family_19h_Zen4_Temp+0x2c4/0x400 [corefreqk]
[ 1048.213476]  ? report_bug+0xd5/0x110
[ 1048.213477]  ? handle_bug+0x39/0x90
[ 1048.213478]  ? exc_invalid_op+0x19/0x70
[ 1048.213478]  ? asm_exc_invalid_op+0x1b/0x20
[ 1048.213480]  ? CCD_AMD_Family_19h_Zen4_Temp+0x2c4/0x400 [corefreqk]
[ 1048.213482]  ? CCD_AMD_Family_19h_Zen4_Temp+0x2c4/0x400 [corefreqk]
[ 1048.213485]  ? Cycle_AMD_Family_15h+0x450/0x450 [corefreqk]
[ 1048.213488]  ? Start_Uncore_Alderlake+0xa0/0xa0 [corefreqk]
[ 1048.213491]  Cycle_AMD_Family_17h+0x30e/0x570 [corefreqk]
[ 1048.213493]  Cycle_AMD_Zen4_RPL+0xab/0xe0 [corefreqk]
[ 1048.213496]  ? Cycle_AMD_F17h_Zen+0xe0/0xe0 [corefreqk]
[ 1048.213499]  __hrtimer_run_queues+0xc8/0x1d0
[ 1048.213500]  hrtimer_interrupt+0x109/0x230
[ 1048.213501]  __sysvec_apic_timer_interrupt+0x5e/0xe0
[ 1048.213501]  sysvec_apic_timer_interrupt+0x7b/0x90
[ 1048.213502]  </IRQ>
[ 1048.213502]  <TASK>
[ 1048.213503]  asm_sysvec_apic_timer_interrupt+0x1b/0x20
[ 1048.213504] RIP: 0010:cpu_idle_poll+0x33/0xd0
[ 1048.213505] Code: 53 65 8b 15 7f 81 e7 61 66 90 e8 c8 23 3c ff fb 66 0f 1f 44 00 00 65 48 8b 1c 25 c0 fb 01 00 48 8b 03 a8 08 74 0b eb 1c f3 90 <48> 8b 03 a8 08 75 13 8b 05 f0 c9 49 01 85 c0 75 ed e8 57 b2 3e ff
[ 1048.213505] RSP: 0018:ffffadc440163e98 EFLAGS: 00000202
[ 1048.213506] RAX: 0000000000000001 RBX: ffffa001008a2000 RCX: 0000000000000000
[ 1048.213506] RDX: ffffa0083e0f2fe0 RSI: 00000000189d8355 RDI: 00000000002f2f4e
[ 1048.213507] RBP: ffffadc440163ea0 R08: 000000f40970008b R09: 0000000000000000
[ 1048.213507] R10: 0000000000000000 R11: 0000000000000001 R12: ffffa001008a2000
[ 1048.213507] R13: ffffa001008a2000 R14: 0000000000000000 R15: 0000000000000000
[ 1048.213508]  ? cpu_idle_poll+0x18/0xd0
[ 1048.213509]  do_idle+0x67/0x270
[ 1048.213509]  ? complete+0x3f/0x50
[ 1048.213510]  cpu_startup_entry+0x20/0x30
[ 1048.213510]  start_secondary+0x12a/0x180
[ 1048.213511]  secondary_startup_64_no_verify+0xc2/0xcb
[ 1048.213512]  </TASK>
[ 1048.213512] ---[ end trace 86be3d16b8d6bc8b ]---
```

Some info just in case
```
CoreFreq# ./build/corefreq-cli -s -n -z -n -B -n -k -n -C 1
Processor                                  [AMD Ryzen 9 9950X 16-Core Processor]
|- Architecture                                             [Zen5/Granite Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0b404023]
|- Signature                                                           [  BF_44]
[... 463 lines cut ...]
012 5199.10   218  1.3625    0  000000000000257856    3.934570312   3.934570312
013 5199.10   218  1.3625    0  000000000000253957    3.875076294   3.875076294
014 5196.78   218  1.3625    0  000000000000304737    4.649917603   4.649917603
015 5199.82   218  1.3625    0  000000000000302224    4.611572266   4.611572266
016        OFF
017        OFF
018        OFF
019        OFF
020        OFF
021        OFF
022        OFF
023        OFF
024        OFF
025        OFF
026        OFF
027        OFF
028        OFF
029        OFF
030        OFF
031        OFF

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J): 122.030303955  69.666427612   0.000000000   0.000000000   0.000000000
Power(W) : 122.030303955  69.666427612   0.000000000   0.000000000   0.000000000
```

vient, 2025-05-21:
SMT is disabled via kernel command line.

New error after patch
```
[ 4129.774738] CoreFreq: AMD_SMN_Read(0, a50104) TryLock
[ 4129.774739] ------------[ cut here ]------------
[ 4129.774739] Unable to find AMD Northbridge id for 0000:00:18.0
[ 4129.774741] WARNING: CPU: 0 PID: 21025 at arch/x86/include/asm/amd_nb.h:100 AMD_17h_DataFabric.constprop.0.isra.0+0x2bc/0x410 [corefreqk]
[ 4129.774745] Modules linked in: corefreqk(OE+) cpuid msr k10temp exasock(OE) kvm_amd kvm crct10dif_pclmul ghash_clmulni_intel sha256_ssse3 sha1_ssse3 nls_iso8859_1 aesni_intel crypto_simd btusb cryptd btrtl btbcm btintel rapl eeepc_wmi snd_hda_codec_hdmi bluetooth wmi_bmof snd_usb_audio snd_usbmidi_lib snd_hda_intel snd_rawmidi snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codecsnd_seq_device snd_hda_core joydev snd_hwdep ccp input_leds mc snd_pcm snd_timer snd soundcore ecdh_generic ecc exanic(OE) i2c_algo_bit ucsi_acpi typec_ucsi typec mac_hid sch_fq_codel binfmt_misc drm ramoops pstore_blk reed_solomon efi_pstore pstore_zone ip_tables x_tables autofs4 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid0 multipath linear raid1 uas hid_generic usbhid usb_storage hid mfd_aaeon asus_wmi sparse_keymap platform_profile crc32_pclmul i2c_piix4 thunderbolt ahci libahci igc xhci_pci xhci_pci_renesas wmi video [last unloaded: corefreqk]
[ 4129.774757] CPU: 0 PID: 21025 Comm: insmod Tainted: G        W  OE     5.15.0-139-generic #149~20.04.1-Ubuntu
[ 4129.774758] Hardware name: ASUS System Product Name/ROG CROSSHAIR X870E HERO, BIOS 1203 03/04/2025
[ 4129.774758] RIP: 0010:AMD_17h_DataFabric.constprop.0.isra.0+0x2bc/0x410 [corefreqk]
[ 4129.774762] Code: 5d 41 5e 41 5f 5d c3 cc cc cc cc 49 8b b6 20 01 00 00 48 85 f6 75 07 49 8b b6 d0 00 00 00 48 c7 c7 88 22 c5 c0 e8 51 d8 4a dd <0f> 0b 31 db e8 eb b9 85 dc 66 39 c3 0f 83 d9 fe ff ff 31 db eb 03
[ 4129.774762] RSP: 0018:ffffadc4426d3900 EFLAGS: 00010286
[ 4129.774762] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000027
[ 4129.774763] RDX: 0000000000000027 RSI: 0000000000000003 RDI: ffffa0083de20588
[ 4129.774763] RBP: ffffadc4426d3988 R08: ffffa0083de20580 R09: 0000000000000001
[ 4129.774763] R10: ffffffffa0657734 R11: 00000000a0657725 R12: 0000000000000001
[ 4129.774763] R13: 0000000000b50104 R14: ffffa001019b3000 R15: 0000000000000190
[ 4129.774764] FS:  00007ffff7a9f540(0000) GS:ffffa0083de00000(0000) knlGS:0000000000000000
[ 4129.774764] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 4129.774764] CR2: 00007fa70e202ba0 CR3: 000000010cf1a000 CR4: 0000000000750ef0
[ 4129.774765] PKRU: 55555554
[ 4129.774765] Call Trace:
[ 4129.774765]  <TASK>
[ 4129.774765]  ? show_regs.cold+0x1a/0x1f
[ 4129.774766]  ? AMD_17h_DataFabric.constprop.0.isra.0+0x2bc/0x410 [corefreqk]
[ 4129.774770]  ? __warn+0x8b/0xe0
[ 4129.774771]  ? AMD_17h_DataFabric.constprop.0.isra.0+0x2bc/0x410 [corefreqk]
[ 4129.774774]  ? report_bug+0xd5/0x110
[ 4129.774775]  ? handle_bug+0x39/0x90
[ 4129.774776]  ? exc_invalid_op+0x19/0x70
[ 4129.774776]  ? asm_exc_invalid_op+0x1b/0x20
[ 4129.774777]  ? AMD_17h_DataFabric.constprop.0.isra.0+0x2bc/0x410 [corefreqk]
[ 4129.774781]  ? pci_bus_find_capability+0x80/0xa0
[ 4129.774782]  ? AMD_UMC_Normalize_Channels+0x2c0/0x2c0 [corefreqk]
[ 4129.774786]  AMD_DataFabric_Raphael+0x5c/0x80 [corefreqk]
[ 4129.774790]  CoreFreqK_ProbePCI+0x5b/0xb0 [corefreqk]
[ 4129.774793]  CoreFreqK_Ignition_Level_Up.cold+0xd7/0x174 [corefreqk]
[ 4129.774797]  CoreFreqK_StartUp+0xee/0x152 [corefreqk]
[ 4129.774801]  ? CoreFreqK_StartUp+0xee/0x152 [corefreqk]
[ 4129.774805]  ? CoreFreqK_Alloc_Public_Level_Up+0x70/0x70 [corefreqk]
[ 4129.774809]  ? CoreFreqK_SetBoost+0xa0/0xa0 [corefreqk]
[ 4129.774813]  ? CoreFreqK_Make_Device_Level_Up+0x50/0x50 [corefreqk]
[ 4129.774816]  ? CoreFreqK_Create_Device_Level_Up+0x70/0x70 [corefreqk]
[ 4129.774820]  ? CoreFreqK_Register_NMI+0x200/0x200 [corefreqk]
[ 4129.774824]  ? CoreFreqK_Alloc_Private_Level_Up+0x50/0x50 [corefreqk]
[ 4129.774827]  ? CoreFreqK_Alloc_Processor_RO_Level_Up+0x60/0x60 [corefreqk]
[ 4129.774831]  ? CoreFreqK_Alloc_Processor_RW_Level_Up+0x80/0x80 [corefreqk]
[ 4129.774835]  ? CoreFreqK_ResetChip+0x50/0x50 [corefreqk]
[ 4129.774838]  ? BaseClock_AMD_Zen_PerCore+0x2c0/0x2c0 [corefreqk]
[ 4129.774842]  ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x50/0x50 [corefreqk]
[ 4129.774850]  ? SMBIOS_Entries+0x420/0x420 [corefreqk]
[ 4129.774854]  ? PerCore_Intel_Query+0x2f0/0x2f0 [corefreqk]
[ 4129.774858]  ? VendorFromCPUID.constprop.0+0x140/0x140 [corefreqk]
[ 4129.774861]  ? GLK_IMC+0x140/0x140 [corefreqk]
[ 4129.774865]  ? 0xffffffffc0c89000
[ 4129.774865]  CoreFreqK_Init+0xf/0x1000 [corefreqk]
[ 4129.774869]  do_one_initcall+0x46/0x1e0
[ 4129.774870]  ? __cond_resched+0x19/0x40
[ 4129.774870]  ? kmem_cache_alloc_trace+0x15a/0x420
[ 4129.774871]  do_init_module+0x52/0x230
[ 4129.774872]  load_module+0x12ae/0x1520
[ 4129.774874]  __do_sys_finit_module+0xbf/0x120
[ 4129.774875]  ? __do_sys_finit_module+0xbf/0x120
[ 4129.774876]  __x64_sys_finit_module+0x1a/0x20
[ 4129.774877]  x64_sys_call+0x1ac3/0x1fa0
[ 4129.774878]  do_syscall_64+0x54/0xb0
[ 4129.774878]  ? fput+0x13/0x20
[ 4129.774879]  ? ksys_mmap_pgoff+0x14b/0x2a0
[... 317 lines cut ...]
012 5200.10   219  1.3688   58  000000000000259550    3.960418701   3.960418701
013 5197.73   219  1.3688   58  000000000000254773    3.887527466   3.887527466
014 5200.05   219  1.3688   58  000000000000306520    4.677124023   4.677124023
[... cut ...] 9   3.884078979
016        OFF
017        OFF
018        OFF
019        OFF
020        OFF
021        OFF
022        OFF
023        OFF
024        OFF
025        OFF
026        OFF
027        OFF
028        OFF
029        OFF
030        OFF
031        OFF

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J): 123.989761353  73.016235352   0.000000000   0.000000000   0.000000000
Power(W) : 123.989761353  73.016235352   0.000000000   0.000000000   0.000000000
```

vient, 2025-05-21:
Oh right, did not notice that temperature appeared 🙂
So the fix worked

cyring, 2025-05-21:
Great but all these nasty messages in the kernel log... In the past did you have _CoreFreq_ running with no issue in the same kernel setup ?
Also don't you have another agent monitoring AMD hardware: `k10temp` or SMU monitor ?

I have to experiment the disabled SMT as you do. Can list commands you are employing ?

cyring, 2025-05-21:
And I suspect something I don't know from your platform because the UMC is not showing up; and Datafabric failed to probe at the usual PCI address in your first run.

Can you list `lspci -nn`

vient, 2025-05-21:
This is a new machine with ubuntu 20.04. I tried `k10temp` but did not understand what's supposed to be done next - `sensors` still shows nothing. Right now k10temp is not loaded, don't know about SMU monitor, guess I don't have it.

About SMT, it is turned off via BIOS after all. Full cmdline is `BOOT_IMAGE=/boot/vmlinuz-5.15.0-139-generic root=UUID=9e7d59c1-e394-432b-911d-eafda3fbfea9 ro isolcpus=1-15 ipv6.disable=1 processor.max_cstate=0 idle=poll mce=ignore_ce ipmi_si.force_kipmid=0 nmi_watchdog=0 nosoftlockup audit=0 nosmt=full tsc=reliable nohz=off mitigations=off acpi_enforce_resources=lax amd_iommu=off pcie_aspm.policy=performance`

```
# lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14d8]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:14d9]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:02.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:02.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:08.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 71)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e0]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e1]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e2]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e3]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e4]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e5]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e6]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e7]
01:00.0 Ethernet controller [0200]: Exablaze ExaNIC X10 [1ce4:0003]
02:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f4] (rev 01)
03:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
03:08.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
03:0c.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
03:0d.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f4] (rev 01)
06:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:07.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:08.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:0c.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:0d.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
08:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7927]
0a:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 06)
0b:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. Device [10ec:8126] (rev 01)
0d:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:43fd] (rev 01)
0e:00.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f6] (rev 01)
0f:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:43fd] (rev 01)
10:00.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f6] (rev 01)
11:00.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2421] (rev 01)
12:00.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2423] (rev 01)
12:01.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2423] (rev 01)
12:02.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2423] (rev 01)
12:03.0 PCI bridge [0604]: ASMedia Technology Inc. Device [1b21:2423] (rev 01)
77:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2426] (rev 01)
78:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2425] (rev 01)
79:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:13c0] (rev c1)
79:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:1640]
79:00.2 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Device [1022:1649]
79:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b6]
79:00.4 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b7]
7a:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b8]
```

cyring, 2025-05-21:
Thank you

```
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e0]
```
So `14e0` was found by _CoreFreq_ but no timings listed under first controller.
```

                              Zen UMC  [14E0]
Controller #0                                                   Undefined
```

----

There is a combination of parameters in your kernel command line I have to check on my setup
```
isolcpus=1-15
processor.max_cstate=0
idle=poll
mce=ignore_ce
ipmi_si.force_kipmid=0
nosoftlockup
nosmt
tsc=reliable
nohz=off
amd_iommu=off
pcie_aspm.policy=performance
```

----

Meanwhile to verify against my kernel standards, do you mind to boot the _CoreFreq_ ISO image from [Wiki/Live CD](https://github.com/cyring/CoreFreq/wiki/Live-CD)
It's an auto-build and autorun of the stable version you will flash on an USB key with a simple `dd`
If things ok, you can update to the latest commit using the scripts: `corefreq_online_update.sh` then `corefreq_build_run.sh`

cyring, 2025-05-21:
I have tried your kernel parameters within the Live CD boot loader
No issue found with my my `3950X`; no error found in the kernel log.

[image]

After updating to the latest version

[image]

[image]

[image]

cyring, 2025-05-21:
> [ 1048.213438] CoreFreq: AMD_SMN_Read(0, 59808) TryLock

You first log shows a failed attempt to read the thermal register `@ SMU(0x59808)`
https://github.com/cyring/CoreFreq/blob/6a99cfb4b49f034c822651e4b4fd2fe6ab41d0b1/x86_64/amd_reg.h#L1518

cyring, 2025-05-21:
> [ 4129.774738] CoreFreq: AMD_SMN_Read(0, a50104) TryLock

The second log (after patch) shows a failed attempt to probe the 11th `{0xa}` possible memory channel `@ SMU address 0x{0,1,2,3,4,5,6,7,8,9,a,b}50104 `
https://github.com/cyring/CoreFreq/blob/6a99cfb4b49f034c822651e4b4fd2fe6ab41d0b1/x86_64/amd_reg.h#L1745

cyring, 2025-05-21:
OK! I see.
Let's try to safely probe only two channels on Raphael.

Please edit function `AMD_DataFabric_Raphael` and change `MC_MAX_CHA` argument with the value `2`

https://github.com/cyring/CoreFreq/blob/6a99cfb4b49f034c822651e4b4fd2fe6ab41d0b1/x86_64/corefreqk.c#L7352

```c
static PCI_CALLBACK AMD_DataFabric_Raphael(struct pci_dev *pdev)
{
	return AMD_17h_DataFabric(	pdev,
					(const unsigned int[2][2]) {
						{ 0x0, 0x20},
						{0x10, 0x30}
					},
					0x44, 0x90,
					1, 2,
		(const unsigned int[]) {PCI_DEVFN(0x18, 0x0)} );
}
```

Next, rebuild/reload all and post the UMC `corefreq-cli -M`

vient, 2025-05-21:
\> You first log 
\> The second log
Sorry should've clarified that I took one error from many in each case.

Full dmesg log after rebuild
```
[ 8590.695347] CoreFreq: PCI_HSMP_Mailbox(1) Timeout
[ 8590.760403] CoreFreq(0:-1:-1): Processor [ BF_44] Architecture [Zen5/Granite Ridge] CPU [16/32]
[ 8590.760479] pci 0000:00:00.2: can't derive routing for PCI INT A
[ 8590.760480] pci 0000:00:00.2: PCI INT A: not connected
[ 8590.760520] ------------[ cut here ]------------
[ 8590.760521] Unable to find AMD Northbridge id for 0000:00:18.0
[ 8590.760525] WARNING: CPU: 0 PID: 43870 at arch/x86/include/asm/amd_nb.h:100 AMD_17h_DataFabric.constprop.0.isra.0+0x2d5/0x420 [corefreqk]
[ 8590.760537] Modules linked in: corefreqk(OE+) exasock(OE) kvm_amd kvm crct10dif_pclmul ghash_clmulni_intel sha256_ssse3 sha1_ssse3 nls_iso8859_1 aesni_intel crypto_simd btusb btrtl btbcm btintel cryptd rapl bluetooth snd_hda_codec_hdmi eeepc_wmi wmi_bmof snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec snd_hda_core ccp joydev input_leds snd_usb_audio snd_usbmidi_lib snd_hwdep snd_rawmidi snd_seq_device mc snd_pcm snd_timer snd ecdh_generic ecc soundcore exanic(OE) i2c_algo_bit ucsi_acpi typec_ucsi typec mac_hid sch_fq_codel binfmt_misc drm ramoops efi_pstore pstore_blk reed_solomon pstore_zone ip_tables x_tables autofs4 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid0 multipath linea
r raid1 uas hid_generic usbhid usb_storage hid mfd_aaeon asus_wmi sparse_keymap platform_profile crc32_pclmul i2c_piix4 thunderbolt igc ahci libahci xhci_pci xhci_pci_renesas wmi video
[ 8590.760560] CPU: 0 PID: 43870 Comm: insmod Tainted: G           OE     5.15.0-139-generic #149~20.04.1-Ubuntu
[ 8590.760562] Hardware name: ASUS System Product Name/ROG CROSSHAIR X870E HERO, BIOS 1203 03/04/2025
[ 8590.760563] RIP: 0010:AMD_17h_DataFabric.constprop.0.isra.0+0x2d5/0x420 [corefreqk]
[ 8590.760568] Code: 5d 41 5e 41 5f 5d c3 cc cc cc cc 49 8b b5 20 01 00 00 48 85 f6 75 07 49 8b b5 d0 00 00 00 48 c7 c7 88 22 ae c0 e8 38 d8 c1 f3 <0f> 0b 31 db e8 d2 b9 fc f2 66 39 c3 0f 83 dc fe ff ff 31 db eb 03
[ 8590.760569] RSP: 0018:ffffbf0dc3593948 EFLAGS: 00010286
[ 8590.760570] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000027
[ 8590.760571] RDX: 0000000000000027 RSI: 0000000000000003 RDI: ffff9b887de20588
[ 8590.760571] RBP: ffffbf0dc35939d0 R08: ffff9b887de20580 R09: 0000000000000001
[ 8590.760572] R10: ffffffffb6c52204 R11: 00000000b6c521f5 R12: 0000000000000001
[ 8590.760572] R13: ffff9b81419b3000 R14: 0000000000000190 R15: 0000000000050104
[ 8590.760573] FS:  00007ffff7a9f540(0000) GS:ffff9b887de00000(0000) knlGS:0000000000000000
[ 8590.760573] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 8590.760574] CR2: 00007faf8cc22000 CR3: 0000000107da6000 CR4: 0000000000750ef0
[ 8590.760574] PKRU: 55555554
[ 8590.760575] Call Trace:
[ 8590.760576]  <TASK>
[ 8590.760578]  ? show_regs.cold+0x1a/0x1f
[ 8590.760580]  ? AMD_17h_DataFabric.constprop.0.isra.0+0x2d5/0x420 [corefreqk]
[ 8590.760584]  ? __warn+0x8b/0xe0
[ 8590.760587]  ? AMD_17h_DataFabric.constprop.0.isra.0+0x2d5/0x420 [corefreqk]
[ 8590.760591]  ? report_bug+0xd5/0x110
[ 8590.760593]  ? handle_bug+0x39/0x90
[ 8590.760594]  ? exc_invalid_op+0x19/0x70
[ 8590.760595]  ? asm_exc_invalid_op+0x1b/0x20
[ 8590.760597]  ? AMD_17h_DataFabric.constprop.0.isra.0+0x2d5/0x420 [corefreqk]
[ 8590.760600]  ? AMD_17h_DataFabric.constprop.0.isra.0+0x2d5/0x420 [corefreqk]
[ 8590.760604]  ? AMD_UMC_Normalize_Channels+0x2c0/0x2c0 [corefreqk]
[ 8590.760608]  AMD_DataFabric_Raphael+0x63/0x80 [corefreqk]
[ 8590.760612]  CoreFreqK_ProbePCI+0x5b/0xb0 [corefreqk]
[ 8590.760616]  CoreFreqK_Ignition_Level_Up.cold+0xd7/0x174 [corefreqk]
[ 8590.760621]  CoreFreqK_StartUp+0xee/0x152 [corefreqk]
[ 8590.760625]  ? CoreFreqK_StartUp+0xee/0x152 [corefreqk]
[ 8590.760629]  ? CoreFreqK_Alloc_Public_Level_Up+0x70/0x70 [corefreqk]
[ 8590.760633]  ? CoreFreqK_SetBoost+0xa0/0xa0 [corefreqk]
[ 8590.760636]  ? CoreFreqK_Make_Device_Level_Up+0x50/0x50 [corefreqk]
[ 8590.760640]  ? CoreFreqK_Create_Device_Level_Up+0x70/0x70 [corefreqk]
[ 8590.760644]  ? CoreFreqK_Register_NMI+0x200/0x200 [corefreqk]
[ 8590.760648]  ? CoreFreqK_Alloc_Private_Level_Up+0x50/0x50 [corefreqk]
[ 8590.760651]  ? CoreFreqK_Alloc_Processor_RO_Level_Up+0x60/0x60 [corefreqk]
[ 8590.760655]  ? CoreFreqK_Alloc_Processor_RW_Level_Up+0x80/0x80 [corefreqk]
[ 8590.760659]  ? CoreFreqK_ResetChip+0x50/0x50 [corefreqk]
[ 8590.760662]  ? BaseClock_AMD_Zen_PerCore+0x2c0/0x2c0 [corefreqk]
[ 8590.760666]  ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x50/0x50 [corefreqk]
[ 8590.760670]  ? SMBIOS_Entries+0x420/0x420 [corefreqk]
[ 8590.760673]  ? PerCore_Intel_Query+0x2f0/0x2f0 [corefreqk]
[ 8590.760677]  ? VendorFromCPUID.constprop.0+0x140/0x140 [corefreqk]
[ 8590.760681]  ? GLK_IMC+0x140/0x140 [corefreqk]
[ 8590.760685]  ? 0xffffffffc0b19000
[ 8590.760685]  CoreFreqK_Init+0xf/0x1000 [corefreqk]
[ 8590.760690]  do_one_initcall+0x46/0x1e0
[ 8590.760691]  ? __cond_resched+0x19/0x40
[ 8590.760692]  ? kmem_cache_alloc_trace+0x15a/0x420
[ 8590.760694]  do_init_module+0x52/0x230
[ 8590.760696]  load_module+0x12ae/0x1520
[ 8590.760698]  __do_sys_finit_module+0xbf/0x120
[... 95 lines cut ...]
[ 8590.760861]  ? exc_page_fault+0x89/0x170
[ 8590.760861]  entry_SYSCALL_64_after_hwframe+0x6c/0xd6
[ 8590.760862] RIP: 0033:0x7ffff7be195d
[... cut ...]
[ 8590.760864] R13: 0000000000000000 R14: 0000555555580840 R15: 0000000000000000
[ 8590.760865]  </TASK>
[ 8590.760865] ---[ end trace 1c8d37777a9bfcd7 ]---
[ 8590.760865] CoreFreq: AMD_SMN_Read(0, 150104) TryLock
[ 8601.656721] CoreFreq: Unload
```
Not much changed in cli
```
# build/corefreq-cli -M
                              Zen UMC  [14E0]
Controller #0                                                   Undefined
```

> do you mind to boot the CoreFreq ISO image

I don't have physical access to the server, so not possible right now, sorry.

cyring, 2025-05-21:
It appears that `9950X` reports I received were coming from `X670` platform chipset
https://gist.github.com/cyring/1d43eef7dada2854f5fcdbe39ce0e5d8

Your board has a `X870` according to kernel log and thus this first try of _CoreFreq_ I can read about.

Differences may exist between the 2 platforms, especially if SMU firmware differs.

----

@madoverlord40 
@notpentadactyl 

Hello,

Can you please run the latest version of the `master` branch with your `9950X` or `9950X3D` for stability tests ?

Check for _CoreFreq_ errors or warnings in the kernel log.

Can you still use _CoreFreq_ as before and read hardware data: monitoring, sensors, UMC ?

cyring, 2025-05-22:
@nooberfsh Hello,
Can you test with your 9950X the latest version of _CoreFreq_ like [above](https://github.com/cyring/CoreFreq/issues/548#issuecomment-2899443264)
Thank you

cyring, 2025-05-22:
@vient Hello
```
[ 9612.824347] Unable to find AMD Northbridge id for 0000:00:18.0
[ 9612.824349] WARNING: CPU: 14 PID: 0 at arch/x86/include/asm/amd_nb.h:100 CCD_AMD_Family_19h_Zen4_Temp+0x2c4/0x400 [corefreqk]
```
This warning comes from this line of code within the kernel:
https://elixir.bootlin.com/linux/v6.11.6/source/arch/x86/include/asm/amd_nb.h#L98

Do you have built a kernel without this prerequisite `CONFIG_AMD_NB` ?

You may grep for it in `/proc/config*` or some config files at `/usr/src/`

vient, 2025-05-22:
It's the stock Ubuntu 20.04 kernel
```
# grep CONFIG_AMD_NB /boot/config-5.15.0-139-generic
CONFIG_AMD_NB=y
```

cyring, 2025-05-22:
> It's the stock Ubuntu 20.04 kernel (I know that using 4 year old kernel on latest platform is not ideal...)
> 
> ```
> # grep CONFIG_AMD_NB /boot/config-5.15.0-139-generic
> CONFIG_AMD_NB=y
> ```

Ok, let me know when you have a chance to test with a recent kernel post version 6 like my boot CD

Because so far I think that using NorthBridge API from old kernel with one of latest Ryzen may be an issue.

We found above a workaround with my patch because it restores the use of my own set of asm functions to access PCI.
But those are not doing great with Genoa EPYC multi-sockets.

Only a new kernel, driving  your 9950X with unmodified _CoreFreq_ source code, may tell us if it is the root cause or not.

vient, 2025-05-22:
Looks fine, I guess
```
# CoreFreq/build/corefreq-cli -M
                              Zen UMC  [14E0]
Controller #0                                               Single Channel
 Bus Rate  3200 MHz       Bus Speed 3200 MHz           DDR5 Speed 6400 MT/s

 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   32   39   39   39  102  141    8   16   32    8   32   96    9   25
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   30   24   22    8    1   10   12    1   11   13    0    0    0    0
      REFI RFC1 RFC2 RFCsb RCPB RPPB BGS:Alt  Ban  Page  CKE  CMD  GDM  ECC
  #0 12463  943  512  416   0    0    ON OFF  R0W0   0    0   1T    ON   0
      MRD:PDA   MOD:PDA  WRMPR STAG PDM RDDATA WRD  WRL  RDL  XS   XP CPDED
  #0   45  32    45  32    24    7 0:F:0   20   6   18   36  975   24    0

 DIMM Geometry for channel #0
      Slot Bank Rank     Rows   Columns    Memory Size (MB)
       #0    16    3     32768      1024          12288    F5-6400J3239G16G
       #1    32    3     65536      1024          49152    F5-6400J3239G16G
```
Thanks!

closed 2025-05-22 by vient

cyring, 2025-06-11:
> My machine is overclocked, has fixed frequency and voltages, so voltage does not change 🤷

If possible, how much do you read from _CoreFreq_ ? Can I see it please ? B/c I want to check if the Zen4 register I'm getting SoC VID from is correct or wrong.

cyring, 2025-06-11:
> [image]
> 
> [image]

Thanks , what I need is the view Voltages

vient, 2025-06-11:
[image]

cyring, 2025-06-11:
> ...

Sorry the part is like this one in the blue rectangle

[image]

vient, 2025-06-12:
😅

[image]

cyring, 2025-06-12:
> 😅
> 
> [image]

This is great !
Thank you very much