cyring/CoreFreq #553: Compilation failure on kernel 6.15.x

bubbleguuum, 2025-06-04:
Latest v2.0.6 does not compile on kernel 6.15.x because there's no `asm/amd/nb.h` and it is still `asm/amd_nb.h`:

```
~/Downloads/linux-6.15.1> find . -name amd_nb.h
./arch/x86/include/asm/amd_nb.h
```

-------

```
[    7s] # CC [M]  module/corefreqk.o
[    7s]   gcc -Wp,-MMD,module/.corefreqk.o.d -nostdinc -I/usr/src/linux-6.15.0-1/arch/x86/include -I/usr/src/linux-6.15.0-1-obj/x86_64/default/arch/x86/include/generated -I/usr/src/linux-6.15.0-1/include -I/usr/src/linux-6.15.0-1-obj/x86_64/default/include -I/usr/src/linux-6.15.0-1/arch/x86/include/uapi -I/usr/src/linux-6.15.0-1-obj/x86_64/default/arch/x86/include/generated/uapi -I/usr/src/linux-6.15.0-1/include/uapi -I/usr/src/linux-6.15.0-1-obj/x86_64/default/include/generated/uapi -include /usr/src/linux-6.15.0-1/include/linux/compiler-version.h -include /usr/src/linux-6.15.0-1/include/linux/kconfig.h -include /usr/src/linux-6.15.0-1/include/linux/compiler_types.h -D__KERNEL__ -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -march=x86-64 -mtune=generic -mno-red-zone -mcmodel=kernel -mstack-protector-guard-reg=gs -mstack-protector-guard-symbol=__ref_stack_chk_guard -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -fmin-function-alignment=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -fno-builtin-wcslen -Wall -Wextra -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-array-bounds -Wno-stringop-overflow -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wunused -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -Wno-unused-parameter -g -gdwarf-5 -I/home/abuild/rpmbuild/BUILD/CoreFreq-2.0.6-build/CoreFreq-2.0.6/x86_64 -D COREFREQ_MAJOR=2 -D COREFREQ_MINOR=0 -D COREFREQ_REV=6 -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=7125000000 -Wall -Wfatal-errors -D DELAY_TSC=1 -D MSR_CORE_PERF_UCC=MSR_IA32_APERF -D MSR_CORE_PERF_URC=MSR_IA32_MPERF -D DT_VIRTUAL_BOARD='{ "linux,dummy-virt", "riscv-virtio", "qemu,pseries", NULL }'  -DMODULE  -DKBUILD_BASENAME='"corefreqk"' -DKBUILD_MODNAME='"corefreqk"' -D__KBUILD_MODNAME=kmod_corefreqk -c -o module/corefreqk.o module/corefreqk.c  
[    7s] module/corefreqk.c:57:10: fatal error: asm/amd/nb.h: No such file or directory
[    7s]    57 | #include <asm/amd/nb.h>
[    7s]       |          ^~~~~~~~~~~~~~
[    7s] compilation terminated.
[    7s] make[4]: *** [/usr/src/linux-6.15.0-1/scripts/Makefile.build:204: module/corefreqk.o] Error 1
[    7s] make[3]: *** [/usr/src/linux-6.15.0-1/Makefile:2026: .] Error 2

```

cyring, 2025-06-04:
Hello,

Headers have been renamed from one RC to the latest one in the [next](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/?h=next-20250604) branch from kernel.org

Here are the renamed header files:
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/arch/x86/include/asm/amd?h=next-20250604

Can you shou me the sources you are using ?

cyring, 2025-06-05:
So I don't get if that commit https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?h=next-20250604&id=bcbb65559532891148d990527e9df6b8fc98e98d has been merged or not !

cyring, 2025-06-05:
> So I don't get if that commit https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?h=next-20250604&id=bcbb65559532891148d990527e9df6b8fc98e98d has been merged or not !

OK so the change is still part within tags after but not within 6.15
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tag/?h=next-20250526

cyring, 2025-06-05:
> Latest v2.0.6 does not compile on kernel 6.15.x because there's no `asm/amd/nb.h` and it is still `asm/amd_nb.h`:

Please find a fix into release [`v2.0.7`](https://github.com/cyring/CoreFreq/releases/tag/2.0.7)