cyring/CoreFreq #558: Incorrect temperature on 9950x

moetayuko, 2025-06-27:
```
❯ corefreq-cli -C 1
CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)
000   45.69   144  0.9000   34  000000000000017680    0.269775391   0.269775391
001   72.80   144  0.9000   34  000000000000014518    0.221527100   0.221527100
002   55.16   144  0.9000   34  000000000000009318    0.142181396   0.142181396
003   38.04   144  0.9000   34  000000000000012885    0.196609497   0.196609497
004   33.44   144  0.9000   34  000000000000005478    0.083587646   0.083587646
005  111.49   144  0.9000   34  000000000000022974    0.350555420   0.350555420
006   11.37   144  0.9000   34  000000000000002650    0.040435791   0.040435791
007    6.79   144  0.9000   34  000000000000002502    0.038177490   0.038177490
008    1.45   144  0.9000   35  000000000000000955    0.014572144   0.014572144
009    1.72   144  0.9000   35  000000000000000926    0.014129639   0.014129639
010    0.82   144  0.9000   35  000000000000006141    0.093704224   0.093704224
011    1.25   144  0.9000   35  000000000000005767    0.087997437   0.087997437
012    1.00   144  0.9000   35  000000000000001133    0.017288208   0.017288208
013    2.71   144  0.9000   35  000000000000000746    0.011383057   0.011383057
014    1.80   144  0.9000   35  000000000000001279    0.019515991   0.019515991
015    1.16   144  0.9000   35  000000000000000735    0.011215210   0.011215210
016   13.98   144  0.9000   34  000000000000000000    0.000000000   0.000000000
017   18.64   144  0.9000   34  000000000000000000    0.000000000   0.000000000
018    6.12   144  0.9000   34  000000000000000000    0.000000000   0.000000000
019    6.35   144  0.9000   34  000000000000000000    0.000000000   0.000000000
020    4.55   144  0.9000   34  000000000000000000    0.000000000   0.000000000
021  100.82   144  0.9000   34  000000000000000000    0.000000000   0.000000000
022   11.95   144  0.9000   34  000000000000000000    0.000000000   0.000000000
023    5.20   144  0.9000   34  000000000000000000    0.000000000   0.000000000
024    1.03   144  0.9000   35  000000000000000000    0.000000000   0.000000000
025    0.97   144  0.9000   35  000000000000000000    0.000000000   0.000000000
026   84.94   144  0.9000   35  000000000000000000    0.000000000   0.000000000
027   35.21   144  0.9000   35  000000000000000000    0.000000000   0.000000000
028    3.04   144  0.9000   35  000000000000000000    0.000000000   0.000000000
029    2.03   144  0.9000   35  000000000000000000    0.000000000   0.000000000
030    2.54   144  0.9000   35  000000000000000000    0.000000000   0.000000000
031    1.04   144  0.9000   35  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J):  40.838531494   1.612655640   0.000000000   0.000000000   0.000000000
Power(W) :  40.838531494   1.612655640   0.000000000   0.000000000   0.000000000
```
```
❯ sensors k10temp-pci-00c3      
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +52.1°C
```
The temperature reported by k10temp is correct.

Additionally, `corefreq-cli -C 1` sometimes fails to respond, and I need to press Ctrl+C and rerun it multiple times to obtain the output.

<details>

<summary>System Information</summary>

```
❯ corefreq-cli -s                          
Processor                                  [AMD Ryzen 9 9950X 16-Core Processor]
|- Architecture                                             [Zen5/Granite Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0b404032]
|- Signature                                                           [  BF_44]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [100.000]
|- Frequency            (MHz)                      Ratio                        
                 Min   3000.00                    <  30 >                       
                 Max   4299.99                    <  43 >                       
|- Factory                                                             [100.000]
                       4300                       [  43 ]                       
|- Performance                                                                  
[... 213 lines cut ...]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                         DRAM   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                     Platform   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
   |- Power Limit                                                PL2   [    0 W]
   |- Time Window                                                TW2   [   0 ns]
|- Package Power Tracking                                        PPT   [Missing]
|- Electrical Design Current                                     EDC   [Missing]
|- Thermal Design Current                                        TDC   [Missing]
|- Core Thermal Point                                                           
|- Package Thermal Point                                                        
   |- Thermal Monitor Trip                                     Limit   [  115 C]
   |- HTC Temperature Limit                                    Limit   [  127 C]
   |- HTC Temperature Hysteresis                           Threshold   [    2 C]
|- Units                                                                        
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
```

</details>

cyring, 2025-06-27:
Hi the other 9950X did not encounter this issue.

What values are you reading when Cores are fully stressed ?
Are those still different than k10temp ?

moetayuko, 2025-06-27:
> Hi the other 9950X did not encounter this issue.
> 
> What values are you reading when Cores are fully stressed ? Are those still different than k10temp ?

Yes, but they don't vary that much:
```
    ~ ▓▒░·································································································
❯ corefreq-cli -C 1
CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)
000 5426.59   192  1.2000   80  000000000000678095   10.346908569  10.346908569
001 5426.64   192  1.2000   80  000000000000685677   10.462600708  10.462600708
002 5426.64   192  1.2000   80  000000000000675926   10.313812256  10.313812256
003 5426.59   192  1.2000   80  000000000000695846   10.617767334  10.617767334
004 5426.64   192  1.2000   80  000000000000687470   10.489959717  10.489959717
005 5426.64   192  1.2000   80  000000000000695677   10.615188599  10.615188599
006 5426.64   190  1.1875   80  000000000000684540   10.445251465  10.445251465
007 5426.64   190  1.1875   80  000000000000693621   10.583816528  10.583816528
008 5376.40   190  1.1875   81  000000000000693943   10.588729858  10.588729858
009 5376.35   190  1.1875   81  000000000000680012   10.376159668  10.376159668
010 5376.30   190  1.1875   81  000000000000697079   10.636581421  10.636581421
011 5376.35   190  1.1875   81  000000000000685131   10.454269409  10.454269409
012 5376.30   190  1.1875   81  000000000000695940   10.619201660  10.619201660
013 5376.30   190  1.1875   81  000000000000691096   10.545288086  10.545288086
014 5376.35   190  1.1875   81  000000000000701210   10.699615479  10.699615479
015 5376.35   190  1.1875   81  000000000000679342   10.365936279  10.365936279
016 5426.64   190  1.1875   80  000000000000000000    0.000000000   0.000000000
017 5426.64   190  1.1875   80  000000000000000000    0.000000000   0.000000000
018 5426.65   190  1.1875   80  000000000000000000    0.000000000   0.000000000
019 5426.65   190  1.1875   80  000000000000000000    0.000000000   0.000000000
020 5426.65   190  1.1875   80  000000000000000000    0.000000000   0.000000000
021 5426.65   190  1.1875   80  000000000000000000    0.000000000   0.000000000
022 5426.65   190  1.1875   80  000000000000000000    0.000000000   0.000000000
023 5426.65   190  1.1875   80  000000000000000000    0.000000000   0.000000000
024 5376.40   190  1.1875   81  000000000000000000    0.000000000   0.000000000
025 5376.30   190  1.1875   81  000000000000000000    0.000000000   0.000000000
026 5376.30   190  1.1875   81  000000000000000000    0.000000000   0.000000000
027 5376.34   190  1.1875   81  000000000000000000    0.000000000   0.000000000
028 5376.40   190  1.1875   81  000000000000000000    0.000000000   0.000000000
029 5376.35   190  1.1875   81  000000000000000000    0.000000000   0.000000000
030 5376.30   190  1.1875   81  000000000000000000    0.000000000   0.000000000
031 5376.35   190  1.1875   81  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J): 220.729217529 167.909622192   0.000000000   0.000000000   0.000000000
Power(W) : 220.729217529 167.909622192   0.000000000   0.000000000   0.000000000

    ~ ▓▒░·································································································
❯ sensors k10temp-pci-00c3
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +83.1°C
```

> Additionally, `corefreq-cli -C 1` sometimes fails to respond, and I need to press Ctrl+C and rerun it multiple times to obtain the output.

This might be because I have another instance of corefreq running. `corefreq-cli -C 1` returns promptly once I close the other one.

moetayuko, 2025-06-27:
> > Meanwhile, `Power(W)` also seems strange, and the readings for CCD2 are missing
> 
> Are max Power `220 W` Package and `167 W` total Cores what you expect ? Can you report about the specifications values ?

Guessing so, but why is the per-core power shown as ~10W?
> 
> Where do you find CCD2 ? Don't know which register it is referring to ?

oops, maybe CPU 0-15 are physical cores while 16-31 are hyperthreading logical cores so their power is shown as zero? I mistakenly thought 16-31 were CCD2...

cyring, 2025-06-27:
> Guessing so, but why is the per-core power shown as ~10W?
>
>
> oops, maybe CPU 0-15 are physical cores while 16-31 are hyperthreading logical cores so their power is shown as zero? I mistakenly thought [...]

----

Indeed you won't find RAPL registers in a logical Core
In short, CoreFreq is doing the sum of all Physical Cores.
Package watt is coming from SMU and may the consumption of other IPs

----

Back to the temperature issue: I have no clue why idle value makes the gape between _CoreFreq_ and k10temp. Don't you have other source of software to show ? Even temperature reporting from tools for Windows ?

moetayuko, 2025-06-27:
> Back to the temperature issue: I have no clue why idle value makes the gape between _CoreFreq_ and k10temp. Don't you have other source of software to show ? Even temperature reporting from tools for Windows ?

hwinfo and aida64 idle temperatures are consistent with k10temp, so CoreFreq must be problematic.

cyring, 2025-06-27:
@vient
@madoverlord40
@huajian628

Hello,

Can you tell if latest version of  _CoreFreq_ reports **wrong temperature** with your 9950X ?
Especially low temperature during the idle states.

vient, 2025-06-27:
Started with idle, then run `stress 0-3` for 5s, `stress 8-15` for 5s, `stress 0-15` for 10s.

https://github.com/user-attachments/assets/1a07d821-730a-4b7d-b918-8131d3ab8b34

Kinda looks like k10temp has an exponential smoothing, with larger alpha when real temperature is bigger than current approximation. Overall looks pretty similar.

cyring, 2025-06-27:
> Overall looks pretty similar.

Thank you for your well documented answer

vient, 2025-06-27:
Just in case you really want to dig in, 6.11 has [some code](https://github.com/torvalds/linux/blob/v6.11/drivers/hwmon/k10temp.c#L499) about 1Ah models which was removed in 6.12. `ccd_offset` is missing so I guess my k10temp only reports Tctl while CoreFreq reports Tccd1 and Tccd2 (?)

vient, 2025-06-27:
Right, [README mentions](https://github.com/torvalds/linux/blob/2c183963fb5fdd849cda66ddf2d93d88d2296a75/Documentation/hwmon/k10temp.rst#description) that Tctl is some "synthetic" value used for fan control (guess that means that smoothing is performed by chip itself).

I also like the note in driver's code
> The information [about converting raw sensor values to temperature] is not confirmed from chip datasheets, but experiments suggest that it provides reasonable temperature values.

cyring, 2025-06-27:
@moetayuko but before I go into code investigation, can you provide the idle and stressed temperature values from a third party software. Preferably a Windows temperature software provided by your mainboard manufacturer ?

moetayuko, 2025-06-27:
> @moetayuko but before I go into code investigation, can you provide the idle and stressed temperature values from a third party software. Preferably a Windows temperature software provided by your mainboard manufacturer ?

idle temperature in windows is around 52℃, close to k10temp

stressed temperature is another story.
I use cinebench r23 in windows, and the temperature reaches the 95℃ threshold as set in bios, meanwhile package power  reaches 270w.
in linux, the `stress` command is used, the temperature reaches only 81-83℃ as shown in both k10temp and corefreq, and the package power given by corefreq is only 220w.
I can't tell if the stressed temperature reported by k10temp is wrong, or `stress` failed to provide enough stress as cinebench did.

vient, 2025-06-27:
Should be right, `stress` is extremely simple. You can try `stress-ng` instead, something like
```
docker run --rm ghcr.io/colinianking/stress-ng --tz --rapl -t 60 --vnni -1
```

cyring, 2025-06-27:
> idle temperature in windows is around 52℃

Thank you. 
Too me that's the main issue.

cyring, 2025-06-27:
@moetayuko 

Based on latest version, can you edit source code for your architecture `[AMD_Zen5_Eldora]` at this line:
https://github.com/cyring/CoreFreq/blob/deb3da7f0049f41794ba27854591f84bac6cc9e2/x86_64/corefreqk.h#L13381

you replace with that line:
```c
.Query = Query_AMD_F17h_PerCluster,
```

then you rebuild:
```sh
make clean; make
```

restart CoreFreq and please post the idle temperature

cyring, 2025-06-27:
> Don't know what k10temp is showing, Zen 5 support [was added](https://github.com/torvalds/linux/commit/2c183963fb5fdd849cda66ddf2d93d88d2296a75) in 6.16 (so not even released yet) while I'm running 6.11.

I have checked: _CoreFreq_ is already the same register address offset
https://github.com/cyring/CoreFreq/blob/deb3da7f0049f41794ba27854591f84bac6cc9e2/x86_64/amd_reg.h#L212

moetayuko, 2025-06-28:
> [@moetayuko](https://github.com/moetayuko)
> 
> Based on latest version, can you edit source code for your architecture `[AMD_Zen5_Eldora]` at this line:
> 
> [CoreFreq/x86_64/corefreqk.h](https://github.com/cyring/CoreFreq/blob/deb3da7f0049f41794ba27854591f84bac6cc9e2/x86_64/corefreqk.h#L13381)
> 
> Line 13381 in [deb3da7](/cyring/CoreFreq/commit/deb3da7f0049f41794ba27854591f84bac6cc9e2)
>  .Query = Query_AMD_F19h_61h_PerCluster, 
> 
> you replace with that line:
> 
> .Query = Query_AMD_F17h_PerCluster,
> 
> then you rebuild:
> 
> make clean; make
> 
> restart CoreFreq and please post the idle temperature

all temperatures become zero after this to 2.0.7

cyring, 2025-06-28:
> Just in case you really want to dig in, 6.11 has [some code](https://github.com/torvalds/linux/blob/v6.11/drivers/hwmon/k10temp.c#L499) about 1Ah models which was removed in 6.12. `ccd_offset` is missing so I guess my k10temp only reports Tctl while CoreFreq reports Tccd1 and Tccd2 (?)

Indeed in [`k10temp_read_temp()`](https://elixir.bootlin.com/linux/v6.15.3/source/drivers/hwmon/k10temp.c#L216) function `k10temp` can switch to `/* Tctl */` or `/* Tdie */` or `/* Tccd{1-12} */` based on its channel numbering

On his side, _CoreFreq_ sticks with its CCD temperature function.

So what would be the best to report temperature **per Core** ? Tctl or Tccd ?

moetayuko, 2025-06-28:
> Specific to 9950X, edit file at this line:
> 
> [CoreFreq/x86_64/corefreqk.h](https://github.com/cyring/CoreFreq/blob/deb3da7f0049f41794ba27854591f84bac6cc9e2/x86_64/corefreqk.h#L9288)
> 
> Line 9288 in [deb3da7](/cyring/CoreFreq/commit/deb3da7f0049f41794ba27854591f84bac6cc9e2)
>  .Param.Offset = {95, 0, 0}, 
> 
> and change offset parameter as bellow
> 
> .Param.Offset = {95, 0, -18},
> 
> Rebuild and restart _CoreFreq_ to test the temperature
> 

something wrong with this change
```
❯ corefreq-cli -C 1
*** buffer overflow detected ***: terminated
[1]    7045 IOT instruction (core dumped)  corefreq-cli -C 1
```

> By the way, I will also need the topology to check the CCD/CCX clusters. Please provide the output of `corefreq-cli -m`
> 

```
❯ corefreq-cli -m
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0   0  0   0  0      32  8        48 12      1024 16 i   65536 16w 
001:  0    2   0  0   1  0      32  8        48 12      1024 16 i   65536 16w 
002:  0    4   0  0   2  0      32  8        48 12      1024 16 i   65536 16w 
003:  0    6   0  0   3  0      32  8        48 12      1024 16 i   65536 16w 
004:  0    8   0  1   4  0      32  8        48 12      1024 16 i   65536 16w 
005:  0   10   0  1   5  0      32  8        48 12      1024 16 i   65536 16w 
006:  0   12   0  1   6  0      32  8        48 12      1024 16 i   65536 16w 
007:  0   14   0  1   7  0      32  8        48 12      1024 16 i   65536 16w 
008:  0   16   1  2   8  0      32  8        48 12      1024 16 i   65536 16w 
009:  0   18   1  2   9  0      32  8        48 12      1024 16 i   65536 16w 
010:  0   20   1  2  10  0      32  8        48 12      1024 16 i   65536 16w 
011:  0   22   1  2  11  0      32  8        48 12      1024 16 i   65536 16w 
012:  0   24   1  3  12  0      32  8        48 12      1024 16 i   65536 16w 
013:  0   26   1  3  13  0      32  8        48 12      1024 16 i   65536 16w 
014:  0   28   1  3  14  0      32  8        48 12      1024 16 i   65536 16w 
015:  0   30   1  3  15  0      32  8        48 12      1024 16 i   65536 16w 
016:  0    1   0  0   0  1      32  8        48 12      1024 16 i   65536 16w 
017:  0    3   0  0   1  1      32  8        48 12      1024 16 i   65536 16w 
018:  0    5   0  0   2  1      32  8        48 12      1024 16 i   65536 16w 
019:  0    7   0  0   3  1      32  8        48 12      1024 16 i   65536 16w 
020:  0    9   0  1   4  1      32  8        48 12      1024 16 i   65536 16w 
021:  0   11   0  1   5  1      32  8        48 12      1024 16 i   65536 16w 
022:  0   13   0  1   6  1      32  8        48 12      1024 16 i   65536 16w 
023:  0   15   0  1   7  1      32  8        48 12      1024 16 i   65536 16w 
024:  0   17   1  2   8  1      32  8        48 12      1024 16 i   65536 16w 
025:  0   19   1  2   9  1      32  8        48 12      1024 16 i   65536 16w 
026:  0   21   1  2  10  1      32  8        48 12      1024 16 i   65536 16w 
027:  0   23   1  2  11  1      32  8        48 12      1024 16 i   65536 16w 
028:  0   25   1  3  12  1      32  8        48 12      1024 16 i   65536 16w 
029:  0   27   1  3  13  1      32  8        48 12      1024 16 i   65536 16w 
030:  0   29   1  3  14  1      32  8        48 12      1024 16 i   65536 16w 
031:  0   31   1  3  15  1      32  8        48 12      1024 16 i   65536 16w
```

cyring, 2025-06-28:
> something wrong with this change

Can you stop completely CoreFreq
Delete its project directory and then clone it again from latest version.

First build and run straight to check if it is working back.

Next attempt to make the offset change `Param.Offset = {95, 0, -18}`

But make sure to fully recompile and reload everything 
```
make clean
make
rmmod corefreqk
insmod build/corefreqk.ko
```

moetayuko, 2025-06-28:
> Can you stop completely CoreFreq Delete its project directory and then clone it again from latest version.
>
> First build and run straight to check if it is working back.
>
> Next attempt to make the [...]

same. I clean built everything and restarted to ensure kernel module is reloaded.
below are the backtraces:
```
#0  __GI_abort () at abort.c:91
#1  0x00007fc11697d5c9 in __libc_message_impl (fmt=fmt@entry=0x7fc116b09216 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:134
#2  0x00007fc116a7ba09 in __GI___fortify_fail (msg=msg@entry=0x7fc116b091fd "buffer overflow detected") at fortify_fail.c:24
#3  0x00007fc116a7b374 in __GI___chk_fail () at chk_fail.c:28
#4  0x000055e5947a2e8d in memcpy (__dest=0x55e59820ce7d, __src=0x55e59820cea0, __len=89) at /usr/include/bits/string_fortified.h:29
#5  Sensors (iter=<optimized out>) at x86_64/corefreq-cli.c:7285
#6  0x000055e5947879ef in main (argc=3, argv=<optimized out>) at x86_64/corefreq-cli.c:22721
(gdb) bt full
#0  __GI_abort () at abort.c:91
        act = {__sigaction_handler = {sa_handler = 0x0, sa_sigaction = 0x0}, sa_mask = {__val = {18446744073709551615, 0 <repeats 15 times>}}, sa_flags = 0, sa_restorer = 0x0}
#1  0x00007fc11697d5c9 in __libc_message_impl (fmt=fmt@entry=0x7fc116b09216 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:134
        ap = {{gp_offset = 16, fp_offset = 0, overflow_arg_area = 0x7fffe4840c10, reg_save_area = 0x7fffe4840ba0}}
        fd = 2
        iov = {{iov_base = 0x7fc116b09216, iov_len = 4}, {iov_base = 0x7fc116b091fd, iov_len = 24}, {iov_base = 0x7fc116b0921c, iov_len = 17}, {iov_base = 0x7fffe4840b60, iov_len = 9537642091124368896}, {iov_base = 0x7fffe4840b70, iov_len = 9537642091124368896}, {
            iov_base = 0x7fffe4840a50, iov_len = 140467284565916}, {iov_base = 0x1f, iov_len = 152}}
        iovcnt = <optimized out>
        total = <optimized out>
        cp = <optimized out>
#2  0x00007fc116a7ba09 in __GI___fortify_fail (msg=msg@entry=0x7fc116b091fd "buffer overflow detected") at fortify_fail.c:24
No locals.
#3  0x00007fc116a7b374 in __GI___chk_fail () at chk_fail.c:28
No locals.
#4  0x000055e5947a2e8d in memcpy (__dest=0x55e59820ce7d, __src=0x55e59820cea0, __len=89) at /usr/include/bits/string_fortified.h:29
No locals.
#5  Sensors (iter=<optimized out>) at x86_64/corefreq-cli.c:7285
        idx = <optimized out>
        sdx = 77
        pw = <optimized out>
        cpu = <optimized out>
        rdx = <optimized out>
        ldx = 89
        out = 0x55e59820c350 "CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)\n000   45.16   152  0.9500  4294901808  ", '0' <repeats 14 times>, "2947    0.044967651   0.044967651\n001   99.80   152  0.9500  429490180"...
        row = 0x55e59820cea0 "\n", ' ' <repeats 13 times>, "Package[0]    Cores         Uncore        Memory        Platform\nEnergy(J):"
#6  0x000055e5947879ef in main (argc=3, argv=<optimized out>) at x86_64/corefreq-cli.c:22721
        theme = <optimized out>
        _dest = <optimized out>
        tty = <optimized out>
        tty = <optimized out>
        tty = <optimized out>
        tty = <optimized out>
        iter = <optimized out>
        iter = <optimized out>
        iter = <optimized out>
        iter = <optimized out>
        iter = <optimized out>
        iter = <optimized out>
        usrIdx = <optimized out>
        roSize = 290816
        rwSize = <optimized out>
        stat_st = {ro = {st_dev = 24, st_ino = 2, st_nlink = 1, st_mode = 33060, st_uid = 0, st_gid = 0, __pad0 = 0, st_rdev = 0, st_size = 290816, st_blksize = 4096, st_blocks = 568, st_atim = {tv_sec = 1751096343, tv_nsec = 55553474}, st_mtim = {tv_sec = 1751096343, 
              tv_nsec = 55991986}, st_ctim = {tv_sec = 1751096343, tv_nsec = 55991986}, __glibc_reserved = {0, 0, 0}}, rw = {st_dev = 24, st_ino = 3, st_nlink = 1, st_mode = 33206, st_uid = 0, st_gid = 0, __pad0 = 0, st_rdev = 0, st_size = 4096, st_blksize = 4096, st_blocks = 8, 
            st_atim = {tv_sec = 1751096343, tv_nsec = 55553474}, st_mtim = {tv_sec = 1751096343, tv_nsec = 55935175}, st_ctim = {tv_sec = 1751096343, tv_nsec = 55935175}, __glibc_reserved = {0, 0, 0}}}
        fd = {ro = 3, rw = 4}
        idx = 2
        program = 0x55e59820b2a0 "corefreq-cli"
        appName = 0x55e59820b2a0 "corefreq-cli"
        option = 67 'C'
        trailing = 0 '\000'
        reason = <optimized out>
```

cyring, 2025-06-28:
Edit file `x86_64/corefreqk.c` and go arround line 16544 where is function `CCD_AMD_Family_19h_Zen4_Temp` defined.

I want to avoid the `49C` offset by comment in the call of function `Core_AMD_Zen_Filter_Temp`

Please replace with:
```c
static void CCD_AMD_Family_19h_Zen4_Temp(CORE_RO *Core)
{
	TCCD_REGISTER TccdSensor = {.value = 0};

	Core_AMD_SMN_Read(	TccdSensor,
				(SMU_AMD_THM_TCTL_CCD_REGISTER_F19H_61H
				+ (Core->T.Cluster.CCD << 2)),
				PRIVATE(OF(Zen)).Device.DF );
/*
	Core_AMD_Zen_Filter_Temp( Core, TccdSensor.CurTmp,
					TccdSensor.CurTempRangeSel == 1 );
*/
	Core_AMD_Family_17h_ThermTrip(Core);
}
```

Next rebuild, reload and test temperature results

moetayuko, 2025-06-28:
> [@moetayuko](https://github.com/moetayuko)
> 
> Edit file `x86_64/corefreqk.c` and go arround line 16462 where is function `Core_AMD_Zen_Filter_Temp` defined.
> 
> I want to avoid the `49C` offset, please replace with the following:
> 
> static void Core_AMD_Zen_Filter_Temp( CORE_RO *Core,	unsigned int CurTmp,
> 							bool scaleRangeSel )
> {
> 	/* Register: SMU::THM::THM_TCON_CUR_TMP - Bit 19: CUR_TEMP_RANGE_SEL
> 		0 = Report on 0C to 225C scale range.
> 		1 = Report on -49C to 206C scale range.
> 	*/
> 	Core->PowerThermal.Sensor = CurTmp;
> /*
> 	if (scaleRangeSel == true) {
> 	    if (Core->PowerThermal.Sensor >= (49 << 3))
> 	    {
> 		Core->PowerThermal.Param.Offset[THERMAL_OFFSET_P1] = 49;
> 
> 		if (Core->PowerThermal.Sensor > (255 << 3)) {
> 			Core->PowerThermal.Events[eLOG] |= EVENT_THOLD2_LOG;
> 		}
> 	    } else {
> 		Core->PowerThermal.Param.Offset[THERMAL_OFFSET_P1] = 0;
> 
> 		Core->PowerThermal.Events[eLOG] |= EVENT_THOLD1_LOG;
> 	    }
> 	} else {
> 		Core->PowerThermal.Param.Offset[THERMAL_OFFSET_P1] = 0;
> 
> 		if (Core->PowerThermal.Sensor > (225 << 3)) {
> 			Core->PowerThermal.Events[eLOG] |= EVENT_THOLD2_LOG;
> 		}
> 	}*/
> }
> 
> Next rebuild, reload and test temperature results

```
❯ corefreq-cli -C 1
CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)
000   17.53   153  0.9563   80  000000000000001667    0.025436401   0.025436401
001   24.44   153  0.9563   80  000000000000002279    0.034774780   0.034774780
002    3.51   153  0.9563   80  000000000000001868    0.028503418   0.028503418
003    6.51   153  0.9563   80  000000000000018931    0.288864136   0.288864136
004   19.49   153  0.9563   80  000000000000002311    0.035263062   0.035263062
005    6.57   153  0.9563   80  000000000000001229    0.018753052   0.018753052
006    1.49   153  0.9563   80  000000000000001873    0.028579712   0.028579712
007    3.86   153  0.9563   80  000000000000002834    0.043243408   0.043243408
008    2.11   153  0.9563   81  000000000000001090    0.016632080   0.016632080
009    9.36   153  0.9563   81  000000000000002972    0.045349121   0.045349121
010    4.19   153  0.9563   81  000000000000001084    0.016540527   0.016540527
011    1.64   153  0.9563   81  000000000000001007    0.015365601   0.015365601
012    2.15   153  0.9563   81  000000000000001130    0.017242432   0.017242432
013    1.07   153  0.9563   81  000000000000001095    0.016708374   0.016708374
014    1.70   153  0.9563   81  000000000000001007    0.015365601   0.015365601
015    1.44   153  0.9563   81  000000000000001007    0.015365601   0.015365601
016    2.43   153  0.9563   80  000000000000000000    0.000000000   0.000000000
017    2.22   153  0.9563   80  000000000000000000    0.000000000   0.000000000
018    7.63   153  0.9563   80  000000000000000000    0.000000000   0.000000000
019   36.78   153  0.9563   80  000000000000000000    0.000000000   0.000000000
020    2.10   153  0.9563   80  000000000000000000    0.000000000   0.000000000
021    2.28   153  0.9563   80  000000000000000000    0.000000000   0.000000000
022    2.61   153  0.9563   80  000000000000000000    0.000000000   0.000000000
023    2.36   153  0.9563   80  000000000000000000    0.000000000   0.000000000
024    1.43   153  0.9563   81  000000000000000000    0.000000000   0.000000000
025   24.02   153  0.9563   81  000000000000000000    0.000000000   0.000000000
026    2.04   153  0.9563   81  000000000000000000    0.000000000   0.000000000
027    1.50   153  0.9563   81  000000000000000000    0.000000000   0.000000000
028    2.20   153  0.9563   81  000000000000000000    0.000000000   0.000000000
029    1.87   153  0.9563   81  000000000000000000    0.000000000   0.000000000
030    1.57   153  0.9563   81  000000000000000000    0.000000000   0.000000000
031    1.34   153  0.9563   81  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J):  22.547164917   0.661987305   0.000000000   0.000000000   0.000000000
Power(W) :  22.547164917   0.661987305   0.000000000   0.000000000   0.000000000

    ~ ▓▒░·············································································································································································································································································
❯ sensors k10temp-pci-00c3                                        
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +49.1°C
```

cyring, 2025-06-28:
@moetayuko 
Thank you for this attempt but obviously nothing better

You can rollback the modified file.

```sh
git checkout x86_64/corefreqk.c
```

----

Can you show me all the sensors while idling ?

```sh
sensors
```

I'm not sure if sensors has debug output, please add them.

moetayuko, 2025-06-29:
> Can you show me all the sensors while idling ?

```
❯ sensors                 
spd5118-i2c-10-53
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +36.2°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

r8169_0_d00:00-mdio-0
Adapter: MDIO adapter
temp1:        +49.0°C  (high = +110.0°C)

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      725.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3500 RPM)
edge:         +49.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +51.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +50.0°C  (crit = +105.0°C, hyst = -273.1°C)
                       (emerg = +110.0°C)
PPT:          42.00 W  (cap = 220.00 W)
pwm1:              0%
sclk:         137 MHz 
mclk:         875 MHz 

nvme-pci-1100
Adapter: PCI adapter
Composite:    +28.9°C  (low  =  -0.1°C, high = +74.8°C)
                       (crit = +79.8°C)

spd5118-i2c-10-51
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +35.8°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +54.8°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)
```