cyring/CoreFreq #560: Compatibility with Xeon Scalable 4th gen

a084ed22, 2025-07-11:
Is corefreq compatible with Sapphire Rapids CPUs? I'm getting the following error when attempting to insmod the kernel module:
```
[lug11 08:31] CoreFreq(18:42:-1): Processor [ 06_8F] Architecture [Sapphire Rapids] SMT [48/48]
[  +0,001388] Oops: general protection fault, maybe for address 0x0: 0000 [#1] PREEMPT SMP NOPTI
[  +0,000021] CPU: 18 UID: 0 PID: 1256901 Comm: insmod Tainted: P           OE      6.12.32+bpo-amd64 #1  Debian 6.12.32-1~bpo12+1
[  +0,000017] Tainted: [P]=PROPRIETARY_MODULE, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[  +0,000009] Hardware name: HPE ProLiant DL380 Gen11/ProLiant DL380 Gen11, BIOS 2.48 03/11/2025
[  +0,000009] RIP: 0010:PerCore_Skylake_Query+0x2d1/0x690 [corefreqk]
[  +0,000032] Code: 44 24 20 65 48 2b 04 25 28 00 00 00 0f 85 d1 03 00 00 48 83 c4 28 5b 41 5c 41 5d 41 5e 41 5f 5d c3 cc cc cc cc b9 b0 06 00 00 <0f> 32 4c 8b 2d be 26 03 00 48 c1 e2 20 89 c0 48 09 d0 41 f7 c5 00
[  +0,000020] RSP: 0018:ff2ad1aa8f8aba18 EFLAGS: 00010046
[  +0,000009] RAX: 0000000000000000 RBX: ff251421be21a000 RCX: 00000000000006b0
[  +0,000331] RDX: 0000000000000400 RSI: ff251428d62c0000 RDI: 0000000000000012
[  +0,000266] RBP: ff2ad1aa8f8aba68 R08: 0000000000002020 R09: 0000000000002020
[  +0,000265] R10: 0000000000002020 R11: 0000000000000020 R12: ff25147ed2432000
[  +0,000260] R13: ff251428d62c05c0 R14: ffffffffc3606c32 R15: ff25148461655938
[  +0,000256] FS:  00007f9b5dcce040(0000) GS:ff25148f7ef00000(0000) knlGS:0000000000000000
[  +0,000256] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0,000254] CR2: 00007f075235b170 CR3: 00000028bb2ee002 CR4: 0000000000f73ef0
[  +0,000254] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  +0,000251] DR3: 0000000000000000 DR6: 00000000fffe07f0 DR7: 0000000000000400
[  +0,000246] PKRU: 55555554
[  +0,000243] Call Trace:
[  +0,000240]  <TASK>
[  +0,000237]  Start_Skylake+0x6f/0x2d0 [corefreqk]
[  +0,000247]  ? __pfx_Start_Skylake+0x10/0x10 [corefreqk]
[  +0,000243]  generic_exec_single+0x6d/0x130
[  +0,000232]  smp_call_function_single+0xbf/0x140
[  +0,000231]  Controller_Start+0xef/0x120 [corefreqk]
[  +0,000237]  CoreFreqK_Ignition_Level_Up+0x433/0x7a0 [corefreqk]
[  +0,000234]  ? __pfx_CoreFreqK_Init+0x10/0x10 [corefreqk]
[  +0,000233]  CoreFreqK_StartUp+0xef/0x160 [corefreqk]
[  +0,000230]  ? __pfx_CoreFreqK_Alloc_Features_Level_Up+0x10/0x10 [corefreqk]
[  +0,000230]  ? __pfx_CoreFreqK_Query_Features_Level_Up+0x10/0x10 [corefreqk]
[  +0,000225]  ? __pfx_CoreFreqK_Alloc_Device_Level_Up+0x10/0x10 [corefreqk]
[  +0,000223]  ? __pfx_CoreFreqK_Make_Device_Level_Up+0x10/0x10 [corefreqk]
[  +0,000220]  ? __pfx_CoreFreqK_Create_Device_Level_Up+0x10/0x10 [corefreqk]
[  +0,000219]  ? __pfx_CoreFreqK_Alloc_Public_Level_Up+0x10/0x10 [corefreqk]
[  +0,000215]  ? __pfx_CoreFreqK_Alloc_Private_Level_Up+0x10/0x10 [corefreqk]
[  +0,000214]  ? __pfx_CoreFreqK_Alloc_Processor_RO_Level_Up+0x10/0x10 [corefreqk]
[  +0,000211]  ? __pfx_CoreFreqK_Alloc_Processor_RW_Level_Up+0x10/0x10 [corefreqk]
[  +0,000206]  ? __pfx_CoreFreqK_Scale_And_Compute_Level_Up+0x10/0x10 [corefreqk]
[  +0,000204]  ? __pfx_CoreFreqK_Alloc_Public_Cache_Level_Up+0x10/0x10 [corefreqk]
[  +0,000203]  ? __pfx_CoreFreqK_Alloc_Private_Cache_Level_Up+0x10/0x10 [corefreqk]
[  +0,000201]  ? __pfx_CoreFreqK_Alloc_Per_CPU_Level_Up+0x10/0x10 [corefreqk]
[  +0,000197]  ? __pfx_CoreFreqK_Ignition_Level_Up+0x10/0x10 [corefreqk]
[  +0,000195]  ? __pfx_CoreFreqK_User_Ops_Level_Up+0x10/0x10 [corefreqk]
[  +0,000198]  CoreFreqK_Init+0x12/0xff0 [corefreqk]
[  +0,000196]  do_one_initcall+0x58/0x320
[  +0,000191]  do_init_module+0x60/0x270
[  +0,000185]  init_module_from_file+0x89/0xe0
[  +0,000180]  idempotent_init_module+0x11d/0x310
[  +0,000175]  __x64_sys_finit_module+0x5e/0xb0
[  +0,000170]  do_syscall_64+0x82/0x190
[  +0,000170]  ? __count_memcg_events+0x53/0xf0
[  +0,000166]  ? count_memcg_events.constprop.0+0x1a/0x30
[  +0,000161]  ? handle_mm_fault+0xae/0x2d0
[  +0,000154]  ? do_user_addr_fault+0x17f/0x670
[  +0,000153]  ? exc_page_fault+0x76/0x190
Errore di segmentazione
[  +0,000149]  entry_SYSCALL_64_after_hwframe+0x76/0x7e
[  +0,000153] RIP: 0033:0x7f9b5d7207d9
[  +0,000149] Code: 08 89 e8 5b 5d c3 66 2e 0f 1f 84 00 00 00 00 00 90 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d f7 05 0d 00 f7 d8 64 89 01 48
[  +0,000317] RSP: 002b:00007ffcfe20ea08 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  +0,000160] RAX: ffffffffffffffda RBX: 0000564d6e5637e0 RCX: 00007f9b5d7207d9
[  +0,000160] RDX: 0000000000000000 RSI: 0000564d391ec4a0 RDI: 0000000000000003
[  +0,000158] RBP: 0000564d391ec4a0 R08: 0000000000000000 R09: 00007f9b5d7f92a0
[  +0,000159] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[  +0,000159] R13: 0000000000000000 R14: 0000564d6e5637b0 R15: 0000000000000000
[  +0,000158]  </TASK>
[  +0,000156] Modules linked in: corefreqk(OE+) nf_conntrack_netlink veth xt_conntrack xt_MASQUERADE bridge stp llc xt_set ip_set nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype nft_compat nf_tables nfnetlink xfrm_user xfrm_algo overlay tcp_diag inet_diag scsi_transport_iscsi binfmt_misc intel_rapl_msr intel_rapl_common intel_uncore_frequency intel_uncore_frequency_common intel_ifs skx_edac_common nfit libnvdimm nls_ascii nls_cp437 vfat fat x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_hdmi wireguard kvm_intel snd_hda_intel snd_intel_dspcfg libchacha20poly1305 snd_intel_sdw_acpi kvm chacha_x86_64 snd_hda_codec poly1305_x86_64 snd_hda_core curve25519_x86_64 libcurve25519_generic drm_ttm_helper libchacha snd_hwdep xhci_pci ip6_udp_tunnel ttm ehci_pci snd_pcm snd_timer ext4 dm_service_time ses snd irqbypass crct10dif_pclmul crc32_pclmul crc16 ghash_clmulni_intel mbcache enclosure sha512_ssse3 jbd2 udp_tunnel
[  +0,000053]  sha256_ssse3 video xhci_hcd ehci_hcd tg3 sha1_ssse3 aesni_intel soundcore usbcore gf128mul crypto_simd cryptd ipmi_ssif isst_if_mbox_pci isst_if_mmio usb_common isst_if_common rapl mpt3sas intel_cstate deflate raid_class hpwdt intel_uncore iaa_crypto libphy watchdog mei_me pmt_telemetry intel_sdsi pmt_class intel_vsec mei hpilo idxd scsi_transport_sas pcspkr evdev sg idxd_bus cfg80211 acpi_power_meter rfkill acpi_ipmi ipmi_si ipmi_devintf ipmi_msghandler button acpi_tad wmi dm_multipath coretemp efi_pstore loop configfs ip_tables x_tables autofs4 xfs efivarfs raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor xor async_tx raid6_pq libcrc32c crc32c_generic raid0 dm_mod raid1 md_mod sd_mod mgag200 i2c_algo_bit ahci drm_shmem_helper libahci drm_kms_helper libata drm crc32c_intel scsi_mod scsi_common
[  +0,003364] ---[ end trace 0000000000000000 ]---
[  +0,109458] RIP: 0010:PerCore_Skylake_Query+0x2d1/0x690 [corefreqk]
[  +0,000354] Code: 44 24 20 65 48 2b 04 25 28 00 00 00 0f 85 d1 03 00 00 48 83 c4 28 5b 41 5c 41 5d 41 5e 41 5f 5d c3 cc cc cc cc b9 b0 06 00 00 <0f> 32 4c 8b 2d be 26 03 00 48 c1 e2 20 89 c0 48 09 d0 41 f7 c5 00
[  +0,000535] RSP: 0018:ff2ad1aa8f8aba18 EFLAGS: 00010046
[  +0,000271] RAX: 0000000000000000 RBX: ff251421be21a000 RCX: 00000000000006b0
[  +0,000272] RDX: 0000000000000400 RSI: ff251428d62c0000 RDI: 0000000000000012
[  +0,000271] RBP: ff2ad1aa8f8aba68 R08: 0000000000002020 R09: 0000000000002020
[  +0,000273] R10: 0000000000002020 R11: 0000000000000020 R12: ff25147ed2432000
[  +0,000273] R13: ff251428d62c05c0 R14: ffffffffc3606c32 R15: ff25148461655938
[  +0,000273] FS:  00007f9b5dcce040(0000) GS:ff25148f7ef00000(0000) knlGS:0000000000000000
[  +0,000278] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0,000279] CR2: 00007f075235b170 CR3: 00000028bb2ee002 CR4: 0000000000f73ef0
[  +0,000281] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  +0,000283] DR3: 0000000000000000 DR6: 00000000fffe07f0 DR7: 0000000000000400
[  +0,000283] PKRU: 55555554
[  +0,000281] note: insmod[1256901] exited with irqs disabled
[  +0,000307] note: insmod[1256901] exited with preempt_count 1
```

cyring, 2025-07-11:
> `[36173.913644] RAX: 0000000000000000 RBX: ff38623652aac000 RCX: 0000000000000640`

This is register `MSR_PP1_POWER_LIMIT(0x00000640)`

----

Please edit function `PerCore_Skylake_Query`

https://github.com/cyring/CoreFreq/blob/638faa37cc8a2a6cc74468e25e5707b3ac3c33ac/x86_64/corefreqk.c#L13871

and comment the register as bellow:

```c
static void PerCore_Skylake_Query(void *arg)
{
	CORE_RO *Core = (CORE_RO *) arg;

	Intel_Platform_Info(Core->Bind);
	Intel_Turbo_Config(Core, Intel_Turbo_Cfg8C_PerCore, Assign_8C_Boost);

	SystemRegisters(Core);

	Intel_Mitigation_Mechanisms(Core);

	Intel_VirtualMachine(Core);

	Intel_Microcode(Core);

	Dump_CPUID(Core);

	Intel_DCU_Technology(Core);

	SpeedStep_Technology(Core);

	TurboBoost_Technology(	Core,
				Set_SandyBridge_Target,
				Get_SandyBridge_Target,
				Cmp_Skylake_Target,
				Core->Boost[BOOST(TDP)] > 0 ?
				Core->Boost[BOOST(TDP)]:Core->Boost[BOOST(1C)],
				Core->Boost[BOOST(MAX)] );

	Query_Intel_C1E(Core);

	if (Core->T.ThreadID == 0) {				/* Per Core */
		Intel_CStatesConfiguration(CSTATES_SKL, Core);
	} else {
		CSTATE_IO_MWAIT CState_IO_MWAIT = {.value = 0};
		RDMSR(CState_IO_MWAIT, MSR_PMG_IO_CAPTURE_BASE);
	/*	Store the C-State Base Address used by I/O-MWAIT	*/
		Core->Query.CStateBaseAddr = CState_IO_MWAIT.LVL2_BaseAddr;
	}
	BITSET_CC(BUS_LOCK, PUBLIC(RO(Proc))->CC6_Mask, Core->Bind);

	BITSET_CC(BUS_LOCK,	PUBLIC(RO(Proc))->PC6_Mask,
				PUBLIC(RO(Proc))->Service.Core);

	BITSET_CC(BUS_LOCK, PUBLIC(RO(Proc))->BTC_NOBR_Mask , Core->Bind);

	PowerThermal(Core);

	ThermalMonitor_Set(Core);
	CorePerfLimitReasons(Core);
	GraphicsPerfLimitReasons(Core);
	RingPerfLimitReasons(Core);

	Intel_Turbo_Activation_Ratio(Core);
	Intel_Turbo_TDP_Config(Core);

	if (Core->Bind == PUBLIC(RO(Proc))->Service.Core) {
		Intel_DomainPowerLimit( MSR_PKG_POWER_LIMIT,
					PKG_POWER_LIMIT_LOCK_MASK,
					PWR_DOMAIN(PKG) );

		Intel_DomainPowerLimit( MSR_PP0_POWER_LIMIT,
					PPn_POWER_LIMIT_LOCK_MASK,
					PWR_DOMAIN(CORES) );
/*
		Intel_DomainPowerLimit( MSR_PP1_POWER_LIMIT,
					PPn_POWER_LIMIT_LOCK_MASK,
					PWR_DOMAIN(UNCORE) );
*/
		Intel_DomainPowerLimit( MSR_PLATFORM_POWER_LIMIT,
					PKG_POWER_LIMIT_LOCK_MASK,
					PWR_DOMAIN(PLATFORM) );

		Intel_Watchdog(Core);
	}
}
```

Now rebuild, reload and test

Thank you

a084ed22, 2025-07-14:
I might need a couple of days to get access to the machine while it's in a state it's okay to crash the kernel for testing.

In the meantime, here's the output - admittedly, I'm not sure what I should be looking at :

```
lspci -nn | grep 8086
00:00.0 System peripheral [0880]: Intel Corporation Ice Lake Memory Map/VT-d [8086:09a2] (rev 20)
00:00.1 System peripheral [0880]: Intel Corporation Ice Lake Mesh 2 PCIe [8086:09a4] (rev 20)
00:00.2 System peripheral [0880]: Intel Corporation Ice Lake RAS [8086:09a3] (rev 20)
00:00.4 Generic system peripheral [0807]: Intel Corporation Device [8086:0b23]
00:0d.0 PCI bridge [0604]: Intel Corporation Device [8086:1bbd] (rev 11)
00:14.0 USB controller [0c03]: Intel Corporation Emmitsburg (C740 Family) USB 3.2 Gen 1 xHCI Controller [8086:1bcd] (rev 11)
00:14.2 RAM memory [0500]: Intel Corporation Device [8086:1bce] (rev 11)
00:14.4 Host bridge [0600]: Intel Corporation Device [8086:1bfe] (rev 11)
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:1be0] (rev 11)
00:17.0 SATA controller [0106]: Intel Corporation Device [8086:1ba2] (rev 11)
00:19.0 SATA controller [0106]: Intel Corporation Sapphire Rapids SATA AHCI Controller [8086:1bd2] (rev 11)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:1b81] (rev 11)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device [8086:1bca] (rev 11)
11:00.0 System peripheral [0880]: Intel Corporation Ice Lake Memory Map/VT-d [8086:09a2] (rev 20)
11:00.1 System peripheral [0880]: Intel Corporation Ice Lake Mesh 2 PCIe [8086:09a4] (rev 20)
11:00.2 System peripheral [0880]: Intel Corporation Ice Lake RAS [8086:09a3] (rev 20)
11:00.4 Generic system peripheral [0807]: Intel Corporation Device [8086:0b23]
11:01.0 PCI bridge [0604]: Intel Corporation Device [8086:352a] (rev 04)
11:05.0 PCI bridge [0604]: Intel Corporation Device [8086:352c] (rev 04)
25:00.0 System peripheral [0880]: Intel Corporation Ice Lake Memory Map/VT-d [8086:09a2] (rev 20)
25:00.1 System peripheral [0880]: Intel Corporation Ice Lake Mesh 2 PCIe [8086:09a4] (rev 20)
25:00.2 System peripheral [0880]: Intel Corporation Ice Lake RAS [8086:09a3] (rev 20)
25:00.4 Generic system peripheral [0807]: Intel Corporation Device [8086:0b23]
25:01.0 PCI bridge [0604]: Intel Corporation Device [8086:352a] (rev 04)
39:00.0 System peripheral [0880]: Intel Corporation Ice Lake Memory Map/VT-d [8086:09a2] (rev 20)
39:00.1 System peripheral [0880]: Intel Corporation Ice Lake Mesh 2 PCIe [8086:09a4] (rev 20)
39:00.2 System peripheral [0880]: Intel Corporation Ice Lake RAS [8086:09a3] (rev 20)
39:00.4 Generic system peripheral [0807]: Intel Corporation Device [8086:0b23]
39:01.0 PCI bridge [0604]: Intel Corporation Device [8086:352a] (rev 04)
39:05.0 PCI bridge [0604]: Intel Corporation Device [8086:352c] (rev 04)
4d:00.0 System peripheral [0880]: Intel Corporation Ice Lake Memory Map/VT-d [8086:09a2] (rev 20)
4d:00.1 System peripheral [0880]: Intel Corporation Ice Lake Mesh 2 PCIe [8086:09a4] (rev 20)
4d:00.2 System peripheral [0880]: Intel Corporation Ice Lake RAS [8086:09a3] (rev 20)
4d:00.4 Generic system peripheral [0807]: Intel Corporation Device [8086:0b23]
4d:01.0 PCI bridge [0604]: Intel Corporation Device [8086:352a] (rev 04)
4d:03.0 PCI bridge [0604]: Intel Corporation Device [8086:352b] (rev 04)
4d:05.0 PCI bridge [0604]: Intel Corporation Device [8086:352c] (rev 04)
4d:07.0 PCI bridge [0604]: Intel Corporation Device [8086:352d] (rev 04)
61:00.0 System peripheral [0880]: Intel Corporation Ice Lake Memory Map/VT-d [8086:09a2] (rev 20)
61:00.1 System peripheral [0880]: Intel Corporation Ice Lake Mesh 2 PCIe [8086:09a4] (rev 20)
61:00.2 System peripheral [0880]: Intel Corporation Ice Lake RAS [8086:09a3] (rev 20)
61:00.4 Generic system peripheral [0807]: Intel Corporation Device [8086:0b23]
61:01.0 PCI bridge [0604]: Intel Corporation Device [8086:352a] (rev 04)
61:03.0 PCI bridge [0604]: Intel Corporation Device [8086:352b] (rev 04)
61:05.0 PCI bridge [0604]: Intel Corporation Device [8086:352c] (rev 04)
61:07.0 PCI bridge [0604]: Intel Corporation Device [8086:352d] (rev 04)
75:00.0 System peripheral [0880]: Intel Corporation Ice Lake Memory Map/VT-d [8086:09a2] (rev 20)
75:00.1 System peripheral [0880]: Intel Corporation Ice Lake Mesh 2 PCIe [8086:09a4] (rev 20)
75:00.2 System peripheral [0880]: Intel Corporation Ice Lake RAS [8086:09a3] (rev 20)
75:00.4 Generic system peripheral [0807]: Intel Corporation Device [8086:0b23]
75:01.0 System peripheral [0880]: Intel Corporation Device [8086:0b25]
75:03.0 System peripheral [0880]: Intel Corporation Ice Lake MSM [8086:09a6]
75:03.1 System peripheral [0880]: Intel Corporation Ice Lake PMON MSM [8086:09a7]
7e:00.0 System peripheral [0880]: Intel Corporation Device [8086:3250]
7e:00.1 System peripheral [0880]: Intel Corporation Device [8086:3251]
7e:00.2 System peripheral [0880]: Intel Corporation Device [8086:3252]
7e:00.3 Host bridge [0600]: Intel Corporation Ice Lake IEH [8086:0998]
7e:00.5 System peripheral [0880]: Intel Corporation Device [8086:3255]
7e:01.0 System peripheral [0880]: Intel Corporation Device [8086:3240]
7e:01.1 System peripheral [0880]: Intel Corporation Device [8086:3241]
7e:01.2 System peripheral [0880]: Intel Corporation Device [8086:3242]
7e:02.0 System peripheral [0880]: Intel Corporation Device [8086:3240]
7e:02.1 System peripheral [0880]: Intel Corporation Device [8086:3241]
7e:02.2 System peripheral [0880]: Intel Corporation Device [8086:3242]
[... cut ...]
ff:1e.1 System peripheral [0880]: Intel Corporation Device [8086:3259] (rev 08)
ff:1e.2 System peripheral [0880]: Intel Corporation Device [8086:325a] (rev 08)
ff:1e.3 System peripheral [0880]: Intel Corporation Device [8086:325b] (rev 08)
ff:1e.4 System peripheral [0880]: Intel Corporation Device [8086:325c] (rev 08)
ff:1e.5 System peripheral [0880]: Intel Corporation Device [8086:325d] (rev 08)
ff:1e.6 System peripheral [0880]: Intel Corporation Device [8086:325e] (rev 08)
ff:1e.7 System peripheral [0880]: Intel Corporation Device [8086:325f] (rev 08)
```

cyring, 2025-07-14:
> I might need a couple of days to get access to the machine while it's in a state it's okay to crash the kernel for testing.

Whenever it's possible you should start with the entries [fix](https://github.com/cyring/CoreFreq/issues/560#issuecomment-3063244529) for `[Sapphire_Rapids]`

> In the meantime, here's the output - admittedly, I'm not sure what I should be looking at :

Thank you 
Among your list, some Device ID will be probe for the [IMC](https://github.com/cyring/CoreFreq/blob/edc23f20c71ba6136cdc48dd5e0745fb6f95994b/x86_64/corefreq-api.h#L1380) Memory Controller

a084ed22, 2025-07-16:
I can confirm that the fix allows loading the module cleanly and that the CPU is correctly identified:
```
CoreFreq(19:43:-1): Processor [ 06_8F] Architecture [Sapphire Rapids] SMT [48/48]
```

cyring, 2025-07-16:
> I can confirm that the fix allows loading the module cleanly and that the CPU is correctly identified:
> 
> ```
> CoreFreq(19:43:-1): Processor [ 06_8F] Architecture [Sapphire Rapids] SMT [48/48]
> ```

This is great.
Is it possible to see screenshots.

a084ed22, 2025-07-17:
Sure, do tell what specific parts you'd want me to capture.

[image]

cyring, 2025-07-17:
> Sure, do tell what specific parts you'd want me to capture

Thank you.

Footer shows no processor voltage `V[0.0]`. 

Can you select the View Voltage and show if the per CPU voltages are displayed or not ?

a084ed22, 2025-07-17:
[image]

It's shown per-cpu.

cyring, 2025-07-17:
> It's shown per-cpu.

Great.  
At least, I can provide an aggregation as a Package value.

cyring, 2025-07-17:
> Sure, do tell what specific parts you'd want me to capture.

Can you post the output of these options:
```sh
corefreq-cli -s -k -n -m -n -C 1 -n -z -n -M
```

a084ed22, 2025-07-17:
```
corefreq-cli -s -k -n -m -n -C 1 -n -z -n -M
Processor                                        [Intel(R) Xeon(R) Silver 4410Y]
|- Architecture                                                [Sapphire Rapids]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x2b000639]
|- Signature                                                           [  06_8F]
|- Stepping                                                            [      8]
|- Online CPU                                                          [ 48/ 48]
|- Base Clock                                                          [ 99.998]
|- Frequency            (MHz)                      Ratio
                 Min    799.99                    <   8 >
                 Max   1999.97                    <  20 >
|- Factory                                                             [100.000]
                       2000                       [  20 ]
|- Performance
                 TGT    799.99                    <   8 >
   |- HWP
                 Min    799.99                    <   8 >
                 Max   3899.94                    <  39 >
                 TGT      AUTO                    <   0 >
|- Turbo Boost                                                         [ UNLOCK]
                  1C   3899.94                    <  39 >
                  2C   3899.94                    <  39 >
                  3C   3399.94                    <  34 >
                  4C   2999.95                    <  30 >
                  5C   2799.95                    <  28 >
                  6C   2799.95                    <  28 >
                  7C   2799.95                    <  28 >
                  8C   2799.95                    <  28 >
                  9C    399.99                    <   4 >
                 10C    599.99                    <   6 >
                 11C    799.99                    <   8 >
                 12C    999.98                    <  10 >
                 13C   1199.98                    <  12 >
                 14C   1199.98                    <  12 >
                 15C   1199.98                    <  12 >
                 16C   1199.98                    <  12 >
|- Uncore                                                              [ UNLOCK]
                 Min    799.99                    <   8 >
                 Max   1999.97                    <  20 >
|- TDP                                                           Level <  0:3  >
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   1999.97                    [  20 ]
               Turbo      AUTO                    <   0 >

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AMX-BF16     [Y]     AMX-TILE [Y]     AMX-INT8 [Y]    AMX-FP16 [N]
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y]
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y]
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N]
|- AVX512-BF16  [Y] AVX-VNNI-VEX [Y] AVX-VNN-INT8 [N] AVX-NE-CONV [N]
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [Y]   MOVDIR64B [Y]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- ENQCMD       [Y]         GFNI [Y]        OSPKE [Y]     WAITPKG [Y]
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y]
|- SERIALIZE    [Y]      SYSCALL [Y]        RDPID [Y]         SGX [Y]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
[... 398 lines cut ...]
   |- CPU #26    500.00 (  5)   799.99 (  8)  1999.99 ( 20)  3899.98 ( 39)
   |- CPU #27    500.01 (  5)   800.01 (  8)  2000.02 ( 20)  3900.05 ( 39)
   |- CPU #28    500.01 (  5)   800.01 (  8)  2000.02 ( 20)  3900.05 ( 39)
   |- CPU #29    500.01 (  5)   800.01 (  8)  2000.03 ( 20)  3900.05 ( 39)
   |- CPU #30    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3900.00 ( 39)
   |- CPU #31    500.00 (  5)   800.00 (  8)  1999.99 ( 20)  3899.99 ( 39)
   |- CPU #32    500.00 (  5)   800.00 (  8)  2000.01 ( 20)  3900.01 ( 39)
   |- CPU #33    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3899.99 ( 39)
   |- CPU #34    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3899.99 ( 39)
   |- CPU #35    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3900.00 ( 39)
   |- CPU #36    500.00 (  5)   799.99 (  8)  1999.98 ( 20)  3899.96 ( 39)
   |- CPU #37    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3900.00 ( 39)
   |- CPU #38    500.00 (  5)   799.99 (  8)  1999.98 ( 20)  3899.96 ( 39)
   |- CPU #39    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3900.00 ( 39)
   |- CPU #40    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3899.99 ( 39)
[... cut ...]
   |- CPU #42    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3900.00 ( 39)
   |- CPU #43    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3899.99 ( 39)
   |- CPU #44    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3900.00 ( 39)
   |- CPU #45    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3899.99 ( 39)
   |- CPU #46    500.00 (  5)   800.00 (  8)  1999.99 ( 20)  3899.99 ( 39)
   |- CPU #47    500.00 (  5)   800.00 (  8)  2000.00 ( 20)  3900.00 ( 39)

                            GenuineIntel  [   0]
```

cyring, 2025-07-17:
Thank you.

For my todo list, I see things I need to  complete
1. Memory Controller
2. Total power consumed by Cores
3. `VT-d`
4. `TCO`

----

From the UI, in "Settings" window, you can toggle on the "Experimental" mode.
Next, you can enable ODCM then a BIAS Hint.
Each percent you choose will cap accordingly the max frequency when system is stressed.  
Just disable "ODCM" to restore processor to its nominal frequency.

If ODCM is well executed on your architecture then I will mark as supported into the kernel module source code aka non-experimental.

cyring, 2025-07-17:
> I don't believe it is working as intended, I see no difference in peak frequency whether ODCM is enabled or not:

Thank you for trying.
At most I can mark for ODCM a successful MSR register access from SPR architecture.

But now remembering that HWP/CPPC  supercedes ODCM.  
To get it to work we have to prevent HWP aka SpeedShift to be enabled by UEFI/BIOS.

Whilst HWP is running then capping is this tome acheive via: 
`Energy Policy    HWP EPP`

a084ed22, 2025-07-17:
I don't seem to be able to apply any change to the Energy Policy HWP EPP. No matter the value chosen, it doesn't get set.

a084ed22, 2025-07-17:
I believe I've implemented the required changes:

```
# cat /proc/cmdline
... nmi_watchdog=0 modprobe.blacklist=i5500_temp,coretemp,intel_idle,intel_cstate,intel_uncore,acpi_cpufreq,pcc_cpufreq,intel_powerclamp,i7core_edac,intel_rapl_msr,intel_rapl_common,intel_pmc_bxt idle=halt intel_pstate=disable intel_idle.max_cstate=0 tsc=unstable nowatchdog
```

However, it seems the CPU-FREQ driver is still intel_pstate, and any change to the energy policy fails to apply.

cyring, 2025-07-17:
> I believe I've implemented the required changes:
>
> ```
> # cat /proc/cmdline
> ... nmi_watchdog=0 [...]

`intel_pstate=disable` should do the trick.

In the past we fixed something in issue #402 

```
initcall_blacklist=acpi_cpufreq_init
```

Perhaps also this one but it may prevent _CoreFreq_ to register
```
cpufreq.off=0
```

a084ed22, 2025-07-26:
Sorry for the long delay. I've tried both suggestions (initcall_blacklist=acpi_cpufreq_init and cpufreq.off=0) but I'm still getting interference in that corefreqk fails to register itself as the frequency driver

cyring, 2025-07-26:
> Sorry for the long delay. I've tried both suggestions (initcall_blacklist=acpi_cpufreq_init and cpufreq.off=0) but I'm still getting interference in that corefreqk fails to register itself as the frequency driver

Could be this Xeon comes with its own module which takes registration priority over any other module.
I'm not aware of your server software stack but I would unload any module one by one until the room is free for use. Especially those listed with `intel*` , `acpi*` , `pstate*` from command `lsmod`

Also observe that `corefreqk.ko` module arguments are case sensitive. Have a look in the kernel log if it was misspelled.

cyring, 2025-08-01:
Hello,
Fix is now available in release [`2.0.8`](https://github.com/cyring/CoreFreq/releases/tag/2.0.8)
If you don't observe a regression in this version; feel free to close your issue.
Regards

cyring, 2025-08-08:
I believe fix is now working with your architecture?

cyring, 2025-08-17:
> In the meantime, here's the output - admittedly, I'm not sure what I should be looking at :

[Datasheet volume 2](https://edc.intel.com/content/www/us/en/design/products-and-solutions/processors-and-chipsets/eagle-stream/4th-gen-xeon-processor-scalable-family-code-name-sapphire-rapids-data/integrated-memory-controller-imc-registers/)

> Integrated Memory Controller (IMC) Registers
> The 4th Gen Intel® Xeon® Processor Scalable Family, codename Sapphire Rapids > implements four Integrated Memory Controllers (IMC). Each IMC is capable of > controlling two DDR5 memory channels, and two DIMMs per channel.
>
> The IMC Host Configuration Space Registers are implemented in Bus: B(30), Device: 12-13-14-15, Function: 0
>
> Device 12 applies to IMC 0
> Device 13 applies to IMC 1
> Device 14 applies to IMC 2
> Device 15 applies to IMC 3

closed 2025-08-17 by cyring