cyring/CoreFreq #561: AMD RYZEN AI MAX+ 395

cyring, 2025-07-11:
This is the work in progress of the Zen5/Strix Halo implementation in _CoreFreq_

Please refer [here](https://github.com/deseven/strixhalo-homelab/issues/1) for the issue history

cyring, 2025-07-13:
@deseven Hello,

Commit aa140f5c1ca124a5741be649391398ff0830db3a is bringing a code specialization for the STX families. Any future registers should be added easier.

* You should no more read garbage values in `TDP PPT EDC TDC` but **zero** (aka **unknown** value)

* `TjMax` should be adjusted to specs

* The previous computed Turbo Boost frequency ratios should match the `5.1 GHz`
```
   |- Boost                                                            [   LOCK]
                 XFR   6400.16                    [  64 ]
                 CPB   6300.16                    [  63 ]
```

* The CPU voltage is still the P-state voltage and the SoC voltage is now zero (unknown).

Feel free to post an update of the `corefreq-cli -s` to check if non regression happened

----

Please let me know if you have a chance to take a photo of [Memtest86+](https://github.com/memtest86plus/memtest86plus). Or any other software which provide the memory controller timings and specs.

You can also try the easy command `sudo dmidecode -t memory` which will output parts of data.  
Be aware however this command probe the DIMM and not necessary the MC.

----

I would also like to see the Collaborative Processor Performance Control, if any or not.
```sh
corefreq-cli -z
```

deseven, 2025-07-13:
Yes, TDP, PPT, EDC and TDC values are marked as `missing` now.

`corefreq-cli -z`:
```
|- Collaborative Processor Performance Control                  CPPC       < ON>
   |- Capabilities     Lowest      Efficient     Guaranteed        Highest
   |- CPU #0     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.11 ( 47)
   |- CPU #1     400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4300.15 ( 43)
   |- CPU #2     400.01 (  4)  1300.02 ( 13)  1900.03 ( 19)  4600.07 ( 46)
   |- CPU #3     400.01 (  4)  1300.02 ( 13)  1900.03 ( 19)  4100.06 ( 41)
   |- CPU #4     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4200.10 ( 42)
   |- CPU #5     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4400.10 ( 44)
   |- CPU #6     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.11 ( 47)
   |- CPU #7     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4500.11 ( 45)
   |- CPU #8     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3800.09 ( 38)
   |- CPU #9     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3300.08 ( 33)
   |- CPU #10    400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4000.14 ( 40)
   |- CPU #11    400.01 (  4)  1300.02 ( 13)  1900.04 ( 19)  3700.07 ( 37)
   |- CPU #12    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3500.08 ( 35)
   |- CPU #13    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3400.09 ( 34)
   |- CPU #14    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3600.09 ( 36)
   |- CPU #15    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3900.10 ( 39)
   |- CPU #16    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.12 ( 47)
   |- CPU #17    400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4300.15 ( 43)
   |- CPU #18    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4600.11 ( 46)
   |- CPU #19    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4100.10 ( 41)
   |- CPU #20    400.01 (  4)  1300.02 ( 13)  1900.03 ( 19)  4200.06 ( 42)
   |- CPU #21    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4400.11 ( 44)
   |- CPU #22    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.12 ( 47)
   |- CPU #23    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4500.11 ( 45)
   |- CPU #24    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3800.09 ( 38)
   |- CPU #25    400.01 (  4)  1300.03 ( 13)  1900.04 ( 19)  3300.08 ( 33)
   |- CPU #26    400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4000.14 ( 40)
   |- CPU #27    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3700.09 ( 37)
   |- CPU #28    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3500.08 ( 35)
   |- CPU #29    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3400.09 ( 34)
   |- CPU #30    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3600.09 ( 36)
   |- CPU #31    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3900.10 ( 39)
```
CPPC was originally off and I turned it on in Perf. Capabilities.

But what's interesting is that I can't disable it now, there is no "Disable" option (I also tried pressing left-right arrows on enable):

[image]

Can it only be changed once per boot maybe?

---

Can't boot into memtest right now, I'll try doing it next week.

For now here's the dmidecode output:
```
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.7.0 present.
# SMBIOS implementations newer than version 3.5.0 are not
# fully supported by this version of dmidecode.

Handle 0x0011, DMI type 16, 23 bytes
Physical Memory Array
	Location: System Board Or Motherboard
	Use: System Memory
	Error Correction Type: None
	Maximum Capacity: 64 GB
	Error Information Handle: 0x0010
	Number Of Devices: 8

Handle 0x0014, DMI type 17, 100 bytes
Memory Device
[... 254 lines cut ...]
	Type: LPDDR5
	Type Detail: Synchronous Unbuffered (Unregistered)
	Speed: 8532 MT/s
	Manufacturer: Micron Technology
	Serial Number: 000000004D543632
	Asset Tag: Not Specified
	Part Number: MT62F4G32D8DV-023 WT
	Rank: 2
	Configured Memory Speed: 8000 MT/s
	Minimum Voltage: 0.5 V
	Maximum Voltage: 0.5 V
	Configured Voltage: 0.5 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version: Unknown
	Module Manufacturer ID: Bank 1, Hex 0x2C
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 16 GB
	Cache Size: None
	Logical Size: None
```
Note that my system is configured to have 64GB of RAM and 64GB of VRAM in the bios.

deseven, 2025-07-13:
^ added `corefreqcli -s` output as well

cyring, 2025-07-13:
> `corefreq-cli -z`:
> 
> ```
> |- Collaborative Processor Performance Control                  CPPC       < ON>
>    |- Capabilities     Lowest      Efficient     Guaranteed        Highest
>    |- CPU #0     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.11 ( 47)
>    |- CPU #1     400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4300.15 ( 43)
>    |- CPU #2     400.01 (  4)  1300.02 ( 13)  1900.03 ( 19)  4600.07 ( 46)
>    |- CPU #3     400.01 (  4)  1300.02 ( 13)  1900.03 ( 19)  4100.06 ( 41)
>    |- CPU #4     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4200.10 ( 42)
>    |- CPU #5     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4400.10 ( 44)
>    |- CPU #6     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.11 ( 47)
>    |- CPU #7     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4500.11 ( 45)
>    |- CPU #8     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3800.09 ( 38)
>    |- CPU #9     400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3300.08 ( 33)
>    |- CPU #10    400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4000.14 ( 40)
>    |- CPU #11    400.01 (  4)  1300.02 ( 13)  1900.04 ( 19)  3700.07 ( 37)
>    |- CPU #12    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3500.08 ( 35)
>    |- CPU #13    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3400.09 ( 34)
>    |- CPU #14    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3600.09 ( 36)
>    |- CPU #15    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3900.10 ( 39)
>    |- CPU #16    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.12 ( 47)
>    |- CPU #17    400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4300.15 ( 43)
>    |- CPU #18    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4600.11 ( 46)
>    |- CPU #19    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4100.10 ( 41)
>    |- CPU #20    400.01 (  4)  1300.02 ( 13)  1900.03 ( 19)  4200.06 ( 42)
>    |- CPU #21    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4400.11 ( 44)
>    |- CPU #22    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4700.12 ( 47)
>    |- CPU #23    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  4500.11 ( 45)
>    |- CPU #24    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3800.09 ( 38)
>    |- CPU #25    400.01 (  4)  1300.03 ( 13)  1900.04 ( 19)  3300.08 ( 33)
>    |- CPU #26    400.01 (  4)  1300.05 ( 13)  1900.07 ( 19)  4000.14 ( 40)
>    |- CPU #27    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3700.09 ( 37)
>    |- CPU #28    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3500.08 ( 35)
>    |- CPU #29    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3400.09 ( 34)
>    |- CPU #30    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3600.09 ( 36)
>    |- CPU #31    400.01 (  4)  1300.03 ( 13)  1900.05 ( 19)  3900.10 ( 39)
> ```
> 
> CPPC was originally off and I turned it on in Perf. Capabilities.
> 
> But what's interesting is that I can't disable it now, there is no "Disable" option (I also tried pressing left-right arrows on enable):
> 
> <img alt="Image" width="436" height="198" src="https://private-user-images.githubusercontent.com/1761803/465739137-98ff5746-ee8e-4a52-b920-8b6955c0f745.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI0MTQ4OTcsIm5iZiI6MTc1MjQxNDU5NywicGF0aCI6Ii8xNzYxODAzLzQ2NTczOTEzNy05OGZmNTc0Ni1lZThlLTRhNTItYjkyMC04YjY5NTVjMGY3NDUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDcxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MTNUMTM0OTU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yjg4YTMxNzUzNDUwMGM4NmJmNzIyOTcyNDJiZmFkNmU4ODZiZGMxYTZhMzRjOWRmY2U0NjliZDY0NDMxZGEzZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.3uQtb4P7ypJXafwHxSny60IhNYtUXS-lO6e1yAk7vUo">
> Can it only be changed once per boot maybe?

CPPC has two implementations:
1. Hardware via MSR registers
2. Firmware via manufacturer protocol 

Hardware which is apparently yours can only be disabled with a processor reset.
If you find it all the time enabled it means that an agent is doing so. It can be one of those, you will have to disable.
* UEFI option
* Kernel module
For example `amd_pstate=disable`

See my [3950X](https://gist.github.com/cyring/3d038a537cfae0383602823a94136ab6) output for the CPPC Firmware

deseven, 2025-07-13:
I see, not a problem then.

It's disabled for me on boot and I disabled amd_pstate as well, everything works under CoreFreq :)

[image]

deseven, 2025-07-14:
Booted into memtest86+, but I'm unsure where to get any information about memory apart from what is already displayed on the main screen. The only available menu that could be opened by pressing F1 doesn't contain anything on that matter. I also tried searching their docs with some keywords like "latency", "timings", "spec", "info" but there is nothing relevant.

[image]

What am I missing?

deseven, 2025-07-15:
```bash
# ./zencli smu 0x50200
[0x00050200] READ(smu) = 0x7a0903e8 (2047411176)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0111 1010 0000 1001 0000 0011 1110 1000
```

cyring, 2025-07-15:
> `10 1000`

Strangely if I just consider the last 6 bits (like it was with the DDR4 specs), it converts to decimal `40` multiplied by `100` BCLK again multiplied by two b/c DDR then we are getting `8000`

deseven, 2025-07-15:
I could switch to other speed in the bios (later today or tomorrow) and get you these values again if you'd like to confirm.

deseven, 2025-07-15:
Did all of them just in case.

<details>
<summary>5500MT/s</summary>

```bash
# for i in {0..7}; do ./zencli smu 0x${i}50200; done
[0x00050200] READ(smu) = 0x4a0902af (1242104495)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0010 1010 1111
[0x00150200] READ(smu) = 0x4a0902af (1242104495)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0010 1010 1111
[0x00250200] READ(smu) = 0x4a0902af (1242104495)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0010 1010 1111
[0x00350200] READ(smu) = 0x4a0902af (1242104495)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0010 1010 1111
[0x00450200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00550200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00650200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00750200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
```

</details>

<details>
<summary>6400MT/s</summary>

```bash
# for i in {0..7}; do ./zencli smu 0x${i}50200; done
[0x00050200] READ(smu) = 0x4a090320 (1242104608)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0011 0010 0000
[0x00150200] READ(smu) = 0x4a090320 (1242104608)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0011 0010 0000
[0x00250200] READ(smu) = 0x4a090320 (1242104608)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0011 0010 0000
[0x00350200] READ(smu) = 0x4a090320 (1242104608)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0100 1010 0000 1001 0000 0011 0010 0000
[0x00450200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00550200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00650200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00750200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
```

</details>

<details>
<summary>7500MT/s</summary>

```bash
# for i in {0..7}; do ./zencli smu 0x${i}50200; done
[0x00050200] READ(smu) = 0x7a0903a9 (2047411113)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0111 1010 0000 1001 0000 0011 1010 1001
[0x00150200] READ(smu) = 0x7a0903a9 (2047411113)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0111 1010 0000 1001 0000 0011 1010 1001
[0x00250200] READ(smu) = 0x7a0903a9 (2047411113)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0111 1010 0000 1001 0000 0011 1010 1001
[0x00350200] READ(smu) = 0x7a0903a9 (2047411113)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0111 1010 0000 1001 0000 0011 1010 1001
[0x00450200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00550200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00650200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
[0x00750200] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111
```

</details>

deseven, 2025-07-15:
So yeah, you're right, it's 687, 800, 937 and 1000 multiplied by 8.

cyring, 2025-07-15:
> So yeah, you're right, it's basically 687, 800, 937 and 1000 multiplied by 8. The multiplier could be related to something else as well.

Consider your architecture can come with a new bits layout, undocumented yet. Thus we are free to select any bit t'il the formula is accurate enough.

cyring, 2025-07-16:
@deseven 

Above formula is fixed to add a correction if the extraction of bits [2-0] is not zero

```math
F(x) = 8 \times \text{low16}(x) + 4 \times \text{extract}(x,0,3) \not = 0
```

----

You can now pull and test the **last** commit d2fd467fdf726830fd981d57354b41e7c3c8694b

Using a Zen5/Phoenix I have injected your register dump raw values to simulate the compute as it ought to be on yours

[image]

[image]

[image]

[image]

* Also observe the followings;
1. `LPDDR5` as mentioned by `dmidecode`
2. `Bus Speed` is weighted by the Base Clock

---

Feel free to show me screenshots and/or outputs of the UMC then I can check the results and no regression happened.

Thank you

cyring, 2025-07-16:
Great to see this result, I spent night coding it.

----

Yes UI lag may happen in circonstances I've not reproduced: I believe through ssh or network terminal: the scankey function may get confused.
Most of the time, I'm using local Xterm or console.

deseven, 2025-07-16:
Great work!

Yeah you're right, it's a coincidence, something started being noisy on my 5Ghz wi-fi network, which did lead to latency spikes. Switched the channel to another one and everything is great again.

cyring, 2025-07-17:
> ```
> # ./zencli smu 0x50040
> [0x00050040] READ(smu) = 0x00150708 (1378056)
>    60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
>  0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0111 0000 1000
> ```
> 
> Looks like a bunch of magic numbers to me :)

I should told you the remaining DIMMs is laying among address offsets `40,44,48,4c`

The field `NumBankGroups bits [3-2]` in the output value `0x00150708` decodes as `2` decimal (aka `10` binary)

See the Register field comment  
`/* 0=None; 1=2x; 2=4x; 3=8x BGs */`,  
a `2` `NumBankGroups` means a `BG` of `4X`

No going back to the Intel specs, can we say if a `4X` `BG` specifies LPDDR5x ? I would say no.

cyring, 2025-08-01:
Hello,
Enhancements are now available in release [2.0.8](https://github.com/cyring/CoreFreq/releases/tag/2.0.8)
I will come back on remaining subjects when specs provide more register details.
If you don't observe a regression in this version; feel free to close your issue.
Regards

deseven, 2025-08-01:
All good, thanks a lot for your work and I enjoyed tagging along!

P. S. No permission to close this issue.

closed 2025-08-01 by cyring