cyring/CoreFreq #566: -Wasm-operand-widths warnings when building with clang

srcshelton, 2025-08-12:
When building on an ARM64 host (with 16k kernel page-size) and using clang as the compiler, I get around 125 warnings in the form of:

```clang
  CC [build/corefreq-cli-extra.o]
aarch64/corefreq-ui.c:2041:30: warning: value size does not match register size specified by the constraint and modifier [-Wasm-operand-widths]
 2041 |     if (!BITVAL(Dump.Status, 0))
      |                              ^
aarch64/corefreq-ui.c:2041:10: note: use constraint modifier "w"
 2041 |     if (!BITVAL(Dump.Status, 0))
      |          ^
aarch64/bitasm.h:479:6: note: expanded from macro 'BITVAL'
  479 |                                         BITVAL_2xPARAM, /*2*/           \
      |                                         ^
aarch64/corefreq-ui.c:2070:34: warning: value size does not match register size specified by the constraint and modifier [-Wasm-operand-widths]
 2070 |                         BITSET(LOCKLESS, Dump.Status, 0);
      |                                                       ^
aarch64/corefreq-ui.c:2070:4: note: use constraint modifier "w"
 2070 |                         BITSET(LOCKLESS, Dump.Status, 0);
      |                         ^
aarch64/bitasm.h:452:4: note: expanded from macro 'BITSET'
  452 |         :       _BITSET_GPR(_lock, _base, _offset)                      \
      |                 ^
aarch64/bitasm.h:171:18: note: expanded from macro '_BITSET_GPR'
  171 |                 "lsl    x12, x12, %[offset]"    "\n\t"                  \
      |                                   ^
```

cyring, 2025-08-12:
> When building on an ARM64 host (with 16k kernel page-size) and using clang as the compiler, I get around 125 warnings

Fyi, from the kernel module `corefreqk.c` to the daemon `corefreqd`, the VMA memories are shared with a rounded length of the same page size. So you have to make sure the page size reports 16K page size on these lines:

https://github.com/cyring/CoreFreq/blob/7a1aeb4a7a82606c26b1ed3c50c3b1740f08a6a5/aarch64/corefreqd.c#L37

Various lines where `PAGE_SHIFT` is employed; kernel must also compute `PAGE_SHIFT` from a 16K pge size
https://github.com/cyring/CoreFreq/blob/7a1aeb4a7a82606c26b1ed3c50c3b1740f08a6a5/aarch64/corefreqk.c#L5690C47-L5690C57

----

Back to the assembly compilation issue, can you change the type of member `Status`

https://github.com/cyring/CoreFreq/blob/7a1aeb4a7a82606c26b1ed3c50c3b1740f08a6a5/aarch64/corefreq-ui.c#L574

as bellow:

```c
struct {
    union {
		Bit64	Reset __attribute__ ((aligned (8)));
	struct {
		Bit64	Status;
		int	Tick;
	};
    };
		time_t	StartedAt;
		FILE	*Handle;
		char	*Buffer;
		void	(*Header)(void);
		void	(*Write)(char*, int);
		void	(*Break)(void);
} Dump __attribute__ ((aligned (8))) = {
	.StartedAt = 0,
	.Reset	= 0x0,
	.Handle = NULL,
	.Buffer = NULL,
	.Header = JSON_Header,
	.Write	= JSON_Page,
	.Break	= JSON_Break
};
```

srcshelton, 2025-08-12:
Ah, so likely an issue here will be that the build-system is currently running 4k pages (which will be reflected in the sysconf() result) but the kernel sources targeted are built for 16k pages.

Should the `CoreFreq` build system perhaps ideally retrieve the kernel-source concept of page-size rather than relying on the runtime state of the build host?