cyring/CoreFreq #567: corefreqk.ko fails on ARM64

srcshelton, 2025-08-12:
```
# make CC=clang LD=ld.lld WARNING="-Wno-asm-operand-widths -Wall -Wfatal-errors" CORE_COUNT=64 OPTIM_LVL=0 MSR_CORE_PERF_UCC= MSR_CORE_PERF_URC= info
HW [aarch64]
CC [clang: /usr/lib/clang /etc/clang /usr/lib/llvm/20/bin/clang]
WARNING [-Wno-asm-operand-widths -Wall -Wfatal-errors]
PWD [/var/tmp/CoreFreq-2.0.8]
BUILD [build]
KERNELDIR [/lib/modules/6.16.0-gentoo/build]
PREFIX [/usr]
LEGACY []
UBENCH [0]
FEAT_DBG []
DELAY_TSC [0]
OPTIM_LVL [0]
CORE_COUNT [64]
TASK_ORDER [5]
MAX_FREQ_HZ [7125000000]
HWM_CHIPSET []
MSR_CORE_PERF_UCC []
MSR_CORE_PERF_URC []
ARCH_PMC []
NO_HEADER []
NO_FOOTER []
NO_UPPER []
NO_LOWER []
SILENT [0]
```

```
# make CC=clang LD=ld.lld WARNING="-Wno-asm-operand-widths -Wall -Wfatal-errors" CORE_COUNT=64 OPTIM_LVL=0 MSR_CORE_PERF_UCC= MSR_CORE_PERF_URC= corefreqk.ko
  MD [build]
  MD [build/module]
  LN [build/Makefile]
  LN [build/module/corefreqk.c]
  CC [M]  module/corefreqk.o
  LD [M]  corefreqk.o
ld.lld: error: ./module/corefreqk.o <inline asm>:4:12: expected readable system register
        mrs     x15 ,   fpcr
                        ^

ld.lld: error: ./module/corefreqk.o <inline asm>:5:12: expected readable system register
        mrs     x16 ,   fpsr
                        ^

make[3]: *** [/usr/src/linux-6.16.0-gentoo/scripts/Makefile.build:502: corefreqk.o] Error 1
make[2]: *** [/usr/src/linux-6.16.0-gentoo/Makefile:2003: .] Error 2
make[1]: *** [/usr/src/linux-6.16.0-gentoo/Makefile:248: __sub-make] Error 2
make: *** [Makefile:181: build/corefreqk.ko] Error 2
```

cyring, 2025-08-13:
> > ```
> > # make CC=clang LD=ld.lld WARNING="-Wno-asm-operand-widths -Wall -Wfatal-errors" CORE_COUNT=64 OPTIM_LVL=0 MSR_CORE_PERF_UCC= MSR_CORE_PERF_URC= info
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > `MSR_CORE_PERF_UCC` `MSR_CORE_PERF_URC` are reserved registers of the x86 ISA
> > I will suggest you run a basic `make` which will employ the registers made for ARM
> 
> FYI, without any options and on ARM64 I see:
> 
> ```
> # make info
> HW [aarch64]
> …
> MSR_CORE_PERF_UCC [MSR_IA32_APERF]
> MSR_CORE_PERF_URC [MSR_IA32_MPERF]
> ```
> 
> … is this correct?

Sure. But don't expect too much from a `Makefile`. Here it was just showing the values of any variable whatever the arch is.

----

So to build _CoreFreq_ with `<clang` your command looks enough:
```sh
make CC=clang LD=ld.lld
```

Of course you will have to apply the patch we shared previously.

Does it work ?

cyring, 2025-08-13:
Another solution to solve the unknown symbols in assembler would be `LLVM_IAS=0`

Quoting [Gentoo wiki](https://wiki.gentoo.org/wiki/LLVM)

> In the past, it was necessary to pass LLVM_IAS=1 to use the Clang internal assembler for a complete LLVM toolchain built kernel. This is no longer required since LLVM=1 now defaults to include the Clang internal assembler. Use LLVM_IAS=0 to disable the internal assembler if desired, otherwise stick to the default behavior.

srcshelton, 2025-08-14:
I've tried setting `LLVM_IAS=0` (no change), `AS` to `/usr/bin/as` and `LD` to `/usr/bin/ld` - all which either resulted in the same error or compilation failures (`ld: unrecognised emulation mode: llvm`).

Further, I'm afraid that with the patch above applied I now get the error message:

```
# LD [M]  corefreqk.o
  ld.lld -EL  -maarch64elf -z norelro -mllvm -import-instr-limit=5 --compress-debug-sections=zstd -z noexecstack   -r -o corefreqk.o @corefreqk.mod 
ld.lld: error: ./module/corefreqk.o <inline asm>:4:12: expected readable system register
        mrs     x15 ,   xzr
                        ^

ld.lld: error: ./module/corefreqk.o <inline asm>:5:12: expected readable system register
        mrs     x16 ,   xzr
                        ^

make[4]: *** [/usr/src/linux-6.16.0/scripts/Makefile.build:502: corefreqk.o] Error 1
```

cyring, 2025-08-16:
Possibly to remove 64-bits operands issue with clang/lld I have this patch giving some results and removing warnings in my BITOPS functions

```patch
diff --git a/aarch64/bitasm.h b/aarch64/bitasm.h
index ee94f0f7..39b07a77 100644
--- a/aarch64/bitasm.h
+++ b/aarch64/bitasm.h
@@ -174,7 +174,7 @@ ASM_RDTSC_PMCx1(x14, x15, ASM_RDTSC, mem_tsc, __VA_ARGS__)
                _BITSET_POST_INST_##_lock                               \
                : [ret] "=m" (_ret)                                     \
                : [addr] "r" (&_base),                                  \
-                 [offset] "r" (_offset)                                \
+                 [offset] "r" ((uint64_t)(_offset))                    \
                _BITSET_CLOBBERS_##_lock                                \
        );                                                              \
        _ret;                                                           \
@@ -239,7 +239,7 @@ ASM_RDTSC_PMCx1(x14, x15, ASM_RDTSC, mem_tsc, __VA_ARGS__)
                _BITCLR_POST_INST_##_lock                               \
                : [ret] "=m" (_ret)                                     \
                : [addr] "r" (&_base),                                  \
-                 [offset] "r" (_offset)                                \
+                 [offset] "r" ((uint64_t)(_offset))                    \
                _BITCLR_CLOBBERS_##_lock                                \
        );                                                              \
        _ret;                                                           \
@@ -303,7 +303,7 @@ ASM_RDTSC_PMCx1(x14, x15, ASM_RDTSC, mem_tsc, __VA_ARGS__)
                _BIT_TEST_POST_INST_##_lock                             \
                : [ret] "=m" (_ret)                                     \
                : [addr] "r" (&_base),                                  \
-                 [offset] "r" (_offset)                                \
+                 [offset] "r" ((uint64_t)(_offset))                    \
                _BIT_TEST_CLOBBERS_##_lock                              \
        );                                                              \
        _ret;                                                           \
```

closed 2025-09-09 by cyring